reinhard-brandstaedter/solarflow-control #64: Topic House is not used anymore after last update

AzraelsDisk, 2023-10-07:
[image]
[image]
[image]

AzraelsDisk, 2023-10-07:
It looks like some of the variables are ignored, is there an other way no to set them?

reinhard-brandstaedter, 2023-10-07:
There is now a new (additional) way to provide configuration via a ```config.ini``` file in the program directory. Maybe I've introduced a bug here.
Please try providing a config.ini (template found in the src dir)

AzraelsDisk, 2023-10-07:
How could this be done via Synology container manager? There is could only add variables. Thx

reinhard-brandstaedter, 2023-10-07:
It's the same as with other docker runtimes, you can mount a single file as a "volume" into a single file into the container as well: https://docs.docker.com/storage/volumes/

But there is indeed a problem when there is no config file present, default values are applied.... needs a fix

closed 2023-12-18 by reinhard-brandstaedter