reinhard-brandstaedter/solarflow-control #77: DEV: System leaving charge through however should not be the case

tuxianerDE, 2023-10-20:
The logs are indicated that path 3.2.2 is taken and thus leaving charge through, however criteria logically should not be met.

Note: We are in low light (low watt situation), direct panels <10

Actual situation should be:
- Take direct panels if available
- Remain in chargethrough and thus set SF to 0 to ensure SF solar power goes to batteries

[solarFlow.log](https://github.com/reinhard-brandstaedter/solarflow-control/files/13051056/solarFlow.log)

reinhard-brandstaedter, 2023-10-20:
"leaving charge through mode" is probably a bit misleading here.
What happens is that, while the SolarInput is below the minimum reserved charge power (MINIMUM_CHARGE_LEVEL) it will not discharge from batteries even though leaving charge through.
Assuming that throughout the day we still want to use the produced solar energy directly while using only the overproduction for charging (why store something - with losses - when we can make use of it directly)
However, I agree in cases where additional panels are connected directly it would make more sense to use everything from the SF Input to charge and leverage the direct panels for household supply.

On the contrary the same principle would apply here: use as much as possible for direct consumption and only store overproduction, meaning also the SF input could supply to the current demand (without further discharging)?

tuxianerDE, 2023-10-20:
True that last sentence is valid, i.e. take the direct panels (fully like in 3.2.1) and add anything above the minimum charge level from the SF to the house - if needed by the demand.

The current situation is that it will always follow path 3.2.2 which leads the system to limit itself to 0 completely instead of the smart combination of direct connected panels which we have in 3.2.1

I stopped the container today morning to see if the direct panels deliver (and they do around 40W) so in theory it suffices to let them through and regulate the solarflow down.

reinhard-brandstaedter, 2023-10-20:
[This commit](https://github.com/reinhard-brandstaedter/solarflow-control/commit/fda6ed815e1b0903e5931b223a2591d11ebcecaf) should fix this behavior

tuxianerDE, 2023-10-20:
pulled dev container. seems to operate now fine with decision path 3.2.2

solarFlowDev  | 2023-10-20 13:50:25,851:INFO: Sun: 08:03 - 18:30, H_SM: [ 159, 160, 160, 160, 154], H_D: 250W, SF_S: 105W, I_DP: 95.9 (47.6W|48.3W), I_OP: 91.8W, SF_H: 0W, SF_B: 15% (13%|8%|24%), CH_T: off, charging: 97W => Limit: 0W - [ 0, 0, 0, 0, 0] - decisionpath: 3.2.2
solarFlowDev  | 2023-10-20 13:50:25,852:INFO: Setting solarflow output limit to 0 W
solarFlowDev  | 2023-10-20 13:50:25,852:INFO: Setting inverter output limit to 210.0 W (105 x 1 / (2/4)

closed 2023-11-27 by tuxianerDE