reinhard-brandstaedter/solarflow-control #78: FEATURE: Revised steering logic aligned with time

tuxianerDE, 2023-10-22:
Overarching rule:

Battery Low:
- To validate shall we use packSoC or for every single battery? Reason: We cannot distinguish which battery is the top battery, this battery provides power to the hub during night, we probably dont want to drain that completely?

If packSoC/single battery is below configured BATTERY_LOW ensure that solarFlow charges only (SF limit 0).
If direct panels are connected inverters limited is demand + 50W.

Since the inverter steers the limit on all MPPTS equally the following math determine the factor applied to the limit:
1) how many INVERTER_INPUTS_USED
2) How many MPPTS are connected to the INPUTS USED?
-> Should we introduce a new config variable instead? i.e. steering factor? Since the math is not consistent.

Night:
Time window: sunset till sunrise

If CHARGE_THROUGH_THRESHOLD is No (default):
- Use SolarFlow and steer via Inverter
- Set SolarFlow to MAX_DISCHARGE_LEVEL
- Set Inverter Limit to demand-20W (max. MAX_DISCHARGE_LEVEL)

If CHARGE_THROUGH_THRESHOLD is Yes:
- Set Limit to 0 (Inverter + SF)

Sunrise:
Time window: sunrise till sunrise + offset (i.e. 60min)

If CHARGE_THROUGH_THRESHOLD is No (default):
- Use SolarFlow and steer via Inverter
- Set SolarFlow to MAX_DISCHARGE_LEVEL; Reduce SolarFlow by Amount of DIRECT CONNECTED PANELS
- Set Inverter Limit to demand-20W (max. MAX_DISCHARGE_LEVEL)

If CHARGE_THROUGH_THRESHOLD is Yes:
- Set SolarFlow output to 0
- Set Inverter Limit to demand-20W (use residual solar if there is any)
- set CHARGE_THROUGH_THRESHOLD indicator to YES

Day:
Time window: (sunrise + offset (i.e. 60 min)) till (sunset - offset (i.e. 60 min))

- SolarFlow should charge up; Discharging from SF should only take place if BATTERY_HIGH or DAY_DISCHARGE_LIMIT or bypass (automatically bs SF) is met (and the CHARGE_THROUGH_INDICATOR is set to No. Otherwise SF is set to be 0W
Set Inverter Limit to demand + 80W (allowing direct connected panels to steer) This logic also works if no direct panels are connected since it steers above condition.

If packSoC is larger then CHARGE_THROUGH_THRESHOLD or packSoC is larger or equal to DAY_DISCHARGE_LIMIT:
	set CHARGE_THROUGH_THRESHOLD indicator to No

Sunset:
Time window: (sunset - offset (i.e. 60 min)) till sunset
If packSoc is greater then CHARGE_THROUGH_THRESHOLD:
- Use SolarFlow and steer via Inverter
- Set SolarFlow to MAX_DISCHARGE_LEVEL
- Set SolarFlow to MAX_DISCHARGE_LEVEL; Reduce SolarFlow by Amount of DIRECT CONNECTED PANELS
- Set Inverter Limit to demand-20W (max. MAX_DISCHARGE_LEVEL)

If packSoC is smaller then CHARGE_THROUGH_THRESHOLD:
- Set SolarFlow output to 0
- Set Inverter Limit to demand-20W (use residual solar if there is any)
- set CHARGE_THROUGH_THRESHOLD indicator to YES

reinhard-brandstaedter, 2023-10-24:
The initial idea of the steering logic had two goals:
* ensure that we never enter a bypass, where suddenly too much energy is passed into the household that cannot be used (because demand is low). Instead ensure steady charging and discharging
* ensure we only feed to the household what is needed

**Revised Goals**
* ensure that we only feed to household what is needed (demand)
* ensure that batteries are completely discharged (0%) and charged (100%) from time to time, even across multiple days
* avoid entering bypass
* support a mix of charging panels and panels that feed directly to the household, prefer the use of direct panels for demand over charging panels

***Ideas***
* create a memory: learn from night usage how much capacity is typically needed during night time, in combination with battery capacity this could help in decisions of charge/discharge limits to ensure battery 0 - 100% charging. (dynamic max-limiting). This could create steady full charge/discharge cycles (battery health)
*

reinhard-brandstaedter, 2023-10-24:
**Idea for Charge-Through**
Instead of working with thresholds on battery level, we can remember the timestamp when the battery reached 100% and then take the decision if it should be discharged again. e.g. only allow discharging if 100% has been reached within 7 days.

On the other hand we could also ensure that the battery will be discharged to 0% from time to time, but I guess that happens anyway...

tuxianerDE, 2023-10-24:
ChargeThrough:
might be difficult especially during the "dark" months of Winter. So making it configurable (if not set or timestamp cannot be determined start counter from 0) dont take decision and let system "just" run.

closed 2023-11-27 by tuxianerDE