reinhard-brandstaedter/solarflow-control #80: REFACTORING: provide control parameters via mqtt

reinhard-brandstaedter, 2023-10-28:
Set control parameters on the fly via mqtt

reinhard-brandstaedter, 2023-10-31:
### These control parameters can be set already:
| Topic | Value | Meaning |
|---|---|
| solarflow-hub/<device ID>/control/chargeThrough | ON|OFF | turns Charge-through mode on or off |

tuxianerDE, 2024-04-26:
Bringing up an old enhancement again.

I would vouch to have other parameters possible to be changed on the fly as well.

Specifically thinking of offset (sunrise and sunset) as this might change over the course of the year and one could use an HA automation to set this value every week.

But also the minimum charge level, for example for people are absent for a longer time i.e. vacation one could simply change this value on the fly.

Whenever the application is restarted the config values are taken as fundamental basis.

mavo, 2024-07-18:
Hello @reinhard-brandstaedter,

may I ask if there is any plan to add more control parameters via MQTT, such as sunset/sunrise offsets? Why do I ask, if I am on vacation in summer time, but there is a day with very bad weather in the morning, it can happen that I consume energy from the grid whilst in the afternoon the battery gets filled to 100% and I gift energy to the grid, which is not ideal.
It would be cool to be able to steer how long the offsets are via MQTT, as @tuxianerDE suggested.

Another option, just for days as "on the fly" change, to have a control parameter which would allow to consume from the battery, even if the sunset/sunrise offsets are not matched. This again I would use on demand like a day today, where I can predict to have my battery full in the afternoon, but currently consume power from the grid as the dishwasher is running whilst the battery is already filled up to 50%.

Do you think you will invest time into this in the future?

Thanks in advance, 
MaVo

reinhard-brandstaedter, 2024-07-19:
There are now additional control parameters that can be changed via MQTT (no HA templates yet):

| Topic | Value | Meaning |
|---|---|---|
| solarflow-hub/{deviceId}/control/sunriseOffset | <int> | minutes after sunrise to be considered night (allow battery consumption) |
| solarflow-hub/{deviceId}/control/sunsetOffset | <int> | minutes before sunset to be considered night (allow battery consumption) |
| solarflow-hub/{deviceId}/control/minChargePower | <int> |  minimum charge power reserved for battery |
| solarflow-hub/{deviceId}/control/maxDischargePower | <int> | maximum discharge power drawn from battery |
| solarflow-hub/{deviceId}/control/dischargeDuringDaytime | ON|OFF | allow discharging during the day / outside offset timeframe |

mavo, 2024-07-23:
@reinhard-brandstaedter Thank you for this quick action! I appreciate it 🙇 

Can you ping me, when you've created a new master release with this new additions?
I currently use a slightly modified dev image which I like to preserve as is, so it would be easier if those changes would be on master 😇 

MaVo

mavo, 2024-07-25:
Hello @reinhard-brandstaedter,

did I misunderstood sth perhaps? I pulled master yesterday (after you released v0.71) and ran it. It works well, but for some reason, I did expect to have the new switch for daylightDischarge in HA, but I do not. The template file is in the image, but its seems to just not be published 🤔 
Anything I missed?

MaVo

reinhard-brandstaedter, 2024-11-20:
Adressed with #302

closed 2024-11-20 by reinhard-brandstaedter