reinhard-brandstaedter/solarflow-control #95: Safe last state on restart

ricariel, 2023-11-09:
Every time the container restarts (automatically if new image is present on docker hub) important values are lost.
Example:
solarflow.py -> 
self.lastFullTS
self.lastEmptyTS

Would be nice if these and perhaps other values get stored somewhere.

reinhard-brandstaedter, 2023-11-09:
Actually exactly these values are stored as retained messages in MQTT:

[image]

As those are retained messages they are read in again upon startup. However it could be that the mqtt broker is not persisting these retained messages upon restart?

ricariel, 2023-11-09:
Read the code again. You are correct. Code just never reached that state with control running. Pardon.

closed 2023-11-09 by ricariel