reinhard-brandstaedter/solarflow-control #114: FEATURE REQUEST: Rapid change scenario

tuxianerDE, 2024-01-12:
When setting the max discharge to 600/800 in the config file, the steering for rapid changes in demand (heater; toaster etc) lead to massive output to the grid as it take time to adjust. 

Thought: Introduce a config parameter RAPID_CHANGE_THRESHOLD (lets say default is 200W).

If a change in demand >= to this threshold is detected and we are on battery the moving average of the smartmeter gets rest and the inverter output gets set to the new demand directly and thus rapidly reduced as well.

[image]

tuxianerDE, 2024-02-05:
Using the current dev build:

Using the average calculation shows good results when demand changes and thus a quick change is required. During steady run I see that the normal average introduces more "smaller changes to the overall system".

Based on this an idea (to document). During normal run the weigthed average is being used; a configurable rapid change value is introduced and validated. If a rapid change is detected (change in home power W >= 150W?) the average changes from weighted to normal average for a duration of Z minutes (1 minute?) and the counter resets when ever the rapid change event occurs again. When the counter is 0, the average resumes to its normal state of weigthed average.

mavo, 2024-02-23:
Just because I have quite some spikes in my house consumption (and also use the battery within the day) I would be interested to get this feature somehow done.
For example, I just send 0.05kwh to the grid because the power was not regulated down in a timely manner after I stopped using the heater while changing the babies pants^^

Can I ask for the current state of this idea? I would like to see it in the tool if possible 😇

MaVo

tuxianerDE, 2024-02-23:
We are currently testing a possibility in our dev tree. Running through 2-3 charge/discharge cycle to see consistent results.

So yes working on it - still some patience.

tuxianerDE, 2024-03-01:
I have done a few tests with the new rapid change detection.

The following screenshot outlines always the same timeframe during a discharge cycle (23:00:00 - 06:00:00)

The following has been configured:
Average calculation set to: avg
rapid_change_diff: 200

[image]

Average calculation set to: wavg
rapid_change_diff: 200
[image]

The wavg behaved better at least in my calculations as less energy was actually "spilled" to the grid when a change was executed. 

Suggestion would be to switch the average calculation from qwavg to wavg together with this feature!

tuxianerDE, 2024-03-08:
I my view issue can be closed. Rapid change works well, configuration is to adjust individually, overall quick adjustment and steering.

closed 2024-03-08 by tuxianerDE