reinhard-brandstaedter/solarflow-control #117: Shelly3EM or ShellyEM3 - none does work for me

mavo, 2024-01-26:
Hi @reinhard-brandstaedter,

first of all, thank you for providing this. I think it will be great - when I get it working.

However, when I try to set up my system I am stuck, as it seems that the config/code are not 100% clear on weather they want the shelly named 3em or em3.
I tried both scenarios-  here are the results:

config.ini with `smartmeter_type = ShellyEM3`
```
Traceback (most recent call last):
File "/solarflow/solarflow-control.py", line 407, in <module>
main(sys.argv[1:])
File "/solarflow/solarflow-control.py", line 404, in main 
                      run()
File "/solarflow/solarflow-control.py", line 329, in run
                      smt_opts = getOpts(smtType)
File "/solarflow/solarflow-control.py", line 315, in getOpts 
                      opts.update({opt:opt_type(converter(configtype.__name__.lower(),opt))})
File "/usr/local/lib/python3.10/configparser.py", line 810, in _get_conv
                      return self._get(section, conv, option, raw=raw, vars=vars,
File "/usr/local/lib/python3.10/configparser.py", line 805, in _get
                      return conv(self.get(section, option, **kwargs))
                                                                                                                                                                                                                      File "/usr/local/lib/python3.10/configparser.py", line 783, in get
                      d = self._unify_values(section, vars)
File "/usr/local/lib/python3.10/configparser.py", line 1154, in _unify_values
                       raise NoSectionError(section) from None
                       configparser.NoSectionError: No section: 'shellyem3'
```

config.ini with `smartmeter_type = Shelly3EM`
```
Traceback (most recent call last):
File "/solarflow/solarflow-control.py", line 407, in <module>
                       main(sys.argv[1:])
File "/solarflow/solarflow-control.py", line 404, in main
                       run()
File "/solarflow/solarflow-control.py", line 328, in run
                       smtType = getattr(smartmeters, SMT_TYPE)
AttributeError: module 'smartmeters' has no attribute 'Shelly3EM'. Did you mean: 'ShellyEM3'?
```

Can you mabe check if there is a mistake somewhere? If you need more information, feels free I will answer as much as I can :-)

Regards,
MaVo

reinhard-brandstaedter, 2024-01-26:
Hi @mavo ,

ShellyEM3 is the correct type.
Did you check that you config.ini contains a section [shellyem3]?
Like this (note the lowercase here):
```
[shellyem3]
# The MQTT base topic your Shelly 3EM (Pro) is posting it's telemetry data to
# Note: you have to configure your Shelly to use MQTT
base_topic = shellies/shellyem3/
```

Example:
https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/config.ini

mavo, 2024-01-26:
Ahh the mistake comes from the main readme. In there, its states the "wrong" shelly name. See [here](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/README.md?plain=1#L121).

Will change it in my config and check again. Thanks for the quick reply 🙇

mavo, 2024-01-26:
Also, the main readme states the wrong shelly name in [this line](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/README.md?plain=1#L54) as well

reinhard-brandstaedter, 2024-01-26:
Oh, thanks for pointing that out! I updated it.
The whole Shelly config started with a similar typo on my side when I misspelled the Shelly product name :-)

mavo, 2024-01-26:
Aaaand I found more :-D
It should not have a `/` in the topic for shelly [here](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/README.md?plain=1#L124) as the shelly topic subscription is adding a leading `/` as well [here](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/smartmeters.py#L114C19-L114C38).

reinhard-brandstaedter, 2024-01-27:
a case of sanitizing, indeed!

mavo, 2024-01-27:
Cool, will close this for now and probably open one more regarding the configuration options.
Or do you like to have one issue discussing multiple different things?

Regards and thanks again for your replies and effort you've put in!

closed 2024-01-27 by mavo