reinhard-brandstaedter/solarflow-control #119: Question: Configuration settings

mavo, 2024-01-29:
Hi @reinhard-brandstaedter,
as mentioned in my last ticket, I'll probably open one more. I do not yet understand the configuration fully, so I kindly ask for some clarification 🙇 

**Setup:**
- 4 Solar Modules (400W each)
- SF with 2 Batteries (AB2000 - not yet delivered, but should arrive soon)
- 2 Modules will be connected to the Hub
- 2 Modules will be connected to inverter directly

**Goals:**
- Prefer home power delivery from panels with a given, fixed battery charge value
- Zero feed in is not important to me. I am fine to export stuff atm, as throwing it away in my view is wasting potential to have green energy in the grid.

**Questions:**
- Why is setting a minimum input for the batteries influencing the maximum [inverter output](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/solarflow-control.py#L77)? Or where's my issue on thinking?
- What will [this config](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/README.md?plain=1#L131) do when set to a value (e. g. 50)?
- How to configure the system to achieve my goals?

**Current config:**
- `limit_inverter = true` (its set to true as a test for the time being to see how it behaves - should be set to `false` in order to have feed in possible, correct?)
- `max_discharge_power = 800` - as I want to drain the battery when home needs the power
- `max_inverter_limit = 800` - as I want to have maximum output to home when required (direct solar power might be that high and/or sf hub can provide the power from batteries)
- `min_charge_power = 200` - I like to have some fixed input to the batteries

reinhard-brandstaedter, 2024-01-29:
Hi @mavo ,

that is generally the same setup I'm running (intend to running as I'm waiting for addtl. panels).
Currently I have 1 panel directly connected and one through the SF hub.
Regarding feeding to the grid my goal is to ensure nothing that is stored in the battery goes to the grid but is discharged when needed at home. The panels connected directly to the inverter can feed-in whatever is not needed at home.

An additional goal for me is to ensure that the battery is rather constantly charged throughout the day (ideally with a optimal min_charge_power set so that it can fully charge from 0 to 100 within a day). The rest should go to home/grid.

As questions:
the MAX_INVERTER_INPUT is not doing anything anymore, so this can be ignored. This is a leftover when controlling the output of the hub wasn't as stable as it is now.

The inverter_min_limit value defines the minimum limit that could be set on a inverter (via Ahoy or OpenDTU). We saw scenarios where setting the limit below a certain level actually completely shut down the inverter (hoymiles HM-1500, I think) and it wouldn't wake up again. So this is a "security" setting to overcome this behavior.

Your current config seems legit and quite similar to what I have. Though a few thoughts:
- With direct connected panels I'm not trying to discharge at full power (800W) but rather have a lower, constant discharge (max_discharge_power) that covers my base load between sunset and sunrise (+/- some offset). So instead of trying to cover peaks, I rather spread the battery consumption during the night. (ideally 100% full at sunset and 0% at sunrise)
- the limit_inverter = true should be left as is. It's actually obligatory, because with direct panels and hub-connected panels you must control the inverter AND the hub to ensure the inverter pulls from the right "source". It's also somewhat historically, as controlling the SF hubs output alone was really slow/lagging.
- for the min_charge_power I have the same requirement to ensure a constant loading of the battery. By the way: the control script also performs a "charge-through" to ensure that the [battery gets fully charged now and then](https://github.com/reinhard-brandstaedter/solarflow-control/blob/963bb4e83b1a925fdda3070b196a6ff889ad7cda/README.md?plain=1#L72) (to allow calibration and increase battery health). This disables any discharging if the battery hasn't been charged to 100% for the given time. Thats especially kicking in in winter.
- I rather discharge during the night. To set what is considered night we use sunset/sunrise times plus [offsets](https://github.com/reinhard-brandstaedter/solarflow-control/blob/963bb4e83b1a925fdda3070b196a6ff889ad7cda/src/config.ini#L11)

Hope that helps!

mavo, 2024-01-29:
Thank you @reinhard-brandstaedter for your detailed explanation, really appreciate it!

>Regarding feeding to the grid my goal is to ensure nothing that is stored in the battery goes to the grid but is discharged when needed at home. The panels connected directly to the inverter can feed-in whatever is not needed at home.

Awesome, exactly what I'd like to have. I think I need to go down to the "buy me a coffee" as you deserve one for sure! 

> The inverter_min_limit value defines the minimum limit that could be set on a inverter (via Ahoy or OpenDTU). We saw scenarios where setting the limit below a certain level actually completely shut [...]

As I have a HM-1600, I better leave it then as is :-)

> the limit_inverter = true should be left as is. It's actually obligatory, because with direct panels and hub-connected panels you must control the inverter AND the hub to ensure the inverter pulls [...]

This is a great hint, so I'll leave as set to true.

> By the way: the control script also performs a "charge-through" to ensure that the [battery gets fully charged now and [...]

Saw this as well, and I like it. Although this could be hard in winter times, as I will have almost 4kWh of battery power. Lets see how it'll be next winter. For now I am looking forward to have more sun over the days ^^

> With direct connected panels I'm not trying to discharge at full power (800W) but rather have a lower, constant discharge (max_discharge_power) that covers my base load between sunset and sunrise [...]

Also connected to my battery size, I cannot really (for the summer days at least) realize a full discharge when sun is down. My house has a base consumption of around 250W, meaning I would need ~13-14hrs of discharge. But this is actually kind of a luxury problem I think so I will go for a higher value.

**One more new question regarding HomeAssistant:**
Another thingy regarding the Home Assistant value publishing. Is there anything like "daily power pack input in kwh" (and output) or something similar? Is there any trick to get the values published to HA, so that I can set up the battery in the energy dashboard for example? 
To my current understanding, and please correct me if I am wrong, I can only use the packInput and packOutput power to calculate how much has been added to / drained from the battery with a riemann calculation which I can then use to feed HA's energy dashboard?!

MaVo

reinhard-brandstaedter, 2024-01-30:
Here is an example how the charge-through works, over the course of 4 days:

[image]

After Day 1, once the battery is full it is allowed to discharge, it will do this at the max discharge level. The next day the battery gets charged a bit again and gets discharged (as it has been full within 32hrs before). Note that my discharging will stop during the night due to the SF hub minimum battery capacity set at this time.
In the last two days it is only charging, not discharging as the last-full time was longer ago than my threshold.
You will also note that the discharging starts/ends before/after sunset/sunrise. Thats the "offset" window to the dynamic calculated sunrise/set times. Adjusting those to rather large values would also allow discharging throughout the day.

For Homeassistant daily capacity usage you really have to rely on calculating that on the HA side. The hub itself doesn't provide these values. Do not use the % value that would be far too inaccurate :-)

mavo, 2024-01-31:
Thanks for the example, thats helpful.
I wonder if its actually possible to disable the sunrise/sunset mechanism to control when the battery is discharged? It's good to charge the battery every now and then, but for me its less important to have it only on discharge in the night. I can just use 20hrs of offset before and after sunrise, which would kinda result in what I want, but is there a smarter way which I did not see so far?
Is there a particular reason why you only wanna discharge after sunset? Less stress on the battery? 

Perhaps you know, as I did not find a definitive answer yet for the following scenario:
- Input from direct panels: 200W
- Input from hub connected panels: 200W
- Home usage/outputToHome: 500W
Question: Will SF pass the energy through the battery, even when it will be directly taken from it again to feed the house, or will it bypass the battery pack in this scenario?

Another question: how often does the hub publish its data into MQTT? I think, this is on the BT-manager side, but I'll ask here anyway. (please don't kill me for opening a ticket in the BT manager as well, as I think there is a small flaw - or something I misunderstood 😇)

reinhard-brandstaedter, 2024-02-01:
Assuming you have direct connected panels it makes sense to use the battery supply only during night +/- offset. Generally it's not very efficient trying to cover (short) peaks from the battery as there is always a slight delay when switching between input and no input, it's better to perform constant discharging. In the end it doesn't matter when you use the battery power, only when your setup is able to fully charge and discharge again throughout the day (without feeding to the grid) it could make sense to discharge while also direct connected panels produce power. 
But if you also want to discharge throughout the day the approach is to set the offset big enough. 

> Question: Will SF pass the energy through the battery, even when it will be directly taken from it again to feed the house, or will it bypass the battery pack in this scenario?
If bypass is enabled and triggers it will feed everything to home.
If bypass is not on it will use min_charge_level to charge the battery and the rest goes to home as needed.

Maybe there is a misunderstanding: the BT-manager should be just used once to take the hub offline, then it doesn't need to run anymore. I do not recommend to run it in proxy mode, that is unreliable and just adds another component that isn't needed. Once the hub is "offline" it will report in arbitrary intervals to MQTT (as data changes, e.g. solarinput is reported if there is change). That is not controlled, it's built into the SF firmware.
Additionally the control script will periodically trigger a "reporting heartbeat" to ensure data that doesn't change that frequently gets reported regularily

mavo, 2024-02-14:
Hello, me again.

Regarding my third question, I think I found the answer in the manual already. As it states that the battery should be drained completely and then filled completely once again.
Is there a way to convince the sf-control to do this, aka deplete the battery and only then start charging again, or should I rather stop sf-control, connect to the hub manually and set it to discharge until it's empty? The full loading should be covered by sf-control again afterwards, correct?

Sorry for the amount of questions. I am really happy about any answers I can get  😇

Mavo

reinhard-brandstaedter, 2024-02-16:
Ad 1.) I can only assume that the reported value for panel power by SF is done slightly different than what OpenDTU does. Also it might be that the Hub itself is using some power and that is already subtracted from the actual panel power. Wouldn't worry too much about it, only if that is the case in high sunshine scenarios - but also keep in mind that the hub has a max input of 400W.

Ad 2.) I'd not run with a min_charge_power of 0 generally, as you always want to make sure the battery gets charged at least a little bit. But there is no fallback if it's set to 0. Zero should be just considered as such.

Ad 3.) From time to time it can happen that the battery shows 98/99% for a long time until it finally jumps to 100%. That is normal and is just part of the calibration game.

Ad 4.) I have a pipeline, whenever something is merged to dev/master new images are built with tags: master or :dev

closed 2024-02-26 by mavo