reinhard-brandstaedter/solarflow-control #121: DEV: Oversteering (DTU Limit too high; shuts down inverter)

tuxianerDE, 2024-02-05:
I have made the following observation. If the inverter via the DTU is steered to high (like the screenshot attached) the inverter actually almost shuts down.

The current implementation is a multiplication based on the number of channels (in my case 4). However may be the number of MPPTs is the one to be used for the inverter. I checked OpenDTU and they allow for a limit max of 150% of the overall inverter limit i.e. HM-1500 (max. 150% -> 2250W)

My config values:
# the max output power of your inverter, used to calculate correct absolute valu                                                                                                                                                             es
inverter_max_power = 1100
max_discharge_power = 800
max_inverter_limit = 800

[image]

tuxianerDE, 2024-02-13:
Resolved. With latest change in DEV inverter limits at 150% as reported from DTU and thus no oversteering.

[image]

closed 2024-02-13 by tuxianerDE

tuxianerDE, 2024-02-23:
Noticed that in solarFlow operation the inverter oversteering limit is getting ignored. Logs are below. The system is not oversteering however the limit to protect should apply here too as it does not make a difference between direct solar panels and battery mode.

Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,677:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0,0.0,0.0 ], B: 39% (39|40|39|38), V:48.6V (48.4|48.5|48.7|48.7), C:-789W, P:False, F:3.0h, E:197.9h, H:752W, L:746W#033[0m
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,677:INFO: #033[33;20mINV: AC:716.5W, DC:759.2W (0.1|0.1|389.2|369.8), L:2917.5W [800W]#033[0m
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,678:INFO: #033[33;32mSMT: T:Smartmeter P:9.0W [ 100.0,96.0,93.0,93.0,93.0,85.0,85.0,55.0,46.0,33.0,32.0,14.0,9.0,9.0,9.0,9.0 ]#033[0m
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,678:INFO: Direct connected panels can't cover demand 0.2W/740.8W, trying to get rest from hub.
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,678:INFO: Checking if Solarflow is willing to contribute 740.6W ...
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,680:INFO: Based on time, solarpower ( 0.0W) minimum charge power (500W) and bypass state (False), hub could contribute 740.6W - Decision path: 2.1.
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,681:INFO: Setting solarflow output limit to 737.0W
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,681:INFO: Solarflow is willing to contribute 737.0W!
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,681:INFO: Adjusting inverter limit from 0.1W to 5.2W
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,682:INFO: Direct connected panel limit is 5W.
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,683:INFO: Setting inverter output limit to 1200.0W (1 min moving average of 727W x 4)
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,684:INFO: Sun: 07:29 - 18:00 Demand: 740.8W, Panel DC: (0.1|0.1), Hub DC: (389.2|369.8), Inverter Limit: 1200.0W, Hub Limit: 737.0W
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,694:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0,0.0,0.0 ], B: 39% (38|40|39|38), V:48.6V (48.4|48.5|48.7|48.7), C:-755W, P:False, F:3.0h, E:197.9h, H:720W, L:737W#033[0m
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,694:INFO: #033[33;20mINV: AC:660.5W, DC:639.9W (0.1|0.1|328.0|311.7), L:1200.0W [3600W]#033[0m
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,694:INFO: #033[33;32mSMT: T:Smartmeter P:109.0W [ 93.0,85.0,85.0,55.0,46.0,33.0,32.0,14.0,9.0,9.0,9.0,9.0,5.0,9.0,1.0,1.0,17.0,109.0 ]#033[0m
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,695:INFO: Direct connected panels can't cover demand 0.2W/686.2W, trying to get rest from hub.
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,695:INFO: Checking if Solarflow is willing to contribute 686.0W ...
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,696:INFO: Based on time, solarpower ( 0.0W) minimum charge power (500W) and bypass state (False), hub could contribute 686.0W - Decision path: 2.1.
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,697:INFO: Setting solarflow output limit to 731.0W
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,698:INFO: Solarflow is willing to contribute 731.0W!
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,698:INFO: Adjusting inverter limit from 0.1W to 5.2W
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,699:INFO: Direct connected panel limit is 5W.
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,700:INFO: Setting inverter output limit to 2912W (1 min moving average of 721W x 4)
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,701:INFO: Sun: 07:29 - 18:00 Demand: 686.2W, Panel DC: (0.1|0.1), Hub DC: (328.0|311.7), Inverter Limit: 2912.0W, Hub Limit: 731.0W

reinhard-brandstaedter, 2024-02-24:
There is two interesting lines here:
```
Feb 23 19:36:08 babylon solarFlow[11127]: 2024-02-23 19:36:08,677:INFO: INV: AC:716.5W, DC:759.2W (0.1|0.1|389.2|369.8), L:2917.5W [800W]
```
and

```
Feb 23 19:36:23 babylon solarFlow[11127]: 2024-02-23 19:36:23,694:INFO: INV: AC:660.5W, DC:639.9W (0.1|0.1|328.0|311.7), L:1200.0W [3600W]
```
The total capacity of the inverter is first detected as [800W] and then suddenly as [3600W]. This is strange as the calculation is based on the relative and absolute limit it reads from OpenDTU.

`capacity = limitAbsolute/limitRelative*100`

The only explanation i have for that is that the data from OpenDTU was not correct. Can you provide more logs (especially only the "INV" lines for a larger timeframe?

tuxianerDE, 2024-02-24:
Here you go. Once the full log of the time frame with more "before". And an extract of the INV section.

[solarFlow.log](https://github.com/reinhard-brandstaedter/solarflow-control/files/14392318/solarFlow.log)
[solarFlow-INVOnly.log](https://github.com/reinhard-brandstaedter/solarflow-control/files/14392319/solarFlow-INVOnly.log)

reinhard-brandstaedter, 2024-02-24:
Hm. seems the limit was detected stable at 1500W before ~19:33, then there was "hickup" where it jumped between 800 and 3600W and it returned to stable 1500W again. That is indeed weird and can only happen with incorrect data from OpenDTU when the absolute and relative limit presented by OpenDTU do not match....
Do you see more such "events" or is this a one-time observation?
You haven't restarted OpenDTU or the inverter around this time?

tuxianerDE, 2024-02-24:
Nope did not touch any of the components as I wanted to ensure that the test with the rapid change detection and avg is correctly working.

This has been a one time occurrence when I look at the graphs, at least nothing observable

reinhard-brandstaedter, 2024-02-24:
Hm, maybe there could be a race condition. We are determining the maxPower of the inverter by reading the absolute and the relative values we get from the DTU.
MQTT topics:
```
            f'{self.base_topic}/status/limit_absolute',
            f'{self.base_topic}/status/limit_relative'
```
The maxPower is always calculated when there is an update to the limitRelative value:

```
def updLimitRelative(self, value:float):
        self.limitRelative = value
        self.maxPower = int(round(self.limitAbsolute/self.limitRelative*100,-2))
```

So in the case that the DTU has not yet updated the absolute value (at the same time or before the relative value) this could lead to a wrong calculation. though this is really surprising, those value should instantly change if either one changes.

Maybe it's good enough to determine the maxPower only once on startup and then never again....

tuxianerDE, 2024-02-25:
So dalla here we go. 

[image]

That is a valid thought you have there, determining this once during startup would be sufficient as the inverter cannot change during run.

tuxianerDE, 2024-02-25:
[image]
Absolute steady

[image]

tuxianerDE, 2024-03-03:
I agree. Determining maxPower once at startup would be sufficient, given that you cannot add a new inverter at runtime and that the serial is different.

tuxianerDE, 2024-03-04:
Observed it again today evening:

Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,427:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0 ], B: 15% (15|15|15|15), V:47.8V (47.7|47.6|47.9|47.9), C:-824W, P:False, F:24.1h, E:77.9h, H:792W, L:826W#033[0m
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,427:INFO: #033[33;20mINV: AC:764.1W, DC:806.6W (1.7|1.6|411.7|391.6), L:1500.0W [2700W]#033[0m
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,428:INFO: #033[33;32mSMT: T:Smartmeter P:1543.0W [ 1000.0 ]#033[0m
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,428:INFO: Direct connected panels can't cover demand 3.3W/1764.1W, trying to get rest from hub.
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,428:INFO: Checking if Solarflow is willing to contribute 1760.8W ...
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,429:INFO: Based on time, solarpower ( 0.0W) minimum charge power (500W) and bypass state (False), hub could contribute 850.0W - Decision path: 2.1.
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,430:INFO: Setting solarflow output limit to 850.0W
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,431:INFO: Solarflow is willing to contribute 850.0W!
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,431:INFO: Adjusting inverter limit from 1.7W to 9.2W
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,431:INFO: Direct connected panel limit is 9W.
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,432:INFO: Setting inverter output limit to 3228W (1 min moving average of 840W x 4)
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,433:INFO: Sun: 07:08 - 18:17 Demand: 1764.1W, Panel DC: (1.7|1.6), Hub DC: (411.7|391.6), Inverter Limit: 3228.0W, Hub Limit: 850.0W
Mar  4 18:00:45 babylon solarFlow[11127]: 2024-03-04 18:00:45,090:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:46 babylon solarFlow[11127]: 2024-03-04 18:00:46,287:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:47 babylon solarFlow[11127]: 2024-03-04 18:00:47,409:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:48 babylon solarFlow[11127]: 2024-03-04 18:00:48,587:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:49 babylon solarFlow[11127]: 2024-03-04 18:00:49,743:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:50 babylon solarFlow[11127]: 2024-03-04 18:00:50,903:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:52 babylon solarFlow[11127]: 2024-03-04 18:00:52,080:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:53 babylon solarFlow[11127]: 2024-03-04 18:00:53,241:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:54 babylon solarFlow[11127]: 2024-03-04 18:00:54,407:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:55 babylon solarFlow[11127]: 2024-03-04 18:00:55,579:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:56 babylon solarFlow[11127]: 2024-03-04 18:00:56,737:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:57 babylon solarFlow[11127]: 2024-03-04 18:00:57,920:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,083:INFO: Rapid rise in demand detected, clearing buffer!
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,449:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0 ], B: 15% (15|15|15|15), V:47.8V (47.7|47.6|47.9|47.9), C:-834W, P:False, F:24.1h, E:77.9h, H:803W, L:850W#033[0m
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,449:INFO: #033[33;20mINV: AC:763.3W, DC:803.0W (1.7|1.6|409.9|389.8), L:3228.0W [700W]#033[0m
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,450:INFO: #033[33;32mSMT: T:Smartmeter P:1538.0W [ 1000.0 ]#033[0m
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,450:INFO: Direct connected panels can't cover demand 3.3W/1763.3W, trying to get rest from hub.
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,450:INFO: Checking if Solarflow is willing to contribute 1760.0W ...
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,451:INFO: Based on time, solarpower ( 0.0W) minimum charge power (500W) and bypass state (False), hub could contribute 850.0W - Decision path: 2.1.
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,452:INFO: Not setting solarflow output limit to 850.0W as it is identical to current limit!
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,452:INFO: Solarflow is willing to contribute 850.0W!
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,452:INFO: Adjusting inverter limit from 1.7W to 9.2W
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,453:INFO: Direct connected panel limit is 9W.
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,453:INFO: Setting inverter output limit to 1050.0W (1 min moving average of 840W x 4)
Mar  4 18:00:59 babylon solarFlow[11127]: 2024-03-04 18:00:59,455:INFO: Sun: 07:08 - 18:17 Demand: 1763.3W, Panel DC: (1.7|1.6), Hub DC: (409.9|389.8), Inverter Limit: 1050.0W, Hub Limit: 850.0W
Mar  4 18:00:44 babylon solarFlow[11127]: 2024-03-04 18:00:44,427:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0 ], B: 15% (15|15|15|15), V:47.8V (47.7|47.6|47.9|47.9), C:-824W, P:False, F:24.1h, E:77.9h, H:792W, L:826W#033[0m
[... cut ...]
Mar  4 18:02:44 babylon solarFlow[11127]: 2024-03-04 18:02:44,594:INFO: Adjusting inverter limit from 1.6W to 9.0W
Mar  4 18:02:44 babylon solarFlow[11127]: 2024-03-04 18:02:44,595:INFO: Direct connected panel limit is 8W.
Mar  4 18:02:44 babylon solarFlow[11127]: 2024-03-04 18:02:44,595:INFO: Setting inverter output limit to 3360W (1 min moving average of 840W x 4)
Mar  4 18:02:44 babylon solarFlow[11127]: 2024-03-04 18:02:44,596:INFO: Sun: 07:08 - 18:17 Demand: 1148.3W, Panel DC: (1.6|1.5), Hub DC: (80.5|76.7), Inverter Limit: 3360.0W, Hub Limit: 850.0W

reinhard-brandstaedter, 2024-03-07:
This should be fixed on the dev branch, now the inverter capacity is determined only one at startup by using 5 samples of absolute and relative limits.

tuxianerDE, 2024-03-08:
Issue resolved. oversteering did not occur based on the monitoring results.