reinhard-brandstaedter/solarflow-control #131: MIN_CHARGE_POWER set to 0 is not working

mavo, 2024-02-21:
Hi,

when setting MIN_CHARGE_POWER to 0 it will jump back to the default of 125. This is a bit unclear to me why it happens, but when you set it either to 1 or -1 (or seems like any other senseable int value), it works.

Following my config.ini and the log output of the config on startup when set to 0:
````
[config.ini]
min_charge_power = 0

==============
[log output]
2024-02-21 21:52:47,783:INFO: Limit via inverter: True
2024-02-21 21:52:47,783:INFO: Control Parameters:
2024-02-21 21:52:47,784:INFO:   MIN_CHARGE_POWER = 125
2024-02-21 21:52:47,784:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-02-21 21:52:47,784:INFO:   MAX_INVERTER_LIMIT = 800
2024-02-21 21:52:47,784:INFO:   MAX_INVERTER_INPUT = 675
````

And with any other value (-1 here):
````
[config.ini]
min_charge_power = -1

==============
[log output]
2024-02-21 21:57:15,154:INFO: Limit via inverter: True
2024-02-21 21:57:15,154:INFO: Control Parameters:
2024-02-21 21:57:15,154:INFO:   MIN_CHARGE_POWER = -1
2024-02-21 21:57:15,154:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-02-21 21:57:15,154:INFO:   MAX_INVERTER_LIMIT = 800
2024-02-21 21:57:15,154:INFO:   MAX_INVERTER_INPUT = 801
````

reinhard-brandstaedter, 2024-02-22:
Setting it to 1 is the most feasible solution ATM, SF will most likely not charge at this level anyway....

mavo, 2024-02-23:
Using -1 now, dunno why. Just want to make sure it does not try to not feed the house 🙈

closed 2024-02-23 by mavo