reinhard-brandstaedter/solarflow-control #134: ISSUE: Batteries empty during night; wake up power output hub error

tuxianerDE, 2024-02-25:
Have made the following observation not repeadetly. 

Situation:

Batteries are discharging overnight; They are actually empty before dawn.
When the Hub wakes up in the morning with first sufficient light. It discharges some residual energy it found somewhere in the batteries for a few minutes.
When then the sun actually comes around the hub does not start charging again. I have to manually restart the hub.

Assumption: 
The hub due to the short burst of energy goes into overheat warning (or some other warning); and that only gets resolved either with the next day (loosing the entire day) OR with the restart like I do it.

To be honest this needs to be fixed in the FW however following thought:

If the system is not in ChargeThrough based on the lastFull timestamp.

Might be worth to set the hub output to 0W (keep it there) when it has reached 0% (or becomes) unreachable and keep it at this level until dawn+offset is reached so that enough solar energy is coming and we dont see another burst.

Not sure if this could be done via the chargeThrough mechanism in a temporary manner? In the attached screenshot one can see the situation. When I restarted (at the end) it does start taking solar energy directly and start charging the batteries.
[image]

reinhard-brandstaedter, 2024-02-25:
I have had this multiple times (going to zero at noght/evening, finding some energy left in the morning), but it always started normally and charging  throughout the day with no issues. I haven’t restarted the hub in months…

tuxianerDE, 2024-02-25:
> [Verfolgungsbild][https://github.com/notifications/beacon/ALMQT6FK3QGPQNH7ETLHVXTYVMGIPA5CNFSM6AAAAABDYWXZKGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTU75AES.gif]

Mmmh interesting

tuxianerDE, 2024-03-02:
Following test setup has been prepared:

full_charge_interval has been set to 36hrs. Based on the last time fullcharge timestamp I generated that would mean around 5AMish in the morning of the 2nd day.

Expected/hopped behaviour:

- SF will discharge batteries on the second day and they will - hopefully - reach 0% somewhere during the night of the second day
- When SF wakes up in the morning with first light. SF-Control sets the output of SF to 0W due to the chargeThrough window being active
- SF will start but batteries will not discharge the residual energy they magically find in some cells.
- SF will keep on running, batteries will not go into undervoltage and will start charging once enough sun is there. i.e.

tuxianerDE, 2024-03-05:
First test has been successful.

Hub has been 0% at around 22.00 last night. Today morning iwth first light, Hub woke up, no energy discharged as chargeThrough has become active. 

Will plan another test towards the end of the week to revalidate as we currently have very low light conditions.

tuxianerDE, 2024-03-10:
Seems to be alright, before closing @reinhard-brandstaedter  is the expected behaviour that a limit 0 with SoC 0 is actually gradually introduced as seen below in the logs or is it supposed to be a hhard cut?

`Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,233:INFO: Battery is empty: 0
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,234:INFO: Battery is empty: 0
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,236:INFO: Reading last empty time: 2024-03-10 06:51:46
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,237:INFO: Reading battery target mode: charging
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,277:INFO: Reading last empty time: 2024-03-10 06:51:46
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,277:INFO: Reading battery target mode: charging
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,973:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0 ], B: 0% ( 0| 0| 0| 0), V:42.0V (42.0|41.8|42.0|42.0), C: -36W, P:False, F:157.0h, E:0.0h, H: 35W, L: 22W#033[0m
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,974:INFO: #033[33;20mINV: AC:51.4W, DC:37.4W (0.9|0.8|18.3|17.4), L:108.0W [1500.0W]#033[0m
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,974:INFO: #033[33;32mSMT: T:Smartmeter P:78.0W [ 37.0,51.0,61.0,63.0,66.0,67.0,72.0,72.0,72.0,69.0,69.0,69.0,67.0,75.0,75.0,75.0,77.0,78.0 ]#033[0m
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,975:INFO: Direct connected panels can't cover demand 1.7W/124.8W, trying to get rest from hub.
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,975:INFO: Checking if Solarflow is willing to contribute 123.1W ...
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,976:INFO: Based on time, solarpower ( 0.0W) minimum charge power (500W) and bypass state (False), hub could contribute 123.1W - Decision path: 2.1.
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,976:INFO: Setting solarflow output limit to 11.0W
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,977:INFO: Solarflow is willing to contribute 11.0W!
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,977:INFO: Adjusting inverter limit from 0.9W to 7.2W
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,977:INFO: Direct connected panel limit is 7W.
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,978:INFO: Setting inverter output limit to 68W (1 min moving average of 10W x 4)
Mar 10 06:51:46 babylon solarFlow[11127]: 2024-03-10 06:51:46,979:INFO: Sun: 06:55 - 18:27 Demand: 124.8W, Panel DC: (0.9|0.8), Hub DC: (18.3|17.4), Inverter Limit: 68.0W, Hub Limit: 11.0W
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,979:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0,0.0,0.0 ], B: 0% ( 0| 0| 0| 0), V:42.0V (42.0|41.8|42.0|42.0), C: -11W, P:False, F:157.0h, E:0.0h, H: 11W, L: 11W#033[0m
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,980:INFO: #033[33;20mINV: AC:28.5W, DC:10.2W (0.7|0.7|4.5|4.3), L:67.5W [1500.0W]#033[0m
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,980:INFO: #033[33;32mSMT: T:Smartmeter P:113.0W [ 67.0,72.0,72.0,72.0,69.0,69.0,69.0,67.0,75.0,75.0,75.0,77.0,78.0,78.0,78.0,113.0,113.0,113.0 ]#033[0m
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,980:INFO: Direct connected panels can't cover demand 1.4W/119.5W, trying to get rest from hub.
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,980:INFO: Checking if Solarflow is willing to contribute 118.1W ...
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,981:INFO: Based on time, solarpower ( 0.0W) minimum charge power (500W) and bypass state (False), hub could contribute 118.1W - Decision path: 2.1.
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,982:INFO: Setting solarflow output limit to 3.0W
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,982:INFO: Solarflow is willing to contribute 3.0W!
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,983:INFO: Adjusting inverter limit from 0.7W to 6.7W
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,983:INFO: Direct connected panel limit is 6W.
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,983:INFO: Setting inverter output limit to 48W (1 min moving average of 10W x 4)
Mar 10 06:52:01 babylon solarFlow[11127]: 2024-03-10 06:52:01,985:INFO: Sun: 06:55 - 18:27 Demand: 119.5W, Panel DC: (0.7|0.7), Hub DC: (4.5|4.3), Inverter Limit: 48.0W, Hub Limit: 3.0W
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,985:INFO: #033[31;20mHUB: S:0.0W [ 0.0,0.0,0.0,0.0,0.0,0.0 ], B: 0% ( 0| 0| 0| 0), V:42.0V (42.0|41.8|42.0|42.0), C: 0W, P:False, F:157.0h, E:0.0h, H: 0W, L: 3W#033[0m
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,985:INFO: #033[33;20mINV: AC:28.5W, DC:10.2W (0.7|0.7|4.5|4.3), L:48.0W [1500.0W]#033[0m
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,985:INFO: #033[33;32mSMT: T:Smartmeter P:114.0W [ 69.0,69.0,69.0,67.0,75.0,75.0,75.0,77.0,78.0,78.0,78.0,113.0,113.0,113.0,113.0,114.0,114.0,114.0 ]#033[0m
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,986:INFO: Direct connected panels can't cover demand 1.4W/133.2W, trying to get rest from hub.
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,986:INFO: Checking if Solarflow is willing to contribute 131.8W ...
[... cut ...]
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,988:INFO: Adjusting inverter limit from 0.7W to 6.7W
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,988:INFO: Direct connected panel limit is 6W.
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,989:INFO: Setting inverter output limit to 42W (1 min moving average of 10W x 4)
Mar 10 06:52:16 babylon solarFlow[11127]: 2024-03-10 06:52:16,990:INFO: Sun: 06:55 - 18:27 Demand: 133.2W, Panel DC: (0.7|0.7), Hub DC: (4.5|4.3), Inverter Limit: 42.0W, Hub Limit: 0.0W
`

reinhard-brandstaedter, 2024-03-13:
Hm, yes limits are always smoothened. Maybe this could be changed now generally, but it applies to 0 as well.

tuxianerDE, 2024-03-13:
I would consider this as beautification as the overall solution works to reduce with SoC 0 the limit, so from that perspective issue is resolved.

closed 2024-03-13 by tuxianerDE