reinhard-brandstaedter/solarflow-control #140: BUG(?) - Battery low is not used?!

mavo, 2024-03-01:
Hello @reinhard-brandstaedter,

I still have issues with my batteries. They really struggle to get their calibration right as it seems. I had them loaded fully now 2 times and also discharged at least 2 times, still when they are at 100% the battery level makes sense when being discharged, however when charging they still think 0,8kWh is ~97% (my batteries have ~4kWh of capacity). But this is not the problem for your app.

Today, as my batteries where thinking they are fuller then they actually are, the control script did only charge with 50% due to the wrongly reported values (as it will charge slower when over 95% filled) - which also is nothing I criticise on this app. However, as the batteries drastically drained as the sun was down I saw, that my batteries were reported with only 7% remaining energy and they still were discharged. 
Why's this the case? There is a control parameter which is defined as BATTERY_LOW and is set to 10%. Where is this actually used? I did not find it 🤔 

Can you help me understand this?

MaVo

reinhard-brandstaedter, 2024-03-01:
@mavo 
battery_low is indeed a leftover from the first version and is not used anymore. It stems back to a time where there was no charge through.

The SoC reporting of the hub is indeed sometimes strange. I see the inconsistent drop and rise of SoC values at different speeds (as do others as well).
The only values that are absolutely correct is absolute 0 and 100%. Sometimes it takes hours to get from 97% to 100% and sometimes the SoC drops from 100% to 10% and then stays there for a long time at the same discharge rate.

mavo, 2024-03-01:
Hey @reinhard-brandstaedter,

how's the tool making sure then, that it does not discharge to 0 all the time? Or is this actually what is wanted?
I thought this is rather not nice to the battery, or isn't it?

reinhard-brandstaedter, 2024-03-02:
The behavior of the SoC is rather inconsistent. Sometimes it will jump from 95 to 100 in no time, sometimes it stays at 98 for a long time.
After a few months of observation I’m thinking of removing the special handlung at the top end.

Discharging to 0 isn’t really a problem for the battery, however we also have observed a behavior that the hub doesn’t properly start sometimes. (I haven’t seen it personally but @tuxianerDE has).

mavo, 2024-03-02:
I would be in favor of removing the special handling.
Also, I have not seen the behavior that @tuxianerDE has described in his issue. But I have my system not very long...

Btw, do you have any idea if the hub2000 will work with your app as well? I would be somewhat surprised if it would not, but you never know 🙈

mavo, 2024-03-02:
If you give me access, I would also change the code and open a pr for it :-)

mavo, 2024-03-04:
hello @reinhard-brandstaedter,

today - I saw something strange looking to me. I have 100% battery, the bypass was enabled and still the app decided 50% of passed on power is plenty.
```
2024-03-04 14:26:47,828:INFO: Direct connected panels can't cover demand 0.0W/369.1W, trying to get rest from hub.
2024-03-04 14:26:47,828:INFO: Checking if Solarflow is willing to contribute 369.1W ...
2024-03-04 14:26:47,829:INFO: Based on time, solarpower (180.4W) and minimum charge power (200W), hub could contribute 90.0W - Decision path: 2.2.
2024-03-04 14:26:47,829:INFO: Not setting solarflow output limit to 90.0W as it is identical to current limit!
2024-03-04 14:26:47,829:INFO: Solarflow is willing to contribute 90.0W!
2024-03-04 14:26:47,829:INFO: Raising inverter limit from 0.0W to 5.0W
2024-03-04 14:26:47,829:INFO: Direct connected panel limit is 5W.
2024-03-04 14:26:47,829:INFO: Setting inverter output limit to 320W (1 min moving average of 80W x 4)
2024-03-04 14:26:47,830:INFO: Sun: 06:47 - 17:52 Demand: 369.1W, Panel DC: (0.0|0.0), Hub DC: (95.9|94.2), Inverter Limit: 320.0W, Hub Limit: 90.0W
2024-03-04 14:26:52,835:INFO: HUB: S:181.1W [ 176.0,177.0,178.0,179.0,180.0,181.0,182.0,183.0,183.0 ], B:100% (100|100), C: 0W, P:True, F:0.0h, E:53.3h, H:182W, L: 90W
2024-03-04 14:26:52,836:INFO: INV: AC:179.3W, DC:190.6W (0.0|0.0|96.0|94.6), L:1600.0W
2024-03-04 14:26:52,836:INFO: SMT: T:ShellyEM3 P:187.7W [ 190.1,188.2,190.7,190.7,193.2,190.1,191.5,189.7,189.9,189.7,195.4,195.6,195.8,195.2,197.0,196.8,202.9,201.6,202.0,192.8,192.6,192.4,188.5,189.4,189.4,189.4,189.4,189.2,195.4,195.7,195.8,197.7,197.1,197.1,189.2,190.5,190.2,187.5,187.8,186.8,190.8,189.0,189.4,187.5,189.2,189.7,192.1,193.1,192.9,194.8,194.0,193.9,188.7,187.9,187.7,190.7,190.6,191.2,189.3,193.9,193.5,193.8,189.7,189.5,180.0,177.2,177.2,177.2,177.2,177.2,183.2,185.8,185.6,191.5,188.5,188.5,184.5,184.7 ]
2024-03-04 14:26:52,836:INFO: Direct connected panels can't cover demand 0.0W/367.0W, trying to get rest from hub.
2024-03-04 14:26:52,836:INFO: Checking if Solarflow is willing to contribute 367.0W ...
2024-03-04 14:26:52,836:INFO: Based on time, solarpower (181.1W) and minimum charge power (200W), hub could contribute 90.0W - Decision path: 2.2.
2024-03-04 14:26:52,837:INFO: Not setting solarflow output limit to 90.0W as it is identical to current limit!
2024-03-04 14:26:52,837:INFO: Solarflow is willing to contribute 90.0W!
2024-03-04 14:26:52,837:INFO: Raising inverter limit from 0.0W to 5.0W
2024-03-04 14:26:52,837:INFO: Direct connected panel limit is 5W.
2024-03-04 14:26:52,837:INFO: Setting inverter output limit to 320W (1 min moving average of 80W x 4)
2024-03-04 14:26:52,837:INFO: Sun: 06:47 - 17:52 Demand: 367.0W, Panel DC: (0.0|0.0), Hub DC: (96.0|94.6), Inverter Limit: 320.0W, Hub Limit: 90.0W
2024-03-04 14:26:57,843:INFO: HUB: S:181.3W [ 176.0,177.0,178.0,179.0,180.0,181.0,182.0,183.0,183.0,182.0 ], B:100% (100|100), C: 0W, P:True, F:0.0h, E:53.3h, H:182W, L: 90W
2024-03-04 14:26:57,843:INFO: INV: AC:179.8W, DC:191.0W (0.0|0.0|96.4|94.6), L:1600.0W
2024-03-04 14:26:57,843:INFO: SMT: T:ShellyEM3 P:187.3W [ 191.5,189.7,189.9,189.7,195.4,195.6,195.8,195.2,197.0,196.8,202.9,201.6,202.0,192.8,192.6,192.4,188.5,189.4,189.4,189.4,189.4,189.2,195.4,195.7,195.8,197.7,197.1,197.1,189.2,190.5,190.2,187.5,187.8,186.8,190.8,189.0,189.4,187.5,189.2,189.7,192.1,193.1,192.9,194.8,194.0,193.9,188.7,187.9,187.7,190.7,190.6,191.2,189.3,193.9,193.5,193.8,189.7,189.5,180.0,177.2,177.2,177.2,177.2,177.2,183.2,185.8,185.6,191.5,188.5,188.5,184.5,184.7,184.5,186.4,187.7 ]
2024-03-04 14:26:57,843:INFO: Direct connected panels can't cover demand 0.0W/367.1W, trying to get rest from hub.
2024-03-04 14:26:57,843:INFO: Checking if Solarflow is willing to contribute 367.1W ...
2024-03-04 14:26:57,844:INFO: Based on time, solarpower (181.3W) and minimum charge power (200W), hub could contribute 90.0W - Decision path: 2.2.
2024-03-04 14:26:57,844:INFO: Not setting solarflow output limit to 90.0W as it is identical to current limit!
2024-03-04 14:26:57,844:INFO: Solarflow is willing to contribute 90.0W!
2024-03-04 14:26:57,844:INFO: Raising inverter limit from 0.0W to 5.0W
2024-03-04 14:26:57,844:INFO: Direct connected panel limit is 5W.
2024-03-04 14:26:57,844:INFO: Setting inverter output limit to 320W (1 min moving average of 80W x 4)
2024-03-04 14:26:57,845:INFO: Sun: 06:47 - 17:52 Demand: 367.1W, Panel DC: (0.0|0.0), Hub DC: (96.4|94.6), Inverter Limit: 320.0W, Hub Limit: 90.0W
2024-03-04 14:27:02,849:INFO: HUB: S:183.1W [ 178.0,179.0,180.0,181.0,182.0,183.0,183.0,182.0,185.0,186.0 ], B:100% (100|100), C: 0W, P:True, F:0.0h, E:53.3h, H:183W, L: 90W
2024-03-04 14:27:02,850:INFO: INV: AC:180.2W, DC:189.7W (0.0|0.0|95.9|93.8), L:1600.0W
2024-03-04 14:27:02,850:INFO: SMT: T:ShellyEM3 P:187.3W [ 191.5,189.7,189.9,189.7,195.4,195.6,195.8,195.2,197.0,196.8,202.9,201.6,202.0,192.8,192.6,192.4,188.5,189.4,189.4,189.4,189.4,189.2,195.4,195.7,195.8,197.7,197.1,197.1,189.2,190.5,190.2,187.5,187.8,186.8,190.8,189.0,189.4,187.5,189.2,189.7,192.1,193.1,192.9,194.8,194.0,193.9,188.7,187.9,187.7,190.7,190.6,191.2,189.3,193.9,193.5,193.8,189.7,189.5,180.0,177.2,177.2,177.2,177.2,177.2,183.2,185.8,185.6,191.5,188.5,188.5,184.5,184.7,184.5,186.4,187.7 ]
```
[image]

Is this something I do not understand fully, or is this not desired behavior?

MaVo

mavo, 2024-03-04:
Hey @tuxianerDE,
I am not using the dev build, image was pulled not that long ago, maybe 1-1 1/2 weeks. Image id is `75566f4e166f` (at least in my container runtime).
[image]

reinhard-brandstaedter, 2024-03-05:
Hi @mavo ,
in Bypass mode the hub ignores any limit. So from the logs you can see that the Hub still puts out all Solarpower to home.
Though I'm surprised a bit by this part:
```
2024-03-04 14:26:57,844:INFO: Setting inverter output limit to 320W (1 min moving average of 80W x 4)
2024-03-04 14:26:57,845:INFO: Sun: 06:47 - 17:52 Demand: 367.1W, Panel DC: (0.0|0.0), Hub DC: (96.4|94.6), Inverter Limit: 320.0W, Hub Limit: 90.0W
2024-03-04 14:27:02,849:INFO: HUB: S:183.1W [ 178.0,179.0,180.0,181.0,182.0,183.0,183.0,182.0,185.0,186.0 ], B:100% (100|100), C: 0W, P:True, F:0.0h, E:53.3h, H:183W, L: 90W
2024-03-04 14:27:02,850:INFO: INV: AC:180.2W, DC:189.7W (0.0|0.0|95.9|93.8), L:1600.0W
```
The inverter limit is set to 1600 although before it tried to set it to 320? Does setting the inverter limit even work?

mavo, 2024-03-05:
Hi @reinhard-brandstaedter,

I have to admit, I never saw any limiting of the inverter the webUI of my ahoydtu. Do you know if the change should be reflected there?

Other question: is there any way I can contribute to this repo?
I'd like to cleanup some old stuff and also propose a change to remove the slow charging towards the end of the battery SoC.
As long as your repo requires pull request reviews there should be no harm to be done. If you want to be the last one to have a look, you can also adjust the repo to only grant yourself approve permissions.

MaVo

mavo, 2024-03-08:
Closing this as you added it in the dev branch. Really appreciate this 💜

closed 2024-03-08 by mavo