reinhard-brandstaedter/solarflow-control #143: Battery full - solar production dropped when no direct connected panel exists

mavo, 2024-03-07:
Hi @reinhard-brandstaedter,

its me again, with another question.
Today, I faced the situation, that I have full sunshine and should get (and got right before the battery was full) around 500W from the solar panels (almost same weather as yesterday, so I can take the values from there). My battery was empty a few days ago, so charge-through was kicking in and did its job. It finished around 12:30 mid day today.
Since then, my solar production did drop severely (one panel seems to be sleeping at all and the other one only produces 50ish watts. My home consumption was around 350W but the SF-output value was only set by your app to 30W as it reported such low solar production - although as mentioned I had full sunshine...
I "fixed" it by connecting one of my 3 panels to the inverter directly again. But its actually a mystery to me what happened.

As I will use the hub2000 when it arrives and I wanted to test how the app works with only hub connected panels, I connected 2 of my current panels by using a Y-adapter into PV-input 1 of the hub and the last panel onto PV-input 2. This is ok as I have 2 Panels which in current season barely exceed 400W together which is no problem for the hub - and it was anyway only a test. I was using this setup for almost a week, so its not a setup-issue I'd say.

Is this a problem of the application that it kinda needs a direct connected panel somehow? 
Do you need any more information from my end to "debug" this?

MaVo

rostchri, 2024-03-07:
I think it is important to create some visualisation (for example) using telegraf/influxdb/grafana. To have a better overview what happened und to compare it with other days. Here are some dashboard-images of my current setup. I have currently 2 PV-Modules (PV-Module 3 & 4) directly attached to my Hoymiles Inverter and 4 PV-Modules attached to the Solarflow-Hub.
I'm not using the Inverter to limit the Energy. I'm only limiting on the Solarflow-Hub (using solarflow-control) by reading Energy-Levels from smart power-sockets (flashed with Tasmota-Firmware) via mqtt.

 The first image is from yesterday, where it was quite cloudy (nearly no sun). Only 6 percent SOC for the complete day (I have currently ~4 kw/h solarflow batteries installed). The battery starts to work (deliver energy) at around 17:00 (green) for 1 hour. The second image is from today (lot better).
 
I'm normally in the situation where the battery is never full (because less sunshine). So the battery will start to deliver when the sun is set. As soon as we will have more sun, and my SOC is reaching 100% before sunset, I will send some newer images. But so far I'm very pleased. The setup is working as expected for nearly 3 weeks now.

[image]
[image]

mavo, 2024-03-07:
Hello, thanks for the replies.

> Did the ByPass turn on on the hub so that everything gets fed to the inverter?

yes, bypass was on (checked by connecting to the hub via BT for a minute)

> log output

Is this somewhere stored? I did not find it, other than the log output of the container which does not go back that much into the past, sadly.

> config.ini

```
# Offset in minutes after sunrise/before sunset. Can be used to set the duration of what is considered "night"
sunrise_offset = 60
sunset_offset = 60

# The time interval in hours that solarflow-control will try to ensure a full battery
full_charge_interval = 36

[ahoydtu]
base_topic = ahoydtu
inverter_id = 1
sf_inverter_channels = [3,4]

# the max output power of your inverter, used to calculate correct absolute values
inverter_max_power = 1600

# The name of the inverter in AhoyDTU
inverter_name = HM-1600

[shellyem3]
# The MQTT base topic your Shelly 3EM (Pro) is posting it's telemetry data to
# Note: you have to configure your Shelly to use MQTT
base_topic = shellies/shellyem3

[control]
min_charge_power = 200
max_discharge_power = 800
max_inverter_limit = 800
limit_inverter = true
inverter_min_limit = 5
```

@rostchri 
Here are 2 images, best I can create on the fly using HA's history graphs.

From yesterday:
[image]
From today:
[image]

Adding almost the same images, just without power consumption added to have a better scale:
Yesterday:
[image]
Today:
[image]

reinhard-brandstaedter, 2024-03-07:
Hi @mavo ,

this seems correct. I think what happens here is that the hub enters/leaves the bypass mode multiple times. Yes this happens. Once it is in Bypass mode the hub-home-output follows your home consumption (purple and light blue line are almost identical)

[image]

With "docker logs <container name>" you should be able to get all logs, unless your docker runtime is configured to only keep a very low amount of logs (usually a few megabytes should be fine?)

mavo, 2024-03-07:
Hi @reinhard-brandstaedter,

in the graph, from 14:09 I again had the setup of 2 panels to the hub and 1 to the inverter. From then onwards the solar production (blue line) came back to expected levels.

The time between 12:30 and 14:00 is the frame where its not that easy to explain. In this time, there was no cloud and full sunshine (as it mostly was the full day) and you can see that when the hub-output-limit started to go up the solar production went down. It had some spikes when I restarted the hub as I was thinking the hub must be in a bugged state. This helped temporarily as the production came back to expected levels shortly only to then drop again :-/

> With "docker logs " you should be able to get all logs, unless your docker runtime is configured to only keep a very low amount of logs (usually a few megabytes should be fine?)

I'll check if the command as available, as I am running the app on my NAS in a provided app and there is no native docker available when SSH-ing into the system.

mavo, 2024-03-07:
As I do not know how the hub should (and does) behave when you set a output limit very low but there is more production available, it looks like it does limit the panel input to what its being told to deliver. 
With this as presumption it would explain the behaviour. Output limit is set to 50% of the solar input (as battery level is > 95% but bypass is not enabled for whatever reason), the hub reduces what it accepts from the panels resulting in a lower input on next evaluation of the values in the app. This then results in the app setting an even lower limit of home output, causing the hub to scale down even more until 30W minimum is reached.

Does this makes sense to you @reinhard-brandstaedter / (and maybe @rostchri)?

mavo, 2024-03-07:
Well... turns out I am an idiot 🙈 

My compose file was not pointing to latest tag of the image, thus my container decided to stay on the older version that I pulled beginning of feb. Now, after pointing it to latest it updated the files and now seems to use the newer code. 
I did check the old container and it was indeed running older versions of the app, which for example did not had the fix for the newer firmware, thus setting a limit very low when bypass is enabled.
So I think mystery solved, I am the one to blame here 🤷‍♂️ 

Sorry for wasting your time, I thought the freaking app would use the newer image when I pull it via the app explicitly... But yeah.

tuxianerDE, 2024-03-08:
No biggy :) good that it got resolved/figured out. Can you then close the issue (you can always open if need be)

closed 2024-03-08 by mavo