reinhard-brandstaedter/solarflow-control #145: Rapid change on shellyem3

mavo, 2024-03-08:
New day, new issue.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/solarflow/solarflow-control.py", line 133, in on_message
    smartmeter.handleMsg(msg)
  File "/solarflow/smartmeters.py", line 64, in handleMsg
    self.updPower()
  File "/solarflow/smartmeters.py", line 47, in updPower
    if diff > self.rapid_change_diff:
AttributeError: 'ShellyEM3' object has no attribute 'rapid_change_diff'
```

Happened after updating to the feb 20 build of the app 🤔

rostchri, 2024-03-08:
Same error applies to all: PowerOpti, VZLogger. Except Smartmeter (this is the only one which is working).

reinhard-brandstaedter, 2024-03-08:
Yes, the last master build is broken :-(
rapid_change was only introduced for generic smartmeter.

Can I encourage you to try the dev build?

mavo, 2024-03-08:
Sure. But only when I am back from vacation. I fixed it by changing the file in the docker image directly...
When I am back I'll check the dev build :-)

rostchri, 2024-03-08:
For me, it is no real problem. I easily fixed this by myself a while ago and sure I will switch to your dev build soon.

reinhard-brandstaedter, 2024-03-09:
Fixed on dev

closed 2024-03-09 by reinhard-brandstaedter

mavo, 2024-03-15:
Hello @reinhard-brandstaedter,
I am sorry, but it's not fixed I think.
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/configparser.py", line 805, in get
    value = d[option]
            ~^^^^^^^^
  File "/usr/local/lib/python3.11/collections/__init__.py", line 1006, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/collections/__init__.py", line 998, in __missing__
    raise KeyError(key)
KeyError: 'rapid_change_diff'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 437, in <module>
    main(sys.argv[1:])
  File "/solarflow/solarflow-control.py", line 434, in main
    run()
  File "/solarflow/solarflow-control.py", line 358, in run
    smt_opts = getOpts(smtType)
               ^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 332, in getOpts
    opts.update({opt:opt_type(converter(configtype.__name__.lower(),opt))})
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/configparser.py", line 834, in getint
    return self._get_conv(section, option, int, raw=raw, vars=vars,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/configparser.py", line 824, in _get_conv
    return self._get(section, conv, option, raw=raw, vars=vars,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/configparser.py", line 819, in _get
    return conv(self.get(section, option, **kwargs))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/configparser.py", line 808, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'rapid_change_diff' in section: 'shellyem3'
```

Happens on latest master and on dev build as well :-(

I tried to disable the rapid-change config in the config.ini, with no sucess.

reinhard-brandstaedter, 2024-03-18:
@mavo did you make sure your config.ini contains this? Just dummy tested it and it works.
```
 [shellyem3]
    base_topic = < your base topic>
    rapid_change_diff = 500
```

mavo, 2024-03-18:
Stupid me, I did not see the change in config.ini required. Its also not in the examples section 😇 
Will test with the config added later this day. Cheers!

mavo, 2024-03-18:
Works. Thx!