reinhard-brandstaedter/solarflow-control #158: [Question]

Dielee, 2024-03-16:
First, thanks for this integration!

Is it possible to use this addon with a deye inverter? They expose the data via TCP modbus. Maybe I can help you with the development of this use case ?

reinhard-brandstaedter, 2024-03-16:
Hi @Dielee ,

it is not directly supported but I think it could be achieved rather easy by using a modbus to mqtt bridge like this one:

https://github.com/kbialek/deye-inverter-mqtt

As this also supports limiting the inverter this should work.

reinhard-brandstaedter, 2024-03-16:
I personally wouldn’t dive into modbus decoding as this has been done before (in above project).
This seems advanced and tested enough and as all interaction with inverters is already done via MQTT this is probably the easiest way without the need to go through a lot of testing/errors.
Once the modbus to mqtt works its just a matter of a few lines of code to make it work.

Dielee, 2024-03-17:
Alright, maybe a solution with da docker compose file would be nice. Will will take a look into the code and the mqtt addon and submit a PR if you want.

reinhard-brandstaedter, 2024-03-18:
Sure thing!
I browsed the Deye/Mqtt documentation a bit. As it only allows setting the limit in % of the maximum, what is likely required in a "deye" DTU class is a transformation between absolute and relative limits, which shouldn't be too difficult since the max capacity of the Deye is known as well. Then the sf-control would be able to continue as usual without much extra handling.
I guess this could be really helpful as quite some people have Deye (or equivalent) inverters.