reinhard-brandstaedter/solarflow-control #160: DEV: Battery got set to 0 at night switched off inverter

tuxianerDE, 2024-03-16:
I observed that during a night run, and a drop in load (dryer finished). The skript determined the new limit.

It set the HUB to 0 and the INV to 40W. Problem is this switched off the inverter. Ideally if such a situation occurs (reason is why because there is still "ground" load). The minimum on battery should be 40W. This should only be the case if it is not day or during the offset times, thus only during night time.

[solarFlow-stalled.txt](https://github.com/reinhard-brandstaedter/solarflow-control/files/14622807/solarFlow-stalled.txt)

reinhard-brandstaedter, 2024-03-16:
That is indeed a interesting case. A short drop in demand to below zero of course led to the decision to stop the hub from contributing. Thus the decision to provide 0 contribution at night will stop the inverter (no power input to inverter).

Though in your case 10s later the demand was back (rapid rise even) which immediately triggered output from battery again and turned on the inverter as well and normal operation continued?

tuxianerDE, 2024-03-16:
I observed the inverter stall - was out of town - connected via VPN and quickly switched to stable to keep the night running. 

When I checked the OpenDTU UI the system was stalling, so all I saw was the WR/Hub did not pick it up directly

tuxianerDE, 2024-03-25:
Issue was not observed anymore

closed 2024-03-25 by tuxianerDE