reinhard-brandstaedter/solarflow-control #164: INV: Did not increase direct panels though solar radition increased drastically.

tuxianerDE, 2024-03-17:
I observed that today afternoon the sun - finally - came out behind the cloud. The inverter in this case did not increase the limit to better match the new direct irridation.

I manually entered in openDTU the limit of 100%, which directly increased the output of the INV accordingly. Afterwards the script properly picked the new limit out and steered it - supposedly correctly.

I have attached a graph showing that it is was not due to any interval limitation as the increased power was present long enough to get detected. Further I attached a screenshot from my DTU showing that the limit was maxed out by the output and yet the script did not trigger any acction up until I manually intervened.

[image]
[solarflow-control-directpanel_increase.log](https://github.com/reinhard-brandstaedter/solarflow-control/files/14627940/solarflow-control-directpanel_increase.log)

tuxianerDE, 2024-03-17:
[image]

tuxianerDE, 2024-03-17:
After manually changing the limit to 100%
[image]

tuxianerDE, 2024-03-17:
Is it possible that the diff limit of 30W is too large and hence the INV did not trigger?

reinhard-brandstaedter, 2024-03-17:
That was my thought too....

tuxianerDE, 2024-04-11:
Issue resolved. Given the limit function & fixed interval in case no triggered received as a health check

closed 2024-04-11 by tuxianerDE