reinhard-brandstaedter/solarflow-control #165: BUG: when limiting SF via inverter only it can happen that the limit/output is fluctuating

reinhard-brandstaedter, 2024-03-17:
This scenario happened today when there was constant load in the house during late sunset. The limiting started to fluctuate heavily between invverter and solarflow limit. Resulting in a rather harmonic swing.

[image]

Upon investigation I suspect that this has to do with the very low (but still positive DC values of direct panels). Once the DTU reported absolute 0 for DC panels this went back to normal by itself.

tuxianerDE, 2024-03-20:
Adding the current state from 20MAR24 test.
[image]
[solarFlow.zip](https://github.com/reinhard-brandstaedter/solarflow-control/files/14664964/solarFlow.zip)

tuxianerDE, 2024-03-26:
Update provided from 26.03.2024
In the morning when the offset kicked it and first input came from the direct panels I observed the behaviour again. Given that I require a long offset time in the morning it goes on for some time.

[image]
[solarFlow-26MAR24-zickZack.zip](https://github.com/reinhard-brandstaedter/solarflow-control/files/14758189/solarFlow-26MAR24-zickZack.zip)

closed 2024-04-01 by reinhard-brandstaedter