reinhard-brandstaedter/solarflow-control #167: Inverter limit not set apparently

mavo, 2024-03-18:
Hey @reinhard-brandstaedter, 

I am looking at the logs once again, seeing something odd.
You kinda saw it in one of my other issues already, but now it seems to be a "problem" as it appears to ignore the limit for my setup.

```
2024-03-18 15:11:15,994:INFO: Direct connected panels can't cover demand 193.9W/277.7W, trying to get rest from hub.
2024-03-18 15:11:15,994:INFO: Checking if Solarflow is willing to contribute 83.8W ...
2024-03-18 15:11:15,995:INFO: Based on time, solarpower (312.2W) minimum charge power (200W) and bypass state (False), hub could contribute 83.8W - Decision path: 1.1.
2024-03-18 15:11:15,995:INFO: Setting solarflow output limit to 81.0W
2024-03-18 15:11:15,995:INFO: Solarflow is willing to contribute 81.0W!
2024-03-18 15:11:15,995:INFO: Adjusting inverter limit from 155.5W to 261.1W
2024-03-18 15:11:15,995:INFO: Direct connected panel limit is 261W.
2024-03-18 15:11:15,995:INFO: Setting inverter output limit to 10W (1 min moving average of 261W x 4)
2024-03-18 15:11:15,995:INFO: Sun: 06:14 - 18:18 Demand: 277.7W, Panel DC: (38.4|155.5), Hub DC: (27.7|27.4), Inverter Limit: 10.0W, Hub Limit: 81.0W
2024-03-18 15:11:16,618:INFO: DTU triggers limit function: 214.4 : 220.4
2024-03-18 15:11:16,618:INFO: HUB: S:312.2W [ 291.0,301.0,303.0,304.0,306.0,308.0,309.0,310.0,311.0,312.0,313.0,314.0,315.0,316.0,315.0,315.0 ], B: 99% (99|99), V:50.5V (50.5|50.5), C: 288W, P:False, F:118.8h, E:8.7h, H: 11W, L: 80W
2024-03-18 15:11:16,622:INFO: INV: AC:236.6W, AC_Prediction: 203.1W, DC:249.0W, DC_prediction: 214.4W (38.4|155.5|27.7|27.4), L: 0W [ 0W]
```
The weird line is actually this one `2024-03-18 15:11:15,995:INFO: Setting inverter output limit to 10W (1 min moving average of 261W x 4)`. It, for some reason, falls back to it lower limit of 10 which I do not really understand.

I am using an AhoyDTU. I cannot find any indication in the UI of the DTU, that there is any limit set after all.
[image]
It actually should not have a max AC-power of 1k, if its using the limits set by law...?

Thanks again,
MaVo

mavo, 2024-03-18:
I did some log.info debugging in the container. For some reason, self.maxPower is always 0 for me.
This explains at least the strange 10W limit it tries to set.
```
2024-03-18 15:45:35,609:INFO: DTU triggers limit function: 172.8 : 177.8
2024-03-18 15:45:35,609:INFO: HUB: S:247.7W [ 249.0,248.0,249.0,248.0,247.0,247.0 ], B: 99% (99|99), V:50.5V (50.5|50.4), C: 219W, P:False, F:119.4h, E:9.3h, H: 16W, L: 60W
2024-03-18 15:45:35,613:INFO: INV: AC:152.7W, AC_Prediction: 162.1W, DC:160.7W, DC_prediction: 172.8W (28.5|126.6|2.6|2.9), L: 0W [ 0W]
2024-03-18 15:45:35,615:INFO: SMT: T:ShellyEM3 P:151.6W [ 130.8,131.2,153.0,153.1,168.7,121.0,121.1,115.2,130.1,130.1,116.1,102.7,102.1,101.7,148.4,148.6,158.8,160.6,160.6,156.1,135.0,135.1,128.4,112.8,112.4,113.9,151.2,151.1,150.8,107.1,107.1,112.7,109.9,110.2,105.3,124.0,124.0,122.7,152.9,152.9,153.3,152.2,151.9,151.8,104.4,104.0,107.8,150.2,150.4,153.9,115.0,114.6,107.2,151.6 ] Predict: 133.0W
2024-03-18 15:45:35,618:INFO: Direct connected panels can't cover demand 155.1W/295.1W, trying to get rest from hub.
2024-03-18 15:45:35,618:INFO: Checking if Solarflow is willing to contribute 140.0W ...
2024-03-18 15:45:35,618:INFO: Based on time, solarpower (247.7W) minimum charge power (200W) and bypass state (False), hub could contribute 47.7W - Decision path: 2.2.
2024-03-18 15:45:35,619:INFO: Not setting solarflow output limit to 60.0W as it is identical to current limit!
2024-03-18 15:45:35,619:INFO: Solarflow is willing to contribute 60.0W!
2024-03-18 15:45:35,619:INFO: Adjusting inverter limit from 126.6W to 220.7W
2024-03-18 15:45:35,619:INFO: Direct connected panel limit is 220W.
2024-03-18 15:45:35,619:INFO: Limit to set 220
2024-03-18 15:45:35,619:INFO: Limit to set parsed 220
2024-03-18 15:45:35,619:INFO: Limit absolute buffer 878.7142857142857
2024-03-18 15:45:35,619:INFO: New limit even value is 880
2024-03-18 15:45:35,619:INFO: self.maxPower 0
2024-03-18 15:45:35,619:INFO: New limit to set 0.0
2024-03-18 15:45:35,619:INFO: Setting inverter output limit to 10W (1 min moving average of 220W x 4)
2024-03-18 15:45:35,620:INFO: Sun: 06:14 - 18:18 Demand: 295.1W, Panel DC: (28.5|126.6), Hub DC: (2.6|2.9), Inverter Limit: 10.0W, Hub Limit: 60.0W
```

> Do you have a full log of the history up to this point?

How much log do you want to see? 

> Maybe also a history of the actual inverter limit reported by Ahoy?

Not really, I only use it with its UI, and its not per default reported to home assistant where I get my history from usually.

> Have you tried setting the limit via MQTT manually? What happens, is it applied?

No, and to be honest, I would not even know how to do it in this instance.

> To be sure the limit topic in the config is correct for the mqtt topic that Ahoy expects?

I don't understand. What Information can I provide to you to determine this - or can you guide me to find what we need?

reinhard-brandstaedter, 2024-03-18:
A log from the startup on would be helpful. As you identified that the maxpower is not detected correctly and neither any limit is set, we need to check the mqtt topic paths vs. what is actually in mqtt explorer.

reinhard-brandstaedter, 2024-03-18:
I found a possible reason for the behavior. Since the power autodetection for DTUs was introduced, we also need to update the relative limit on Ahoy. The inverter_max_power will become obsolete then.

mavo, 2024-03-18:
Here's the startup log:
```
2024-03-18 19:14:15,251:INFO: Control Parameters:
2024-03-18 19:14:15,251:INFO:   MIN_CHARGE_POWER = 200
2024-03-18 19:14:15,251:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-03-18 19:14:15,251:INFO:   MAX_INVERTER_LIMIT = 800
2024-03-18 19:14:15,251:INFO:   MAX_INVERTER_INPUT = 600
2024-03-18 19:14:15,251:INFO:   SUNRISE_OFFSET = 60
2024-03-18 19:14:15,251:INFO:   SUNSET_OFFSET = 60
2024-03-18 19:14:15,441:INFO: Using AhoyDTU: Base topic: ahoydtu, Limit topic: ahoydtu/ctrl/limit/1, SF Channels: [3, 4]
2024-03-18 19:14:15,441:INFO: Using ShellyEM3: Base topic: shellies/shellyem3
2024-03-18 19:14:15,441:INFO: Connected to MQTT Broker!
2024-03-18 19:14:15,441:INFO: Hub subscribing: /73bkTV/vSU06042/properties/report
2024-03-18 19:14:15,441:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/solarInputPower
2024-03-18 19:14:15,441:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/electricLevel
2024-03-18 19:14:15,441:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputPackPower
2024-03-18 19:14:15,441:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/packInputPower
2024-03-18 19:14:15,441:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputHomePower
2024-03-18 19:14:15,442:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputLimit
2024-03-18 19:14:15,442:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/masterSoftVersion
2024-03-18 19:14:15,442:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/pass
2024-03-18 19:14:15,442:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/batteries/+/socLevel
2024-03-18 19:14:15,442:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/batteries/+/totalVol
2024-03-18 19:14:15,442:INFO: Hub subscribing: solarflow-hub/vSU06042/control/#
2024-03-18 19:14:15,443:INFO: DTU subscribing: ahoydtu/HM-1600/+/P_DC
2024-03-18 19:14:15,443:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/P_AC
2024-03-18 19:14:15,443:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/active_PowerLimit
2024-03-18 19:14:15,443:INFO: DTU subscribing: ahoydtu/status
2024-03-18 19:14:15,443:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-03-18 19:14:15,443:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/0/power
2024-03-18 19:14:15,444:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/1/power
2024-03-18 19:14:15,444:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/2/power
```

Topic for power limit looks actually good to me.

mavo, 2024-03-18:
This is funny... After connecting via a different device, and running the mqtt explorer for a couple of minutes, I do not see the topic for ch0/active_PowerLimit anymore. However, I found this (red box)
[image]

reinhard-brandstaedter, 2024-03-18:
The HM1600 topic tree seems to be very old. The timestamp for last_success dates back to January 20, 07:28. Likely a leftover from some Ahoy DTU config change? But that is a side topic.

The limit 10 likely stems from no proper max_capacity being used (self.maxPower = 0).

I just commited to dev a change that should address this. Unfortunately I do not have an AhoyDTU to test it.

mavo, 2024-03-18:
Well, tried the dev build, but now it seems to be stuck somehow. It starts, logs some rate limit event logs and then.. nothing anymore...
```
2024-03-18 21:33:16,516:INFO: Limit via inverter: True
2024-03-18 21:33:16,516:INFO: Control Parameters:
2024-03-18 21:33:16,516:INFO:   MIN_CHARGE_POWER = 200
2024-03-18 21:33:16,516:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-03-18 21:33:16,516:INFO:   MAX_INVERTER_LIMIT = 800
2024-03-18 21:33:16,516:INFO:   MAX_INVERTER_INPUT = 600
2024-03-18 21:33:16,516:INFO:   SUNRISE_OFFSET = 60
2024-03-18 21:33:16,516:INFO:   SUNSET_OFFSET = 60
2024-03-18 21:33:16,720:INFO: Using AhoyDTU: Base topic: ahoydtu, Limit topic: ahoydtu/ctrl/limit/1, SF Channels: [3, 4]
2024-03-18 21:33:16,720:INFO: Using ShellyEM3: Base topic: shellies/shellyem3
2024-03-18 21:33:16,720:INFO: Connected to MQTT Broker!
2024-03-18 21:33:16,721:INFO: Hub subscribing: /73bkTV/vSU06042/properties/report
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/solarInputPower
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/electricLevel
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputPackPower
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/packInputPower
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputHomePower
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputLimit
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/masterSoftVersion
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/pass
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/batteries/+/socLevel
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/batteries/+/totalVol
2024-03-18 21:33:16,721:INFO: Hub subscribing: solarflow-hub/vSU06042/control/#
2024-03-18 21:33:16,722:INFO: DTU subscribing: ahoydtu/HM-1600/+/P_DC
2024-03-18 21:33:16,722:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/P_AC
2024-03-18 21:33:16,722:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/active_PowerLimit
2024-03-18 21:33:16,722:INFO: DTU subscribing: ahoydtu/status
2024-03-18 21:33:16,722:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-03-18 21:33:16,722:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/0/power
2024-03-18 21:33:16,722:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/1/power
2024-03-18 21:33:16,722:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/2/power
2024-03-18 21:33:16,723:INFO: Reading last empty time: 2024-03-18 06:29:24
2024-03-18 21:33:16,723:INFO: Reading last full time: 2024-03-13 16:23:22
2024-03-18 21:33:18,538:INFO: DTU triggers limit function: 497.3 : 0
2024-03-18 21:33:18,538:INFO: HUB: S:-1.0W [ ], B: -1% (-1), V:-1.0V (-1.0), C:-494W, P:False, F:125.2h, E:15.1h, H:471W, L: -1W
2024-03-18 21:33:18,538:INFO: INV: AC:472.4W, AC_Prediction: 472.4W, DC:497.3W, DC_prediction: 497.3W (0.0), L: 0W [ -1W]
2024-03-18 21:33:18,538:INFO: SMT: T:ShellyEM3 P:0.0W [ ] Predict: 0.0W
2024-03-18 21:33:28,183:INFO: SMT triggers limit function: 75.16 : 0
2024-03-18 21:33:28,183:INFO: Rate limit on trigger function, last call was only 9.644973s ago!
2024-03-18 21:33:28,183:INFO: Rapid rise in demand detected, clearing buffer!
2024-03-18 21:33:28,183:INFO: SMT triggers limit function: 390.24 : 75.16
2024-03-18 21:33:28,183:INFO: Rate limit on trigger function, last call was only 9.645356s ago!
2024-03-18 21:33:28,184:INFO: Rapid drop in demand detected, clearing buffer!
2024-03-18 21:33:28,184:INFO: SMT triggers limit function: 17.150000000000034 : 390.24
2024-03-18 21:33:28,184:INFO: Rate limit on trigger function, last call was only 9.645893s ago!
2024-03-18 21:34:16,723:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-18 21:35:16,723:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-18 21:36:16,723:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
```

mavo, 2024-03-18:
EDIT: Well... It started once, then I restarted the container and since then I do get an error on startup.

```
2024-03-18 21:53:21,081:INFO: Limit via inverter: True
2024-03-18 21:53:21,081:INFO: Control Parameters:
2024-03-18 21:53:21,081:INFO:   MIN_CHARGE_POWER = 200
2024-03-18 21:53:21,081:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-03-18 21:53:21,081:INFO:   MAX_INVERTER_LIMIT = 800
2024-03-18 21:53:21,081:INFO:   MAX_INVERTER_INPUT = 600
2024-03-18 21:53:21,081:INFO:   SUNRISE_OFFSET = 60
2024-03-18 21:53:21,081:INFO:   SUNSET_OFFSET = 60
2024-03-18 21:53:21,269:INFO: Using AhoyDTU: Base topic: ahoydtu, Limit topic: ahoydtu/ctrl/limit/1, SF Channels: [3, 4]
2024-03-18 21:53:21,269:INFO: Using ShellyEM3: Base topic: shellies/shellyem3
2024-03-18 21:53:21,269:INFO: Connected to MQTT Broker!
2024-03-18 21:53:21,270:INFO: Hub subscribing: /73bkTV/vSU06042/properties/report
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/solarInputPower
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/electricLevel
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputPackPower
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/packInputPower
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputHomePower
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/outputLimit
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/masterSoftVersion
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/pass
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/batteries/+/socLevel
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/telemetry/batteries/+/totalVol
2024-03-18 21:53:21,270:INFO: Hub subscribing: solarflow-hub/vSU06042/control/#
2024-03-18 21:53:21,271:INFO: DTU subscribing: ahoydtu/HM-1600/+/P_DC
2024-03-18 21:53:21,271:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/P_AC
2024-03-18 21:53:21,271:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/active_PowerLimit
2024-03-18 21:53:21,271:INFO: DTU subscribing: ahoydtu/status
2024-03-18 21:53:21,271:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-03-18 21:53:21,272:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/0/power
2024-03-18 21:53:21,272:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/1/power
2024-03-18 21:53:21,272:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/2/power
2024-03-18 21:53:21,273:INFO: Reading last empty time: 2024-03-18 06:29:24
2024-03-18 21:53:21,273:INFO: Reading last full time: 2024-03-13 16:23:22
2024-03-18 21:53:28,923:INFO: SMT triggers limit function: 79.19 : 0
Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 453, in <module>
2024-03-18 21:53:28,923:INFO: HUB: S:-1.0W [ ], B: -1% (-1), V:-1.0V (-1.0), C: 0W, P:False, F:125.5h, E:15.4h, H: -1W, L: -1W
    main(sys.argv[1:])
  File "/solarflow/solarflow-control.py", line 450, in main
    run()
  File "/solarflow/solarflow-control.py", line 381, in run
    client.loop_forever()
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/solarflow/solarflow-control.py", line 146, in on_message
    smartmeter.handleMsg(msg)
  File "/solarflow/smartmeters.py", line 80, in handleMsg
    self.updPower()
  File "/solarflow/smartmeters.py", line 72, in updPower
    self.trigger_callback(self.client)
  File "/solarflow/solarflow-control.py", line 361, in limit_callback
    limitHomeInput(client)
  File "/solarflow/solarflow-control.py", line 262, in limitHomeInput
    log.info(f'{inv}')
             ^^^^^^^^
  File "/solarflow/dtus.py", line 48, in __str__
    AC:{self.getCurrentACPower():>3.1f}W, AC_Prediction: {self.getPredictedACPower():>3.1f}W, \
        ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/dtus.py", line 120, in getCurrentACPower
    return self.acPower.last()
           ^^^^^^^^^^^^^^^^^^^
  File "/solarflow/utils.py", line 62, in last
    return self.values[-1][1]
           ~~~~~~~~~~~^^^^
IndexError: list index out of range
```

Probably its because the inverter turned off, as the app determined its time for chargeThrough before I restarted the container.

reinhard-brandstaedter, 2024-03-18:
OK, that needs more robustness indeed

mavo, 2024-03-18:
Well, its not about the inverter being offline. I connected via BT, set some output to home and disconnected. Started the app again, same error but ahoy dtu reports power being provided.

mavo, 2024-03-18:
Funny(?), same thing now happens also on latest image 👀

mavo, 2024-03-18:
Now its really crazy. Started the container another 25 times and it came up - only to now go crazy as it does not get the demand calculated somehow. It drops to 0 but the app things lets give the hub full steam and discharge as crazy with 800W.

```
2024-03-18 22:41:11,520:INFO: SMT triggers limit function: 10.7 : 50.1
2024-03-18 22:41:11,520:INFO: HUB: S:0.0W [ 0.0 ], B: 60% (57|63), V:48.5V (48.5|48.5), C:-712W, P:False, F:126.3h, E:16.2h, H:679W, L:800W
2024-03-18 22:41:11,523:INFO: INV: AC:703.3W, AC_Prediction: 444.0W, DC:740.3W, DC_prediction: 466.4W (0.0|0.0|374.8|365.5), L: 0W [ -1W]
2024-03-18 22:41:11,525:INFO: SMT: T:ShellyEM3 P:-258.3W [ 47.0,48.1,22.0,22.2,23.4,1.2,0.9,-2.1,-26.4,-26.4,-24.9,-57.0,-57.1,-48.9,-77.9,-77.7,-84.4,-93.3,-93.4,-96.8,-120.6,-120.4,-116.3,-148.2,-148.2,-152.8,-189.6,-190.1,-193.6,-224.1,-224.0,-233.6,-258.3,-258.3 ] Predict: 10.7W
2024-03-18 22:41:11,529:INFO: Direct connected panels can't cover demand 0.0W/454.7W, trying to get rest from hub.
2024-03-18 22:41:11,529:INFO: Checking if Solarflow is willing to contribute 454.7W ...
2024-03-18 22:41:11,529:INFO: Based on time, solarpower ( 0.0W) minimum charge power (200W) and bypass state (False), hub could contribute 454.7W - Decision path: 2.1.
2024-03-18 22:41:11,529:INFO: Hub should contribute more (454.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-18 22:41:11,529:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-18 22:41:11,529:INFO: Setting inverter output limit to 960W (1 min moving average of 227W x 4)
2024-03-18 22:41:11,530:INFO: Sun: 06:14 - 18:18 Demand: 454.7W, Panel DC: (0.0|0.0), Hub DC: (374.8|365.5), Inverter Limit: 960.0W, Hub Limit: 800.0W
2024-03-18 22:41:11,539:INFO: DTU triggers limit function: 466.4 : 435.9
2024-03-18 22:41:11,539:INFO: Rate limit on trigger function, last call was only 0.018828s ago!
2024-03-18 22:41:17,529:INFO: DTU triggers limit function: 497.0 : 466.4
2024-03-18 22:41:17,529:INFO: Rate limit on trigger function, last call was only 6.009509s ago!
2024-03-18 22:41:23,065:INFO: SMT triggers limit function: -42.4 : 10.7
2024-03-18 22:41:23,065:INFO: HUB: S:0.0W [ 0.0 ], B: 60% (57|63), V:48.1V (48.1|48.1), C:-769W, P:False, F:126.3h, E:16.2h, H:733W, L:800W
2024-03-18 22:41:23,069:INFO: INV: AC:755.6W, AC_Prediction: 503.7W, DC:795.3W, DC_prediction: 530.2W (0.0|0.0|402.7|392.6), L: 0W [ -1W]
2024-03-18 22:41:23,071:INFO: SMT: T:ShellyEM3 P:-272.1W [ 0.9,-2.1,-26.4,-26.4,-24.9,-57.0,-57.1,-48.9,-77.9,-77.7,-84.4,-93.3,-93.4,-96.8,-120.6,-120.4,-116.3,-148.2,-148.2,-152.8,-189.6,-190.1,-193.6,-224.1,-224.0,-233.6,-258.3,-258.3,-248.5,-248.4,-248.5,-251.3,-272.5,-272.1 ] Predict: -42.4W
2024-03-18 22:41:23,074:INFO: Direct connected panels can't cover demand 0.0W/0.0W, trying to get rest from hub.
2024-03-18 22:41:23,074:INFO: Checking if Solarflow is willing to contribute 0.0W ...
2024-03-18 22:41:23,074:INFO: Based on time, solarpower ( 0.0W) minimum charge power (200W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.1.
2024-03-18 22:41:23,074:INFO: Hub should contribute more (0.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-18 22:41:23,074:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-18 22:41:23,074:INFO: Setting inverter output limit to 540W (1 min moving average of 10W x 4)
2024-03-18 22:41:23,075:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (402.7|392.6), Inverter Limit: 540.0W, Hub Limit: 800.0W
2024-03-18 22:41:23,531:INFO: DTU triggers limit function: 530.2 : 497.0
2024-03-18 22:41:23,531:INFO: Rate limit on trigger function, last call was only 0.465543s ago!
2024-03-18 22:41:34,528:INFO: DTU triggers limit function: 597.7 : 530.2
2024-03-18 22:41:34,528:INFO: HUB: S:0.0W [ 0.0 ], B: 60% (57|63), V:48.1V (48.1|48.1), C:-825W, P:False, F:126.3h, E:16.2h, H:786W, L:800W
2024-03-18 22:41:34,532:INFO: INV: AC:799.4W, AC_Prediction: 565.9W, DC:841.4W, DC_prediction: 597.7W (0.0|0.0|426.0|415.4), L: 0W [ -1W]
2024-03-18 22:41:34,534:INFO: SMT: T:ShellyEM3 P:-322.4W [ -26.4,-24.9,-57.0,-57.1,-48.9,-77.9,-77.7,-84.4,-93.3,-93.4,-96.8,-120.6,-120.4,-116.3,-148.2,-148.2,-152.8,-189.6,-190.1,-193.6,-224.1,-224.0,-233.6,-258.3,-258.3,-248.5,-248.4,-248.5,-251.3,-272.5,-272.1,-280.7,-319.1,-305.0,-301.8,-300.2,-308.7,-310.0,-322.4 ] Predict: -72.3W
2024-03-18 22:41:34,537:INFO: Direct connected panels can't cover demand 0.0W/0.0W, trying to get rest from hub.
2024-03-18 22:41:34,537:INFO: Checking if Solarflow is willing to contribute 0.0W ...
2024-03-18 22:41:34,537:INFO: Based on time, solarpower ( 0.0W) minimum charge power (200W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.1.
2024-03-18 22:41:34,537:INFO: Hub should contribute more (0.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-18 22:41:34,538:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-18 22:41:34,538:INFO: Setting inverter output limit to 272W (1 min moving average of 10W x 4)
2024-03-18 22:41:34,538:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (426.0|415.4), Inverter Limit: 272.0W, Hub Limit: 800.0W
2024-03-18 22:41:37,534:INFO: DTU triggers limit function: 628.7 : 597.7
[... cut ...]
2024-03-18 22:42:05,539:INFO: Hub should contribute more (0.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-18 22:42:05,539:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-18 22:42:05,539:INFO: Setting inverter output limit to 56W (1 min moving average of 10W x 4)
2024-03-18 22:42:05,539:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (420.6|410.2), Inverter Limit: 56.0W, Hub Limit: 800.0W
```

mavo, 2024-03-19:
Hi @reinhard-brandstaedter 

> As mentioned https://github.com/reinhard-brandstaedter/solarflow-control/issues/168#issuecomment-2005010062 the hub has a very slow reaction to setting a limit, thus limiting via the inverter is implemented where possible.

This I know. And its not about the rate limiting itself. It's that it was reading 0 demand but still decided to give it maximum battery output. And it did not back-off from it.

> Shortly after startup there should be a line like below "Determined inverter's max capacity..."

This indeed does not get logged for me, and also, the inverter is not limited. At least neither the UI nor the MQTT explorer shows it as limited. I could see that the correct values has been set in the ctrl topic, but those seemed to be ignored from the inverter.

Is there anything I can provide you to track the issue further?

reinhard-brandstaedter, 2024-03-19:
A working limit command is essential, so lets try to figure that out...

While the control script is off, can you try to set the ctrl/limit/0 topic to something like 100W and see if the inverter applies the limit:
[image]

You should then see it applied:
[image]

If that works, please pull the latest dev build, I specifically added the maxpower detection from settings and not detection. See if that helps.

reinhard-brandstaedter, 2024-03-19:
Hi @mavo!
I just noticed something. The inverter_id has to be "0" for the first inverter. I'm pretty sure at some point this was 1 ...
In your screenshot above we can see that the limit is applied to inverter "1" which doesn't exist.

Can you change your config to this:
```
[ahoydtu]
# The MQTT base topic your AhoyDTU reports to (as configured in AhoyDTU UI)
base_topic = ahoydtu

# The inverter ID in AhoyDTU: typically 0 for the first inverter
inverter_id = 0
# List of indices of the inverter channels/ports (as reported in the DTU) that the Solarflow Hub is connected to
# typically the index starts at 1 as 0 is the output channel of the inverter
# e.g. 1,3 or 3 or [1,3]
sf_inverter_channels = < adjust for you>

# the max output power of your inverter, used to calculate correct absolute values
inverter_max_power = 1600

# The name of the inverter in AhoyDTU
inverter_name = HM-1600
```

mavo, 2024-03-19:
[image]
This looks good, after changing to `0` with fresh pulled dev image 👍

mavo, 2024-03-19:
Good news @reinhard-brandstaedter,
the limit of the inverter works now as expected, I can see the value being changed.

However, now I am back to the state where the app sets the limit of the hub to 800W and sticks to it, although my household demand is only ~400W with the PC running.
```
2024-03-19 19:29:06,691:INFO: DTU triggers limit function: 320.6 : 196.3
2024-03-19 19:29:06,691:INFO: HUB: S:0.0W [ 0.0 ], B: 86% (87|85), V:49.2V (49.1|49.2), C:-340W, P:False, F:1.9h, E:37.0h, H:324W, L:800W
2024-03-19 19:29:06,691:INFO: INV: AC:304.6W, AC_Prediction: 304.6W, DC:320.6W, DC_prediction: 320.6W (0.0|0.0|177.2|172.8), L:649.6W [1600W]
2024-03-19 19:29:06,694:INFO: SMT: T:ShellyEM3 P:57.1W [ 63.7,-62.4,137.4,144.9,79.9 ] Predict: 168.6W
2024-03-19 19:29:06,696:INFO: Direct connected panels can't cover demand 0.0W/473.2W, trying to get rest from hub.
2024-03-19 19:29:06,696:INFO: Checking if Solarflow is willing to contribute 473.2W ...
2024-03-19 19:29:06,696:INFO: Based on time, solarpower ( 0.0W) minimum charge power (200W) and bypass state (False), hub could contribute 473.2W - Decision path: 2.1.
2024-03-19 19:29:06,696:INFO: Hub should contribute more (473.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-19 19:29:06,696:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-19 19:29:06,696:INFO: Setting inverter output limit to 828W (1 min moving average of 236W x 4)
2024-03-19 19:29:06,697:INFO: Sun: 06:12 - 18:19 Demand: 473.2W, Panel DC: (0.0|0.0), Hub DC: (177.2|172.8), Inverter Limit: 828.0W, Hub Limit: 800.0W
2024-03-19 19:29:09,913:INFO: SMT triggers limit function: 71.46000000000001 -> 149.1
2024-03-19 19:29:09,913:INFO: Rate limit on trigger function, last call was only 3.2s ago!
2024-03-19 19:29:09,917:INFO: SMT triggers limit function: 11.5 -> 89.1
2024-03-19 19:29:09,917:INFO: Rate limit on trigger function, last call was only 3.2s ago!
2024-03-19 19:29:11,692:INFO: DTU triggers limit function: 380.4 : 320.6
2024-03-19 19:29:11,692:INFO: Rate limit on trigger function, last call was only 5.0s ago!
2024-03-19 19:29:12,692:INFO: DTU triggers limit function: 418.2 : 380.4
2024-03-19 19:29:12,692:INFO: Rate limit on trigger function, last call was only 6.0s ago!
2024-03-19 19:29:18,693:INFO: DTU triggers limit function: 455.7 : 418.2
2024-03-19 19:29:18,693:INFO: HUB: S:0.0W [ 0.0 ], B: 86% (87|85), V:49.2V (49.1|49.2), C:-402W, P:False, F:1.9h, E:37.0h, H:383W, L:800W
2024-03-19 19:29:18,697:INFO: INV: AC:418.9W, AC_Prediction: 433.2W, DC:440.9W, DC_prediction: 455.7W (0.0|0.0|223.3|217.6), L:649.6W [1600W]
2024-03-19 19:29:18,697:INFO: SMT: T:ShellyEM3 P:-26.1W [ 137.4,144.9,64.5,-25.2 ] Predict: -25.2W
2024-03-19 19:29:18,700:INFO: Direct connected panels can't cover demand 0.0W/0.0W, trying to get rest from hub.
2024-03-19 19:29:18,700:INFO: Checking if Solarflow is willing to contribute 0.0W ...
2024-03-19 19:29:18,701:INFO: Based on time, solarpower ( 0.0W) minimum charge power (200W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.1.
2024-03-19 19:29:18,701:INFO: Hub should contribute more (0.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-19 19:29:18,701:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-19 19:29:18,702:INFO: Setting inverter output limit to 434W (1 min moving average of 10W x 4)
2024-03-19 19:29:18,702:INFO: Sun: 06:12 - 18:19 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (223.3|217.6), Inverter Limit: 434.0W, Hub Limit: 800.0W
```

Is the hub output limit not relevant, when its being "controlled" by the inverter output?
What I also can see is some drastic jumps on the inverter limits, which makes the household consumption kind of a mess.
The consumption is actually very even currently, there is no on/off-consumer active, only steadily consuming items.
[image]

reinhard-brandstaedter, 2024-03-19:
Ok we got at least one thing sorted out.
The other one is an issue that I’m already tracking.

Can you try to set your 800W limit ( max inverter input) to something lower than your baseload would be. I believe it has to do something wirh the fact when the smartmeter goes below zero, then i’m doung a wrong calculation on demand.

I can only verify this once my battery is able to discharge…

Regarding the hub limit at max: yes, since the hub is so slow in applying a limit we open it up and limit via inverter (which is MUCH faster, and also works below 100W). The hub is just not made for quick lumit adoption.

mavo, 2024-03-19:
> I can only verify this once my battery is able to discharge…

You can modify the solarflow.py in the container and set [this](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/solarflow.py#L43) to false. Worked at least for me yesterday ^^

reinhard-brandstaedter, 2024-03-20:
that doesn't help if the battery is empty :-)

Btw. no need to go to the code for that, you can control chargethrough via MQTT:
[image]

mavo, 2024-03-20:
> that doesn't help if the battery is empty :-)

Well xD
I am currently not able to do this as I am not really at home and don't want to break it while I cannot properly fix it. Thats why I did not test as of yet.
But the assumption that its related to negative values sounds reasonable.

mavo, 2024-03-23:
Did you perhaps had time.to check when the battery is discharging?
I am still blocked from it, as I am not at home really and there is very little sun atm...

reinhard-brandstaedter, 2024-03-28:
There have now been some improvements regarding the limit calculation esp. when current feed in is happening.
There was indeed a bug that led to fluctuationg control settings. These should now be resolved and tests have been positive.

mavo, 2024-03-28:
Awesome, will pull the latest master build later today and check it.

Also, my hub2k will arrive sometime next week. I hope it just works also with your great work in this project.

closed 2024-03-31 by mavo