reinhard-brandstaedter/solarflow-control #168: "Ghost"-Demand drop to 0, causing battery to lower discharge when demand is actually still the same

mavo, 2024-03-18:
Hello @reinhard-brandstaedter,

me again with another one. Do you have an idea, why I see sudden drops in demand (actually to 0 as reported) when my home consumption is actually not really changed?

Here is an example, the chart shows the reported power on the three phases of my shellyem3 (turquoise, orange and yellow lines) and the output to home from the batteries (as its dark and the batteries have plenty of juice thx to the sun today) in blue and also the combined consumption of all three shelly phases (purple line).
[image]

Here is the corresponding log output from the time. Note the difference in reported output when "it happens".
```
2024-03-18 19:49:22,560:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.5V (48.5|48.5), C:-502W, P:False, F:123.4h, E:13.3h, H:479W, L:474W
2024-03-18 19:49:22,563:INFO: INV: AC:468.5W, AC_Prediction: 463.4W, DC:493.2W, DC_prediction: 488.0W (0.0|0.0|249.6|243.6), L:1600.0W [ 0W]
2024-03-18 19:49:22,565:INFO: SMT: T:ShellyEM3 P:13.8W [ 34.9,34.5,6.3,6.3,8.5,8.6,8.4,10.8,13.8 ] Predict: 12.2W
2024-03-18 19:49:22,569:INFO: Direct connected panels can't cover demand 0.0W/475.6W, trying to get rest from hub.
2024-03-18 19:49:22,569:INFO: Checking if Solarflow is willing to contribute 475.6W ...
2024-03-18 19:49:22,569:INFO: Based on time, solarpower ( 0.0W) minimum charge power (200W) and bypass state (False), hub could contribute 475.6W - Decision path: 2.1.
2024-03-18 19:49:22,569:INFO: Not setting solarflow output limit to 474.0W as it is identical to current limit!
2024-03-18 19:49:22,569:INFO: Solarflow is willing to contribute 474.0W!
2024-03-18 19:49:22,569:INFO: Adjusting inverter limit from 0.0W to 5.0W
2024-03-18 19:49:22,569:INFO: Direct connected panel limit is 5W.
2024-03-18 19:49:22,569:INFO: Setting inverter output limit to 10W (1 min moving average of 464W x 4)
2024-03-18 19:49:22,570:INFO: Sun: 06:14 - 18:18 Demand: 475.6W, Panel DC: (0.0|0.0), Hub DC: (249.6|243.6), Inverter Limit: 10.0W, Hub Limit: 474.0W
2024-03-18 19:49:37,053:INFO: SMT triggers limit function: -13.8 : 6.1
2024-03-18 19:49:37,053:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.5V (48.5|48.5), C:-501W, P:False, F:123.4h, E:13.3h, H:478W, L:474W
2024-03-18 19:49:37,056:INFO: INV: AC:468.3W, AC_Prediction: 464.4W, DC:492.9W, DC_prediction: 489.4W (0.0|0.0|249.5|243.4), L:1600.0W [ 0W]
2024-03-18 19:49:37,058:INFO: SMT: T:ShellyEM3 P:-86.1W [ 8.4,10.8,13.8,14.0,9.0,9.1,9.1,-86.1 ] Predict: -13.8W
2024-03-18 19:49:37,062:INFO: Setting inverter output limit to 10W (1 min moving average of 10W x 4)
2024-03-18 19:49:37,062:INFO: Setting solarflow output limit to 379.0W
2024-03-18 19:49:37,062:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (249.5|243.4), Inverter Limit: 10.0W, Hub Limit: 379.0W
2024-03-18 19:49:41,049:INFO: SMT triggers limit function: -23.9 : -13.8
2024-03-18 19:49:41,049:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.5V (48.5|48.5), C:-502W, P:False, F:123.4h, E:13.3h, H:479W, L:379W
2024-03-18 19:49:41,052:INFO: INV: AC:469.1W, AC_Prediction: 465.7W, DC:493.7W, DC_prediction: 490.9W (0.0|0.0|249.9|243.8), L:1600.0W [ 0W]
2024-03-18 19:49:41,054:INFO: SMT: T:ShellyEM3 P:-83.1W [ 8.4,10.8,13.8,14.0,9.0,9.1,9.1,-86.1,-83.3,-83.1 ] Predict: -23.9W
2024-03-18 19:49:41,057:INFO: Setting inverter output limit to 10W (1 min moving average of 10W x 4)
2024-03-18 19:49:41,058:INFO: Setting solarflow output limit to 310.0W
2024-03-18 19:49:41,058:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (249.9|243.8), Inverter Limit: 10.0W, Hub Limit: 310.0W
2024-03-18 19:49:45,564:INFO: DTU triggers limit function: 494.0 : 488.0
2024-03-18 19:49:45,565:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.5V (48.5|48.5), C:-496W, P:False, F:123.4h, E:13.3h, H:473W, L:310W
2024-03-18 19:49:45,568:INFO: INV: AC:443.4W, AC_Prediction: 469.4W, DC:466.7W, DC_prediction: 494.0W (0.0|0.0|236.0|230.7), L:1600.0W [ 0W]
2024-03-18 19:49:45,570:INFO: SMT: T:ShellyEM3 P:1.9W [ 8.4,10.8,13.8,14.0,9.0,9.1,9.1,-86.1,-83.3,-83.1,-107.1,-87.3,-87.3,-36.2,1.9 ] Predict: -19.7W
2024-03-18 19:49:45,573:INFO: Setting inverter output limit to 10W (1 min moving average of 10W x 4)
2024-03-18 19:49:45,574:INFO: Setting solarflow output limit to 258.0W
2024-03-18 19:49:45,574:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (236.0|230.7), Inverter Limit: 10.0W, Hub Limit: 258.0W
2024-03-18 19:49:50,561:INFO: DTU triggers limit function: 500.1 : 494.0
2024-03-18 19:49:50,561:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.6V (48.6|48.6), C:-476W, P:False, F:123.4h, E:13.3h, H:454W, L:258W
2024-03-18 19:49:50,565:INFO: INV: AC:402.4W, AC_Prediction: 475.1W, DC:423.4W, DC_prediction: 500.1W (0.0|0.0|214.4|209.0), L:1600.0W [ 0W]
2024-03-18 19:49:50,567:INFO: SMT: T:ShellyEM3 P:86.2W [ 8.4,10.8,13.8,14.0,9.0,9.1,9.1,-86.1,-83.3,-83.1,-107.1,-87.3,-87.3,-36.2,1.9,1.4,92.5,102.6,102.5,90.5,97.9,97.9,69.0,86.2 ] Predict: -24.6W
2024-03-18 19:49:50,570:INFO: Adjusting inverter limit from 0.0W to 5.0W
2024-03-18 19:49:50,570:INFO: Setting inverter output limit to 10W (1 min moving average of 10W x 4)
2024-03-18 19:49:50,570:INFO: Setting solarflow output limit to 218.0W
2024-03-18 19:49:50,571:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (214.4|209.0), Inverter Limit: 10.0W, Hub Limit: 218.0W
2024-03-18 19:49:53,561:INFO: DTU triggers limit function: 506.0 : 500.1
2024-03-18 19:49:53,561:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.6V (48.6|48.6), C:-459W, P:False, F:123.4h, E:13.3h, H:438W, L:218W
2024-03-18 19:49:53,564:INFO: INV: AC:373.9W, AC_Prediction: 480.7W, DC:393.5W, DC_prediction: 506.0W (0.0|0.0|199.1|194.4), L:1600.0W [ 0W]
2024-03-18 19:49:53,566:INFO: SMT: T:ShellyEM3 P:132.7W [ 8.4,10.8,13.8,14.0,9.0,9.1,9.1,-86.1,-83.3,-83.1,-107.1,-87.3,-87.3,-36.2,1.9,1.4,92.5,102.6,102.5,90.5,97.9,97.9,69.0,86.2,86.4,113.8,132.7 ] Predict: -26.6W
2024-03-18 19:49:53,569:INFO: Adjusting inverter limit from 0.0W to 5.0W
2024-03-18 19:49:53,570:INFO: Setting inverter output limit to 10W (1 min moving average of 10W x 4)
2024-03-18 19:49:53,570:INFO: Setting solarflow output limit to 187.0W
2024-03-18 19:49:53,570:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (199.1|194.4), Inverter Limit: 10.0W, Hub Limit: 187.0W
2024-03-18 19:49:56,559:INFO: DTU triggers limit function: 511.9 : 506.0
2024-03-18 19:49:56,560:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.6V (48.6|48.6), C:-446W, P:False, F:123.4h, E:13.3h, H:425W, L:187W
2024-03-18 19:49:56,563:INFO: INV: AC:344.5W, AC_Prediction: 486.3W, DC:362.5W, DC_prediction: 511.9W (0.0|0.0|183.6|178.9), L:1600.0W [ 0W]
2024-03-18 19:49:56,565:INFO: SMT: T:ShellyEM3 P:157.7W [ 14.0,9.0,9.1,9.1,-86.1,-83.3,-83.1,-107.1,-87.3,-87.3,-36.2,1.9,1.4,92.5,102.6,102.5,90.5,97.9,97.9,69.0,86.2,86.4,113.8,132.7,132.5,136.8,148.5,148.7,149.1,157.7 ] Predict: -30.2W
2024-03-18 19:49:56,568:INFO: Adjusting inverter limit from 0.0W to 5.0W
2024-03-18 19:49:56,568:INFO: Setting inverter output limit to 10W (1 min moving average of 10W x 4)
2024-03-18 19:49:56,569:INFO: Setting solarflow output limit to 146.0W
2024-03-18 19:49:56,569:INFO: Sun: 06:14 - 18:18 Demand: 0.0W, Panel DC: (0.0|0.0), Hub DC: (183.6|178.9), Inverter Limit: 10.0W, Hub Limit: 146.0W
2024-03-18 19:50:01,559:INFO: DTU triggers limit function: 519.0 : 511.9
2024-03-18 19:50:01,559:INFO: HUB: S:0.0W [ 0.0 ], B: 81% (80|83), V:48.6V (48.6|48.6), C:-406W, P:False, F:123.4h, E:13.3h, H:387W, L:146W
2024-03-18 19:50:01,564:INFO: INV: AC:282.4W, AC_Prediction: 493.2W, DC:297.2W, DC_prediction: 519.0W (0.0|0.0|150.4|146.8), L:1600.0W [ 0W]
2024-03-18 19:50:01,567:INFO: SMT: T:ShellyEM3 P:204.0W [ 14.0,9.0,9.1,9.1,-86.1,-83.3,-83.1,-107.1,-87.3,-87.3,-36.2,1.9,1.4,92.5,102.6,102.5,90.5,97.9,97.9,69.0,86.2,86.4,113.8,132.7,132.5,136.8,148.5,148.7,149.1,157.7,157.6,145.4,154.2,154.1,142.0,171.4,171.3,190.5,204.0 ] Predict: -23.7W
[... 110 lines cut ...]
2024-03-18 19:50:42,041:INFO: INV: AC:45.7W, AC_Prediction: 418.4W, DC:48.1W, DC_prediction: 441.5W (0.0|0.0|24.2|23.9), L:1600.0W [ 0W]
2024-03-18 19:50:42,043:INFO: SMT: T:ShellyEM3 P:404.8W [ -87.3,-36.2,1.9,1.4,92.5,102.6,102.5,90.5,97.9,97.9,69.0,86.2,86.4,113.8,132.7,132.5,136.8,148.5,148.7,149.1,157.7,157.6,145.4,154.2,154.1,142.0,171.4,171.3,190.5,204.0,204.0,195.5,218.1,218.0,218.3,227.9,228.1,227.4,247.2,247.0,250.8,283.0,283.1,267.1,274.9,275.0,287.7,308.8,309.0,304.9,324.2,324.1,322.9,345.5,345.5,352.7,368.9,368.9 [... cut ...]
2024-03-18 19:50:42,571:INFO: Setting solarflow output limit to 186.0W
2024-03-18 19:50:42,571:INFO: Solarflow is willing to contribute 186.0W!
2024-03-18 19:50:42,571:INFO: Adjusting inverter limit from 0.0W to 5.0W
2024-03-18 19:50:42,571:INFO: Direct connected panel limit is 5W.
2024-03-18 19:50:42,571:INFO: Setting inverter output limit to 10W (1 min moving average of 176W x 4)
2024-03-18 19:50:42,572:INFO: Sun: 06:14 - 18:18 Demand: 477.6W, Panel DC: (0.0|0.0), Hub DC: (24.2|23.9), Inverter Limit: 10.0W, Hub Limit: 186.0W
```

reinhard-brandstaedter, 2024-03-18:
What you are running into here is a issue with too quick updates to the SF limit. Notice the 1-2second interval where it triggers a new limit? I have made this observation as well and in combination with too quick new values from the smartmeter (seems also every 1s interval). This leads to false predictive values. Kind of a vicious circle that I'm investigating.

We have already built a "rate-limit" for the SF limit switching, to accommodate for it's slow reaction time (it takes 30-60s until a limit is actually applied on the SF side).
I think that little dip and jump in the orange phase actually triggered that behavior.

mavo, 2024-03-20:
As this seems also somewhat relates to my other issue or is very close and we touched upon this in the other issue, I'll close this one for now.

closed 2024-03-20 by mavo