reinhard-brandstaedter/solarflow-control #185: App does not trigger any updates on latest master

mavo, 2024-03-28:
Hello @reinhard-brandstaedter,

I was testing latest master and noticed, that the app somehow stopped receiving events. Here are the logs to the time frame in the screenshot.
[image]

```
2024-03-28 21:20:27,294:INFO: DTU triggers limit function: 402.2 : 438.3
2024-03-28 21:20:27,294:INFO: HUB: S:0.0W [ 0.0,0.0 ], B: 65% (64|65), V:46.9V (46.9|46.9), C:-473W, P:False, F:76.6h, E:13.2h, H:451W, L:800W
2024-03-28 21:20:27,298:INFO: INV: AC:422.6W, AC_Prediction: 382.1W, DC:444.8W, DC_prediction: 402.2W (0.0|0.0|225.4|219.4), L:1075.2W [1600W]
2024-03-28 21:20:27,299:INFO: SMT: T:ShellyEM3 P:-1.4W [ -127.0,-114.7,-4.3,-6.0,-0.9 ] Predict: 93.8W
2024-03-28 21:20:27,302:INFO: Direct connected panels (0.0W) can't cover demand (475.9W), trying to get rest from hub.
2024-03-28 21:20:27,303:INFO: Checking if Solarflow is willing to contribute 475.9W ...
2024-03-28 21:20:27,303:INFO: Based on time, solarpower ( 0.0W) minimum charge power (400W) and bypass state (False), hub could contribute 475.9W - Decision path: 2.1.
2024-03-28 21:20:27,303:INFO: Hub should contribute more (475.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-28 21:20:27,303:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 800, new limit: 800
2024-03-28 21:20:27,303:INFO: Setting inverter output limit to 948W (1 min moving average of 237W x 4)
2024-03-28 21:20:27,303:INFO: Sun: 05:51 - 18:35 Demand: 475.9W, Panel DC: (0.0|0.0), Hub DC: (225.4|219.4), Inverter Limit: 948.0W, Hub Limit: 800.0W
2024-03-28 21:20:34,292:INFO: DTU triggers limit function: 461.6 : 402.2
2024-03-28 21:20:34,292:INFO: Rate limit on trigger function, last call was only 7.0s ago!
2024-03-28 21:20:58,298:INFO: DTU triggers limit function: 526.1 : 461.6
2024-03-28 21:20:58,298:INFO: HUB: S:0.0W [ 0.0,0.0 ], B: 65% (64|65), V:47.0V (46.9|47.0), C:-510W, P:False, F:76.6h, E:13.2h, H:486W, L:800W
2024-03-28 21:20:58,302:INFO: INV: AC:477.3W, AC_Prediction: 501.3W, DC:502.4W, DC_prediction: 526.1W (0.0|0.0|255.3|248.4), L:1075.2W [1600W]
2024-03-28 21:20:58,303:INFO: SMT: T:ShellyEM3 P:-58.2W [ -4.3,-6.0,-1.3,-58.0,-57.6 ] Predict: -88.9W
2024-03-28 21:20:58,306:INFO: Direct connected panels (0.0W) can't cover demand (412.4W), trying to get rest from hub.
2024-03-28 21:20:58,306:INFO: Checking if Solarflow is willing to contribute 412.4W ...
2024-03-28 21:20:58,307:INFO: Based on time, solarpower ( 0.0W) minimum charge power (400W) and bypass state (False), hub could contribute 412.4W - Decision path: 2.1.
2024-03-28 21:20:58,307:INFO: Hub should contribute more (412.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-28 21:20:58,307:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-28 21:20:58,307:INFO: Setting inverter output limit to 824W (1 min moving average of 206W x 4)
2024-03-28 21:20:58,307:INFO: Sun: 05:51 - 18:35 Demand: 412.4W, Panel DC: (0.0|0.0), Hub DC: (255.3|248.4), Inverter Limit: 824.0W, Hub Limit: 800.0W
2024-03-28 21:21:05,886:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:21:59,293:INFO: DTU triggers limit function: 461.7 : 526.1
2024-03-28 21:21:59,293:INFO: HUB: S:0.0W [ 0.0,0.0 ], B: 65% (64|65), V:47.0V (47.0|47.0), C:-449W, P:False, F:76.6h, E:13.2h, H:428W, L:800W
2024-03-28 21:21:59,293:INFO: INV: AC:438.6W, AC_Prediction: 438.6W, DC:461.7W, DC_prediction: 461.7W (0.0|0.0|223.3|217.3), L:824.0W [1600W]
2024-03-28 21:21:59,293:INFO: SMT: T:ShellyEM3 P:-1.6W [ -58.4,-6.4,-3.4,-1.6 ] Predict: -1.6W
2024-03-28 21:21:59,293:INFO: Direct connected panels (0.0W) can't cover demand (437.0W), trying to get rest from hub.
2024-03-28 21:21:59,293:INFO: Checking if Solarflow is willing to contribute 437.0W ...
2024-03-28 21:21:59,294:INFO: Based on time, solarpower ( 0.0W) minimum charge power (400W) and bypass state (False), hub could contribute 437.0W - Decision path: 2.1.
2024-03-28 21:21:59,294:INFO: Hub should contribute more (437.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-03-28 21:21:59,294:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-03-28 21:21:59,294:INFO: Setting inverter output limit to 872W (1 min moving average of 218W x 4)
2024-03-28 21:21:59,294:INFO: Sun: 05:51 - 18:35 Demand: 437.0W, Panel DC: (0.0|0.0), Hub DC: (223.3|217.3), Inverter Limit: 872.0W, Hub Limit: 800.0W
2024-03-28 21:22:05,887:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:23:05,887:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:24:05,887:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:25:05,888:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:26:05,888:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:27:05,888:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:28:05,888:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:29:05,889:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:30:05,889:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:31:05,889:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:32:05,890:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:33:05,890:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:34:05,890:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:35:05,891:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:36:05,891:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:37:05,891:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:38:05,892:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:39:05,892:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:40:05,892:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:41:05,892:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:42:05,893:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:43:05,893:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:44:05,893:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:45:05,894:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:46:05,894:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:47:05,894:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:48:05,895:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 21:49:05,895:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
[... 43 lines cut ...]
2024-03-28 22:02:58,000:INFO: DTU subscribing: ahoydtu/status
2024-03-28 22:02:58,000:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-03-28 22:02:58,000:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/0/power
2024-03-28 22:02:58,000:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/1/power
2024-03-28 22:02:58,000:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/2/power
2024-03-28 22:02:58,001:INFO: Reading last empty time: 2024-03-28 08:10:39
2024-03-28 22:02:58,001:INFO: Reading last full time: 2024-03-25 16:45:03
2024-03-28 22:02:58,001:INFO: Reading battery target mode: charging
2024-03-28 22:02:59,291:INFO: DTU triggers limit function: 461.8 : 0
2024-03-28 22:02:59,291:INFO: HUB: S:-1.0W [ ], B: -1% (-1), V:-1.0V (-1.0), C:-471W, P:False, F:77.3h, E:13.9h, H:449W, L: -1W
2024-03-28 22:02:59,291:INFO: INV: AC:438.8W, AC_Prediction: 438.8W, DC:461.8W, DC_prediction: 461.8W (0.0), L: 0W [1600W]
2024-03-28 22:02:59,291:INFO: SMT: T:ShellyEM3 P:0.0W [ ] Predict: 0.0W
2024-03-28 22:02:59,550:INFO: Rate limit on trigger function, last call was only 0.3s ago!
2024-03-28 22:02:59,551:INFO: Rapid rise in demand detected, clearing buffer!
[... cut ...]
2024-03-28 22:04:58,000:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 22:05:58,001:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 22:06:58,001:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
2024-03-28 22:07:58,001:INFO: Triggering telemetry update: iot/73bkTV/vSU06042/properties/read
```

I shut down my PC hoping to trigger any update, but was not successful. After changing back to dev build from a couple of days ago (23.03.) it seems to work again 🤔

reinhard-brandstaedter, 2024-03-28:
Do you see anywhere in the logs something like this:

2024-03-29 00:16:46,199:INFO: SMT triggers limit function: 215.7 -> 228.1

I’m wondering if the Shelly doesn’t send usage updates at a regular interval (<10s).
Also interesting that you got disconnected from MQTT at some point?

reinhard-brandstaedter, 2024-03-29:
Hi @mavo 
sorry I completely f'up the last master build (and some branches as well). Something went wrong with my own PRs and merges.
I have no idea what exactly went wrong, but I reset various branches now and the new master has been built again.

mavo, 2024-03-29:
No worries, glad you found the issue. Will try the latest master build later today again and close this issue when all is as expected :-)

mavo, 2024-03-30:
Looks good so far, will close the issue in the evening when battery discharging can be checked as well as the bat was very empty yesterday.

closed 2024-03-31 by mavo