reinhard-brandstaedter/solarflow-control #191: HUB2K support

mavo, 2024-04-05:
Hi @reinhard-brandstaedter,

today my hub2k did arrive, and ofc I installed it asap.
So, its now running, I can see data in the sf-app on my smartphone and I disconnected it from the cloud using BT manager.
I can see a small amount of messages in mqtt explorer, however, the control-app seems to have no idea.

```
2024-04-05 10:35:58,597:INFO: MQTT User is not set, assuming authentication not needed
2024-04-05 10:35:58,597:INFO: Solarflow Hub: A8yh63/ETFTke36
2024-04-05 10:35:58,598:INFO: Limit via inverter: True
2024-04-05 10:35:58,598:INFO: Control Parameters:
2024-04-05 10:35:58,598:INFO:   MIN_CHARGE_POWER = 100
2024-04-05 10:35:58,598:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-04-05 10:35:58,598:INFO:   MAX_INVERTER_LIMIT = 700
2024-04-05 10:35:58,598:INFO:   MAX_INVERTER_INPUT = 800
2024-04-05 10:35:58,598:INFO:   SUNRISE_OFFSET = 60
2024-04-05 10:35:58,598:INFO:   SUNSET_OFFSET = 60
2024-04-05 10:35:58,803:INFO: Using AhoyDTU: Base topic: ahoydtu, Limit topic: ahoydtu/ctrl/limit/0, SF Channels: [3, 4]
2024-04-05 10:35:58,803:INFO: Using ShellyEM3: Base topic: shellies/shellyem3
2024-04-05 10:35:58,803:INFO: Connected to MQTT Broker!
2024-04-05 10:35:58,803:INFO: Hub subscribing: /A8yh63/ETFTke36/properties/report
2024-04-05 10:35:58,803:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/solarInputPower
2024-04-05 10:35:58,803:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/electricLevel
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputPackPower
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/packInputPower
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputHomePower
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputLimit
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/inverseMaxPower
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/masterSoftVersion
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/pass
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/batteries/+/socLevel
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/batteries/+/totalVol
2024-04-05 10:35:58,804:INFO: Hub subscribing: solarflow-hub/ETFTke36/control/#
2024-04-05 10:35:58,805:INFO: DTU subscribing: ahoydtu/HM-1600/+/P_DC
2024-04-05 10:35:58,805:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/P_AC
2024-04-05 10:35:58,805:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/active_PowerLimit
2024-04-05 10:35:58,805:INFO: DTU subscribing: ahoydtu/status
2024-04-05 10:35:58,805:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-04-05 10:35:58,805:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/0/power
2024-04-05 10:35:58,805:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/1/power
2024-04-05 10:35:58,805:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/2/power
2024-04-05 10:35:59,070:INFO: HUB: S:-1.0W [ ], B: -1% (-1), V:-1.0V (-1.0), C: 0W, P:False, F:-1.0h, E:-1.0h, H: -1W, L: -1W
2024-04-05 10:35:59,070:INFO: INV: AC:438.9W, AC_Prediction: 438.9W, DC:462.1W, DC_prediction: 462.1W (), L: 0W [1600W]
2024-04-05 10:35:59,070:INFO: SMT: T:ShellyEM3 P:51.6W [ 51.6 ] Predict: 51.6W
2024-04-05 10:35:59,071:INFO: Rapid drop in demand detected, clearing buffer!
2024-04-05 10:35:59,071:INFO: Rate limit on trigger function, last call was only 0.0s ago!
2024-04-05 10:36:00,887:INFO: DTU triggers limit function: 462.1 : 0
2024-04-05 10:36:00,887:INFO: Rate limit on trigger function, last call was only 1.8s ago!
2024-04-05 10:36:58,805:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-04-05 10:37:08,914:INFO: SMT triggers limit function: -6.2 -> -72.8
2024-04-05 10:37:08,914:INFO: HUB: S:-1.0W [ ], B: -1% (-1), V:-1.0V (-1.0), C: 0W, P:False, F:-1.0h, E:-1.0h, H: -1W, L: -1W
2024-04-05 10:37:08,918:INFO: INV: AC:437.5W, AC_Prediction: 437.5W, DC:460.7W, DC_prediction: 460.7W (0.1|0.1|234.0|227.9), L: 0W [1600W]
2024-04-05 10:37:08,920:INFO: SMT: T:ShellyEM3 P:53.9W [ -116.0,-109.7,-126.8,-128.5,-125.6,-123.8,-6.2 ] Predict: -72.8W
```

I found how to set the product_id in the config file and did it. I also additionally had to modify the solarflow.py file in the container, as it contains the hard coded [prod-id of the smaller hub](https://github.com/reinhard-brandstaedter/solarflow-control/blob/0bed4e5a0521472f06e1dd65d197e84a81c48137/src/solarflow/solarflow.py#L18).

However, I cannot see much more + the app is currently not working. What data can I provide you to help me find the mistake I made? 😇 
This is what mqtt explorer puts out atm.
[image]

MaVo

mavo, 2024-04-05:
Firmware update fixed it 🎉

closed 2024-04-05 by mavo

mavo, 2024-04-05:
Me2, same version.
Do you have issues?

I summarized the changes I did to make it work #192

olomberg, 2024-04-06:
HI @mavo.
No. Works as expected so far. Only the official Zendure MQTT-API sucks. But I found a way to work around it.

Newbie-2, 2024-04-11:
Hallo mavo,
gibt es schon eine Lösung um nach der Trennung vom Zendure Server die Daten des HUB-2000 per MQTT zB. im MQTT-Explorer zu empfangen ? 

Bei mir hat zwar die Trennung geklappt, aber ich bekomme keine Daten ?
[image]

Siehe auch Issue im BT-Manager
https://github.com/reinhard-brandstaedter/solarflow-bt-manager/issues/12

mavo, 2024-04-13:
Hi @Newbie-2,
After I did the firmware update to 3.xx it just worked.
Did you check if you have a queue named solarflow-hub in your mqtt explorer?

As said, after doing the update it just worked. I did nothing more to the queue config, only changing properties to use hub2 in the app.
 
MaVo