reinhard-brandstaedter/solarflow-control #192: Hub2k configuration

mavo, 2024-04-05:
Hello @reinhard-brandstaedter,

after sorting out the initial hickups I had I'd still like to raise the problems I found, hoping you can incorporate them into the software properly.

What I did in order to get the hub2k to work is:
- added param `product_id = A8yh63` in config.ini
- Modified container:
  - adjusted file [solarflow.py and changed the product-id](https://github.com/reinhard-brandstaedter/solarflow-control/blob/0bed4e5a0521472f06e1dd65d197e84a81c48137/src/solarflow/solarflow.py#L18) 
  - adjusted [template file](https://github.com/reinhard-brandstaedter/solarflow-control/blob/0bed4e5a0521472f06e1dd65d197e84a81c48137/src/solarflow/homeassistant/number.outputLimit.json) to match product-id 
  - adjusted [template file](https://github.com/reinhard-brandstaedter/solarflow-control/blob/0bed4e5a0521472f06e1dd65d197e84a81c48137/src/solarflow/homeassistant/switch.passMode.json) to match product-id as well

There is one more thing - as I for times with a lot of sun I'd like to set the min_charge to 0, but this still does not work and it will fall back to the default. Can I somehow convince you to have the default on 0 as well, so that 0 actually works?

Cheers, MaVo

mavo, 2024-04-05:
One more request I'd have:
In [here](https://github.com/reinhard-brandstaedter/solarflow-control/blob/0bed4e5a0521472f06e1dd65d197e84a81c48137/src/solarflow/solarflow.py#L300), there's a comment already saying either give 0 OR whats left after min_charge is taken into account.

Can this actually use `availableInput - min_charge` instead of 0? I have no direct connected panels anymore due to the change to hub2k, I'd actually like to still feed something to my home, even in CHARGE_THROUGH... Wdyt?

reinhard-brandstaedter, 2024-04-05:
> One more request I'd have: In [here](https://github.com/reinhard-brandstaedter/solarflow-control/blob/0bed4e5a0521472f06e1dd65d197e84a81c48137/src/solarflow/solarflow.py#L300), there's a comment already saying either give 0 OR whats left after min_charge is taken into account.
> 
> Can this actually use `availableInput - min_charge` instead of 0? I have no direct connected panels anymore due to the change to hub2k, I'd actually like to still feed something to my home, even in CHARGE_THROUGH... Wdyt? Maybe have a property to set this via the config file would be great 🤔

That would then be the "normal" behavior if chargeThrough is set to OFF?

reinhard-brandstaedter, 2024-04-05:
> There is one more thing - as I for times with a lot of sun I'd like to set the min_charge to 0, but this still does not work and it will fall back to the default. Can I somehow convince you to have the default on 0 as well, so that 0 actually works?

I have now set the default of the min_charge to 0, that should resolve this problem and I think people do customize that property anyway so 0 as default is OK

mavo, 2024-04-05:
> That would then be the "normal" behavior if chargeThrough is set to OFF?

It is, but as said, would it be an option, that you make that configurable as well? So that I can define, that even when `chargeThrough` is ON it only feeds the battery what is produced more than consumed.
This would, as said, fit my use case with no direct panels anymore. 
I almost hear you: "Why this is relevant you ask?" For me, it will now happen that I have fed nothing to the house but all to the battery in the afternoon. Meaning the battery is now charged to ~3kWh and tomorrow its predicted to be a sunny day. Which means, it will feed all to the battery in the early hours, only to then feed the grid with the power that is produced but the house consumption is low. Its just a matter of "not really be able to optimize my setup to the given environmental circumstances".

MaVo

mavo, 2024-04-07:
@reinhard-brandstaedter,

is my dream of this new [feature](https://github.com/reinhard-brandstaedter/solarflow-control/issues/192#issuecomment-2040463426) something you'd consider worth implementing, or rather not?
I saw a couple of references to this issue in different commits, which seem to have addressed all the hub2k related stuff already.
Now I wonder if you anyway not consider implementing this if I in this case should close the issue to not just have it open for fun.

Cheers & happy Sunday, MaVo

boennemann, 2024-04-09:
Hey all,

my opendtu compatible inverter will arrive soon and I'm preparing to setup solarflow-control.

I'm getting confused reading this issue. I feel like the original issue is about PV Hub 2000 (in)compatibility and only on a side-note about a different feature request – yet the discussion is only about the feature request.

As I understand it the PV Hub 2000 is not supported out of the box right now and requires running a forked image.

It doesn't seem like a real issue, more like that the product id config isn't consistently applied. I'm not mega firm in python, otherwise I'd try to send a PR.

@reinhard-brandstaedter Would it be possible to change these 3 instances to refer to the variable instead?

reinhard-brandstaedter, 2024-04-11:
The hub 2000 is already supported on the dev build, will soon be merged to master.

closed 2024-04-14 by reinhard-brandstaedter