reinhard-brandstaedter/solarflow-control #194: Hub in bypass does not allow 100% power output

mavo, 2024-04-06:
Hello @reinhard-brandstaedter,

today I noticed a behaviour, I never saw before.
My battery is full and I am in bypass mode. However, the SF-control app does limit my home output to ~600W (80% of inverter capacity).

```
2024-04-06 08:26:42,222:INFO: HUB: S:629.7W [ 629.2,631.2,631.3,629.3,628.7 ], B:100% (100|100), V:50.5V (50.5|50.5), C: 0W, P:True, F:0.0h, E:-1.0h, H:660W, L:800W
2024-04-06 08:26:42,225:INFO: INV: AC:634.0W, AC_Prediction: 632.5W, DC:667.4W, DC_prediction: 666.2W (0.0|0.0|338.7|329.7), L:1264.0W [1600W]
2024-04-06 08:26:42,227:INFO: SMT: T:ShellyEM3 P:-115.6W [ -369.6,-372.8,-209.5,-257.9,-129.6,-156.2,-115.7 ] Predict: -93.6W
2024-04-06 08:26:42,230:INFO: Direct connected panels (0.0W) can't cover demand (538.9W), trying to get rest from hub.
2024-04-06 08:26:42,230:INFO: Checking if Solarflow is willing to contribute 538.9W ...
2024-04-06 08:26:42,230:INFO: Adjusting inverter limit from 629.7W to 635.3W
2024-04-06 08:26:42,230:INFO: Based on time, solarpower (629.7W) minimum charge power (0W) and bypass state (True), hub could contribute 635.0W - Decision path: 1.1.
2024-04-06 08:26:42,230:INFO: Hub should contribute more (635.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-06 08:26:42,230:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 800, new limit: 800
2024-04-06 08:26:42,230:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-06 08:26:42,231:INFO: Sun: 06:30 - 19:51 Demand: 538.9W, Panel DC: (0.0|0.0), Hub DC: (338.7|329.7), Inverter Limit: 1268.0W, Hub Limit: 800.0W
2024-04-06 08:26:42,234:INFO: SMT triggers limit function: -142.3 -> -106.0
2024-04-06 08:26:42,234:INFO: Rate limit on trigger function, last call was only 0.0s ago!
2024-04-06 08:26:42,238:INFO: SMT triggers limit function: -150.3 -> -109.7
2024-04-06 08:26:42,238:INFO: Rate limit on trigger function, last call was only 0.0s ago!
2024-04-06 08:26:42,413:INFO: SMT triggers limit function: -153.4 -> -111.1
2024-04-06 08:26:42,413:INFO: Rate limit on trigger function, last call was only 0.2s ago!
2024-04-06 08:26:42,416:INFO: SMT triggers limit function: -139.6 -> -104.7
2024-04-06 08:26:42,416:INFO: Rate limit on trigger function, last call was only 0.2s ago!
2024-04-06 08:26:42,420:INFO: SMT triggers limit function: -134.5 -> -102.3
2024-04-06 08:26:42,420:INFO: Rate limit on trigger function, last call was only 0.2s ago!
2024-04-06 08:26:44,232:INFO: SMT triggers limit function: -132.6 -> -101.4
2024-04-06 08:26:44,233:INFO: Rate limit on trigger function, last call was only 2.0s ago!
2024-04-06 08:26:44,236:INFO: SMT triggers limit function: -138.2 -> -104.0
2024-04-06 08:26:44,237:INFO: Rate limit on trigger function, last call was only 2.0s ago!
2024-04-06 08:26:44,240:INFO: SMT triggers limit function: -141.6 -> -105.6
2024-04-06 08:26:44,240:INFO: Rate limit on trigger function, last call was only 2.0s ago!
2024-04-06 08:26:46,245:INFO: SMT triggers limit function: -142.9 -> -67.7
2024-04-06 08:26:46,245:INFO: Rate limit on trigger function, last call was only 4.0s ago!
2024-04-06 08:26:46,248:INFO: SMT triggers limit function: -160.41000000000003 -> -110.8
2024-04-06 08:26:46,248:INFO: Rate limit on trigger function, last call was only 4.0s ago!
2024-04-06 08:26:46,252:INFO: SMT triggers limit function: -158.8 -> -110.0
2024-04-06 08:26:46,252:INFO: Rate limit on trigger function, last call was only 4.0s ago!
2024-04-06 08:26:46,409:INFO: SMT triggers limit function: -157.8 -> -109.5
2024-04-06 08:26:46,409:INFO: Rate limit on trigger function, last call was only 4.2s ago!
2024-04-06 08:26:46,412:INFO: SMT triggers limit function: -162.7 -> -111.8
2024-04-06 08:26:46,413:INFO: Rate limit on trigger function, last call was only 4.2s ago!
2024-04-06 08:26:46,416:INFO: SMT triggers limit function: -161.9 -> -111.4
2024-04-06 08:26:46,417:INFO: Rate limit on trigger function, last call was only 4.2s ago!
2024-04-06 08:26:48,257:INFO: SMT triggers limit function: -161.5 -> -111.3
2024-04-06 08:26:48,258:INFO: Rate limit on trigger function, last call was only 6.0s ago!
2024-04-06 08:26:48,261:INFO: SMT triggers limit function: -161.2 -> -111.1
2024-04-06 08:26:48,261:INFO: Rate limit on trigger function, last call was only 6.0s ago!
2024-04-06 08:26:48,265:INFO: SMT triggers limit function: -162.1 -> -111.5
2024-04-06 08:26:48,266:INFO: Rate limit on trigger function, last call was only 6.0s ago!
2024-04-06 08:26:50,360:INFO: SMT triggers limit function: -162.7 -> -111.8
2024-04-06 08:26:50,360:INFO: Rate limit on trigger function, last call was only 8.1s ago!
2024-04-06 08:26:50,363:INFO: SMT triggers limit function: -165.5 -> -113.1
2024-04-06 08:26:50,364:INFO: Rate limit on trigger function, last call was only 8.1s ago!
2024-04-06 08:26:50,367:INFO: SMT triggers limit function: -166.5 -> -113.6
2024-04-06 08:26:50,367:INFO: Rate limit on trigger function, last call was only 8.1s ago!
2024-04-06 08:26:52,403:INFO: SMT triggers limit function: -166.9 -> -113.8
2024-04-06 08:26:52,403:INFO: HUB: S:629.7W [ 631.2,631.3,629.3,628.7,629.8 ], B:100% (100|100), V:50.5V (50.5|50.5), C: 0W, P:True, F:0.0h, E:-1.0h, H:661W, L:800W
2024-04-06 08:26:52,407:INFO: INV: AC:632.0W, AC_Prediction: 631.4W, DC:665.3W, DC_prediction: 664.5W (0.0|0.0|337.1|328.2), L:1264.0W [1600W]
2024-04-06 08:26:52,408:INFO: SMT: T:ShellyEM3 P:-167.1W [ -372.8,-209.5,-257.9,-129.6,-156.2,-142.9,-166.9 ] Predict: -113.8W
2024-04-06 08:26:52,411:INFO: Direct connected panels (0.0W) can't cover demand (517.6W), trying to get rest from hub.
2024-04-06 08:26:52,412:INFO: Checking if Solarflow is willing to contribute 517.6W ...
2024-04-06 08:26:52,412:INFO: Adjusting inverter limit from 629.7W to 635.3W
2024-04-06 08:26:52,412:INFO: Based on time, solarpower (629.7W) minimum charge power (0W) and bypass state (True), hub could contribute 635.0W - Decision path: 1.1.
2024-04-06 08:26:52,412:INFO: Hub should contribute more (635.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-06 08:26:52,412:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-04-06 08:26:52,412:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-06 08:26:52,413:INFO: Sun: 06:30 - 19:51 Demand: 517.6W, Panel DC: (0.0|0.0), Hub DC: (337.1|328.2), Inverter Limit: 1268.0W, Hub Limit: 800.0W
2024-04-06 08:26:52,416:INFO: SMT triggers limit function: -165.8 -> -113.3
[... 175 lines cut ...]
2024-04-06 08:33:05,093:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-04-06 08:33:08,215:INFO: Battery is full: 100
2024-04-06 08:33:08,215:INFO: Reading last full time: 2024-04-06 08:33:08
2024-04-06 08:33:08,256:INFO: Reading battery target mode: discharging
```

When I disable the app and set the inverter to 100%, it will deliver the 100%. However, the app as soon as it runs will fight this again.
Here's my config.

```
[control]
# min charge 0 is fine, as I want to feed the home whenever possible
min_charge_power = 0
# change also, discharge can be actually 800W
max_discharge_power = 800
max_inverter_limit = 800
# I don't want 0-feed-in
limit_inverter = true
inverter_min_limit = 5
```

Any Idea why this is happening?

Cheers,
MaVo

reinhard-brandstaedter, 2024-04-06:
Your hub has probably set the max inverter input to 600W (inverter overload protection)?

mavo, 2024-04-06:
The hub is set to 800w max output.
At least this is what I configured when I initially set it up.

As said, as soon as I set the inverter to 100%, it will pass the full 800w. But then the app strikes back and regulates down again.

mavo, 2024-04-06:
Here's a screenshot from the app settings.
[image]

From the logs you can also see, that the inverter is limited, not the hub output.

mavo, 2024-04-06:
Hello @reinhard-brandstaedter,

was using the time while kids doing a nap to dig a bit. 🙈 

I changed [this line](https://github.com/reinhard-brandstaedter/solarflow-control/blob/0bed4e5a0521472f06e1dd65d197e84a81c48137/src/solarflow/solarflow-control.py#L244) back to its original value, which is using max inverter power. I see the commit msg, which seems to manage concerns when the bypass is about to end.

To make sure its not the hub having set the 600W somewhere, I changed it back and since then the hub happily contributes 800W.
```
2024-04-06 11:31:16,965:INFO: SMT triggers limit function: -480.6 -> -490.8
2024-04-06 11:31:16,965:INFO: HUB: S:801.3W [ 799.4,795.2,773.8,800.3,810.5,810.4 ], B:100% (100|100), V:49.8V (49.9|49.8), C: 0W, P:True, F:0.0h, E:-1.0h, H:847W, L:800W
2024-04-06 11:31:16,969:INFO: INV: AC:801.5W, AC_Prediction: 801.2W, DC:843.6W, DC_prediction: 843.3W (0.0|0.0|427.3|416.4), L:1600.0W [1600W]
2024-04-06 11:31:16,971:INFO: SMT: T:ShellyEM3 P:-480.3W [ -479.2,-416.8,-482.0,-481.9,-482.2,-480.6 ] Predict: -490.8W
2024-04-06 11:31:16,974:INFO: Direct connected panels (0.0W) can't cover demand (310.4W), trying to get rest from hub.
2024-04-06 11:31:16,974:INFO: Checking if Solarflow is willing to contribute 310.4W ...
2024-04-06 11:31:16,974:INFO: Based on time, solarpower (801.3W) minimum charge power (0W) and bypass state (True), hub could contribute 800.0W - Decision path: 1.2.
2024-04-06 11:31:16,974:INFO: Hub should contribute more (800.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-06 11:31:16,975:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 800, new limit: 800
2024-04-06 11:31:16,975:INFO: Current inverter AC output is higher than configured output limit (ac_limit), reducing limit to {inv_limit}
2024-04-06 11:31:16,975:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-06 11:31:16,975:INFO: Sun: 06:30 - 19:51 Demand: 310.4W, Panel DC: (0.0|0.0), Hub DC: (427.3|416.4), Inverter Limit: 1598.0W, Hub Limit: 800.0W
2024-04-06 11:31:27,028:INFO: SMT triggers limit function: -487.4 -> -506.7
2024-04-06 11:31:27,028:INFO: HUB: S:805.9W [ 795.2,773.8,800.3,810.5,810.8,814.1 ], B:100% (100|100), V:49.8V (49.9|49.8), C: 0W, P:True, F:0.0h, E:-1.0h, H:850W, L:800W
2024-04-06 11:31:27,032:INFO: INV: AC:805.4W, AC_Prediction: 806.6W, DC:847.7W, DC_prediction: 849.0W (0.0|0.0|429.3|418.4), L:1600.0W [1600W]
2024-04-06 11:31:27,034:INFO: SMT: T:ShellyEM3 P:-487.8W [ -416.8,-482.0,-481.9,-482.2,-480.6,-487.4 ] Predict: -506.7W
2024-04-06 11:31:27,038:INFO: Direct connected panels (0.0W) can't cover demand (299.9W), trying to get rest from hub.
2024-04-06 11:31:27,038:INFO: Checking if Solarflow is willing to contribute 299.9W ...
2024-04-06 11:31:27,038:INFO: Based on time, solarpower (805.9W) minimum charge power (0W) and bypass state (True), hub could contribute 800.0W - Decision path: 1.2.
2024-04-06 11:31:27,038:INFO: Hub should contribute more (800.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-06 11:31:27,038:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 800, new limit: 800
2024-04-06 11:31:27,038:INFO: Current inverter AC output is higher than configured output limit (ac_limit), reducing limit to {inv_limit}
2024-04-06 11:31:27,039:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-06 11:31:27,039:INFO: Sun: 06:30 - 19:51 Demand: 299.9W, Panel DC: (0.0|0.0), Hub DC: (429.3|418.4), Inverter Limit: 1598.0W, Hub Limit: 800.0W
```

Is the limitedRise function somehow holding itself back? To me, it looks like the app never sets the inverter to 100%, therefore, the hub reported solar input never goes up to the allowed max of 800W, therefore, the set limit function will never realize, that more power is actually available 🤔

reinhard-brandstaedter, 2024-04-06:
Hi @mavo ,
the limited rise function is a "limited exponential growth" function which uses the max_inverter_limit setting as limit.
In bypass mode this uses the current available solarinput and sets the inverter limit so that it should very close to that allow the solarinput to pass into the inverter AC output.

This example from your logs seems to pretty match this precisely:

```
2024-04-06 08:27:12,525:INFO: HUB: S:629.9W [ 629.3,628.7,630.6,630.1 ], B:100% (100|100), V:50.5V (50.5|50.5), C: 0W, P:True, F:0.0h, E:-1.0h, H:658W, L:800W
...
2024-04-06 08:27:12,534:INFO: Adjusting inverter limit from 629.9W to 635.4W
...
2024-04-06 08:27:12,534:INFO: Sun: 06:30 - 19:51 Demand: 497.3W, Panel DC: (0.0|0.0), Hub DC: (339.2|330.3), Inverter Limit: 1268.0W, Hub Limit: 800.0W
```

The inverter limit is set to 1268W (317W) which is actually higher than what the hub is currently reporting on solar input (629.9W), so in theory (keeping delays in reported data out of the calculation) the inverter will pass though everything it gets from the hub.

Now as with the nature of an exponential function the actual limit 800W (max_inverter_limit) is never precisely reached, maybe something like 799. And here comes the "top edge" that you are reporting in your last log into play: it's approx 800W the hub is feeding to the inverter. The hub's solarInput is slightly above 800W e.g. 805W, you can also see that the hub is not exactly feeding 800W to the inverter (the inverter is e.g. reporting 429.3W and 418.4W and the two y-cable-connected channels). In reality you are not missing any real power, it's all about the minimal skew in measuring and reporting.
Additionally there is another difference in DC and AC (channels report DC, inverter limit is applied to AC)

What you can try is rising the max_inverter_limit to something like 820, that should push your limit on the inverter in bypass mode up.

mavo, 2024-04-06:
Hey @reinhard-brandstaedter,

so I get what you explain on how it should work. But this does not match what I can see.
In the screenshot you can see the time when I used the rise function, the power was maxing at around 630w. When I modified the code to simply use max inverter power, it jumped to the expected ~800w.
[image]

So it looks to me as if something is keeping the inverter too much at pace so that it fails to grow...

MaVo

reinhard-brandstaedter, 2024-04-06:
Also please note this typical bypass behavior:
The hub usually "reduces" reported SolarInput before the bypass is enabled. Once the bypass is active it goes back up:

[image]

mavo, 2024-04-07:
> Also please note this typical bypass behavior:
> The hub usually "reduces" reported SolarInput before the bypass is enabled. Once the bypass is active it goes back up:

Good morning @reinhard-brandstaedter,

that's true, I noticed as well. But my bypass was active since around 9:30am.
Notice the hard drop around this time 
[image]

The one short spike up to ~800 you can see once is where I was fighting the app to convince it to get to full steam.

reinhard-brandstaedter, 2024-04-07:
Please set max_inverter_limit to 920, this will top you out at 800W with the current limit function on max input..

mavo, 2024-04-07:
> Please set max_inverter_limit to 920, this will top you out at 800W with the current limit function on max input..

Thx, will try this. Although it's hard to actually test because I need a very sunny day and bypass enabled. But I'll try anyway.

One question for this: As I still use my hms-1600-4t I double the value, meaning 1840, correct?
Does this in any way allow an overpowering of the inverter, in case the house consumption is high whilst preparing lunch in combination with high radiation from the sun? 🤔 

MaVo

reinhard-brandstaedter, 2024-04-07:
No, keep it at 920 do not double it.
This impacts the top off of the limit function and the per-channel input. In theory if you get 400W input per channel you will get a 1600W total output, but
there is another safety that throttles the overall limit based on the AC output (if this goes over a safe zone). you can see this in your last log even.

mavo, 2024-04-15:
Is this still required after the latest changes on master?

mavo, 2024-04-17:
Seems to work without the setting to 920 now, thx!

closed 2024-04-17 by mavo