reinhard-brandstaedter/solarflow-control #195: Dev: Direct panel limit does not increased in the morning offset

tuxianerDE, 2024-04-07:
Hi,

today morning I observed that whilst in offset + direct panels the limit did not get increased whilst the battery was anyways in CT mode.

Attached a screenshot 
[image]

I have also extracted the log, once the offset was over the limit directly jumped (offset value sunrise 240). Expectation would be that the limit gets increased as the direct panels start to increasingly produce power.

Ideally this should be the case both scenarions:
1) When batteries in CT mode the more we get from the direct panels the better
2) When not in CT mode but running actively on battery (SoC != 0) the more we can use the direct panels the less we have to use whatever is left of the battery. 

[solarFlow-directPanelIssue.log](https://github.com/reinhard-brandstaedter/solarflow-control/files/14896898/solarFlow-directPanelIssue.log)

reinhard-brandstaedter, 2024-04-07:
It is actually trying to limit the grid feed-in to zero at this time as it still has to assume power is coming (also) from batteries.
Can you confirm this if you smartmeter readings at this time are around/below zero?

reinhard-brandstaedter, 2024-04-07:
I think it behaves correctly, see this log snippet (5 minutes before the Offset time ended at 10:53):
During offset it kept to the zero feed in policy (SMT values are close to zero).

Then the demand increased to 10kW, and we tried to increase the inverter limit to it's maximum.
At 10:48:01,634 it sent the command to increase the inverter limit to 1700
Only 4 seconds later the demand dropped again and another 3 seconds later the demand rose again. The script tried to set the inverter limit correctly again to 1700W. But then demand dropped again only after 3 seconds...

```
2024-04-07 10:47:05,475:INFO: HUB: S:238.5W [ 244.6,244.6,244.8,244.2,240.6,238.5 ], B: 98% (98|99|98|99), V:51.2V (51.2|51.2|51.0|51.2), C: 232W, P:False, F:304.6h, E:50.2h, H: 0W, L: 0W
2024-04-07 10:47:05,475:INFO: INV: AC:222.9W, AC_Prediction: 222.9W, DC:234.7W, DC_prediction: 234.7W (123.9|127.5|0.0|0.0), L:523.5W [1500.0W]
2024-04-07 10:47:05,480:INFO: SMT: T:Smartmeter P:-9.0W [ 5.0,5.0,4.2,6.5,4.2,4.4 ] Predict: 6.5W
2024-04-07 10:47:05,630:INFO: Triggering telemetry update: iot/73bkTV/68aE3bUQ/properties/read
2024-04-07 10:48:01,628:INFO: Rapid rise in demand detected, clearing buffer!
2024-04-07 10:48:01,629:INFO: HUB: S:216.2W [ 229.8,229.8,228.1,225.2,219.9,216.2 ], B: 98% (98|99|98|99), V:51.1V (51.2|51.1|50.9|51.1), C: 201W, P:False, F:304.7h, E:50.3h, H: 0W, L: 0W
2024-04-07 10:48:01,629:INFO: INV: AC:221.2W, AC_Prediction: 221.2W, DC:232.8W, DC_prediction: 232.8W (113.7|123.4|0.0|0.0), L:523.5W [1500.0W]
2024-04-07 10:48:01,630:INFO: SMT: T:Smartmeter P:10889.0W [ 10889.0,10889.0 ] Predict: 10889.0W
2024-04-07 10:48:01,630:INFO: Direct connected panels (237.1W) can't cover demand (11110.2W), trying to get rest from hub.
2024-04-07 10:48:01,631:INFO: Checking if Solarflow is willing to contribute 10873.1W ...
2024-04-07 10:48:01,632:INFO: Based on time, solarpower (216.2W) minimum charge power (500W) and bypass state (False), hub could contribute 850.0W - Decision path: 2.1.
2024-04-07 10:48:01,632:INFO: Hub should contribute more (850.0W) than what we currently get from panels (237.1W), we will use the inverter for fast/precise limiting!
2024-04-07 10:48:01,633:INFO: Battery hasn't fully charged for 304.7 hours! To ensure it is fully charged at least every 238hrs not discharging now!
2024-04-07 10:48:01,633:INFO: Not setting solarflow output limit to 0.0W as it is identical to current limit!
2024-04-07 10:48:01,634:INFO: Setting inverter output limit to 1700W (1 min moving average of 425W x 4)
2024-04-07 10:48:01,635:INFO: Sun: 06:53 - 20:13 Demand: 11110.2W, Panel DC: (113.7|123.4), Hub DC: (0.0|0.0), Inverter Limit: 1700.0W, Hub Limit: 0.0W
2024-04-07 10:48:01,635:INFO: SMT triggers limit function: 10889.0 -> 10889.0: executed
2024-04-07 10:48:05,475:INFO: HUB: S:211.9W [ 225.7,225.7,224.5,221.0,217.0,211.9 ], B: 98% (98|99|98|99), V:51.1V (51.2|51.1|50.9|51.1), C: 193W, P:False, F:304.7h, E:50.3h, H: 0W, L: 0W
2024-04-07 10:48:05,475:INFO: INV: AC:221.2W, AC_Prediction: 221.2W, DC:232.8W, DC_prediction: 232.8W (113.7|123.4|0.0|0.0), L:523.5W [1500.0W]
2024-04-07 10:48:05,476:INFO: SMT: T:Smartmeter P:10868.0W [ 10887.0,10887.0,10886.6 ] Predict: 10886.6W
2024-04-07 10:48:05,631:INFO: Triggering telemetry update: iot/73bkTV/68aE3bUQ/properties/read
2024-04-07 10:48:05,745:INFO: Rapid drop in demand detected, clearing buffer!
2024-04-07 10:48:05,746:INFO: HUB: S:211.9W [ 225.7,225.7,224.5,221.0,217.0,211.9 ], B: 98% (98|99|98|99), V:51.1V (51.2|51.1|50.9|51.1), C: 193W, P:False, F:304.7h, E:50.3h, H: 0W, L: 0W
2024-04-07 10:48:05,746:INFO: INV: AC:221.2W, AC_Prediction: 221.2W, DC:232.8W, DC_prediction: 232.8W (113.7|123.4|0.0|0.0), L:523.5W [1500.0W]
2024-04-07 10:48:05,747:INFO: SMT: T:Smartmeter P:33.0W [ 33.0,33.0 ] Predict: 33.0W
2024-04-07 10:48:05,747:INFO: Direct connected panels (237.1W) can't cover demand (254.2W), trying to get rest from hub.
2024-04-07 10:48:05,747:INFO: Checking if Solarflow is willing to contribute 17.1W ...
2024-04-07 10:48:05,748:INFO: Based on time, solarpower (211.9W) minimum charge power (500W) and bypass state (False), hub could contribute 17.1W - Decision path: 2.1.
2024-04-07 10:48:05,748:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 0, new limit: 17
2024-04-07 10:48:05,748:INFO: Solarflow is willing to contribute 0.0W!
2024-04-07 10:48:05,749:INFO: Adjusting inverter limit from 123.4W to 229.3W
2024-04-07 10:48:05,749:INFO: Direct connected panel limit is 229W.
2024-04-07 10:48:05,750:INFO: Setting inverter output limit to 916W (1 min moving average of 229W x 4)
2024-04-07 10:48:05,751:INFO: Sun: 06:53 - 20:13 Demand: 254.2W, Panel DC: (113.7|123.4), Hub DC: (0.0|0.0), Inverter Limit: 916.0W, Hub Limit: 0.0W
2024-04-07 10:48:05,751:INFO: SMT triggers limit function: 33.0 -> 33.0: executed
2024-04-07 10:48:08,029:INFO: Rapid rise in demand detected, clearing buffer!
2024-04-07 10:48:08,031:INFO: HUB: S:211.9W [ 225.7,225.7,224.5,221.0,217.0,211.9 ], B: 98% (98|99|98|99), V:51.1V (51.2|51.1|50.9|51.1), C: 193W, P:False, F:304.7h, E:50.3h, H: 0W, L: 0W
2024-04-07 10:48:08,031:INFO: INV: AC:221.2W, AC_Prediction: 221.2W, DC:232.8W, DC_prediction: 232.8W (113.7|123.4|0.0|0.0), L:523.5W [1500.0W]
2024-04-07 10:48:08,032:INFO: SMT: T:Smartmeter P:1635.0W [ 1635.0,1635.0 ] Predict: 1635.0W
2024-04-07 10:48:08,032:INFO: Direct connected panels (237.1W) can't cover demand (1856.2W), trying to get rest from hub.
[... cut ...]
2024-04-07 10:48:09,090:INFO: SMT: T:Smartmeter P:6567.0W [ 6567.0,6567.0 ] Predict: 6567.0W
```

I think with such fast-pace on/off we can't keep up.
And during offset times (basically a manual night) we kind of have to maintain the zero-feed-in policy because it would mean feeding in from battery if you "open" the inverter and would still allow the battery to feed in as well.
Only when the hub is currently in charge through and it's output is limited to 0 we could have this case as "inverter open".

tuxianerDE, 2024-04-08:
Here is the SMT values, I am not sure what the definition of close to 0, but I would stipulate we are gravitating around it.

That said, I agree that during the offset in the morning the 0 feed in policy is suitable WHEN on battery, as I was in CT (and we can add the SoC = 0) situation that would be unnecessary steering.

[image]

closed 2024-04-15 by tuxianerDE