reinhard-brandstaedter/solarflow-control #201: Additional "MAX_FEED_IN" config

boennemann, 2024-04-09:
Hey @reinhard-brandstaedter,

thanks for this amazing project.

I'm seeing the `MAX_INVERTER_LIMIT` variable, but I was wondering whether it's possible to introduce something like a `MAX_FEED_IN`, too.

What I'm aiming for is something like this:

MAX_INVERTER_LIMIT = 1000 (Let's pretend this is what my inverter supports)
MAX_FEED_IN = 600 

Now what I'm aiming for is that the inverter would give up to 1000W to the house when we know the house is consuming that. If the load in the house is low and we know we're feeding into the grid the inverter limit should be limited in such a way that only MAX_FEED_IN goes to the grid.

Examples using above config values and assuming we have a full battery and >1kW available:

Load: 1000 Inverter Limit: 1000
Load: 800 Inverter Limit: 1000 (feed in: 200)
Load: 400 Inverter Limit: 1000 (feed in: 600)
Load: 100 Invert Limit: 700 (feed in: 600)

I hope this makes sense!

tuxianerDE, 2024-04-11:
Hey,

that is in parts a philsophical question :)

What we have done:
- When the SolarFlow feeds data into house (i.e. during night and offset) the Inverter will limit it to the proper usage to ensure that no battery power is feed into grid.
- During the day if there is high demand, and direct panels supply power but not enough AND the panels of the SolarFlow hub produce more that the max_charge_power this excess energy gets feed to the house (as long as it is needed. If the consumption of the house drops the Hub limit will be set to 0 again and the direct panels will supply power.
- If the house does not consume more power then the direct panels provide (or less) our decision was that would not limit the feed to grid when the battery/solarhub is NOT used.

The last point basically goes back to the original idea of the "Balkonkraftwerke" that excess energy gets fed into the grid. We wanted to ensure that we maximize the return on investment or any component where a proper steering makes a difference and that is when the solarFlow Hub feeds in energy.

boennemann, 2024-04-11:
Hey @tuxianerDE, 

thanks for the extensive reply.
My suggestion for this flag is coming from the same place I feel like.

My thinking here (and obviously everything I'm describing is all just hypothetical 🤪):

Right now there is a legal limit of 600W feed-in.
I'd feel confident to surpass this 600W limit within my own house and on my own risk (I know there is an extra line with extra fuse, everything 2,5mm² with cables, yadda yadda).

So as long as I'm using the energy directly within my own house I'd like to keep the inverter limit higher than 600.
If I can't use it up right now (battery full, not at home, no heatpump required etc.) I would want to ensure that I'm feeding a maximum of 600W into the grid on the phase my Balkonkraftwerk is connected to, to not exceed this legal limit.

Does that make sense?