reinhard-brandstaedter/solarflow-control #202: Config Infos and genereal questions

busaku, 2024-04-10:
Inside the Readme and config part are a little typos:
```
# Smartmeter Type: either Smartmeter (generic, Tasmota, Hichi, ...), PowerOpti, Shelly3EM
smartmeter_type = ...
```
But Shelly3EM needs to be ShellyEM3 inside the config

```
[shellyem3]
# The MQTT base topic your Shelly 3EM (Pro) is posting it's telemetry data to
# Note: you have to configure your Shelly to use MQTT
base_topic = shellies/shellyem3...
rapid_change_diff = 200
```
We need to set up rapid_change_diff, but its missed inside the condif files. And its not described anywhere, what is that setting for?

Unfortunately it's not allowed to make your own commits, otherwise I would have simply made a PR with the changes :)

And a last question. Inside Homeassistant i now have the possibility to change the Bypass.
Is this the same setting as inside the app? Because the app shows me a warning not to switch this setting too often?
So, did solarflow-control "take care" of my inverter and the solarflow-hub? 
I mean not to write the Inverter Limit "hard" and doesnt switch the Bypass setting often?

Last question:
Ist it possible to disable the full charge cycle?

Thanks for your work here, the software seems to fit my needs perfectly <3

boennemann, 2024-04-10:
Hey Busaku,

I was also wondering whether it would be possible to have prose explanations for the config values.
Some (most) are self explanatory, but this probably depends on the depth of knowledge of the reader anyway.

E.g. `inverter_min_limit` I don't understand at all.

I'd be willing to contribute explanations as well :)

busaku, 2024-04-10:
Yes, and i got 
[image]

with this settings:
```# the max output power of your inverter, used to calculate correct absolute values
inverter_max_power = 1600

[control]
min_charge_power = 10
max_discharge_power = 800
max_inverter_limit = 800
limit_inverter = true
inverter_min_limit = 10```

it thought max_inverter_limit = 800 sets the output limit to 800. (calculated, not just 50% and all input MPPTs are droppet to 50% even two (zendure) are sending no input)

tuxianerDE, 2024-04-11:
Hey folks,

to your points and let me see if I get all of them.

1) rapid change diff - Inside the control skript is a detection if demands in your home consumption change. If this happend drastically (like a heater/water heater) the rapid change condition gets triggered and it flushes the projection. Setting the value here change the threshold when this condition is met and thus the projection gets flushed.

2) inverter_min_limit - We noticed through our tests that the inverter must not be steered to a 0W limit. As this throttels down the inverter and switches it off. We noticed that a min value is necessary in this case 10W that gets applied so that the script does not limit the inverter and switches it off.

3) The charge through cycle cannot be switched off. Even through Zendure claims that there BMS forced a full charge/zero charge cycle from time to time noone has so far seen this in the telegram group or among us working on this project here. Hence we have introduced this in order to maintain healthy batteries.

4) Inverter_max_power gets detected now automatically when starting up the script. You will see this in the logs there is a line stating "max power detected to be blabla". The skript allows a maximum of 150% of the max power of the inverter detected.

hope this shines first light on this.