reinhard-brandstaedter/solarflow-control #204: Home Assistant Addon

boennemann, 2024-04-10:
Hey all,

I'm planning to run the docker container as a Home Assistant Add-on directly on my Home Assistant Yellow – as I understood an add-on is anyway just a super slim layer of metadata on top of docker containers.

Before I dive into that I wanted to open the discussion already.

1) Has anyone done this before and can share something?
2) Are there any reasons I'm missing why this is either a bad idea or not possible
3) Once I've got it working I'd love to contribute the setup back, ideally getting this into Home Assistant as an Add-On without the need to fork.

Let me know what you think!

busaku, 2024-04-10:
I would love an direct HA-Addon :)

boennemann, 2024-04-11:
In the end it's really just a matter of 3 steps

1) copying the contents of the `src/` folder to `/addons/solarflow-control`
2) adding config.yaml, here is mine: 
```yml
name: "Solarflow Control"
description: "Solarflow Control"
version: "0.0.1"
slug: "solarflow_control"
init: false
arch:
  - aarch64
  - amd64
  - armhf
  - armv7
  - i386
 ```
3) I added this one line to the Dockerfile: `COPY ./config.ini /solarflow/config.ini`

busaku, 2024-04-11:
So it worked without any extras?
You should publish it into official HA-Addon Repo :)
( After the okay from all contributor here :) )

busaku, 2024-04-11:
Wow, that was easy!
Love it :)  

[image]

boennemann, 2024-04-14:
I hope I get to mapping out the config values soon – it's anyway annoying to rebuild the docker container for every config change. Afterwards it might be almost universally applicable.

Maybe a sprinkle of metadata on top and we're good to go :)

busaku, 2024-04-30:
@boennemann any updates here? if not, i would give it a try :)

Sweti934, 2024-05-06:
Hi all thank you for the amazing work so far!! Really awesome. To get it right - I could integrate this into home assistant ? I have all the things reporting already to my home assistant mosquito Mqtt. Would I have to change the topics or could I change the topics inside the docker where this here listeners to?

dfigus, 2024-09-02:
Hi all,

I just started with a HA addon here [dfigus/addon-solarflow-control](https://github.com/dfigus/addon-solarflow-control). It's still in an early state, but you can already configure everything via addon configs and it's runny already for a few days without issues in my setup. 

If you want to give it a try and provide feedback or even contribute, you can install it via my EDGE addon repository: 
[https://github.com/dfigus/hassio-addons-edge](https://github.com/dfigus/hassio-addons-edge)

Once I added a bit of docs, I will promote it to a beta ([https://github.com/dfigus/hassio-addons-beta](https://github.com/dfigus/hassio-addons-beta))  and finally release it ([https://github.com/dfigus/hassio-addons](https://github.com/dfigus/hassio-addons)).

Sweti934, 2024-09-02:
Thanks @dfigus  - Highly appreciated! I am so curious how it works out!

jschroeter, 2024-09-14:
I wanted to give this a try but I'm stuck at allowing anonymous MQTT access to my broker running in HA, how did you guys do this? I do see in the logs that Hyper wants to connect but the connection gets refused due to the credentials.

dfigus, 2024-09-15:
@jschroeter I use the EMQX add-on as this allows anonymous login. The Mosquito add-on does no longer allow anonymous login. The best option would be to spoof the login credentials the hub is sending to the broker, but I didn't have the time to look into this so far...

reinhard-brandstaedter, 2024-09-15:
@jschroeter if the hyper uses the same (stupid) credentials as the rest of the SF devices then you can calculate the password it from the device ID and add that user to your MQTT broker.

reinhard-brandstaedter, 2024-09-15:
@jschroeter @dfigus as sooner or later people will find out the password of their SF/Hyper device anyway here is how you can calculate it from your device ID, without snooping (verified only on SF devices):

```
#!/usr/bin/python
import hashlib
device_id = "<DEVICEID>"
md5_hash = hashlib.md5(device_id.encode()).hexdigest().upper()[8:24]
```

jschroeter, 2024-09-15:
Awesome, will give it a try! Username is the device ID?

reinhard-brandstaedter, 2024-09-15:
username = device ID, correct

heisly, 2024-12-14:
I'm stucking when taking my Hyper 2000 offline and try to connect it to my local MQTT Broker (mosquito on LoxBerry).
First of all I'm not sure about the product ID I should take:
export SF_PRODUCT_ID=“ja72U0“
export SF_PRODUCT_ID=“ja72U0ha"

Then before disconnecting, I setup all the things like WIFI SSID, WIFI PWD, but also
export MQTT_USER=“loxberry_user“
export MQTT_PWD=“loxberry_pwd“

But I didn't get the topic in LoxBerrys Incoming overview.
Then I read about a hardcoded password, so now I'm completely confused.

From my point of view it doesn't make sense to setup MQTT_USER and MQTT_PWD if the Hyper 2000 can only use its hardcoded user (=device id?) and password??

What am I making/thinking wrong?
Anyone have a suggestion?

heisly

reinhard-brandstaedter, 2024-12-14:
If you run solarflow-bt-manager with the -i option you need to provide the SF_PRODUCT_ID.
If you run with the -d option (to disconnect) you need to provide the MQTT Host and Port.
If you run it in proxy mode you need to provide the MQTT user/pwd as well as the script will talk to your broker.
If you just disconnect it the hub will talk to your broker with its default user/pwd, BUT solarflow-control still might need also a user and password to talk to the mqtt broker.

reinhard-brandstaedter, 2024-12-15:
See [here](https://github.com/reinhard-brandstaedter/solarflow-control/tree/master/doc) this is a general overview, which properties can be set and how.

heisly, 2024-12-15:
This was the missing puzzle, many thanks to your support!
It was a hard one, but now I have it up and running!

DB6CP, 2024-12-19:
Just started Using the Addon. Nice to get a clean settup and integration. However i ran into some Problems which may be present in the basis code. 

TypeError: unsupported operand type(s) for -: 'int' and 'NoneType'
[09:51:49] INFO: Running Solarflow Control
Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 82, in <module>
    or int(os.environ.get('MAX_INVERTER_INPUT',MAX_INVERTER_LIMIT - MIN_CHARGE_POWER))
                                               ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~

Seems like MIN_CHARGE_POWER is not set or it is not recognized properly. Can anyone help ?

dfigus, 2025-05-07:
@Grieshof In my opinion the main problem is that you are not using OpenDTU or AhoyDTU for controlling the inverter. Solarflow-control from @reinhard-brandstaedter only supports those two DTUs which send values to specific MQTT topics. For your setup you would need to simulate such a DTU in your MQTT broker in order to make this work. 

Also from the MQTT screenshot above you did not enable MQTT in your Shelly Pro 3EM and also not configure the addon-on to use this (it's to be configured as a generic smartmeter and the Generic Smartmeter Base Topic and Current Accessor are empty). But this doesn't help if you don't have the DTU topic solved.

martin-nrw-85, 2025-05-07:
Hi, ich hoffe, wir kommen auf Deutsch gut miteinander klar 😉.
Ich habe den Hub1200 von der Cloud abgekoppelt und lasse ihn jetzt über WLAN mit meinem MQTT-Broker kommunizieren.

Der Buzzer, das “Home Output Limit” des Hubs, der minimale und maximale Ladestand sowie die maximale Leistung des Wechselrichters (siehe Foto) lassen sich darüber steuern.
 
[image]

Wenn du Fragen hast, kann ich dir hier und da vielleicht weiterhelfen 😉🙌

Grieshof, 2025-05-07:
Ah ok, danke, aber das heißt über die Cloud hast du eine relativ große Verzögerung für die Steuerung zur anpassung für eine 0-Einspeisung oder ?

busaku, 2025-05-07:
Welchen Wechselrichter hast du denn und wie willst du den regeln?

martin-nrw-85, 2025-05-07:
> Ah ok, danke, aber das heißt über die Cloud hast du eine relativ große Verzögerung für die Steuerung zur anpassung für eine 0-Einspeisung oder ?

Cloud??? Er läuft bei mir lokal ohne Cloud.

martin-nrw-85, 2025-05-07:
> Welchen Wechselrichter hast du denn und wie willst du den regeln?

Ich Regel nicht den WR sondern den Hub Ausgang.

Grieshof, 2025-05-07:
Wechselrichter: hoymiles hms-800w-2t   also 800 Watt mit integriertem DTU und Wlan. ;)  was ich so gelesen habe maximal alle 30sek update der Daten und nie auf 0 stellen aber da ich eh immer mindestens 120 Watt Verbrauch habe, wird das auch nicht passieren..

Grieshof, 2025-05-07:
> Hi, ich hoffe, wir kommen auf Deutsch gut miteinander klar 😉. Ich habe den Hub1200 von der Cloud abgekoppelt und lasse ihn jetzt über WLAN mit meinem MQTT-Broker kommunizieren.
>
> Der Buzzer, das [...]

Mich würde es intressieren wie du das geschafft hast ? ;)  ich versuche  grad die Werte aus dem JSON von der MQTT Nachricht als entitäten zu kriegen,...   Wäre super wenns du ne Anleitung und die Codes dazu hättest, wenn du Zeit hättest ?  ;)

martin-nrw-85, 2025-05-07:
> Mich würde es intressieren wie du das geschafft hast ? ;)  ich versuche  grad die Werte aus dem JSON von der MQTT Nachricht als entitäten zu kriegen,...   Wäre super wenns du ne Anleitung und die [...]

Eigentlich hat Reinhard alles schon hier src/solarflow/homeassistant abgelegt. Ich füge noch ein Foto meiner mqtt Datei an. 

Ich habe damals das Problem gehabt das mein WR mit der identischen Leistung auf den MPPT-Trackern nicht klar kam. Daher habe ich gebastelt bis ich es hinbekommen hatte. 

[image]