reinhard-brandstaedter/solarflow-control #206: High waves in limiting the Inverter

busaku, 2024-04-11:
Instead of doing a "flat" wave of the inverter limit (doesn't care if i disable inverter limit), the limit jumps between really low and really hight numbers (e.g. 80 - 2000 - 80 - 600 - 400 - 80 - 2000).

Here are my settings:
```
[ahoydtu]
# The MQTT base topic your AhoyDTU reports to (as configured in AhoyDTU UI)
base_topic = inverter
# The inverter ID in AhoyDTU: typically 1 for the first inverter
inverter_id = 0
# List of indices of the inverter channels/ports (as reported in the DTU) that the Solarflow Hub is connected to
# typically the index starts at 1 as 0 is the output channel of the inverter
# e.g. 1,3 or 3 or [1,3]
sf_inverter_channels = [3,4]

# the max output power of your inverter, used to calculate correct absolute values
inverter_max_power = 1600

# The name of the inverter in AhoyDTU
inverter_name = BKW_Hoymiles

[shellyem3]
# The MQTT base topic your Shelly 3EM (Pro) is posting it's telemetry data to
# Note: you have to configure your Shelly to use MQTT
base_topic = shellies/shellyem3-<mysehlly>
rapid_change_diff = 250

[control]
min_charge_power = 50
max_discharge_power = 500
max_inverter_limit = 800
limit_inverter = true.  # doesn't take any effect (?)
inverter_min_limit = 10
```

<details>
<summary>Logfiles (long)</summary>

```
2024-04-11 05:38:45,348:INFO: HUB: S:0.0W [ 0.0 ], B: 19% (19|20), V:48.3V (48.4|48.3), C: -35W, P:False, F:12.1h, E:-1.0h, H: 34W, L:800W
2024-04-11 05:38:45,349:INFO: INV: AC:81.6W, AC_Prediction: 81.6W, DC:85.7W, DC_prediction: 85.7W (8.2|8.0|24.3|45.2), L:555.2W [1600W]
2024-04-11 05:38:45,358:INFO: SMT: T:ShellyEM3 P:204.5W [ 285.0,281.7,273.3,278.8,199.3,204.5 ] Predict: 189.4W
2024-04-11 05:38:45,368:INFO: Direct connected panels (16.2W) can't cover demand (271.0W), trying to get rest from hub.
2024-04-11 05:38:45,369:INFO: Checking if Solarflow is willing to contribute 254.8W ...
2024-04-11 05:38:45,371:INFO: Based on time, solarpower ( 0.0W) minimum charge power (50W) and bypass state (False), hub could contribute 254.8W - Decision path: 2.1.
2024-04-11 05:38:45,372:INFO: Hub should contribute more (254.0W) than what we currently get from panels (16.2W), we will use the inverter for fast/precise limiting!
2024-04-11 05:38:45,373:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-04-11 05:38:45,374:INFO: Setting inverter output limit to 508W (1 min moving average of 127W x 4)
2024-04-11 05:38:45,377:INFO: Sun: 06:46 - 20:24 Demand: 271.0W, Panel DC: (8.2|8.0), Hub DC: (24.3|45.2), Inverter Limit: 508.0W, Hub Limit: 800.0W
2024-04-11 05:38:45,396:INFO: SMT triggers limit function: 204.6 -> 189.4
2024-04-11 05:38:45,397:INFO: Rate limit on trigger function, last call was only 0.0s ago!
2024-04-11 05:38:47,328:INFO: SMT triggers limit function: 212.2 -> 194.5
2024-04-11 05:38:47,328:INFO: Rate limit on trigger function, last call was only 2.0s ago!
2024-04-11 05:38:47,346:INFO: SMT triggers limit function: 206.8 -> 190.9
2024-04-11 05:38:47,347:INFO: Rate limit on trigger function, last call was only 2.0s ago!
2024-04-11 05:38:47,366:INFO: SMT triggers limit function: 204.7 -> 189.5
2024-04-11 05:38:47,367:INFO: Rate limit on trigger function, last call was only 2.0s ago!
2024-04-11 05:38:48,332:INFO: SMT triggers limit function: 199.4 -> 185.9
2024-04-11 05:38:48,333:INFO: Rate limit on trigger function, last call was only 3.0s ago!
2024-04-11 05:38:48,352:INFO: SMT triggers limit function: 193.7 -> 182.1
2024-04-11 05:38:48,352:INFO: Rate limit on trigger function, last call was only 3.0s ago!
2024-04-11 05:38:48,371:INFO: SMT triggers limit function: 191.8 -> 180.9
2024-04-11 05:38:48,372:INFO: Rate limit on trigger function, last call was only 3.0s ago!
2024-04-11 05:38:50,339:INFO: SMT triggers limit function: 189.6 -> 179.4
2024-04-11 05:38:50,340:INFO: Rate limit on trigger function, last call was only 5.0s ago!
2024-04-11 05:38:50,369:INFO: SMT triggers limit function: 192.4 -> 181.3
2024-04-11 05:38:50,370:INFO: Rate limit on trigger function, last call was only 5.0s ago!
2024-04-11 05:38:50,399:INFO: SMT triggers limit function: 193.5 -> 182.0
2024-04-11 05:38:50,400:INFO: Rate limit on trigger function, last call was only 5.1s ago!
2024-04-11 05:38:53,942:INFO: DTU triggers limit function: 85.7 : -17.3
2024-04-11 05:38:53,943:INFO: Rate limit on trigger function, last call was only 8.6s ago!
[... 569 lines cut ...]
2024-04-11 05:42:59,211:INFO: Rate limit on trigger function, last call was only 6.0s ago!
2024-04-11 05:42:59,233:INFO: SMT triggers limit function: 81.4 -> 64.0
2024-04-11 05:42:59,233:INFO: Rate limit on trigger function, last call was only 6.0s ago!
2024-04-11 05:43:01,487:INFO: SMT triggers limit function: 79.9 -> 63.0
2024-04-11 05:43:01,487:INFO: Rate limit on trigger function, last call was only 8.3s ago!
2024-04-11 05:43:01,514:INFO: SMT triggers limit function: 79.4 -> 62.7
2024-04-11 05:43:01,515:INFO: Rate limit on trigger function, last call was only 8.3s ago!
2024-04-11 05:43:01,533:INFO: SMT triggers limit function: 79.3 -> 62.6
2024-04-11 05:43:01,533:INFO: Rate limit on trigger function, last call was only 8.3s ago!
2024-04-11 05:43:08,960:INFO: DTU triggers limit function: 179.5 : 63.1
2024-04-11 05:43:08,961:INFO: HUB: S:0.0W [ 0.0 ], B: 19% (19|19), V:47.9V (47.9|47.9), C:-200W, P:False, F:12.2h, E:-1.0h, H:191W, L:800W
2024-04-11 05:43:08,992:INFO: INV: AC:296.1W, AC_Prediction: 170.6W, DC:311.5W, DC_prediction: 179.5W (9.9|9.5|101.5|190.8), L:256.0W [1600W]
2024-04-11 05:43:09,006:INFO: SMT: T:ShellyEM3 P:79.2W [ 170.5,4.5,-5.8,229.0,14.0,79.3 ] Predict: 62.6W
2024-04-11 05:43:09,033:INFO: Direct connected panels (19.4W) can't cover demand (233.2W), trying to get rest from hub.
2024-04-11 05:43:09,034:INFO: Checking if Solarflow is willing to contribute 213.8W ...
2024-04-11 05:43:09,036:INFO: Based on time, solarpower ( 0.0W) minimum charge power (50W) and bypass state (False), hub could contribute 213.8W - Decision path: 2.1.
2024-04-11 05:43:09,037:INFO: Hub should contribute more (213.0W) than what we currently get from panels (19.4W), we will use the inverter for fast/precise limiting!
2024-04-11 05:43:09,037:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 800, new limit: 800
2024-04-11 05:43:09,038:INFO: Setting inverter output limit to 424W (1 min moving average of 106W x 4)
2024-04-11 05:43:09,040:INFO: Sun: 06:46 - 20:24 Demand: 233.2W, Panel DC: (9.9|9.5), Hub DC: (101.5|190.8), Inverter Limit: 424.0W, Hub Limit: 800.0W
```

</details>

Any idea what we can do here?

reinhard-brandstaedter, 2024-04-11:
likely a duplicate of what @tuxianerDE is ecperiencing.
Can you confirm that this mostly happens during low radiation on any direct connected panels but not when more sun is out?

reinhard-brandstaedter, 2024-04-11:
FYI the readon why the limit is raised fast esp in low light is that typically radiation increases the power delivered very fast. To avoid cut off this limit function is used:

[image]

busaku, 2024-04-11:
Hmh no, or better maybe.
Herse a screenshot from this night. So no input on the direct connected PV Panels.

[image]

But yes, the log files are from this morning, low light on direct connected PV Panels :)
So, okay i got it but why i drops so fast again?

And is it normal, that the **limit_inverter** setting will be ignored?
(Iam thinking about the battery-health if the inverter limit rise and drops so fast)

reinhard-brandstaedter, 2024-04-11:
The "limit_inverter" setting is indeed a leftover from the past. Where a pure limiting of the solarflow output was possible. However this is now obsolete, because with what solarflow-control is doing now can't be done with purely limiting the hub, so the inverter is always involved.

To you chart, can you maybe also provide what was happening at this time on the demand side? So maybe data from your smartmeter as chart? Did it go up/down/below zero?

There is always a fine balance between delivering power as fast as possible when needed and performing not too many limiting steps and not feeding in energy to the grid from battery.
Imagine one of the most challenging situations: a induction power stove that pulsates up/down within minutes where an algorithm based on multiple input variables tries to cover demand as fast as possible on these changes...and then there is switching delays as well...

Here is an example:
Note that my max_discharge_power is at 400W, thus the difference of the inverterlimit is a bit smaller.
Also note that we are trying to set the limit in a way that also considers the multiple channels of the inverter. a limit of 800W on a 4-channel inverter means max. 200W per channel. If only 2 channels are delivering you need to set the limit to 1600W to actually get 800W (400W per channel).

[image]

busaku, 2024-04-11:
Yes, here it is:
Blue = Shelly3EM
Purple = Zendure to Inverter
0-1 am
[image]

0-0:15 am in detail:
[image]

It looks like the target randomly rised or dropped. No big change in the consumption before the change of zendure->Inverter
[image]

In general it seems that the target "consumption" is a little bit low (around -80 watt)
Is it possible to adjust the target rate?
[image]

Than, my last question is it possible to set a maximal inverter output level?
So if the battery is full and we have full solar, its not allowed to "feed in" more than 600 watt in germany.
I would be awesome to put an absolut maximum output level :)

reinhard-brandstaedter, 2024-04-14:
Hi @busaku,
I added more precision regarding AC/DC conversion using the inverters efficiency telemetry. At least here this eliminates some of the overshooting. Can you try running the latest dev build?

busaku, 2024-04-15:
@reinhard-brandstaedter thanks, i will provide you feedback :)

busaku, 2024-04-15:
@reinhard-brandstaedter we got an error (sometimes)
```
2024-04-15 10:07:05,050:INFO: Reading battery target mode: discharging
2024-04-15 10:07:05,051:INFO: Reading last full time: 2024-04-14 20:19:47
2024-04-15 10:07:44,699:INFO: HUB: S:510.9W [ 509.4,509.4,508.3,510.9 ], B: 39% (-1), V:-1.0V (-1.0), C: 494W, P:False, F:13.8h, E:-1.0h, H: -1W, L: -1W
2024-04-15 10:07:44,699:INFO: INV: AC:501.3W, AC_Prediction: 501.3W, DC:527.5W, DC_prediction: 527.5W (264.3|262.4|0.0|0.0), L: 0W [1600W]
2024-04-15 10:07:44,699:INFO: SMT: T:ShellyEM3 P:-150.6W [ -145.1,-145.1,-166.9 ] Predict: -166.9W
2024-04-15 10:07:44,700:INFO: Direct connected panels (500.5W) can cover demand (313.6W)
2024-04-15 10:07:44,700:INFO: Setting solarflow output limit to 0.0W
2024-04-15 10:07:44,701:INFO: Setting inverter output limit to 1008W (1 min moving average of 252W x 4)
2024-04-15 10:07:44,701:INFO: Grid feed in from panels connected directly to inverter: 186.9!
2024-04-15 10:07:44,703:INFO: Sun: 06:38 - 20:30 Demand: 313.6W, Panel DC: (500.5W), Hub DC: (0.0W), Inverter Limit: 1008.0W, Hub Limit: 0.0W
2024-04-15 10:07:44,703:INFO: SMT triggers limit function: -145.1 -> -166.9: executed
2024-04-15 10:07:44,704:INFO: SMT triggers limit function: -143.8 -> -163.8: skipped
2024-04-15 10:07:44,706:INFO: SMT triggers limit function: -142.7 -> -161.2: skipped
2024-04-15 10:07:47,698:INFO: SMT triggers limit function: -141.5 -> -158.6: skipped
2024-04-15 10:07:47,699:INFO: SMT triggers limit function: -139.3 -> -154.9: skipped
2024-04-15 10:07:47,701:INFO: SMT triggers limit function: -137.2 -> -151.7: skipped
2024-04-15 10:07:51,703:INFO: SMT triggers limit function: -135.2 -> -148.6: skipped
2024-04-15 10:07:51,706:INFO: SMT triggers limit function: -130.8 -> -142.8: skipped
2024-04-15 10:07:51,709:INFO: SMT triggers limit function: -126.7 -> -137.6: skipped
2024-04-15 10:07:56,700:INFO: SMT triggers limit function: -116.4 -> -93.4: skipped
2024-04-15 10:07:56,702:INFO: SMT triggers limit function: -112.9 -> -79.7: skipped
2024-04-15 10:07:56,705:INFO: SMT triggers limit function: -109.6 -> -76.6: skipped
2024-04-15 10:08:01,700:INFO: HUB: S:480.4W [ 498.0,498.0,495.6,493.9,492.0,480.4 ], B: 39% (40), V:50.1V (50.1), C: 395W, P:False, F:13.8h, E:-1.0h, H: -1W, L: 0W
2024-04-15 10:08:01,702:INFO: INV: AC:427.1W, AC_Prediction: 427.1W, DC:449.6W, DC_prediction: 449.6W (186.3|186.1|0.0|0.0), L:1008.0W [1600W]
2024-04-15 10:08:01,702:INFO: SMT: T:ShellyEM3 P:-23.0W [ -101.0,-101.0,-106.5,-60.1 ] Predict: -60.1W
2024-04-15 10:08:01,702:INFO: Direct connected panels (353.8W) can cover demand (273.7W)
2024-04-15 10:08:01,703:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 0, new limit: 0
2024-04-15 10:08:01,703:INFO: Setting inverter output limit to 708W (1 min moving average of 177W x 4)
2024-04-15 10:08:01,704:INFO: Grid feed in from panels connected directly to inverter: 80.1!
2024-04-15 10:08:01,705:INFO: Sun: 06:38 - 20:30 Demand: 273.7W, Panel DC: (353.8W), Hub DC: (0.0W), Inverter Limit: 708.0W, Hub Limit: 0.0W
2024-04-15 10:08:01,705:INFO: SMT triggers limit function: -106.5 -> -60.1: executed
2024-04-15 10:08:01,708:INFO: SMT triggers limit function: -104.2 -> -58.9: skipped
2024-04-15 10:08:01,710:INFO: SMT triggers limit function: -102.0 -> -57.9: skipped
2024-04-15 10:08:04,720:INFO: SMT triggers limit function: -82.1 -> -42.7: skipped
2024-04-15 10:08:04,727:INFO: SMT triggers limit function: -79.6 -> -41.2: skipped
2024-04-15 10:08:04,734:INFO: SMT triggers limit function: -77.3 -> -39.9: skipped
2024-04-15 10:08:04,741:INFO: SMT triggers limit function: -75.2 -> -38.7: skipped
2024-04-15 10:08:04,747:INFO: SMT triggers limit function: -73.2 -> -37.7: skipped
2024-04-15 10:08:04,754:INFO: SMT triggers limit function: -71.4 -> -36.7: skipped
2024-04-15 10:08:05,028:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-04-15 10:08:08,711:INFO: SMT triggers limit function: -57.9 -> -24.9: skipped
2024-04-15 10:08:08,718:INFO: SMT triggers limit function: -55.8 -> -23.4: skipped
2024-04-15 10:08:08,726:INFO: SMT triggers limit function: -53.8 -> -22.1: skipped
2024-04-15 10:08:25,715:INFO: HUB: S:371.1W [ 453.4,453.4,441.7,425.1,395.4,371.1 ], B: 39% (40|38), V:50.2V (50.1|50.2), C: 330W, P:False, F:13.8h, E:-1.0h, H: 0W, L: 0W
2024-04-15 10:08:25,715:INFO: INV: AC:349.3W, AC_Prediction: 349.3W, DC:367.6W, DC_prediction: 367.6W (181.8|181.1|0.0|0.0), L:707.2W [1600W]
2024-04-15 10:08:25,721:INFO: SMT: T:ShellyEM3 P:43.9W [ -60.5,-60.5,-44.6,-16.1,-2.0,17.3 ] Predict: -150.3W
2024-04-15 10:08:25,721:INFO: Direct connected panels (344.8W) can cover demand (342.1W)
2024-04-15 10:08:25,721:INFO: Adjusting inverter limit from 172.7W to 600.4W
2024-04-15 10:08:25,721:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 0, new limit: 0
2024-04-15 10:08:25,722:INFO: Setting inverter output limit to 2400W (1 min moving average of 600W x 4)
2024-04-15 10:08:25,722:INFO: Grid feed in from panels connected directly to inverter: 2.7!
2024-04-15 10:08:25,723:INFO: Sun: 06:38 - 20:30 Demand: 342.1W, Panel DC: (344.8W), Hub DC: (0.0W), Inverter Limit: 2400.0W, Hub Limit: 0.0W
2024-04-15 10:08:25,723:INFO: SMT triggers limit function: -2.0 -> 17.3: executed
2024-04-15 10:08:25,730:INFO: SMT triggers limit function: -0.9 -> 17.9: skipped
2024-04-15 10:08:25,739:INFO: SMT triggers limit function: 0.0 -> 18.3: skipped
[... cut ...]
2024-04-15 10:11:05,037:INFO: Setting solarflow output limit to 60.0W
2024-04-15 10:11:05,038:INFO: Solarflow is willing to contribute 49.0W of the requested 49.0!
2024-04-15 10:11:05,038:INFO: Adjusting inverter limit from 136.7W to 533.6W
2024-04-15 10:11:05,039:INFO: Direct connected panel limit is 533W.
2024-04-15 10:11:05,040:INFO: Setting inverter output limit to 2132W (1 min moving average of 533W x 4)
2024-04-15 10:11:05,041:INFO: Sun: 06:38 - 20:30 Demand: 321.6W, Panel DC: (272.6W), Hub DC: (0.0W), Inverter Limit: 2132.0W, Hub Limit: 60.0W
```

tuxianerDE, 2024-04-15:
Do pull again the latest build. There are still some bugs in it like I have documented it #211

tuxianerDE, 2024-04-17:
Hey,

are you still seeing the behaviour with the latest master build ? The changes done are now upstream from dev to the master build.

busaku, 2024-04-17:
Purple = Net House Power (Shelly)
Blue = Zendure to Inverter

So it looks a waaaay smoother :)
Thanks for your work, I close this issue

23-0
[image]

0-1 am
[image]

1-2 am
[image]

closed 2024-04-17 by busaku