reinhard-brandstaedter/solarflow-control #209: Offset Option

sstidl, 2024-04-14:
Hi,

I might have missed it, but is there an option to set an offset in power control? I often notice that negative total demand ("strom wird verschenkt") isn't corrected quickly enough. In my previous setup, I used hoymileszeroexport.py where I could set the target point with a tolerance range. For instance, I set my target point to 20W with a range of +-15W. With this setup, I hardly ever fed excess energy back into the network.

Thanks for any clarification.

reinhard-brandstaedter, 2024-04-14:
Hi @sstidl ,

yes implementing such an offset came to my mind when I recently improved the conversion between DC and AC data using the inverter's reported efficiency (latest dev build has this).
During rather quiet times (supply from battery during night). I do see some improvement as the target point (0W) is met with +/-5 W almost precisely.
During production times and higher consumption on/off there will always be a little "give-away" after high usage/fluctuations.
Generally with a SF Hub AND additionally connected panels connected to the inverter there is just another moving part that makes things a little bit more complicated.

I'm currently observing if this would actually pay off, but the data right now is rather good (talking about +/- a few W):
[image]

reinhard-brandstaedter, 2024-04-16:
Offset option added, see https://github.com/reinhard-brandstaedter/solarflow-control/blob/f9fb944ecedfb87134cd90866d2abb60f83d4e5a/src/config.ini#L70

closed 2024-04-16 by reinhard-brandstaedter