reinhard-brandstaedter/solarflow-control #210: Feed in instead of charging Battery

boennemann, 2024-04-14:
I keep observing feed in into the grid, even though my battery isn't full.
The battery is at 72% and yet the inverter is set to ~800W with only a ~330W demand.
The battery receives ~350W-ish.
Almost feels like this should be the other way around.

Here are my logs:

```sh
2024-04-14 11:05:31,123:INFO: HUB: S:1274.9W [ 1275.1,1275.1,1275.1,1274.8,1275.1,1274.9 ], B: 72% (72), V:50.6V (50.6), C: 345W, P:False, F:20.2h, E:-1.0h, H:910W, L:1000W
2024-04-14 11:05:31,138:INFO: INV: AC:847.6W, AC_Prediction: 846.5W, DC:896.8W, DC_prediction: 895.7W (443.5|450.6), L:835.0W [1000.0W]
2024-04-14 11:05:31,144:INFO: SMT: T:ShellyEM3 P:-512.6W [ -515.6,-515.6,-516.2,-516.9,-516.3,-515.7 ] Predict: -514.6W
2024-04-14 11:05:31,144:INFO: Direct connected panels (0.0W) can't cover demand (331.9W), trying to get rest from hub.
2024-04-14 11:05:31,144:INFO: Checking if Solarflow is willing to contribute 331.9W ...
2024-04-14 11:05:31,145:INFO: Based on time, solarpower (1274.9W) minimum charge power (0W) and bypass state (False), hub could contribute 331.9W - Decision path: 1.2.
2024-04-14 11:05:31,146:INFO: Hub should contribute more (331.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-14 11:05:31,146:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 11:05:31,146:INFO: Current inverter AC output (847.6) is higher than configured output limit (800), reducing limit to 831.0
2024-04-14 11:05:31,146:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-14 11:05:31,147:INFO: Sun: 06:25 - 20:02 Demand: 331.9W, Panel DC: ( 0), Hub DC: (443.5|450.6), Inverter Limit: 831.0W, Hub Limit: 1000.0W
```

boennemann, 2024-04-14:
Update: It seems to also have been the other way around, when suddenly it flipped:

[image]

reinhard-brandstaedter, 2024-04-14:
Generally it would make more sense to run with a minimum charge power greater than 0.

is your  sf_inverter_channels variable set correctly? You seem to have only two inputs.

Try the latest dev build, there have been some improvements in regards to feed-in handling/information.

boennemann, 2024-04-14:
Thanks for the pointers. I'm updating the config. As my battery is full now, I can only update tomorrow.

> Generally it would make more sense to run with a minimum charge power greater than 0.
Can you explain what this config variable does?

> is your sf_inverter_channels variable set correctly? You seem to have only two inputs.
I have the hub connected to both inverter channels and no modules connected directly to the inverter. 0 is the output and thus I'm setting `sf_inverter_channels = [1,2]`. 

[image]

> Over time the limit should be decreased down to 800 max (AC output 800W). Does your limit then jump back up again? or is it further reduced to match the demand?

If understood you correctly then my 2nd screenshot show that it's staying at somewhere above 800

Thank you 🙏

reinhard-brandstaedter, 2024-04-14:
OK, the inverter channels are correct.

The ```min_charge_power``` variable defines the power that should be used for charging before the hub gives anything to the output. It ensures that the battery is first charged and then the house is supported.
In your case with 1200W input that would surely make sense.

So the expected (my setup is similar, except I'm not hitting the 800W legal limit) behavior would be:
- use the first "min_charge_power" W to charge
- everything else that exceeds goes to the house
- if the house demand is covered (starts grid feeding) the rest is assigned for faster charging.

That works quite flawlessly here, the only difference as said is the legal limit hit.

boennemann, 2024-04-14:
Alright, I'll try 150W then 👍 

In the meantime the next weird thing popped up.
The battery hit 100%, then it stopped feeding the house altogether, no feed-in, but not even the demand from the house. The inverter limit seems to be alright, but the Hub apparently doesn't output anything.

[image]

Might try hopping onto the dev build next.

boennemann, 2024-04-14:
Running the dev branch now.
The hub keeps pretending it doesn't have any solar input 😬 
Perfect sunshine and I'm drawing from neither the battery, nor sun, but the grid 🤔 

DTU is limited to 20W 😕 

```sh
2024-04-14 14:18:00,177:INFO: HUB: S:38.8W [ 570.2,570.2,524.3,350.3,148.8,38.8 ], B:100% (100), V:53.3V (53.3), C: -46W, P:False, F:0.0h, E:-1.0h, H: 44W, L:1000W
2024-04-14 14:18:00,191:INFO: INV: AC:66.0W, AC_Prediction: 46.5W, DC:69.9W, DC_prediction: 49.2W (12.1|12.2), L:20.0W [1000.0W]
2024-04-14 14:18:00,191:INFO: SMT: T:ShellyEM3 P:203.6W [ 147.9,147.9,135.2,159.9 ] Predict: 159.9W
2024-04-14 14:18:00,191:INFO: Direct connected panel are producing 0.0W, trying to get 182.9W from hub.
2024-04-14 14:18:00,192:INFO: Based on time, solarpower (38.8W) minimum charge power (150W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-04-14 14:18:00,192:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 14:18:00,193:INFO: Solarflow is willing to contribute 0.0W of the requested 182.9!
2024-04-14 14:18:00,193:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-14 14:18:00,194:INFO: Sun: 06:25 - 20:02 Demand: 182.9W, Panel DC: (0.0W), Hub DC: (23.0W), Inverter Limit: 20.0W, Hub Limit: 1000.0W
2024-04-14 14:18:00,194:INFO: SMT triggers limit function: 135.2 -> 159.9: executed
2024-04-14 14:18:00,196:INFO: SMT triggers limit function: 137.5 -> 163.5: skipped
2024-04-14 14:18:00,198:INFO: SMT triggers limit function: 139.6 -> 166.7: skipped
2024-04-14 14:18:05,670:INFO: SMT triggers limit function: 146.4 -> 202.4: skipped
2024-04-14 14:18:05,680:INFO: SMT triggers limit function: 152.7 -> 203.7: skipped
2024-04-14 14:18:05,690:INFO: SMT triggers limit function: 155.1 -> 205.0: skipped
2024-04-14 14:18:09,270:INFO: Battery is full: 100
2024-04-14 14:18:09,315:INFO: Reading last full time: 2024-04-14 14:18:09
2024-04-14 14:18:09,315:INFO: Reading battery target mode: discharging
2024-04-14 14:18:10,131:INFO: SMT triggers limit function: 168.1 -> 216.5: skipped
2024-04-14 14:18:10,139:INFO: SMT triggers limit function: 169.4 -> 215.4: skipped
2024-04-14 14:18:10,147:INFO: SMT triggers limit function: 170.5 -> 214.0: skipped
```

reinhard-brandstaedter, 2024-04-14:
What you see is kind of normal when bypass kicks in. It will take a bit untill the hub starts feeding in again in bypass.

boennemann, 2024-04-14:
Hmm 🤔 But surely not two hours?

I even tried `limit_inverter = false` now, but even then the inverter is limited to 20W, even after unblocking it manually.

I'm wondering where this 20W inverter limit is coming from, because it's wasting away energy at peak hours.

According to the logs it's being set to ~200, but when I check OpenDTU it's at 20W.

[image]

boennemann, 2024-04-14:
Found the log now at least:

```sh
2024-04-14 14:54:15,441:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 14:54:15,441:INFO: Solarflow is willing to contribute 2.5W of the requested 213.8!
2024-04-14 14:54:15,442:INFO: Setting inverter output limit to 20W (1 min moving average of 10W x 2)
```

It seems like a logical loop. During bypass if the inverter is limited to 20W there is no way for the hub to produce anything.

reinhard-brandstaedter, 2024-04-14:
Also, make sure the script is running for a bit, as it takes a little initial data collection.

boennemann, 2024-04-14:
I set it to 30W, but doesn't change anything. 
The inverter was also running before though.

Will try something significantly higher now.

boennemann, 2024-04-14:
Seeing "2024-04-14 16:23:30,905:INFO: Setting inverter output limit to 20W (1 min moving average of 10W x 2)" even with Inverter Min set to 150W.

reinhard-brandstaedter, 2024-04-14:
> Seeing "2024-04-14 16:23:30,905:INFO: Setting inverter output limit to 20W (1 min moving average of 10W x 2)" even with Inverter Min set to 150W.

That can’t be if the property is set to 20, the min inverter limit is set to <nr mppt channels>*<inverter_min_limit>

boennemann, 2024-04-14:
> That can’t be if the property is set to 20, the min inverter limit is set to *<inverter_min_limit>

I thought the same – this can't be. Didn't nail down the issue yet, but both example config.ini's seem to define a config value for this that's not referenced anywhere in the source code. https://github.com/search?q=repo%3Areinhard-brandstaedter%2Fsolarflow-control%20inverter_min_limit&type=code

Digging now what the variable would need to be named.

boennemann, 2024-04-14:
I can't find any variable for this. It seems like it's not implemented as a variable. Could only find a hardcoded `10` in the source. Bumped this up now. Will report.

reinhard-brandstaedter, 2024-04-14:
can you attach a full log, from start up on?

boennemann, 2024-04-14:
This is all that I still have, can't reproduce until there is sun again.

Somehow I feel like my Hub is not doing the right thing altogether.

[image]

When I increased the min limit in the code – fair it did provide the min limit.
Then the sun slowly faded out and now it's keeping the battery at 100% even with no sun whatsoever and again fulling everything from the grid.

```sh
2024-04-14 20:46:41,604:INFO: Reading last full time: 2024-04-14 20:46:41
2024-04-14 20:46:41,605:INFO: Reading battery target mode: discharging
2024-04-14 20:46:47,813:INFO: Rapid drop in demand detected, clearing buffer!
2024-04-14 20:46:47,815:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100), V:49.8V (49.8), C: 0W, P:False, F:0.0h, E:-1.0h, H: 0W, L:1000W
2024-04-14 20:46:47,823:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.2W, DC_prediction: 0.3W (0.1|0.1), L:290.0W [1000.0W]
2024-04-14 20:46:47,823:INFO: SMT: T:ShellyEM3 P:1119.5W [ 1119.5,1119.5 ] Predict: 1119.5W
2024-04-14 20:46:47,823:INFO: Direct connected panel are producing 0.0W, trying to get 1119.5W from hub.
2024-04-14 20:46:47,824:INFO: Based on time, solarpower ( 0.0W) minimum charge power (150W) and bypass state (False), hub could contribute 750.0W - Decision path: 2.1.
2024-04-14 20:46:47,824:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 20:46:47,825:INFO: Solarflow is willing to contribute 375.0W of the requested 1119.5!
2024-04-14 20:46:47,825:INFO: Inverter is not reachable/down. Can't set limit
2024-04-14 20:46:47,826:INFO: Sun: 06:25 - 20:02 Demand: 1119.5W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 750.0W, Hub Limit: 1000.0W
2024-04-14 20:46:47,826:INFO: SMT triggers limit function: 1119.5 -> 1119.5: executed
2024-04-14 20:46:49,811:INFO: Rapid drop in demand detected, clearing buffer!
2024-04-14 20:46:49,813:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100), V:49.8V (49.8), C: 0W, P:False, F:0.0h, E:-1.0h, H: 0W, L:1000W
2024-04-14 20:46:49,820:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.2W, DC_prediction: 0.3W (0.1|0.1), L:290.0W [1000.0W]
2024-04-14 20:46:49,820:INFO: SMT: T:ShellyEM3 P:394.4W [ 394.4,394.4,394.4 ] Predict: 394.4W
2024-04-14 20:46:49,820:INFO: Direct connected panel are producing 0.0W, trying to get 394.4W from hub.
2024-04-14 20:46:49,821:INFO: Based on time, solarpower ( 0.0W) minimum charge power (150W) and bypass state (False), hub could contribute 394.4W - Decision path: 2.1.
2024-04-14 20:46:49,821:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 20:46:49,821:INFO: Solarflow is willing to contribute 197.0W of the requested 394.4!
2024-04-14 20:46:49,822:INFO: Inverter is not reachable/down. Can't set limit
2024-04-14 20:46:49,823:INFO: Sun: 06:25 - 20:02 Demand: 394.4W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 394.0W, Hub Limit: 1000.0W
2024-04-14 20:46:49,823:INFO: SMT triggers limit function: 394.4 -> 394.4: executed
2024-04-14 20:47:02,819:INFO: SMT triggers limit function: 423.6 -> 434.4: skipped
2024-04-14 20:47:03,813:INFO: SMT triggers limit function: 430.4 -> 443.2: skipped
2024-04-14 20:47:03,815:INFO: Rapid rise in demand detected, clearing buffer!
2024-04-14 20:47:03,816:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100), V:49.8V (49.8), C: 0W, P:False, F:0.0h, E:-1.0h, H: 0W, L:1000W
2024-04-14 20:47:03,822:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.2W, DC_prediction: 0.3W (0.1|0.1), L:290.0W [1000.0W]
2024-04-14 20:47:03,822:INFO: SMT: T:ShellyEM3 P:1850.0W [ 1850.0,1850.0 ] Predict: 1850.0W
2024-04-14 20:47:03,822:INFO: Direct connected panel are producing 0.0W, trying to get 1850.0W from hub.
2024-04-14 20:47:03,823:INFO: Based on time, solarpower ( 0.0W) minimum charge power (150W) and bypass state (False), hub could contribute 750.0W - Decision path: 2.1.
2024-04-14 20:47:03,823:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 20:47:03,824:INFO: Solarflow is willing to contribute 375.0W of the requested 1850.0!
2024-04-14 20:47:03,824:INFO: Inverter is not reachable/down. Can't set limit
2024-04-14 20:47:03,825:INFO: Sun: 06:25 - 20:02 Demand: 1850.0W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 750.0W, Hub Limit: 1000.0W
2024-04-14 20:47:03,825:INFO: SMT triggers limit function: 1850.0 -> 1850.0: executed
2024-04-14 20:47:17,815:INFO: Rapid drop in demand detected, clearing buffer!
2024-04-14 20:47:17,817:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100), V:49.8V (49.8), C: 0W, P:False, F:0.0h, E:-1.0h, H: 0W, L:1000W
2024-04-14 20:47:17,824:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.2W, DC_prediction: 0.3W (0.1|0.1), L:290.0W [1000.0W]
2024-04-14 20:47:17,825:INFO: SMT: T:ShellyEM3 P:528.7W [ 528.7,528.7 ] Predict: 528.7W
2024-04-14 20:47:17,825:INFO: Direct connected panel are producing 0.0W, trying to get 528.7W from hub.
2024-04-14 20:47:17,827:INFO: Based on time, solarpower ( 0.0W) minimum charge power (150W) and bypass state (False), hub could contribute 528.7W - Decision path: 2.1.
2024-04-14 20:47:17,827:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 20:47:17,828:INFO: Solarflow is willing to contribute 264.0W of the requested 528.7!
2024-04-14 20:47:17,829:INFO: Inverter is not reachable/down. Can't set limit
2024-04-14 20:47:17,830:INFO: Sun: 06:25 - 20:02 Demand: 528.7W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 528.0W, Hub Limit: 1000.0W
[... cut ...]
2024-04-14 20:47:47,833:INFO: SMT triggers limit function: 414.1 -> 414.1: executed
2024-04-14 20:48:36,868:INFO: Battery is full: 100
2024-04-14 20:48:36,871:INFO: Reading last full time: 2024-04-14 20:48:36
2024-04-14 20:48:36,874:INFO: Reading battery target mode: discharging
2024-04-14 20:48:38,776:INFO: Triggering telemetry update: iot/A8yh63/<ID>/properties/read
2024-04-14 20:48:42,029:INFO: Battery is full: 100
2024-04-14 20:48:42,075:INFO: Reading last full time: 2024-04-14 20:48:42
2024-04-14 20:48:42,076:INFO: Reading battery target mode: discharging
```

boennemann, 2024-04-14:
> Then the sun slowly faded out and now it's keeping the battery at 100% even with no sun whatsoever and again fulling everything from the grid.

Okay that one is resolved. Something during debugging today I set the Hub to bypass without enabling auto-recover from bypass. Switched this back now solarflow-control seems to be feeding the right amount from the battery again.

reinhard-brandstaedter, 2024-04-14:
Oh OK that explains why this log happens, I was just about to ask.
Hub limit is set, inverter limit is set but nothing is coming from hub, so inverter is not reachable.

```
2024-04-14 20:46:47,815:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100), V:49.8V (49.8), C: 0W, P:False, F:0.0h, E:-1.0h, H: 0W, L:1000W
2024-04-14 20:46:47,823:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.2W, DC_prediction: 0.3W (0.1|0.1), L:290.0W [1000.0W]
2024-04-14 20:46:47,823:INFO: SMT: T:ShellyEM3 P:1119.5W [ 1119.5,1119.5 ] Predict: 1119.5W
2024-04-14 20:46:47,823:INFO: Direct connected panel are producing 0.0W, trying to get 1119.5W from hub.
2024-04-14 20:46:47,824:INFO: Based on time, solarpower ( 0.0W) minimum charge power (150W) and bypass state (False), hub could contribute 750.0W - Decision path: 2.1.
2024-04-14 20:46:47,824:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-14 20:46:47,825:INFO: Solarflow is willing to contribute 375.0W of the requested 1119.5!
2024-04-14 20:46:47,825:INFO: Inverter is not reachable/down. Can't set limit
```

boennemann, 2024-04-14:
In the meantime I'll pull https://github.com/reinhard-brandstaedter/solarflow-control/commit/6844d8b3bdc7a917575a26aa8adf4bb7b125760e and the rest from dev and then I guess we'll see tomorrow. Hope the weather is good enough to fill the battery again.

boennemann, 2024-04-15:
I'm back to being limited to 20W today, even with the set minimum
It's super cloudy and rainy today, but I'm receiving 300W+ when manually unblocking the inverter.

```sh
2024-04-15 12:06:43,271:INFO: HUB: S:14.5W [ 21.8,21.8,20.1,19.0,18.4,14.5 ], B: 30% (30), V:49.6V (49.6), C:-139W, P:False, F:15.1h, E:-1.0h, H:145W, L:1000W
2024-04-15 12:06:43,284:INFO: INV: AC:216.5W, AC_Prediction: 160.7W, DC:346.5W, DC_prediction: 175.6W (12.1|12.7), L:20.0W [1000.0W]
2024-04-15 12:06:43,292:INFO: SMT: T:ShellyEM3 P:333.0W [ 196.9,196.9,160.3,134.0,96.8,259.8 ] Predict: 255.5W
2024-04-15 12:06:43,293:INFO: Direct connected panel are producing 0.0W, trying to get 283.2W from hub.
2024-04-15 12:06:43,294:INFO: Based on time, solarpower (14.5W) minimum charge power (150W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-04-15 12:06:43,294:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-15 12:06:43,295:INFO: Solarflow is willing to contribute 0.0W of the requested 283.2!
2024-04-15 12:06:43,295:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-15 12:06:43,297:INFO: Sun: 06:23 - 20:03 Demand: 283.2W, Panel DC: (0.0W), Hub DC: (23.4W), Inverter Limit: 20.0W, Hub Limit: 1000.0W
```

reinhard-brandstaedter, 2024-04-15:
But something still doesn’t add up looking at your log snippet:
- the Hub feeds 145W to home “H:145W”
- the inverter is apparently limited to 20W, but reports AC output of 216W
- the inverter reports a DC input og 2x 12W

I hope your log snippet is just a short timestamp and this is due to recent changes/switches and the moving values just haven’t updated/reported yet.
(would also be interesting if the hub2000 reports data maybe at a different pace…)

boennemann, 2024-04-16:
Okay, battery just hit 100% and it's happening again.

```sh
2024-04-16 14:49:17,493:INFO: DTU triggers limit function: 105.3 -> 265.7: skipped
2024-04-16 14:49:22,468:INFO: DTU triggers limit function: 148.9 -> 243.8: skipped
2024-04-16 14:49:22,477:INFO: HUB: S:130.3W [ 615.2,615.2,586.8,500.4,279.3,130.3 ], B:100% (100), V:52.4V (52.4), C: -27W, P:False, F:0.0h, E:-1.0h, H:157W, L:1000W
2024-04-16 14:49:22,488:INFO: INV: AC:148.9W, AC_Prediction: 243.8W, DC:115.1W, DC_prediction: 282.8W (101.1|118.0), L:548.0W [1000.0W]
2024-04-16 14:49:22,495:INFO: SMT: T:ShellyEM3 P:802.1W [ 832.0,832.0,861.6,863.9,857.3 ] Predict: 811.9W
2024-04-16 14:49:22,496:INFO: Direct connected panel are producing 0.0W, trying to get 1064.4W from hub.
2024-04-16 14:49:22,497:INFO: Based on time, solarpower (130.3W) minimum charge power (150W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-04-16 14:49:22,497:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-16 14:49:22,498:INFO: Solarflow is willing to contribute 0.0W of the requested 1064.4!
2024-04-16 14:49:22,498:INFO: Setting inverter output limit to 20W (1 min moving average of 10W x 2)
2024-04-16 14:49:22,499:INFO: Sun: 06:21 - 20:05 Demand: 1064.4W, Panel DC: (0.0W), Hub DC: (207.1W), Inverter Limit: 20.0W, Hub Limit: 1000.0W
2024-04-16 14:49:22,500:INFO: DTU triggers limit function: 148.9 -> 243.8: executed
2024-04-16 14:49:22,505:INFO: DTU triggers limit function: 148.9 -> 243.8: skipped
2024-04-16 14:49:23,403:INFO: SMT triggers limit function: 861.9 -> 851.8: skipped
2024-04-16 14:49:27,446:INFO: DTU triggers limit function: 123.7 -> 245.4: skipped
2024-04-16 14:49:27,464:INFO: DTU triggers limit function: 123.7 -> 245.4: skipped
2024-04-16 14:49:27,484:INFO: DTU triggers limit function: 123.7 -> 245.4: skipped
2024-04-16 14:49:32,459:INFO: DTU triggers limit function: 111.5 -> 168.8: skipped
2024-04-16 14:49:32,470:INFO: DTU triggers limit function: 111.5 -> 168.8: skipped
2024-04-16 14:49:32,490:INFO: DTU triggers limit function: 111.5 -> 168.8: skipped
2024-04-16 14:49:32,644:INFO: SMT triggers limit function: 875.5 -> 858.4: skipped
2024-04-16 14:49:32,651:INFO: SMT triggers limit function: 878.8 -> 864.6: skipped
2024-04-16 14:49:32,661:INFO: SMT triggers limit function: 881.8 -> 870.0: skipped
2024-04-16 14:49:37,452:INFO: DTU triggers limit function: 97.1 -> 173.9: skipped
2024-04-16 14:49:37,476:INFO: DTU triggers limit function: 97.1 -> 173.9: skipped
2024-04-16 14:49:37,494:INFO: HUB: S:100.1W [ 434.6,434.6,299.9,140.1,96.3,100.1 ], B:100% (100), V:52.4V (52.4), C: -40W, P:False, F:0.0h, E:-1.0h, H: 82W, L:1000W
2024-04-16 14:49:37,507:INFO: INV: AC:97.1W, AC_Prediction: 173.9W, DC:121.9W, DC_prediction: 180.0W (12.6|13.6), L:20.0W [1000.0W]
2024-04-16 14:49:37,514:INFO: SMT: T:ShellyEM3 P:977.0W [ 881.9,881.9,885.3,881.8,870.0 ] Predict: 872.3W
2024-04-16 14:49:37,514:INFO: Direct connected panel are producing 0.0W, trying to get 894.8W from hub.
2024-04-16 14:49:37,515:INFO: Based on time, solarpower (100.1W) minimum charge power (150W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-04-16 14:49:37,516:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-16 14:49:37,516:INFO: Solarflow is willing to contribute 0.0W of the requested 894.8!
2024-04-16 14:49:37,516:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-16 14:49:37,517:INFO: Sun: 06:21 - 20:05 Demand: 894.8W, Panel DC: (0.0W), Hub DC: (24.8W), Inverter Limit: 20.0W, Hub Limit: 1000.0W
2024-04-16 14:49:37,518:INFO: DTU triggers limit function: 97.1 -> 173.9: executed
2024-04-16 14:49:42,456:INFO: DTU triggers limit function: 57.7 -> 100.3: skipped
2024-04-16 14:49:42,468:INFO: DTU triggers limit function: 57.7 -> 100.3: skipped
2024-04-16 14:49:42,499:INFO: DTU triggers limit function: 57.7 -> 100.3: skipped
2024-04-16 14:49:47,478:INFO: DTU triggers limit function: 20.5 -> 67.1: skipped
2024-04-16 14:49:47,485:INFO: DTU triggers limit function: 20.5 -> 67.1: skipped
2024-04-16 14:49:47,492:INFO: DTU triggers limit function: 20.5 -> 67.1: skipped
2024-04-16 14:49:50,425:INFO: SMT triggers limit function: 958.8 -> 981.8: skipped
2024-04-16 14:49:50,432:INFO: SMT triggers limit function: 962.6 -> 985.6: skipped
2024-04-16 14:49:50,439:INFO: SMT triggers limit function: 965.9 -> 988.3: skipped
2024-04-16 14:49:52,453:INFO: DTU triggers limit function: 16.6 -> 64.4: skipped
2024-04-16 14:49:52,471:INFO: DTU triggers limit function: 16.6 -> 64.4: skipped
2024-04-16 14:49:52,487:INFO: DTU triggers limit function: 16.6 -> 64.4: skipped
2024-04-16 14:49:57,457:INFO: HUB: S:49.5W [ 246.1,246.1,111.5,75.1,72.4,49.5 ], B:100% (100), V:52.4V (52.4), C: -11W, P:False, F:0.0h, E:-1.0h, H: 11W, L:1000W
2024-04-16 14:49:57,470:INFO: INV: AC:8.5W, AC_Prediction: 63.6W, DC:15.5W, DC_prediction: 68.6W (0.5|0.5), L:20.0W [1000.0W]
2024-04-16 14:49:57,475:INFO: SMT: T:ShellyEM3 P:1001.8W [ 893.1,893.1,890.7,884.5,965.9,988.3 ] Predict: 876.1W
2024-04-16 14:49:57,476:INFO: Direct connected panel are producing 0.0W, trying to get 989.3W from hub.
2024-04-16 14:49:57,477:INFO: Based on time, solarpower (49.5W) minimum charge power (150W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-04-16 14:49:57,477:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
[... cut ...]
2024-04-16 14:51:32,696:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-16 14:51:32,696:INFO: Solarflow is willing to contribute 2.5W of the requested 1055.5!
2024-04-16 14:51:32,697:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-16 14:51:32,698:INFO: Sun: 06:21 - 20:05 Demand: 1055.5W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 20.0W, Hub Limit: 1000.0W
2024-04-16 14:51:32,698:INFO: SMT triggers limit function: 1044.8 -> 1055.5: executed
```

[image]

welli2009, 2024-04-16:
Hi, I had the same Issue and googled quite a long time. Somewhere in the official zendure discussion page ist the solution presented: there is no solution. The reason why this is happening is ig the Relais, which is controlling the Bypass. To not “destroy” the hub or the Relais, they are dimming the solar input until the Relais is jumping over. This happened sometimes two times the day to me, first at the moment, when the battery was fully loaded and the second time a little after sunset, when the hub should use the battery’s for feeding the home. Maybe they are improving this in a later softwareupdate or in the next generation of hubs, the 2k has the same issue. So what could be done by the script is reducing the solar input and then “manually” activating the bypass mode by the python script.
   
  
   Jacob

boennemann, 2024-04-16:
Hey @welli2009, 
thanks for adding this context! Do you have a link to this discussion page?

For me right now I changed the code to have a 160W min limit, and the hub is feeding that in. The hub would provide more however. It's still solarflow-control that limits it to (now) 160W, instead of the demand.

reinhard-brandstaedter, 2024-04-16:
@boennemann do you have the logs after 14:51, maybe until 15:15? That would help me debugging.

boennemann, 2024-04-16:
@reinhard-brandstaedter Argh, no sorry. I'm running through a Home Assistant addon setup, and that doesn't persist the logs. I'll capture them when I hit full battery next time.

I also changed the max battery charge to 98% now, to try and see whether literally hitting the 100% causes the problem, or hitting the max limit causes it.

reinhard-brandstaedter, 2024-04-16:
@boennemann I'm mostly interested in these lines:
```
2024-04-16 14:51:32,694:INFO: Direct connected panel are producing 0.0W, trying to get 1055.5W from hub.
2024-04-16 14:51:32,695:INFO: Adjusting inverter limit from 0.0W to 5.0W
2024-04-16 14:51:32,695:INFO: Based on time, solarpower ( 0.0W) minimum charge power (150W) and bypass state (True), hub could contribute  5.0W - Decision path: 2.2.
2024-04-16 14:51:32,696:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-16 14:51:32,696:INFO: Solarflow is willing to contribute 2.5W of the requested 1055.5!
```

The hub is open, the inverter limit should be increased, it is actually at 20W, so the hub SHOULD deliver 2x10W to the inverter, but the inverter doesn't put anything out on the AC side. If it would you should see another log like:
```
2024-04-16 14:51:32,695:INFO: Adjusting inverter limit from 20.0W to <something exponentially higher>W
```
And then the proper limit adoption should take its course...but with 0W stuck on AC output it's going nowhere.

reinhard-brandstaedter, 2024-04-16:
This is btw what @welli2009 described, in color (the purple lines are the bypass state):
You can see that the hub reduces Solarinput _before_ switching the bypass. The inverter limit goes to 40W (10W per channel in my case). Then there is some back/forth of the hub's bypass (sometimes this takes a few minutes, sometimes it takes half an hour). As soon as the bypass is left the inverter gets adjusted.
And finally (40minutes later!) the hub decides to stay in bypass and really pass through solarpower. The inverter is adjusted accordingly. During that time when the hub sent zero to the inverter, it even turned off and woke up again.

[image]

This doesn't make much sense, to me. That's why I thought of switching the bypass from the script also. But so far didn't dare to do it as the official statement is that too many switches could damage the relais. On the other hand, the hub's firmware seems to do quite som unnecessary switches itself. (16 vs manual 2)

reinhard-brandstaedter, 2024-04-17:
@boennemann another idea: you could try to set the bypass mode to off completely.

boennemann, 2024-04-17:
> @boennemann another idea: you could try to set the bypass mode to off completely.

Sorry for having to ask, how would do that? MQTT call on Homeassistant suffices? Or solarflow-bt-manager?

reinhard-brandstaedter, 2024-04-19:
Actually I think it is (that's at least what I use in the current steering tests):
```
{"properties": { "passMode": 1 }} ==> manual off
{"properties": { "passMode": 2 }} ==> manual on
{"properties": { "passMode": 0 }} ==> automatic (let the hub's firmware decide
```
and then the ```pass``` variable is the current state of the bypass.
It would be really nice if there was a documentation from Zendure somewhere to not have to explore these things in trial/error...

boennemann, 2024-04-21:
The weather is not good enough to reach 100% battery, will update you probably next week.

The rest is working super reliably and as expected however.

boennemann, 2024-04-21:
It seems as if this commit addresses my problem though? https://github.com/reinhard-brandstaedter/solarflow-control/commit/68e731b7e56109a7d72d97d47523132eee221018

Will update the controller later today, too, and give that a spin :)

welli2009, 2024-04-21:
I installed the latest version and reached 100% yesterday, Bypass worked super good, activated itself at 100% and deactivated later in the evening when the solarpower couldn’t feed the home anymore!
   
  
   Good work!

boennemann, 2024-04-22:
Also updated to the latest version. 120h had gone by, so the battery was charged to 100% today anyway. It hit 100%, very shortly went down to 0W and now doing what's expected.

Will observe more, but this looks very promising!
Thank you @reinhard-brandstaedter 🙏

welli2009, 2024-04-23:
[image]
Hi,
my hub hit 100% today in the early afternoon but the bypass eventually didn't worked as expected first, (got down to zero, activated and dimmed the solarinput while bright sunshine and deactivated itself after ~10 minutes) but now at the moment everything works fine

reinhard-brandstaedter, 2024-04-23:
@welli2009 you did pass bypass control to the script,  by setting it to manual before, right?

welli2009, 2024-04-23:
No, it was all automatic

reinhard-brandstaedter, 2024-04-23:
Well then you are playing the Zendure firmware lottery :-)

welli2009, 2024-04-23:
But for now, it works!

mavo, 2024-04-27:
@reinhard-brandstaedter what is expected as the setting for bypass on the hub?
It should not be automatic anymore with the newest version, did I get it right?

reinhard-brandstaedter, 2024-04-27:
Only on the dev build:
control_bypass=true|false
in the [solarflow] config section

mavo, 2024-04-27:
Funny, I am almost certain to not run the dev build and I saw the app complaining on startup, that this parameter is missing.
I am running the latest master build (will verify it again to not talk bs).

You also did not really answer my question about what should be configured on the hub. It should be manual...?

And last but not least, a general question to the audience for Hub2k users... Which Firmware do you use?

MaVo

tuxianerDE, 2024-04-27:
Hi,

The master does already know the config parameter hence the complaining. Other function is not there.

Actually rheinard did answer your question. The answer is does not matter. If you set the bypass option in the config section the mode will be switched to manual and the logic is with the script. If set to false it will assume automatic and the hub will do what the firmware is coded to do.

mavo, 2024-04-27:
Ah ok. Did not understand it this way. Thank you for your clarification.
Will try the dev build, as it may improve the "issue" with bypass enabling and seeing sometimes an hour of no power delivered to the house, even though bp is on and plenty of sun.

boennemann, 2024-04-29:
For me bypass is manual, battery is full, bypass is enabled, but inverter limit is way too low. Overshoot is not fed into the grid AND it's not even matching the demand from the house right now.

I'd expect the inverter limit to be set to max when bypass is on.

```
2024-04-29 14:07:16,025:INFO: HUB: S:174.4W [ 177.2,177.2,177.0,176.4,176.3,174.4 ], B:100% (100), V:49.9V (49.9), C: 0W, P:False (manual), F:0.0h, E:103.1h, H:170W, L:1000W
2024-04-29 14:07:16,033:INFO: INV: AC:198.9W, AC_Prediction: 197.5W, DC:210.4W, DC_prediction: 209.0W (104.9|105.7), L:200.0W [1000.0W]
2024-04-29 14:07:16,033:INFO: SMT: T:ShellyEM3 P:1204.3W [ 1204.3,1204.3 ] Predict: 1204.3W
2024-04-29 14:07:16,034:INFO: Direct connected panel are producing 0.0W, trying to get 1383.3W from hub.
2024-04-29 14:07:16,035:INFO: Based on time, solarpower (174.4W) minimum charge power (50W) and bypass state (False), hub could contribute 124.4W - Decision path: 2.2.
2024-04-29 14:07:16,035:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-29 14:07:16,035:INFO: Solarflow is willing to contribute 62.0W (per channel) of the requested 1383.3!
2024-04-29 14:07:16,035:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-29 14:07:16,036:INFO: Sun: 05:57 - 20:23 Demand: 1383.3W, Panel DC: (0.0W), Hub DC: (199.0W), Inverter Limit: 200.0W, Hub Limit: 1000.0W
2024-04-29 14:07:16,036:INFO: SMT triggers limit function: 1204.3 -> 1204.3: executed
2024-04-29 14:07:22,000:INFO: SMT triggers limit function: 1257.1 -> 1268.0: skipped
2024-04-29 14:07:22,004:INFO: SMT triggers limit function: 1264.0 -> 1276.0: skipped
2024-04-29 14:07:22,008:INFO: SMT triggers limit function: 1271.1 -> 1284.1: skipped
2024-04-29 14:07:23,002:INFO: SMT triggers limit function: 1278.3 -> 1295.0: skipped
2024-04-29 14:07:23,006:INFO: SMT triggers limit function: 1285.6 -> 1303.4: skipped
2024-04-29 14:07:23,009:INFO: SMT triggers limit function: 1293.0 -> 1311.8: skipped
2024-04-29 14:07:24,999:INFO: SMT triggers limit function: 1301.1 -> 1328.3: skipped
2024-04-29 14:07:25,004:INFO: SMT triggers limit function: 1309.6 -> 1338.4: skipped
2024-04-29 14:07:25,008:INFO: SMT triggers limit function: 1318.7 -> 1349.0: skipped
2024-04-29 14:07:27,611:INFO: HUB: S:176.3W [ 176.5,176.5,175.7,176.3 ], B:100% (100), V:49.9V (49.9), C: 0W, P:False (manual), F:0.0h, E:103.1h, H:171W, L:1000W
2024-04-29 14:07:27,624:INFO: INV: AC:198.9W, AC_Prediction: 197.0W, DC:210.5W, DC_prediction: 209.0W (104.4|106.0), L:200.0W [1000.0W]
2024-04-29 14:07:27,625:INFO: SMT: T:ShellyEM3 P:1700.6W [ 1318.7,1318.7,1349.0 ] Predict: 1349.0W
2024-04-29 14:07:27,625:INFO: Direct connected panel are producing 0.0W, trying to get 1527.9W from hub.
2024-04-29 14:07:27,626:INFO: Based on time, solarpower (176.3W) minimum charge power (50W) and bypass state (False), hub could contribute 126.3W - Decision path: 2.2.
2024-04-29 14:07:27,626:INFO: Not setting solarflow output limit to 1000.0W as it is identical to current limit!
2024-04-29 14:07:27,627:INFO: Solarflow is willing to contribute 63.0W (per channel) of the requested 1527.9!
2024-04-29 14:07:27,627:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-29 14:07:27,628:INFO: Sun: 05:57 - 20:23 Demand: 1527.9W, Panel DC: (0.0W), Hub DC: (198.9W), Inverter Limit: 200.0W, Hub Limit: 1000.0W
2024-04-29 14:07:28,537:INFO: Triggering telemetry update: iot/A8yh63/4tS9ry5A/properties/read
```

reinhard-brandstaedter, 2024-04-29:
@boennemann keep in mind that the hub is reducing solarpower when the bypass switch is happening and/or the battery is just close to 100% or just reached the 100%, maybe that is why. At this time the solarpower is reduced. In your case the BP is not yet on [P:False (manual)] as soon as it has switched you should see the solarinput on the hub increase again, and then also the inverter limit.

mavo, 2024-04-29:
This feels like its getting very close to the other issue reported by me [here](https://github.com/reinhard-brandstaedter/solarflow-control/issues/227).
Maybe we find one common one to be used...?

boennemann, 2024-04-29:
Right now I'm in an infinite loop of the battery feeding into my house up until the minimum inverter limit with nothing from the panels. It does that for a few moments until the battery is no longer at 100%, then charges back up to 100% and the cycle starts anew.

Even after hours of waiting, or manually setting the hub to bypass manually via bluetooth nothing happens.

I'm super close to just throwing everything away and going back to the also broken stock CT-Mode.
I entered this whole rabbit-hole because I didn't want to lose feed-in, but now as soon as battery is full I'm losing both feed-in and my house's consumption is coming full-on from the grid.

I now tried resetting my entire config.ini to the exact defaults from the `dev` branch except enabling control_bypass, running the dev branch untampered and letting this sit until the end of the day. If I don't see this bypass mode spring into action I guess I'll have to use the public holiday on Wednesday to revert everything back to how it was and praying that Zendures dev team awakes from hibernation eventually – or find out whether my merchant would still refund me for the Hub.

reinhard-brandstaedter, 2024-04-29:
It really seems that your bypass is switching slow. FYI this is how it usually looks for me (letting sf-control do the BP switch):
[image]

Note that this is still limiting the inverter to demand on bypass. But I have removed this and it is now limiting to SF's max inverter input.

However what I have noticed also is a a strange behavior of the hub itself (just saw this the first time today) when the ```passMode``` is switched to 2 (manual on) the ```pass``` turns on (value: 1) and then after some time the ```passMode``` is set to 0 (automatic) again. Since this is definitely not happening in my logic I assume this happens in the firmware for some reason. So if that is the case it could well be that the firmware then "fights" the manual switch because it is in auto.
@boennemann can you provide logs / graphs of your infinite loop? And maybe also a screenshot of the telemetry/control data in MQTT?

reinhard-brandstaedter, 2024-04-29:
This is a "automatic" bypass behavior:
Note the battery was full at 16:15, the solarinput was decreased before that to take load off the relay) and then the BP only kicked in at 16:45. At 17:00 the hub decided for some reason to turn it OFF again shortly, only to turn it ON again, and voila there was some solarinput passed through finally.
That non-deterministic behavior is exactly what I'm trying to solve by manually switching the BP ON/OFF.

[image]

boennemann, 2024-04-29:
[image]

reinhard-brandstaedter, 2024-04-29:
Looks like BP kicked in and then ~200W were fed into house rather flat. Was this the demand at this time, so that your smartmeter/shelly reported around zero?

boennemann, 2024-04-30:
> Looks like BP kicked in and then ~200W were fed into house rather flat. Was this the demand at this time, so that your smartmeter/shelly reported around zero?

No I changed this line of code:

```py
inv_limit = 200 if inv_limit < 200 else int(inv_limit)
```

to at least not waste everything. Demand was even higher at times (heatpump).

Later I reverted fully back to latest `dev` branch to see if that breaks it somehow.

reinhard-brandstaedter, 2024-04-30:
@boennemann make sure you keep the full logs of today ;-)

reinhard-brandstaedter, 2024-04-30:
@boennemann I think I know what happened in your case with the bypass_control.
Can you check if your battery was fully discharged (hit 0%) the night before the day where the BP should have been enabled?

mavo, 2024-05-01:
Just because I am curious...
What's the connection between the battery reaching 0 and bp on the next day?

Also, isn't the app respecting the discharge limit nowadays again...?

reinhard-brandstaedter, 2024-05-01:
As the bypass control is still somewhat experimental…
I have used a existing flag “battery_target” that is flipped on either 0 or 100 (used for charge through) to identify if the BP can be turned on at 100%.
This was based on the assumption that the battery gets fully discharged over night (always in my case).
So then the BP for me always was enabled.

However if you haven’t reached 0 before that wouldn’t let the BP turn on.
I’ve now changed this.

Keep in mind that I’m trying to fix the firmwares own frequent BP on/off, it requires indeed a few considerations when to turn it on/off to avoid frequent switches.

reinhard-brandstaedter, 2024-05-01:
Bypass control with the latest changes (on dev) now works satisfying, only two switches on and off at sunset-offset.
Note that the hub reduced solarinput down to ~15W BEFORE the 100% battery SoC was reached and then the bypass was enabled:

[image]

Once we reached sunset - offset (120mins in my case) the bypass was disabled again and the hub continued with normal contribution work. No more bypass switching is allowed after this time until after next sunrise.

closed 2024-05-14 by reinhard-brandstaedter