reinhard-brandstaedter/solarflow-control #212: Should the hub also trigger limit function?

mavo, 2024-04-15:
Hi @reinhard-brandstaedter,

I am not sure if this was like it before, I only started to notice today (I changed to latest master yesterday in the morning ours).
For some reason (maybe even wanted) I do not see any "hub triggers limit function" calls, which I would expect when the hub input changes. 
The result for me is, that when the sun is out, household demand is matched, all good. But then a dark cloud happens to appear, as the hub is not triggering any update, my battery is then used to keep the inverter at its desired output rate. It will change until the normal 120s trigger runs and "cleans it".

Here are my startup logs:
```
2024-04-15 13:22:08,713:INFO: Solarflow Hub: A8yh63/ETFTke36
2024-04-15 13:22:08,713:INFO: Limit via inverter: True
2024-04-15 13:22:08,713:INFO: Control Parameters:
2024-04-15 13:22:08,713:INFO:   MIN_CHARGE_POWER = 0
2024-04-15 13:22:08,713:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-04-15 13:22:08,713:INFO:   MAX_INVERTER_LIMIT = 920
2024-04-15 13:22:08,713:INFO:   MAX_INVERTER_INPUT = 920
2024-04-15 13:22:08,713:INFO:   SUNRISE_OFFSET = 60
2024-04-15 13:22:08,713:INFO:   SUNSET_OFFSET = 60
2024-04-15 13:22:08,920:INFO: Using AhoyDTU: Base topic: ahoydtu, Limit topic: ahoydtu/ctrl/limit/0, SF Channels: [3, 4]
2024-04-15 13:22:08,920:INFO: Using ShellyEM3: Base topic: shellies/shellyem3
2024-04-15 13:22:08,921:INFO: Connected to MQTT Broker!
2024-04-15 13:22:08,921:INFO: Hub subscribing: /A8yh63/ETFTke36/properties/report
2024-04-15 13:22:08,921:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/solarInputPower
2024-04-15 13:22:08,921:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/electricLevel
2024-04-15 13:22:08,921:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputPackPower
2024-04-15 13:22:08,921:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/packInputPower
2024-04-15 13:22:08,922:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputHomePower
2024-04-15 13:22:08,922:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputLimit
2024-04-15 13:22:08,922:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/inverseMaxPower
2024-04-15 13:22:08,922:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/masterSoftVersion
2024-04-15 13:22:08,922:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/pass
2024-04-15 13:22:08,922:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/batteries/+/socLevel
2024-04-15 13:22:08,922:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/batteries/+/totalVol
2024-04-15 13:22:08,923:INFO: Hub subscribing: solarflow-hub/ETFTke36/control/#
2024-04-15 13:22:08,923:INFO: DTU subscribing: ahoydtu/HM-1600/+/P_DC
2024-04-15 13:22:08,923:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/P_AC
2024-04-15 13:22:08,923:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/active_PowerLimit
2024-04-15 13:22:08,923:INFO: DTU subscribing: ahoydtu/status
2024-04-15 13:22:08,923:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-04-15 13:22:08,923:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/0/power
2024-04-15 13:22:08,923:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/1/power
2024-04-15 13:22:08,924:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/2/power
2024-04-15 13:22:08,925:INFO: Reading last full time: 2024-04-12 17:10:29
2024-04-15 13:22:08,925:INFO: Reading battery target mode: discharging
```

And here are some logs from the app running:
```
2024-04-15 12:53:28,999:INFO: HUB: S:240.3W [ 251.0,251.0,249.6,248.1,246.4,240.3 ], B: 31% (31|32), V:49.3V (49.3|49.3), C: -27W, P:False, F:67.7h, E:-1.0h, H:259W, L:800W
2024-04-15 12:53:29,000:INFO: INV: AC:258.1W, AC_Prediction: 258.1W, DC:271.6W, DC_prediction: 271.6W (0.0|0.0|137.5|133.9), L:512.0W [1600W]
2024-04-15 12:53:29,002:INFO: SMT: T:ShellyEM3 P:139.4W [ 116.0,116.0,116.2,118.7,124.8,134.9 ] Predict: 92.3W
2024-04-15 12:53:29,002:INFO: Direct connected panels (0.0W) can't cover demand (393.0W), trying to get rest from hub.
2024-04-15 12:53:29,002:INFO: Checking if Solarflow is willing to contribute 393.0W ...
2024-04-15 12:53:29,002:INFO: Based on time, solarpower (240.3W) minimum charge power (0W) and bypass state (False), hub could contribute 240.3W - Decision path: 2.2.
2024-04-15 12:53:29,002:INFO: Hub should contribute more (240.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-15 12:53:29,002:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-04-15 12:53:29,002:INFO: Setting inverter output limit to 480W (1 min moving average of 120W x 4)
2024-04-15 12:53:29,003:INFO: Sun: 06:09 - 20:07 Demand: 393.0W, Panel DC: (0.0|0.0), Hub DC: (137.5|133.9), Inverter Limit: 480.0W, Hub Limit: 800.0W
2024-04-15 12:53:29,003:INFO: SMT triggers limit function: 124.8 -> 134.9: executed
2024-04-15 12:53:29,006:INFO: SMT triggers limit function: 125.0 -> 135.1: skipped
2024-04-15 12:53:33,015:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-04-15 12:54:32,770:INFO: HUB: S:232.6W [ 236.8,236.8,235.9,234.1,233.8,232.6 ], B: 31% (31|32), V:49.3V (49.4|49.3), C: -27W, P:False, F:67.7h, E:-1.0h, H:242W, L:800W
2024-04-15 12:54:32,770:INFO: INV: AC:243.5W, AC_Prediction: 243.5W, DC:256.2W, DC_prediction: 256.2W (0.0|0.0|120.2|117.1), L:480.0W [1600W]
2024-04-15 12:54:32,772:INFO: SMT: T:ShellyEM3 P:145.0W [ 149.9,149.9,149.6,150.3,151.6,147.1 ] Predict: 150.0W
2024-04-15 12:54:32,772:INFO: Direct connected panels (0.0W) can't cover demand (390.6W), trying to get rest from hub.
2024-04-15 12:54:32,773:INFO: Checking if Solarflow is willing to contribute 390.6W ...
2024-04-15 12:54:32,773:INFO: Based on time, solarpower (232.6W) minimum charge power (0W) and bypass state (False), hub could contribute 232.6W - Decision path: 2.2.
2024-04-15 12:54:32,773:INFO: Hub should contribute more (232.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-15 12:54:32,773:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-04-15 12:54:32,773:INFO: Setting inverter output limit to 464W (1 min moving average of 116W x 4)
2024-04-15 12:54:32,774:INFO: Sun: 06:09 - 20:07 Demand: 390.6W, Panel DC: (0.0|0.0), Hub DC: (120.2|117.1), Inverter Limit: 464.0W, Hub Limit: 800.0W
[... 27 lines cut ...]
2024-04-15 12:58:32,776:INFO: SMT: T:ShellyEM3 P:179.0W [ 165.7,165.7,163.3,159.5,155.1,154.9 ] Predict: 176.6W
2024-04-15 12:58:32,776:INFO: Direct connected panels (0.0W) can't cover demand (373.4W), trying to get rest from hub.
2024-04-15 12:58:32,776:INFO: Checking if Solarflow is willing to contribute 373.4W ...
2024-04-15 12:58:32,776:INFO: Based on time, solarpower (223.9W) minimum charge power (0W) and bypass state (False), hub could contribute 223.9W - Decision path: 2.2.
2024-04-15 12:58:32,777:INFO: Hub should contribute more (223.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
2024-04-15 12:58:32,777:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-04-15 12:58:32,777:INFO: Setting inverter output limit to 444W (1 min moving average of 111W x 4)
2024-04-15 12:58:32,777:INFO: Sun: 06:09 - 20:07 Demand: 373.4W, Panel DC: (0.0|0.0), Hub DC: (113.7|110.8), Inverter Limit: 444.0W, Hub Limit: 800.0W
2024-04-15 12:58:33,017:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-04-15 12:58:39,184:INFO: HUB: S:223.1W [ 222.9,222.9,223.2,223.7,223.8,223.1 ], B: 31% (31|32), V:49.3V (49.4|49.3), C: 0W, P:False, F:67.8h, E:-1.0h, H:220W, L:800W
2024-04-15 12:58:39,187:INFO: INV: AC:217.8W, AC_Prediction: 217.6W, DC:229.2W, DC_prediction: 229.1W (0.0|0.0|121.2|118.1), L:435.2W [1600W]
2024-04-15 12:58:39,189:INFO: SMT: T:ShellyEM3 P:160.0W [ 164.6,164.6,160.7,157.7,155.3,165.5 ] Predict: 173.5W
2024-04-15 12:58:39,189:INFO: Direct connected panels (0.0W) can't cover demand (383.3W), trying to get rest from hub.
2024-04-15 12:58:39,189:INFO: Checking if Solarflow is willing to contribute 383.3W ...
2024-04-15 12:58:39,189:INFO: Based on time, solarpower (223.1W) minimum charge power (0W) and bypass state (False), hub could contribute 223.1W - Decision path: 2.2.
2024-04-15 12:58:39,189:INFO: Hub should contribute more (223.0W) than what we currently get from panels (0.0W), we will use the inverter for fast/precise limiting!
[... cut ...]
2024-04-15 12:58:39,190:INFO: SMT triggers limit function: 155.3 -> 165.5: executed
2024-04-15 12:58:39,194:INFO: SMT triggers limit function: 155.4 -> 165.4: skipped
2024-04-15 12:59:33,017:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
```

In the current reported logs, it did not happen that the radiation changed drastically, but it was changing, but the app never reported any hub triggering something. Is this how it should be? If yes, please just close this ticket with a comment so that it does not clutter the other tickets.

reinhard-brandstaedter, 2024-04-15:
I haven’t seen any need to trigger things on hub change. Usually this is reflected in DTU changes or smartmeter changes.

mavo, 2024-04-15:
But if the demand of house does not change (a lot), its not triggering anything. This can happen with no direct connected panels. And in those cases its actually required. 
Is it actually hard to incorporate something like this?

reinhard-brandstaedter, 2024-04-15:
Not difficult, but usually too many triggers (that are linked) creates unnecessary steering noise.
Already today a change happening on the DTU likely triggers a change on the smartmeter as well.
A change on the hub usually also (in combination with min charge power) usually also triggers a change on the inverter.
And as a backup there is still the regular interval trigger.

mavo, 2024-04-15:
> Not difficult, but usually too many triggers (that are linked) creates unnecessary steering noise.

Understood. However, for me (and most likely for all having no direct panels), the DTU does not trigger too much, as it only changes when the SMT changes due to changes in demand.
Maybe the app can recognize when there is no direct connected panels...?

> And as a backup there is still the regular interval trigger.

Which is somewhat slowish.... 🤔

reinhard-brandstaedter, 2024-04-16:
The challenge here is that the hub usually reports it's solarpower very frequently, the more change the faster actually. So if we would want the hub to trigger the limit function as well we will need some kind of fixed time window (not a moving avg) to determine a significant change, otherwise it would just be a a short-interval of minimal changes (and we certainly do not want to trigger on those)
See this reported data:

[image]

mavo, 2024-04-16:
> we will need some kind of fixed time window (not a moving avg)

Which is tricky, as it might happen that the hub does not report anything if it does not change, such as it was the case for me when the shelly did not update with the fixed interval?
As far as I get it, those fixed time window is also used for the usage reported by the SMT, isn't it?

I mean, I get that this is not THE most urgent issue and also it does work and only happens on day time, so you won't feed energy to the grid (at least I did not see this yet and also cannot come with a situation this happens and can be prevented by this trigger).
On the other side, it would be great to have this, as in my config the only trigger thats left is the SMT, and if this does not change I need to wait for the regular interval. I was thinking about some sort of script which changes the DTU ever so slightly in intervals in case there was no change in the inverter reported. However, not only feels wrong, but also is as this would interfere with the whole sf-control eventually 😵

reinhard-brandstaedter, 2024-04-16:
No, the smartmeter uses a moving average window of 10s, since there are typically up/down movements on the smartmeter this works well for triggering adjustments on change. (because the change is usually not +2W every 2 seconds, but rather something like +60W or more instantly).

I'm running with no direct panels for a couple of weeks now and I haven't seen any issue with this setup. You are not loosing any energy. If it doesn't go to the house it goes to the battery, and since you are running with a min_charge level  of 0 (I still don't see the benefit of that) it will be adjusted soon enough. You are just getting a bit of charge power for max 2 mins.

reinhard-brandstaedter, 2024-04-16:
The latest master does have a trigger on hub solarinput change. Try if this changes the steering cadence for you.

mavo, 2024-04-16:
You are an absolute hero!
I'll try, but I probably won't have a result tomorrow, as it's time for a full charge cycle :-)

reinhard-brandstaedter, 2024-04-18:
HUB triggers limit function now correctly

closed 2024-04-18 by reinhard-brandstaedter