reinhard-brandstaedter/solarflow-control #216: Control script unable to set SF output - for 6 days (correlation with update)

AzraelsDisk, 2024-04-17:
Since round about 5 days the control script is not able to set the SF output anymore. 
It is independend if it goes up 0 --> 300 --> 800 or down 800 --> 0 

If is done manuale via the Statuspage it is working 

I use the latest version of control script and statuspage

Option: --Online is used 

> 2024-04-17 03:11:13,022:INFO: HUB: S:0.0W [ 0.0 ], B: 0% ( 0| 0), V:42.1V (42.1|42.1), C:-143W, P:False, F:80.3h, E:0.0h, H:137W, L:800W
2024-04-17 03:11:13,024:INFO: INV: AC:142.4W, AC_Prediction: 142.9W, DC:150.0W, DC_prediction: 150.5W (74.7|74.3|0.0|0.0), L:279.0W [1500.0W]
2024-04-17 03:11:13,025:INFO: SMT: T:ShellyEM3 P:166.7W [ 32.5,32.5,34.3,37.6,41.4,53.7 ] Predict: 13.9W
2024-04-17 03:11:13,025:INFO: Direct connected panel are producing 0.0W, trying to get 175.2W from hub.
2024-04-17 03:11:13,025:INFO: Based on time, solarpower ( 0.0W) minimum charge power (45W) and bypass state (False), hub could contribute 150.0W - Decision path: 2.1.
2024-04-17 03:11:13,026:INFO: Battery is empty! Disabling solaraflow output, setting limit to 0
2024-04-17 03:11:13,026:INFO: Setting solarflow output limit to 0.0W
2024-04-17 03:11:13,026:INFO: Solarflow is willing to contribute 75.0W of the requested 175.2!
2024-04-17 03:11:13,026:INFO: Setting inverter output limit to 300W (1 min moving average of 75W x 4)
2024-04-17 03:11:13,026:INFO: Sun: 06:20 - 20:09 Demand: 175.2W, Panel DC: (0.0W), Hub DC: (141.5W), Inverter Limit: 300.0W, Hub Limit: 0.0W
2024-04-17 03:11:13,026:INFO: SMT triggers limit function: 41.4 -> 53.7: executed
2024-04-17 03:11:13,028:INFO: SMT triggers limit function: 45.4 -> 61.2: skipped
2024-04-17 03:11:13,030:INFO: SMT triggers limit function: 49.1 -> 66.8: skipped
2024-04-17 03:11:14,022:INFO: SMT triggers limit function: 55.3 -> 71.9: skipped
2024-04-17 03:11:14,024:INFO: SMT triggers limit function: 58.6 -> 76.2: skipped
2024-04-17 03:11:14,027:INFO: SMT triggers limit function: 61.7 -> 82.6: skipped
2024-04-17 03:11:17,027:INFO: SMT triggers limit function: 64.7 -> 120.2: skipped
2024-04-17 03:11:17,029:INFO: SMT triggers limit function: 67.4 -> 123.0: skipped
2024-04-17 03:11:17,031:INFO: SMT triggers limit function: 69.9 -> 125.4: skipped
2024-04-17 03:11:30,024:INFO: HUB: S:0.0W [ 0.0 ], B: 0% ( 0| 0), V:42.1V (42.1|42.1), C: -12W, P:False, F:80.3h, E:0.0h, H: 12W, L:800W
2024-04-17 03:11:30,026:INFO: INV: AC:70.8W, AC_Prediction: 49.2W, DC:74.7W, DC_prediction: 52.1W (0.2|0.2|0.0|0.0), L:300.0W [1500.0W]
2024-04-17 03:11:30,027:INFO: SMT: T:ShellyEM3 P:165.9W [ 57.4,57.4,65.7,80.6,144.2,170.6 ] Predict: -2.2W
2024-04-17 03:11:30,028:INFO: Direct connected panel are producing 0.0W, trying to get 150.6W from hub.
2024-04-17 03:11:30,028:INFO: Based on time, solarpower ( 0.0W) minimum charge power (45W) and bypass state (False), hub could contribute 150.0W - Decision path: 2.1.
2024-04-17 03:11:30,028:INFO: Hub has just recently adjusted limit, need to wait until it is set again! Current limit: 800, new limit: 800
2024-04-17 03:11:30,028:INFO: Solarflow is willing to contribute 75.0W of the requested 150.6!
2024-04-17 03:11:30,028:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-17 03:11:30,028:INFO: Sun: 06:20 - 20:09 Demand: 150.6W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 300.0W, Hub Limit: 800.0W
2024-04-17 03:11:30,028:INFO: SMT triggers limit function: 144.2 -> 170.6: executed
2024-04-17 03:11:30,030:INFO: SMT triggers limit function: 145.7 -> 170.6: skipped
2024-04-17 03:11:30,032:INFO: SMT triggers limit function: 147.2 -> 170.7: skipped
2024-04-17 03:11:51,959:INFO: Triggering telemetry update: iot/73bktV/mg773f30/properties/read
2024-04-17 03:12:51,707:INFO: HUB: S:0.0W [ 0.0 ], B: 0% ( 0| 0), V:42.8V (42.8|42.9), C: 0W, P:False, F:80.3h, E:0.0h, H: 0W, L:800W
2024-04-17 03:12:51,710:INFO: INV: AC:70.8W, AC_Prediction: 49.2W, DC:74.7W, DC_prediction: 52.1W (0.2|0.2|0.0|0.0), L:300.0W [1500.0W]
2024-04-17 03:12:51,711:INFO: SMT: T:ShellyEM3 P:164.5W [ 166.1,166.1,165.3,165.8,165.7,164.3 ] Predict: 167.7W
2024-04-17 03:12:51,711:INFO: Direct connected panel are producing 0.0W, trying to get 144.3W from hub.
2024-04-17 03:12:51,711:INFO: Based on time, solarpower ( 0.0W) minimum charge power (45W) and bypass state (False), hub could contribute 144.3W - Decision path: 2.1.
2024-04-17 03:12:51,711:INFO: Battery is empty! Disabling solaraflow output, setting limit to 0
2024-04-17 03:12:51,712:INFO: Setting solarflow output limit to 0.0W
2024-04-17 03:12:51,712:INFO: Solarflow is willing to contribute 72.0W of the requested 144.3!
2024-04-17 03:12:51,712:INFO: Inverter is not reachable/down. Can't set limit
2024-04-17 03:12:51,712:INFO: Sun: 06:20 - 20:09 Demand: 144.3W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 288.0W, Hub Limit: 0.0W
2024-04-17 03:12:51,959:INFO: Triggering telemetry update: iot/73bktV/mg773f30/properties/read
2024-04-17 03:13:51,960:INFO: Triggering telemetry update: iot/73bktV/mg773f30/properties/read
2024-04-17 03:14:51,708:INFO: HUB: S:0.0W [ 0.0 ], B: 0% ( 0| 0), V:43.0V (43.0|43.0), C: 0W, P:False, F:80.3h, E:0.1h, H: 0W, L:800W
2024-04-17 03:14:51,711:INFO: INV: AC:70.8W, AC_Prediction: 49.2W, DC:74.7W, DC_prediction: 52.1W (0.2|0.2|0.0|0.0), L:300.0W [1500.0W]
2024-04-17 03:14:51,712:INFO: SMT: T:ShellyEM3 P:153.1W [ 156.2,156.2,155.8,155.4,154.9,154.7 ] Predict: 158.1W
2024-04-17 03:14:51,712:INFO: Direct connected panel are producing 0.0W, trying to get 134.7W from hub.
2024-04-17 03:14:51,712:INFO: Based on time, solarpower ( 0.0W) minimum charge power (45W) and bypass state (False), hub could contribute 134.7W - Decision path: 2.1.
2024-04-17 03:14:51,712:INFO: Battery is empty! Disabling solaraflow output, setting limit to 0
2024-04-17 03:14:51,713:INFO: Setting solarflow output limit to 0.0W
2024-04-17 03:14:51,713:INFO: Solarflow is willing to contribute 67.0W of the requested 134.7!
2024-04-17 03:14:51,713:INFO: Inverter is not reachable/down. Can't set limit
2024-04-17 03:14:51,713:INFO: Sun: 06:20 - 20:09 Demand: 134.7W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 268.0W, Hub Limit: 0.0W
2024-04-17 03:14:51,961:INFO: Triggering telemetry update: iot/73bktV/mg773f30/properties/read
2024-04-17 03:15:51,961:INFO: Triggering telemetry update: iot/73bktV/mg773f30/properties/read
2024-04-17 03:16:51,435:INFO: Publishing Homeassistant templates...
2024-04-17 03:16:51,470:INFO: Published 26 Homeassistant templates.
2024-04-17 03:16:51,709:INFO: HUB: S:0.0W [ 0.0 ], B: 0% ( 0| 0), V:43.0V (43.0|43.0), C: 0W, P:False, F:80.3h, E:0.1h, H: 0W, L:800W
2024-04-17 03:16:51,711:INFO: INV: AC:70.8W, AC_Prediction: 49.2W, DC:74.7W, DC_prediction: 52.1W (0.2|0.2|0.0|0.0), L:300.0W [1500.0W]
2024-04-17 03:16:51,712:INFO: SMT: T:ShellyEM3 P:167.0W [ 159.4,159.4,161.3,164.3,163.8,164.2 ] Predict: 153.8W
[... 23 lines cut ...]
2024-04-17 03:20:51,713:INFO: SMT: T:ShellyEM3 P:161.8W [ 167.2,167.2,165.6 ] Predict: 165.6W
2024-04-17 03:20:51,713:INFO: Direct connected panel are producing 0.0W, trying to get 145.6W from hub.
2024-04-17 03:20:51,713:INFO: Based on time, solarpower ( 0.0W) minimum charge power (45W) and bypass state (False), hub could contribute 145.6W - Decision path: 2.1.
2024-04-17 03:20:51,713:INFO: Battery is empty! Disabling solaraflow output, setting limit to 0
2024-04-17 03:20:51,713:INFO: Setting solarflow output limit to 0.0W
2024-04-17 03:20:51,714:INFO: Solarflow is willing to contribute 72.5W of the requested 145.6!
2024-04-17 03:20:51,714:INFO: Inverter is not reachable/down. Can't set limit
2024-04-17 03:20:51,714:INFO: Sun: 06:20 - 20:09 Demand: 145.6W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 288.0W, Hub Limit: 0.0W
2024-04-17 03:20:51,965:INFO: Triggering telemetry update: iot/73bktV/mg773f30/properties/read
2024-04-17 03:21:51,965:INFO: Triggering telemetry update: iot/73bktV/mg773f30/properties/read
2024-04-17 03:22:51,710:INFO: HUB: S:0.0W [ 0.0 ], B: 0% ( 0| 0), V:43.0V (43.0|43.0), C: 0W, P:False, F:80.4h, E:0.2h, H: 0W, L:800W
<

**On the statuspage log there is no signal of setting the SF output**

>
2024-04-16 18:17:53,472:INFO: 192.168.188.99 - - [16/Apr/2024 18:17:53] "POST /socket.io/?EIO=4&transport=polling&t=OxdirIn&sid=iyiLfiUsWnwILUhOAAAw HTTP/1.1" 200 -
2024-04-16 18:17:53,474:INFO: 192.168.188.99 - - [16/Apr/2024 18:17:53] "GET /socket.io/?EIO=4&transport=polling&t=OxdirIo&sid=iyiLfiUsWnwILUhOAAAw HTTP/1.1" 200 -
2024-04-16 18:17:53,484:INFO: 192.168.188.99 - - [16/Apr/2024 18:17:53] "GET /socket.io/?EIO=4&transport=polling&t=OxdirJ4&sid=iyiLfiUsWnwILUhOAAAw HTTP/1.1" 200 -
2024-04-16 18:20:50,056:INFO: Client disconnected
2024-04-16 18:20:50,056:INFO: 192.168.188.99 - - [16/Apr/2024 18:20:50] "GET /socket.io/?EIO=4&transport=websocket&sid=iyiLfiUsWnwILUhOAAAw HTTP/1.1" 200 -
2024-04-17 03:11:09,437:WARNING: batteryTarget is type str: charging
2024-04-17 06:17:46,990:WARNING: batteryTarget is type str: charging
2024-04-17 06:42:27,782:WARNING: batteryTarget is type str: charging
<

tuxianerDE, 2024-04-19:
Hey,

 I am seeing the related issue with the battery discharge as the mqtt connection is lost. Are you still in online mode? The likelihood that something weird is happening with the online MQTT connection to Zendure on their site is very likely. 

I know the phenomenon you describe, that is why we added the feature that if the SoC is 0 the limited of the HUB gets set to 0W, otherwise with first light the hubs finds residual energy and bursts it out leading to fact that you have to do a hard reset.

So the functionality to protect you is there appears to be the connecting to Zendure being spotty as ususal.

AzraelsDisk, 2024-04-19:
Yes, I‘m in online mode.

What I would like to stress is, that since this update something is brocken between control-script and Statuspage for setting SF-Output in online mode!

It is never working now. Prior this update, I could see in the Log of Satuspage, that it sends the control command to Zendure when the control-script changes the Output for SF. This is now missing.

I know the workaround is go offline … what I will do in the future… but maybe somebody else don’t like it and it would be very good if you could fix this

Thanks

tuxianerDE, 2024-04-19:
Hey,

ok I might have missed what update are we talking about:
- We had a new master update of the SF control skript last week -> you mean this update?
- Zendure also released a new Firmware version -> you mean this update? 
- Lastly the status page update -> there was a new version last week.

sorry to ask but given that were so many variables flying around just want to be sure that we look at the right thing.

AzraelsDisk, 2024-04-19:
From my point of view it occurs on Contol or Statuspage. Zendure was later so I would exclude thisSorry that i can not be precisely more

closed 2024-11-12 by tuxianerDE

AzraelsDisk, 2024-11-14:
The problem still exists. 
I use the --onlinemode in this case the control commands are not transmitted. In other words, the output limit of the SF is not changed.

If I make this change manually, however, it works.
Would you please look at this again? It's a pity because I can't use many functions. 

Unfortunately, taking it offline is not an option, as the SF1200 can only use WPA2 and I don't want to let it into my main network