reinhard-brandstaedter/solarflow-control #221: Pushing Battery to Grid

AzraelsDisk, 2024-04-18:
Good morning, 

Controlscript actually doesn't limit Inverter if Battery is pused to Grid far to high until its empty

> 2024-04-18 08:30:54,992:INFO: Based on time, solarpower ( 0.0W) minimum charge power (45W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-04-18 08:30:54,992:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-04-18 08:30:54,992:INFO: Solarflow is willing to contribute **### 0.0W** of the requested 110.6!
2024-04-18 08:30:54,992:INFO: Grid feed in from battery: 439.6W!
2024-04-18 08:30:54,992:INFO: Sun: 06:18 - 20:10 Demand: 110.6W, Panel DC: (0.0W), ### **Hub DC: (550.2W)**, **_Inverter Limit: 1500.0W,_** Hub Limit: 800.0W
2024-04-18 08:30:54,992:INFO: SMT triggers limit function: -408.8 -> -419.6: executed
2024-04-18 08:30:54,994:INFO: SMT triggers limit function: -410.8 -> -421.6: skipped
2024-04-18 08:30:54,996:INFO: SMT triggers limit function: -412.7 -> -423.5: skipped
2024-04-18 08:31:05,991:INFO: SMT triggers limit function: -434.0 -> -454.9: skipped
2024-04-18 08:31:05,994:INFO: SMT triggers limit function: -435.9 -> -457.0: skipped
2024-04-18 08:31:05,995:INFO: SMT triggers limit function: -437.6 -> -458.6: skipped
2024-04-18 08:31:06,991:INFO: SMT triggers limit function: -439.4 -> -460.4: skipped
2024-04-18 08:31:06,993:INFO: SMT triggers limit function: -440.8 -> -461.5: skipped
2024-04-18 08:31:06,995:INFO: SMT triggers limit function: -442.1 -> -462.6: skipped
2024-04-18 08:31:07,993:INFO: SMT triggers limit function: -443.3 -> -463.5: skipped
2024-04-18 08:31:07,995:INFO: SMT triggers limit function: -444.6 -> -464.6: skipped
2024-04-18 08:31:07,997:INFO: SMT triggers limit function: -445.9 -> -465.7: skipped
2024-04-18 08:31:12,994:INFO: HUB: S:0.0W [ 0.0 ], B: 13% (13|14), V:45.9V (45.9|45.9), C: -94W, P:False, F:109.6h, E:25.8h, H: 90W, L:800W
2024-04-18 08:31:12,995:INFO: INV: AC:596.4W, AC_Prediction: 638.9W, DC:611.5W, DC_prediction: 611.5W (322.2|319.5|0.0|0.0), L:1500.0W [1500.0W]
2024-04-18 08:31:12,996:INFO: SMT: T:ShellyEM3 P:-490.2W [ -411.3,-411.3,-432.2,-442.1,-456.1,-466.8 ] Predict: -341.9W
<

AzraelsDisk, 2024-04-18:
may this happens due to connectino lost of zendure MQTT... ?

reinhard-brandstaedter, 2024-04-18:
According to that your hub is currently providing ~90W to the inverter, but the inverter says it is providing ~600W.
So either your hub data is not correct or your inverter data is not correct.
Since you are relying on Zendure's MQTT I'd say your hub data is not up to date. Take your hub offline ;-)

closed 2024-05-07 by AzraelsDisk