reinhard-brandstaedter/solarflow-control #222: TypeError: unsupported operand type(s) for *: 'method' and 'float'

busaku, 2024-04-18:
```
2024-04-18 14:33:21,197:INFO: Reading battery target mode: discharging
2024-04-18 14:33:21,197:INFO: Reading last full time: 2024-04-18 13:22:25
2024-04-18 14:33:24,532:INFO: Battery is full: -1
2024-04-18 14:33:24,536:INFO: Reading last full time: 2024-04-18 14:33:24
2024-04-18 14:33:24,537:INFO: Reading battery target mode: discharging
2024-04-18 14:34:00,665:INFO: HUB: S:234.0W [ 234.2,234.2,234.0 ], B:100% (100|100), V:50.3V (50.3|50.3), C: 0W, P:True (auto), F:0.0h, E:-1.0h, H:231W, L: 0W
2024-04-18 14:34:00,665:INFO: INV: AC:513.5W, AC_Prediction: 513.5W, DC:540.4W, DC_prediction: 540.4W (149.7|149.9|83.3|156.6), L: 0W [1600W]
2024-04-18 14:34:00,666:INFO: SMT: T:ShellyEM3 P:-251.8W [ -224.9,-224.9,-259.3 ] Predict: -259.3W
2024-04-18 14:34:00,666:INFO: Direct connected panels (284.7W) can cover demand (53.4W)
Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 548, in <module>
    main(sys.argv[1:])
  File "/solarflow/solarflow-control.py", line 545, in main
    run()
  File "/solarflow/solarflow-control.py", line 483, in run
    client.loop_forever()
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/solarflow/solarflow-control.py", line 133, in on_message
    smartmeter.handleMsg(msg)
  File "/solarflow/smartmeters.py", line 96, in handleMsg
    self.updPower()
  File "/solarflow/smartmeters.py", line 80, in updPower
    log.info(f'SMT triggers limit function: {previous} -> {self.getPower()}: {"executed" if self.trigger_callback(self.client,force=force_trigger) else "skipped"}')
                                                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 449, in limit_callback
    limitHomeInput(client)
  File "/solarflow/solarflow-control.py", line 288, in limitHomeInput
    direct_limit = getDirectPanelLimit(inv,hub,smt)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 187, in getDirectPanelLimit
    direct_panel_power = inv.getDirectACPower() + inv.getHubACPower() if hub.getBypass() else 0
                         ^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/dtus.py", line 163, in getDirectACPower
    return self.getDirectDCPower * (self.getEfficiency()/100)
           ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for *: 'method' and 'float'
```

Did not digging deeper at the moment (battery is 100%, direct pv did not cover power consumption, but zendure + direct pv are enough, so no discharging needed

busaku, 2024-04-18:
Ah yes:
https://github.com/reinhard-brandstaedter/solarflow-control/blame/9452879f77c53f070c316293de0f1f9833a3f360/src/solarflow/dtus.py#L163C21-L163C37

missing ()

reinhard-brandstaedter, 2024-04-18:
jesus what happened to my tests!

closed 2024-04-18 by reinhard-brandstaedter