reinhard-brandstaedter/solarflow-control #225: New Bypass Control

busaku, 2024-04-18:
https://github.com/reinhard-brandstaedter/solarflow-control/blob/68e731b7e56109a7d72d97d47523132eee221018/src/solarflow/solarflow-control.py#L210

How can we be sure, that only happens once ?
So, as I understood, it sets the byPass to false (it ignore the config settings if the user like to control the bypass). But what happened if the hub sets the flow back to true? Or next loop will set it to false again. How can we be sure, that it doesn't trigger the byPass state in a loop?

https://github.com/reinhard-brandstaedter/solarflow-control/blob/68e731b7e56109a7d72d97d47523132eee221018/src/solarflow/solarflow.py#L137

On this point, it will set always the state twice to true, on line 137 and after that on line 142
And: How we can be sure, that you comment on 136 will be secured? If we set the byPass to true and a new metric update runs fast, we don't wait until its true, we set it to true again.

https://github.com/reinhard-brandstaedter/solarflow-control/blob/68e731b7e56109a7d72d97d47523132eee221018/src/solarflow/solarflow.py#L365 

As I see this line correctly: we just set the state to false directly, but on line 137 we are set it to true, so we don't wait.

Please do not take my suggestions or questions as criticism. I just want to help make this project better and better :)
And: Iam not a python pro, so maybe my interpretation aren't correct :)

reinhard-brandstaedter, 2024-04-19:
You are right, that part here should actually be removed, I made a mistake on the merge. The ```batteryTarget``` is used as a context switch.
https://github.com/reinhard-brandstaedter/solarflow-control/blob/68e731b7e56109a7d72d97d47523132eee221018/src/solarflow/solarflow.py#L142

Whenever the 100% is reached the ```batteryTarget``` is flipped from _charging_ to _discharging_ so the bypass is only turned off then.

The reason for directly setting the state here:
https://github.com/reinhard-brandstaedter/solarflow-control/blob/68e731b7e56109a7d72d97d47523132eee221018/src/solarflow/solarflow.py#L365

...is that we want to continue determining limit values in the control part immediately here:
https://github.com/reinhard-brandstaedter/solarflow-control/blob/68e731b7e56109a7d72d97d47523132eee221018/src/solarflow/solarflow-control.py#L217
..after turning the bypass of in line 211.

Keep in mind the bypass-control is still experimental, still need to determin the effects of manually switching on/off.

busaku, 2024-04-19:
Thanks for your feedback and your fix, i close this issue

commit 83f22119 (2024-04-19): Fix #225
closed 2024-04-19 by busaku