reinhard-brandstaedter/solarflow-control #226: Current inverter AC is higher than configured - Limit not set (identical)

busaku, 2024-04-23:
```
2024-04-23 11:28:05,122:INFO: HUB: S:654.5W [ 652.9,652.9,652.8,653.5,654.5 ], B:100% (100|100), V:51.3V (51.3|51.3), C: 0W, P:True (auto), F:0.0h, E:-1.0h, H:637W, L: 0W
2024-04-23 11:28:05,122:INFO: INV: AC:1296.6W, AC_Prediction: 1296.6W, DC:1364.7W, DC_prediction: 1364.7W (357.3|352.6|228.6|427.3), L:2099W (524.8W/channel) [1400W]
2024-04-23 11:28:05,129:INFO: SMT: T:ShellyEM3 P:-863.1W [ -853.9,-853.9,-857.4,-858.3,-858.2,-851.4 ] Predict: -849.7W
2024-04-23 11:28:05,129:INFO: Direct connected panels (674.5W) can cover demand (426.3W)
2024-04-23 11:28:05,130:INFO: Not setting solarflow output limit to 0.0W as it is identical to current limit!
2024-04-23 11:28:05,130:INFO: Current inverter AC output (1296.6) is higher than configured output limit (800), reducing limit to 2095.2
2024-04-23 11:28:05,130:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-23 11:28:05,131:INFO: Grid feed in from unknown: 871.4!
2024-04-23 11:28:05,132:INFO: Sun: 06:21 - 20:43 Demand: 426.3W, Panel DC: (674.5W), Hub DC: (623.2W), Inverter Limit: 2095.2W, Hub Limit: 0.0W
2024-04-23 11:28:05,464:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-04-23 11:28:09,044:INFO: Reading last full time: 2024-04-23 11:28:09
2024-04-23 11:28:09,045:INFO: Reading battery target mode: discharging
2024-04-23 11:28:23,114:INFO: Reading last full time: 2024-04-23 11:28:23
2024-04-23 11:28:23,114:INFO: Reading battery target mode: discharging
2024-04-23 11:28:55,350:INFO: HUB: S:655.0W [ 654.7,654.7,655.0 ], B:100% (100|100), V:51.2V (51.3|51.2), C: 0W, P:True (auto), F:0.0h, E:-1.0h, H:638W, L: 0W
2024-04-23 11:28:55,351:INFO: INV: AC:1297.8W, AC_Prediction: 1297.8W, DC:1366.0W, DC_prediction: 1366.0W (356.3|351.7|228.7|427.3), L:2099W (524.8W/channel) [1400W]
2024-04-23 11:28:55,351:INFO: SMT: T:ShellyEM3 P:-840.4W [ -847.6,-847.6,-857.8 ] Predict: -857.8W
2024-04-23 11:28:55,352:INFO: Direct connected panels (672.6W) can cover demand (418.1W)
2024-04-23 11:28:55,352:INFO: Not setting solarflow output limit to 0.0W as it is identical to current limit!
2024-04-23 11:28:55,353:INFO: Current inverter AC output (1297.8) is higher than configured output limit (800), reducing limit to 2095.2
2024-04-23 11:28:55,353:INFO: Not setting inverter output limit as it is identical to current limit!
2024-04-23 11:28:55,353:INFO: Grid feed in from unknown: 877.8!
2024-04-23 11:28:55,355:INFO: Sun: 06:21 - 20:43 Demand: 418.1W, Panel DC: (672.6W), Hub DC: (623.2W), Inverter Limit: 2095.2W, Hub Limit: 0.0W
2024-04-23 11:28:55,355:INFO: SMT triggers limit function: -847.6 -> -857.8: executed
```

So, batterie is full, but the inverter did not get limited.
It looks like the inverter limit isn't calculated from the maximum output if the ac output is higher as the limit.  But iam not sure.

reinhard-brandstaedter, 2024-04-23:
True those two "borders" need to be matched.
I'm wondering though if the getting close to the ac limit actually triggered at one point. could you maybe share the logs from before?

busaku, 2024-04-23:
Sorry, i don't have the logs anymore.
I have implement my idea with withinRange. Looking good.

So i think you need to change line 230 too in your fix :)

Like:
```
 if self.getCurrentACPower() > self.acLimit and inv_limit > self.acLimit:
            # decrease inverter limit slowly
            inv_limit = self.limitAbsolute - 4
            withinRange = 0
            log.info(f'Current inverter AC output ({self.getCurrentACPower()}) is higher than configured output limit ({self.acLimit}), reducing limit to {inv_limit}')

        # failsafe: if the current AC output is close to the AC limit do not increase the invert limit too much
        if self.getCurrentACPower() < self.acLimit and self.isWithin(self.getCurrentACPower(), self.acLimit, 6):
            # only increase inverter limit a little bit
            inv_limit = self.limitAbsolute + 2
            withinRange = 0
            log.info(f'Current inverter AC output ({self.getCurrentACPower()}) is close to the configured AC output limit ({self.acLimit}), slow limit increase to {inv_limit}') 
        
        #if self.limitAbsolute != inv_limit and self.reachable:
        if not self.isWithin(inv_limit,self.limitAbsolute,withinRange) and self.reachable: 
```

closed 2024-05-14 by reinhard-brandstaedter