reinhard-brandstaedter/solarflow-control #227: Bypass on, but no power delivered to home

mavo, 2024-04-23:
Hi @reinhard-brandstaedter,

for some reason there is no power delivered to my home, even tho the battery is full and bypass got enabled.

[image]

Here is some log output, do you have any idea what's going on?
 
```
2024-04-23 14:54:27,631:INFO: HUB: S:83.0W [ 83.0,83.0 ], B:100% (100|10
0), V:52.8V (52.8|52.7), C: 0W, P:True (auto), F:0.0h, E:-1.0h, H: 0W, L
:800W
2024-04-23 14:54:27,632:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:3.6W
, DC_prediction: 3.6W (0.0|0.0|1.7|1.9), L:416.0W [1600W]
2024-04-23 14:54:27,634:INFO: SMT: T:ShellyEM3 P:218.4W [ 219.8,219.8,21
9.6,219.7,220.0,219.9 ] Predict: 220.2W
2024-04-23 14:54:27,634:INFO: Direct connected panel are producing 0.0W,
 trying to get 209.9W from hub.
2024-04-23 14:54:27,634:INFO: Based on time, solarpower (83.0W) minimum
charge power (0W) and bypass state (True), hub could contribute 209.9W -
 Decision path: 0.2.
2024-04-23 14:54:27,634:INFO: Not setting solarflow output limit to 800.
0W as it is identical to current limit!
2024-04-23 14:54:27,634:INFO: Solarflow is willing to contribute 104.5W
(per channel) of the requested 209.9!
2024-04-23 14:54:27,634:INFO: Not setting inverter output limit as it is
 identical to current limit!
2024-04-23 14:54:27,635:INFO: Sun: 05:52 - 20:21 Demand: 209.9W, Panel D
C: (0.0W), Hub DC: (0.0W), Inverter Limit: 416.0W, Hub Limit: 800.0W
2024-04-23 14:54:30,794:INFO: Reading last full time: 2024-04-23 14:54:3
0
2024-04-23 14:54:30,794:INFO: Reading battery target mode: discharging
2024-04-23 14:54:47,898:INFO: Reading last full time: 2024-04-23 14:54:4
7
2024-04-23 14:54:47,938:INFO: Reading battery target mode: discharging
2024-04-23 14:55:27,603:INFO: Triggering telemetry update: iot/A8yh63/ET
FTke36/properties/read
2024-04-23 14:55:30,215:INFO: Reading last full time: 2024-04-23 14:55:3
0
2024-04-23 14:55:30,215:INFO: Reading battery target mode: discharging
2024-04-23 14:55:48,317:INFO: Reading last full time: 2024-04-23 14:55:4
8
2024-04-23 14:55:48,358:INFO: Reading battery target mode: discharging
2024-04-23 14:56:27,575:INFO: Publishing Homeassistant templates...
2024-04-23 14:56:27,603:INFO: Triggering telemetry update: iot/A8yh63/ET
FTke36/properties/read
2024-04-23 14:56:27,612:INFO: Published 26 Homeassistant templates.
2024-04-23 14:56:27,632:INFO: HUB: S:79.5W [ 79.5,79.5 ], B:100% (100|10
0), V:52.6V (52.6|52.6), C: 0W, P:True (auto), F:0.0h, E:-1.0h, H: 0W, L
:800W
2024-04-23 14:56:27,632:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:4.2W
, DC_prediction: 4.2W (0.0|0.0|2.0|2.2), L:416.0W [1600W]
2024-04-23 14:56:27,634:INFO: SMT: T:ShellyEM3 P:222.9W [ 224.4,224.4,22
4.3,224.1,224.0,223.8 ] Predict: 225.2W
2024-04-23 14:56:27,634:INFO: Direct connected panel are producing 0.0W,
 trying to get 213.8W from hub.
2024-04-23 14:56:27,634:INFO: Based on time, solarpower (79.5W) minimum
charge power (0W) and bypass state (True), hub could contribute 213.8W -
 Decision path: 0.2.
2024-04-23 14:56:27,634:INFO: Not setting solarflow output limit to 800.
0W as it is identical to current limit!
2024-04-23 14:56:27,634:INFO: Solarflow is willing to contribute 106.5W
(per channel) of the requested 213.8!
2024-04-23 14:56:27,635:INFO: Setting inverter output limit to 424W (1 m
in moving average of 106W x 4)
2024-04-23 14:56:27,635:INFO: Sun: 05:52 - 20:21 Demand: 213.8W, Panel D
C: (0.0W), Hub DC: (0.0W), Inverter Limit: 424.0W, Hub Limit: 800.0W
2024-04-23 14:56:30,600:INFO: Reading last full time: 2024-04-23 14:56:3
0
2024-04-23 14:56:30,641:INFO: Reading battery target mode: discharging
2024-04-23 14:56:48,726:INFO: Reading last full time: 2024-04-23 14:56:4
8
2024-04-23 14:56:48,766:INFO: Reading battery target mode: discharging
2024-04-23 14:57:27,603:INFO: Triggering telemetry update: iot/A8yh63/ET
FTke36/properties/read
2024-04-23 14:57:30,954:INFO: Reading last full time: 2024-04-23 14:57:3
0
2024-04-23 14:57:30,955:INFO: Reading battery target mode: discharging
2024-04-23 14:57:49,099:INFO: Reading last full time: 2024-04-23 14:57:4
9
2024-04-23 14:57:49,099:INFO: Reading battery target mode: discharging
2024-04-23 14:58:27,604:INFO: Triggering telemetry update: iot/A8yh63/ET
FTke36/properties/read
2024-04-23 14:58:27,632:INFO: HUB: S:88.5W [ 88.0,88.0,88.5 ], B:100% (1
00|100), V:52.5V (52.5|52.5), C: 0W, P:True (auto), F:0.0h, E:-1.0h, H:
0W, L:800W
2024-04-23 14:58:27,632:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:4.2W
, DC_prediction: 4.2W (0.0|0.0|2.0|2.2), L:424.0W [1600W]
2024-04-23 14:58:27,634:INFO: SMT: T:ShellyEM3 P:223.3W [ 222.7,222.7,22
2.9,223.0,222.7,222.1 ] Predict: 222.8W
2024-04-23 14:58:27,634:INFO: Direct connected panel are producing 0.0W,
 trying to get 212.1W from hub.
2024-04-23 14:58:27,635:INFO: Based on time, solarpower (88.5W) minimum
charge power (0W) and bypass state (True), hub could contribute 212.1W -
 Decision path: 0.2.
2024-04-23 14:58:27,635:INFO: Not setting solarflow output limit to 800.
0W as it is identical to current limit!
2024-04-23 14:58:27,635:INFO: Solarflow is willing to contribute 106.0W
(per channel) of the requested 212.1!
2024-04-23 14:58:27,635:INFO: Not setting inverter output limit as it is
 identical to current limit!
2024-04-23 14:58:27,635:INFO: Sun: 05:52 - 20:21 Demand: 212.1W, Panel D
C: (0.0W), Hub DC: (0.0W), Inverter Limit: 424.0W, Hub Limit: 800.0W
2024-04-23 14:58:30,526:INFO: Reading last full time: 2024-04-23 14:58:3
0
2024-04-23 14:58:30,526:INFO: Reading battery target mode: discharging
2024-04-23 14:58:49,748:INFO: Reading last full time: 2024-04-23 14:58:4
9
2024-04-23 14:58:49,748:INFO: Reading battery target mode: discharging
```

Cheers,
MaVo

reinhard-brandstaedter, 2024-04-23:
Check your DTU if the inverters limit > 0. If thas the case, wait until the hub delivers, if it doesn’t, it’s a zendure issue.

mavo, 2024-04-23:
Inverter limit was set to 400, this was actually ongoing for couple of hours.

It now resolved itself... Strange.
Anyway. Will close the ticket.

Thx for fast response.

mavo, 2024-04-26:
It happens again - right now.
Battery is full, but no power is delivered to home.

Are you running Hub2k? If yes, which firmware version do you run for battery and hub @reinhard-brandstaedter?

mavo, 2024-04-26:
Apparently I only have to write a comment here and then it fixes itself 🤔

mavo, 2024-04-26:
Now, that this has recovered again another question - is it on purpose, to have 0-feed-in when battery is full and one is on pass-through?
The app limits my inverter now to match the house consumption.

reinhard-brandstaedter, 2024-04-26:
Yes, currently in BP it matches demand, thinking of changing that again.

To the original topic. We can’t influence what the hub is doong in BP itself, we can switch it to BP, buth what the hub is doing and delivering once switched depends on the hub.

mavo, 2024-04-26:
> Yes, currently in BP it matches demand, thinking of changing that again.

Make it configurable 😁 
I know, one more config param makes applications mostly not more user friendly. But adds flexibility 😇

AzraelsDisk, 2024-04-28:
> > Yes, currently in BP it matches demand, thinking of changing that again.
> 
> Make it configurable 😁 I know, one more config param makes applications mostly not more user friendly. But adds flexibility 😇

I would also vote for WR > Max if Bypass = YES :-D

AzraelsDisk, 2024-04-29:
Another issue I just noticed... when BP is active and more power is needed than a panel on SF is producing, it looks like the information about the total power available is missing due to the effect that only the power of one panel is reported.

mavo, 2024-04-29:
> Another issue I just noticed... when BP is active and more power is needed than a panel on SF is producing, it looks like the information about the total power available is missing due to the effect that only the power of one panel is reported.

I saw the same. I think this is the transition phase between non-BP and BP. It can last a while, or at least it did for me lately. It actually improved (or it feels like it did - only had it one more time happening) after I switched to latest dev build + letting the app control BP manually. Then it happened "only" for ~10min instead of up to an hour previously.

AzraelsDisk, 2024-04-29:
problem is, that the Script is not trying to get more ;-)

reinhard-brandstaedter, 2024-04-29:
> I would also vote for WR > Max if Bypass = YES :-D

Latest dev now uses the maxInversePower from the hub. Which likely means full steam.
However this will need more consideration as it could also mean trouble in case additional panels are connected to the inverter at the same time.

reinhard-brandstaedter, 2024-04-29:
> Another issue I just noticed... when BP is active and more power is needed than a panel on SF is producing, it looks like the information about the total power available is missing due to the effect that only the power of one panel is reported.

What do you mean by only the power of one panel is reported? Did you observe that the hubs solarInput (total power of both panels) is not correct?

AzraelsDisk, 2024-04-29:
> > I would also vote for WR > Max if Bypass = YES :-D
> 
> Latest dev now uses the maxInversePower from the hub. Which likely means full steam. However this will need more consideration as it could also mean trouble in case additional panels are connected to the inverter at the same time.

Ok, will try to check tomorrow, currently running on:latest (maybe that is already dev?)

AzraelsDisk, 2024-04-29:
> What do you mean by only the power of one panel is reported? Did you observe that the hubs solarInput (total power of both panels) is not correct?

prior to bypass
[image]

after bypass (and limiting from your script)
[image]

so for my it looks like SF is only reporting 1 panel as solarpower 

maybe this problem is gone with your current update

reinhard-brandstaedter, 2024-04-29:
No, that is likely the "match to demand" in bypass mode. If you check your smartmeter reading it's likely hovering around 0 at this time?

reinhard-brandstaedter, 2024-04-30:
Never gets boring with Zendure ;-)
I've never seen this behavior ...

AzraelsDisk, 2024-05-01:
with the last update the behaviour in the bypass has changed, works as it should! :-)
However, I have noticed that the "Switch" is not set correctly in the HA, but the status is set to bypass

mavo, 2024-05-07:
As this one is kinda old, I'll close it.
Will re-open or use a new ticket, if it still exists in latest build - maybe its also something to contact zendure support about as it seem to only happen to me 🤔

closed 2024-05-07 by mavo