reinhard-brandstaedter/solarflow-control #233: inverter_min_limit deprecated ? what about HM800 Efficiency bug ?

ForgetMeOrNot, 2024-05-02:
Version 0.62
I'm not very good at coding, have you considered that the efficiency calculation or transmission by in opendtu / Hoymiles in connection with solarflow hub are partly not valid ?
I see inv.getEfficiency calls in the code but the config.ini variable inverter_min_limit only seems to appear in the config.ini.

I would like to define a minimum HUB power

https://github.com/tbnobody/OpenDTU/discussions/1078#discussioncomment-9050738

reinhard-brandstaedter, 2024-05-02:
No, I'm not aware of this "bug", by quickly skimming over the discussion there is not much that sf-control is able to do about that. I've had a similar conversation with Hoymiles support a while back which resulted in the recommendation of not using y-cables for feeding in from battery (like Solarflow does) as this can create MPPT issues (and maybe leading to the described behavior as well. So that is a topic that should be fixed in hardware.

Regarding min_inverter_limit this is actually a leftover that is not used anymore, you can ignore it.

What do you exactly want to achieve with a minimum hub power?

ForgetMeOrNot, 2024-05-02:
soweit ich es beobachten konnte kommt es zur falschen Ermittlung der Effizenz/Gesamtleitstung, wenn der HUB weniger als 120W liefert und dadurch die Spannung zu gering ist. daher dachte ich nehme das inverter_min_limit. wenn es das nicht mehr gibt könnte man es wieder implementieren ?

not using y-cables heißt ich müsste auf bypass verzichten und die Solarpanels auf teillast in der Sonne fahren, sonst grillt der Hub mit 2 Panels ja den 400W port in meinem Fall an einem einzigen MPPT des hm800

I really appreciate your work best regards

reinhard-brandstaedter, 2024-05-03:
The inverter_min_limit was used to set the minimum power feed to the inverter to keep it running / prevent shutdown.
Below 100W we actually have a conflict then, because the hub cannot be precisely controlled below 100W so we rather use the inverter to do that (so that demands like 45W can be properly matched).
Since sf-control tries to match demand when consuming from battery this HM800 issue could basically make that completely obsolete. As with a minimum contribution of 120W you could get everything from 120W feed in to the grid from battery (worst case) to zero (if there is exactly 120W demand in your house...

IMO, if that behavior can be fixed then the compensation should happen in the DTU

closed 2024-07-12 by reinhard-brandstaedter