reinhard-brandstaedter/solarflow-control #234: No output on thunderstorm (dark weather) v0.62

busaku, 2024-05-02:
Hey,

here are some logs:
```
2024-05-02 18:32:50,699:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100|100), V:49.9V (49.9|49.9), C: 0W, P:True (manual, not possible), F:0.0h, E:-1.0h, H: 0W, L:800W
2024-05-02 18:32:50,700:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:1.8W, DC_prediction: 1.8W (0.6|0.4|0.4|0.4), L:1600W (400.0W/channel) [1600W]
2024-05-02 18:32:50,709:INFO: SMT: T:ShellyEM3 P:193.8W [ 186.9,186.9,189.7,189.4,189.4,191.4 ] Predict: 179.8W
2024-05-02 18:32:50,710:INFO: Direct connected panel are producing 0.0W, trying to get 191.4W from hub.
2024-05-02 18:32:50,712:INFO: Based on time, solarpower ( 0.0W) minimum charge power (30W) and bypass state (True), hub could contribute 800.0W - Decision path: 0.2.
2024-05-02 18:32:50,712:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-02 18:32:50,712:INFO: Solarflow is willing to contribute 400.0W (per channel) of the requested 191.4!
2024-05-02 18:32:50,712:INFO: Not setting inverter output limit as it is identical to current limit!
2024-05-02 18:32:50,713:INFO: Sun: 06:04 - 20:58 Demand: 191.4W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 1600.0W, Hub Limit: 800.0W
2024-05-02 18:32:50,903:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-05-02 18:32:55,007:INFO: Reading last full time: 2024-05-02 18:32:55
2024-05-02 18:32:55,008:INFO: Reading battery target mode: discharging
2024-05-02 18:33:50,904:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-05-02 18:33:54,482:INFO: Reading last full time: 2024-05-02 18:33:54
2024-05-02 18:33:54,482:INFO: Reading battery target mode: discharging
2024-05-02 18:33:58,544:INFO: Reading last full time: 2024-05-02 18:33:58
2024-05-02 18:33:58,546:INFO: Reading battery target mode: discharging
2024-05-02 18:34:50,327:INFO: Publishing Homeassistant templates...
2024-05-02 18:34:50,444:INFO: Published 26 Homeassistant templates.
2024-05-02 18:34:50,701:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100|100), V:49.9V (49.9|49.9), C: 0W, P:True (manual, not possible), F:0.0h, E:-1.0h, H: 0W, L:800W
2024-05-02 18:34:50,701:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:1.8W, DC_prediction: 1.8W (0.6|0.4|0.4|0.4), L:1600W (400.0W/channel) [1600W]
2024-05-02 18:34:50,702:INFO: SMT: T:ShellyEM3 P:196.8W [ 201.6,201.6,201.6,202.1 ] Predict: 202.1W
2024-05-02 18:34:50,702:INFO: Direct connected panel are producing 0.0W, trying to get 202.1W from hub.
2024-05-02 18:34:50,703:INFO: Based on time, solarpower ( 0.0W) minimum charge power (30W) and bypass state (True), hub could contribute 800.0W - Decision path: 0.2.
2024-05-02 18:34:50,703:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-02 18:34:50,703:INFO: Solarflow is willing to contribute 400.0W (per channel) of the requested 202.1!
2024-05-02 18:34:50,704:INFO: Not setting inverter output limit as it is identical to current limit!
2024-05-02 18:34:50,705:INFO: Sun: 06:04 - 20:58 Demand: 202.1W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 1600.0W, Hub Limit: 800.0W
2024-05-02 18:34:50,905:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-05-02 18:34:54,854:INFO: Reading last full time: 2024-05-02 18:34:54
2024-05-02 18:34:54,855:INFO: Reading battery target mode: discharging
2024-05-02 18:34:58,879:INFO: Reading last full time: 2024-05-02 18:34:58
2024-05-02 18:34:58,881:INFO: Reading battery target mode: discharging
2024-05-02 18:35:50,906:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-05-02 18:35:55,172:INFO: Reading last full time: 2024-05-02 18:35:55
2024-05-02 18:35:55,174:INFO: Reading battery target mode: discharging
2024-05-02 18:35:59,206:INFO: Reading last full time: 2024-05-02 18:35:59
2024-05-02 18:35:59,209:INFO: Reading battery target mode: discharging
2024-05-02 18:36:50,701:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100|100), V:49.9V (49.9|49.9), C: 0W, P:True (manual, not possible), F:0.0h, E:-1.0h, H: 0W, L:800W
2024-05-02 18:36:50,702:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:1.8W, DC_prediction: 1.8W (0.6|0.4|0.4|0.4), L:1600W (400.0W/channel) [1600W]
2024-05-02 18:36:50,711:INFO: SMT: T:ShellyEM3 P:192.4W [ 199.8,199.8,199.0,197.6,196.7,195.8 ] Predict: 205.4W
2024-05-02 18:36:50,711:INFO: Direct connected panel are producing 0.0W, trying to get 195.8W from hub.
2024-05-02 18:36:50,712:INFO: Based on time, solarpower ( 0.0W) minimum charge power (30W) and bypass state (True), hub could contribute 800.0W - Decision path: 0.2.
2024-05-02 18:36:50,713:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-02 18:36:50,713:INFO: Solarflow is willing to contribute 400.0W (per channel) of the requested 195.8!
2024-05-02 18:36:50,713:INFO: Not setting inverter output limit as it is identical to current limit!
2024-05-02 18:36:50,714:INFO: Sun: 06:04 - 20:58 Demand: 195.8W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 1600.0W, Hub Limit: 800.0W
2024-05-02 18:36:50,907:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-05-02 18:36:55,075:INFO: Reading last full time: 2024-05-02 18:36:55
2024-05-02 18:36:55,077:INFO: Reading battery target mode: discharging
2024-05-02 18:36:59,093:INFO: Reading last full time: 2024-05-02 18:36:59
2024-05-02 18:36:59,094:INFO: Reading battery target mode: discharging
2024-05-02 18:37:11,864:INFO: Rapid rise in demand detected, clearing buffer!
2024-05-02 18:37:11,866:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100|100), V:49.9V (49.9|49.9), C: 0W, P:True (manual, not possible), F:0.0h, E:-1.0h, H: 0W, L:800W
2024-05-02 18:37:11,866:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:1.8W, DC_prediction: 1.8W (0.6|0.4|0.4|0.4), L:1600W (400.0W/channel) [1600W]
2024-05-02 18:37:11,867:INFO: SMT: T:ShellyEM3 P:712.4W [ 712.4,712.4 ] Predict: 712.4W
2024-05-02 18:37:11,867:INFO: Direct connected panel are producing 0.0W, trying to get 712.4W from hub.
2024-05-02 18:37:11,868:INFO: Based on time, solarpower ( 0.0W) minimum charge power (30W) and bypass state (True), hub could contribute 800.0W - Decision path: 0.2.
2024-05-02 18:37:11,868:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-02 18:37:11,869:INFO: Solarflow is willing to contribute 400.0W (per channel) of the requested 712.4!
2024-05-02 18:37:11,869:INFO: Not setting inverter output limit as it is identical to current limit!
2024-05-02 18:37:11,870:INFO: Sun: 06:04 - 20:58 Demand: 712.4W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 1600.0W, Hub Limit: 800.0W
2024-05-02 18:37:11,871:INFO: SMT triggers limit function: 712.4 -> 712.4: executed
2024-05-02 18:37:14,869:INFO: SMT triggers limit function: 646.6 -> 635.7: skipped
2024-05-02 18:37:14,871:INFO: SMT triggers limit function: 633.3 -> 620.6: skipped
2024-05-02 18:37:14,873:INFO: SMT triggers limit function: 620.8 -> 606.7: skipped
[... 8 lines cut ...]
2024-05-02 18:38:50,702:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100|100), V:49.8V (49.9|49.8), C: 0W, P:True (manual, not possible), F:0.0h, E:-1.0h, H: 0W, L:800W
2024-05-02 18:38:50,703:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:1.8W, DC_prediction: 1.8W (0.6|0.4|0.4|0.4), L:1600W (400.0W/channel) [1600W]
2024-05-02 18:38:50,703:INFO: SMT: T:ShellyEM3 P:254.9W [ 255.6,255.6,256.8 ] Predict: 256.8W
2024-05-02 18:38:50,704:INFO: Direct connected panel are producing 0.0W, trying to get 256.8W from hub.
2024-05-02 18:38:50,705:INFO: Based on time, solarpower ( 0.0W) minimum charge power (30W) and bypass state (True), hub could contribute 800.0W - Decision path: 0.2.
2024-05-02 18:38:50,706:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-02 18:38:50,706:INFO: Solarflow is willing to contribute 400.0W (per channel) of the requested 256.8!
2024-05-02 18:38:50,707:INFO: Not setting inverter output limit as it is identical to current limit!
2024-05-02 18:38:50,709:INFO: Sun: 06:04 - 20:58 Demand: 256.8W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 1600.0W, Hub Limit: 800.0W
2024-05-02 18:38:50,909:INFO: Triggering telemetry update: iot/73bkTV/6y0X53pJ/properties/read
2024-05-02 18:38:54,732:INFO: Reading last full time: 2024-05-02 18:38:54
2024-05-02 18:38:54,733:INFO: Reading battery target mode: discharging
2024-05-02 18:38:58,762:INFO: Reading last full time: 2024-05-02 18:38:58
2024-05-02 18:38:58,763:INFO: Reading battery target mode: discharging
[... cut ...]
2024-05-02 18:39:55,162:INFO: Reading battery target mode: discharging
```

So, maybe the problem is, that bypass is still true ore?
But there is no sun atm (heavy thunderstorm, totally dark here)
Or is the problem the sunset/rise setting and he did not try to change the bypass to false again?

If yes, would it be a better idea to not check the time, instead of checking solar input?
Like a new setting enable_batterie_at = 20w e.g. and if the solar is below of this value, it's like the sun is gone ? :)

busaku, 2024-05-02:
i disabled the bypass mode manually. now it works.

```
2024-05-02 19:01:57,484:INFO: HUB: S:0.0W [ 0.0 ], B:100% (100|100), V:49.8V (49.9|49.8), C:-177W, P:False (auto, not possible), F:0.0h, E:-1.0h, H:169W, L:800W
2024-05-02 19:01:57,485:INFO: INV: AC:163.6W, AC_Prediction: 163.6W, DC:172.1W, DC_prediction: 172.1W (0.8|0.6|101.3|101.5), L:1600W (400.0W/channel) [1600W]
2024-05-02 19:01:57,485:INFO: SMT: T:ShellyEM3 P:-75.8W [ 11.6,11.6,-24.4,-36.9 ] Predict: -36.9W
2024-05-02 19:01:57,485:INFO: Direct connected panel are producing 0.0W, trying to get 155.9W from hub.
2024-05-02 19:01:57,486:INFO: Based on time, solarpower ( 0.0W) minimum charge power (30W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-05-02 19:01:57,487:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-02 19:01:57,487:INFO: Solarflow is willing to contribute 0.0W (per channel) of the requested 155.9!
2024-05-02 19:01:57,487:INFO: Setting inverter output limit to 40W (1 min moving average of 10W x 4)
2024-05-02 19:01:57,488:INFO: Grid feed in from battery: 36.9W!
2024-05-02 19:01:57,489:INFO: Sun: 06:04 - 20:58 Demand: 155.9W, Panel DC: (0.0W), Hub DC: (192.8W), Inverter Limit: 40.0W, Hub Limit: 800.0W
2024-05-02 19:01:57,489:INFO: SMT triggers limit function: -24.4 -> -36.9: executed
2024-05-02 19:01:59,480:INFO: SMT triggers limit function: -26.3 -> -38.5: skipped
2024-05-02 19:01:59,483:INFO: SMT triggers limit function: -19.9 -> -30.8: skipped
```

busaku, 2024-05-02:
okay, worked like a half :D
because of the minimum-charge level and still really dark weather, sc like to charge the batterie instead of feeding the house :(

busaku, 2024-05-02:
Hmm, yes you are right. Just thinking:
What do Ou thinking about an setting like 
`if darkness > x minutes and timeToDarkness > y and dailyByPassSwitch < z disableBypass`

Maybe all settings configurable with min/max values?

reinhard-brandstaedter, 2024-05-03:
Nah that doesn't provide much value, just makes things more complicated.
If you want feed-in from the hub throughout the day, you can adjust the morning/evening offset time-ranges, that will give you the same result.

closed 2024-05-03 by busaku