reinhard-brandstaedter/solarflow-control #239: ByPass-State - should it be enabled?

mavo, 2024-05-06:
Hello @reinhard-brandstaedter,

I switched to newest dev build and I am confused. It said turn on BP, but its still not on.
```
2024-05-06 15:11:54,367:INFO: MQTT User is not set, assuming authentication not needed
2024-05-06 15:11:54,367:INFO: Solarflow Hub: A8yh63/ETFTke36
2024-05-06 15:11:54,367:INFO: Limit via inverter: True
2024-05-06 15:11:54,367:INFO: Control Parameters:
2024-05-06 15:11:54,368:INFO:   MIN_CHARGE_POWER = 0
2024-05-06 15:11:54,368:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-05-06 15:11:54,368:INFO:   MAX_INVERTER_LIMIT = 800
2024-05-06 15:11:54,368:INFO:   MAX_INVERTER_INPUT = 800
2024-05-06 15:11:54,368:INFO:   SUNRISE_OFFSET = 180
2024-05-06 15:11:54,368:INFO:   SUNSET_OFFSET = 120
2024-05-06 15:11:54,369:INFO: Publishing Homeassistant templates...
2024-05-06 15:11:54,411:INFO: Published 26 Homeassistant templates.
2024-05-06 15:11:54,411:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-05-06 15:11:54,411:INFO: Using AhoyDTU: Base topic: ahoydtu, Limit topic: ahoydtu/ctrl/limit/0, SF Channels: [3, 4]
2024-05-06 15:11:54,412:INFO: Using ShellyEM3: Base topic: shellies/shellyem3
2024-05-06 15:11:54,412:INFO: Connected to MQTT Broker!
2024-05-06 15:11:54,412:INFO: Hub subscribing: /A8yh63/ETFTke36/properties/report
2024-05-06 15:11:54,412:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/solarInputPower
2024-05-06 15:11:54,412:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/electricLevel
2024-05-06 15:11:54,412:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputPackPower
2024-05-06 15:11:54,412:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/packInputPower
2024-05-06 15:11:54,412:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputHomePower
2024-05-06 15:11:54,412:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/outputLimit
2024-05-06 15:11:54,413:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/inverseMaxPower
2024-05-06 15:11:54,413:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/masterSoftVersion
2024-05-06 15:11:54,413:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/pass
2024-05-06 15:11:54,413:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/passMode
2024-05-06 15:11:54,413:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/batteries/+/socLevel
2024-05-06 15:11:54,413:INFO: Hub subscribing: solarflow-hub/ETFTke36/telemetry/batteries/+/totalVol
2024-05-06 15:11:54,413:INFO: Hub subscribing: solarflow-hub/ETFTke36/control/#
2024-05-06 15:11:54,413:INFO: Turning hub bypass OFF
2024-05-06 15:11:54,413:INFO: DTU subscribing: ahoydtu/HM-1600/+/P_DC
2024-05-06 15:11:54,413:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/P_AC
2024-05-06 15:11:54,413:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/active_PowerLimit
2024-05-06 15:11:54,413:INFO: DTU subscribing: ahoydtu/HM-1600/ch0/Efficiency
2024-05-06 15:11:54,413:INFO: DTU subscribing: ahoydtu/status
2024-05-06 15:11:54,413:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-05-06 15:11:54,413:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/0/power
2024-05-06 15:11:54,413:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/1/power
2024-05-06 15:11:54,414:INFO: Shelly3EM subscribing: shellies/shellyem3/emeter/2/power
2024-05-06 15:11:54,414:INFO: Reading last full time: 2024-05-06 15:11:36
2024-05-06 15:11:54,415:INFO: Reading battery target mode: discharging
2024-05-06 15:11:57,458:INFO: Bypass control, turning on bypass!
2024-05-06 15:11:57,458:INFO: Turning hub bypass ON
2024-05-06 15:11:57,459:INFO: Reading last full time: 2024-05-06 15:11:57
2024-05-06 15:11:57,459:INFO: Reading battery target mode: discharging
2024-05-06 15:12:28,720:INFO: Reading last full time: 2024-05-06 15:12:28
2024-05-06 15:12:28,721:INFO: Reading battery target mode: discharging
2024-05-06 15:12:54,369:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-05-06 15:12:57,356:INFO: Reading last full time: 2024-05-06 15:12:57
2024-05-06 15:12:57,396:INFO: Reading battery target mode: discharging
2024-05-06 15:13:29,538:INFO: Reading last full time: 2024-05-06 15:13:29
2024-05-06 15:13:29,579:INFO: Reading battery target mode: discharging
2024-05-06 15:13:54,370:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-05-06 15:13:54,412:INFO: HUB: S:126.4W [ 126.4,126.4 ], B:100% (100|100), V:51.0V (51.0|50.9), C: 0W, P:False (manual, not possible), F:0.0h, E:-1.0h, H: 0W, L:800W
2024-05-06 15:13:54,414:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:3.8W, DC_prediction: 4.0W (0.0|0.0|1.8|2.0), L: 0W (0.0W/channel) [1600W]
2024-05-06 15:13:54,416:INFO: SMT: T:ShellyEM3 P:315.6W [ 320.9,320.9,320.7,320.6,320.5,321.2 ] Predict: 320.7W
2024-05-06 15:13:54,416:INFO: Direct connected panel are producing 0.0W, trying to get 311.2W from hub.
2024-05-06 15:13:54,417:INFO: Based on time, solarpower (126.4W) minimum charge power (0W) and bypass state (False), hub could contribute 126.4W - Decision path: 2.2.
2024-05-06 15:13:54,417:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-06 15:13:54,417:INFO: Solarflow is willing to contribute 63.0W (per channel) of the requested 311.2!
2024-05-06 15:13:54,418:INFO: Setting inverter output limit to 252W (1 min moving average of 63W x 4)
2024-05-06 15:13:54,418:INFO: Sun: 05:26 - 20:43 Demand: 311.2W, Panel DC: (0.0W), Hub DC: (0.0W), Inverter Limit: 252.0W, Hub Limit: 800.0W
2024-05-06 15:13:56,994:INFO: Reading last full time: 2024-05-06 15:13:56
2024-05-06 15:13:56,994:INFO: Reading battery target mode: discharging
2024-05-06 15:14:30,435:INFO: Reading last full time: 2024-05-06 15:14:30
[... 123 lines cut ...]
2024-05-06 15:30:31,707:INFO: SMT triggers limit function: 1452.0 -> 1440.2: executed
2024-05-06 15:30:31,711:INFO: SMT triggers limit function: 1451.8 -> 1440.3: skipped
2024-05-06 15:30:31,714:INFO: SMT triggers limit function: 1451.6 -> 1440.4: skipped
2024-05-06 15:30:33,715:INFO: SMT triggers limit function: 1450.2 -> 1438.5: skipped
2024-05-06 15:30:33,718:INFO: SMT triggers limit function: 1450.0 -> 1438.5: skipped
2024-05-06 15:30:33,721:INFO: SMT triggers limit function: 1449.4 -> 1437.9: skipped
2024-05-06 15:30:35,727:INFO: SMT triggers limit function: 1447.0 -> 1436.6: skipped
2024-05-06 15:30:35,731:INFO: SMT triggers limit function: 1446.5 -> 1436.1: skipped
2024-05-06 15:30:35,734:INFO: SMT triggers limit function: 1446.3 -> 1436.1: skipped
2024-05-06 15:30:42,531:INFO: Reading last full time: 2024-05-06 15:30:42
2024-05-06 15:30:42,571:INFO: Reading battery target mode: discharging
2024-05-06 15:30:54,376:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-05-06 15:30:56,833:INFO: Reading last full time: 2024-05-06 15:30:56
2024-05-06 15:30:56,834:INFO: Reading battery target mode: discharging
2024-05-06 15:31:43,133:INFO: Reading last full time: 2024-05-06 15:31:43
2024-05-06 15:31:43,133:INFO: Reading battery target mode: discharging
2024-05-06 15:31:54,370:INFO: Publishing Homeassistant templates...
2024-05-06 15:31:54,377:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-05-06 15:31:54,408:INFO: Published 26 Homeassistant templates.
```

Should the app report the proper bypass state?

Regards,
MaVo

reinhard-brandstaedter, 2024-05-06:
Bad timing :-) to switch release.
Let it run, it will enable BP switching tomorrow at sunrise.

mavo, 2024-05-06:
But it should all work fine for the night with discharge of battery to feed the consumption?

reinhard-brandstaedter, 2024-05-06:
yep, at sunset it will start feeding and reset BP enablement.

reinhard-brandstaedter, 2024-05-06:
Seems like a race-condition. From the flow it is correct to enable the bypass immediately at startup upon first reading of the battery state. However it seems that the command didn't go to the hub...

mavo, 2024-05-07:
Yeah, might be. I could also see, that HA reported BP to be enabled, but the HUB had no idea about it.
Today is sadly very cloudy, so I will not be able to see if it works this time. Maybe tomorrow 🤷

mavo, 2024-05-07:
One more thing @reinhard-brandstaedter , can you maybe describe how the BP currently behaves when controlled by the app?

I am particularly interested in the behaviour when it will be turned off. So as far as I understood it, the BP, when enabled over the day will only get disabled when sunset (without any offset) is reached, correct?
If yes... Why's this the case?
For me, just by the nature of how the panels are located, my house needs battery support from around 7pm, as the solar production then drops below the 200w that the house has on idle power. Now, IF it is how I think it is, it would mean I would consume energy from the grid and have plenty in the batteries the next morning, resulting in sub-ideal usage of the system.

reinhard-brandstaedter, 2024-05-07:
Bypass turns off at sunset - sunset_offset, just like when you normally allow feed in from the hub. So you can accomodate your production situation.

closed 2024-05-13 by reinhard-brandstaedter