reinhard-brandstaedter/solarflow-control #243: Enable feed to grid if battery is full

floriandrott, 2024-05-12:
Hi,

I recognized if

- solar panels are producing more energy as the house consumes
- solarflow battery is fully charged (100%)
- solarflow bypass is on

that still the inverter gets limited by solarflow-control to the consumed power of the house. Is there a way that the inverter gets not limited anymore so that the over-produced energy is feed into the grid? I realized with the current inverter limit that I still consume energy from the grid because of the "slow" limit correction based on the house consumed energy.

Therefore it would be nice to just feed the over-produced energy into the grid.

floriandrott, 2024-05-12:
Maybe this behaviour is because of #244?

reinhard-brandstaedter, 2024-05-12:
Did you try the latest dev build?
Actually just this morning I added some improvements "at the top end" (Bypass on, AC limit reached) that should fix a few things. Here is my successful test of today, which nicely covers a few cases:

[image]

floriandrott, 2024-05-13:
This morning I updated to the DEV build.

The battery reached 100%, but in the logs bypass is not displayed. 
`Based on time, solarpower (215.2W) minimum charge power (175W) and bypass state (False), hub could contribute 226.5W - Decision path: 2.1.`

In the event of the 100%, the inverter goes "down" (marked yellow in OpenDTU). It seems like that for some seconds the energy goes down and the inverter (HMS-800) goes in standby. In standby solarflow-control can't wake it up, so I manually restarted the inverter via OpenDTU. 

Now the inverter is again feeding in energy into the house network. Unfortunately I have the same behavior as before. The inverter gets limited to the current house consumption so that I still consume some energy from the grid (when the consumption of the house changes).

floriandrott, 2024-09-12:
https://github.com/reinhard-brandstaedter/solarflow-control/issues/244 was the root cause of this isse.

closed 2024-09-12 by floriandrott