reinhard-brandstaedter/solarflow-control #244: Bypass state not recognized correctly

floriandrott, 2024-05-12:
If the battery is completely full (100%) the hub goes into bybass mode. The bypass mode is not correctly detected by solarflow.

Home Assistant integration shows
[image]

Based on the MQTT topics it should detect the bypass state:
[image]

The MQTT value goes to "2" which means bypass on but as shown in picture one, the solarflow-control is not detecting it.

reinhard-brandstaedter, 2024-05-12:
Are you just noticing that the HA sensor is not properly reporting, or is the hub actually not in bypass state?
Did you check directly in MQTT what the reported state is?

It should show it like this:
[image]

floriandrott, 2024-05-13:
In the morning I switched over to the DEV build. Today the batterie reached again 100% and the hub went into bypass mode.
In MQTT Explorer I see 
[image]

The hub went into bypass but solarflow-control is not really seeing it. The log is saying
`HUB: S:200.5W [ 200.5,200.5 ], B:100% (100), V:51.5V (51.5), C: 0W, P:False (manual, not possible), F:0.0h, E:-1.0h, H: 0W, L:600W`
`Based on time, solarpower (200.5W) minimum charge power (175W) and bypass state (False), hub could contribute 300.8W - Decision path: 2.1.`

reinhard-brandstaedter, 2024-05-13:
pass=0 means the hub is NOT in bypass (yet).
passMode=2 means it SHOULD switch to bypass on, however it can take a bit until the hub actually switches ( needs to reduce solarinput first )

floriandrott, 2024-05-13:
Ah, okay. Right now I have 360W as solarinput but still pass=0 and passMode=2. Is this normal that it takes so long?

floriandrott, 2024-05-13:
I double checked the values, solarinput went to 0 W but still pass=0 and passMode=2. There was also no energy feed from the battery to the house. I manually turned off "bypass" in HA by the control elements. Now the hub is contributing energy to the house via the battery. 

I'm happy to contribute in any way with log files, etc. Just let me know :)

reinhard-brandstaedter, 2024-05-13:
Actually the HA switch for passMode is not correct (@ricariel ?).
passMode = 2  is manual on
passMode = 1 is manual off
passMode = 0 is auto (let hub firmware decide)
So if you switch it to "off" in HA, you are actually switching the hub to "auto" mode, where we have seen all kinds of delays when the firmware will decide to turn it on ...

If you check the solarflow-control logs you will see various infos about the bypass:
```P:False (manual, not possible)``` : means BP is reported ```off``` but in manual mode (passMode=1), sf-controls own logic has set any change to ```not possible``` meaning it will currently not turn it on (avoiding frequent switches like in the firmware)
```P:False (manual, possible)``` : means it is ```off``` but sf-control will turn it on once the battery reaches 100%
```P:True (manual, possible)```: means it is ```on``` and sf-control is controlling it
```P:True|False (auto, ...)```: means passMode = 0 , the hubs firmware controls it

In your logs you should observe the state of this, usually the hub is switched to ```(manual, possible)``` in the morning at sunrise (if ```control_bypass = true``` is set in your config).
Once the bypass is on, sf-control will turn it off and sunset - offset if demand is higher than what is delivered via bypass (so it can start discharging if needed).

ricariel, 2024-05-14:
I use this switch for Bypass and it works:

   - select:
       name: "Bypass Modus"
       unique_id: "serial-passMode"
       state_topic: "solarflow-hub/serial/telemetry/passMode"
       value_template: "{{ 'Auto' if value=='0' else ('Off' if value=='1' else ('On' if value=='2' else        'Unknown')) }}"                                     
       options:
         - 'Auto'
         - 'Off'
         - 'On'
       command_topic: "iot/73bkTV/serial/properties/write"
       command_template: "{\"properties\": {\"passMode\": {{ '0' if value=='Auto' else ('1' if value=='Off'    else ('2' if value=='On' else '0')) }} }}"          
       device:
         name: "SolarFlow Hub 1200"
         identifiers: "serial"
         manufacturer: "Zendure"
         model: "SmartPV Hub 1200 Controller"

mavo, 2024-05-16:
I face the same issue currently.
For some reason, HA template is telling the BP is on, but the app reports it's not and thus (my probably somewhat faulty hub) only lets a small amount of energy to my house, which is most of the time not enough to feed it completely which is far from the truth as there is not a single cloud (and its not even 3pm).
The combination of the hub not really going into BP mode lets the app reduce the inverter to a too low number in terms of capacity. This boils down to the logic of not using the battery when being in daylight condition. And as the hub is just reporting some 100w of power available, it does not even try to get more...

This is frustrating, but it's less a problem of the app than a problem with my hub, I think. Zendure support is slow AF as well, which lets me hate when my system turns on BP.
Sometimes it works when it reaches the battery level of 100% on a day. But most of the time it's just meh.

Sorry for ranting, I need to let it out somewhere...

mavo, 2024-05-16:
The "fun" part is, as my hub behaves strangely in BP mode, I know it's actually in BP mode.
But the app does not know it...

Is there anything that I can do, or the app to actually get the proper BP state?

mavo, 2024-05-16:
Hello @reinhard-brandstaedter, sorry to bother you again.

In an effort to understand whats happening, I scrolled through the logs of the day and actually noticed, the app apparently never really understand the hubs BP mode for me.
Here is the log of the time when it turned on, but the app never realized.
```
2024-05-16 11:31:12,685:INFO: HUB: S:745.5W [ 969.4,969.4,918.0,867.9,810.0,745.5 ], B: 99% (100|99), V:53.1V (53.3|53.0), C: 444W, P:False (manual, possible), F:17.9h, E:198.6h, H:224W, L:800W
2024-05-16 11:31:12,685:INFO: INV: AC:223.7W, AC_Prediction: 223.7W, DC:235.4W, DC_prediction: 235.4W (0.0|0.0|119.2|115.7), L:448W (112.0W/channel) [1600W]
2024-05-16 11:31:12,687:INFO: SMT: T:ShellyEM3 P:9.7W [ 12.2,12.2,10.1,8.9,10.1,13.4 ] Predict: 15.3W
2024-05-16 11:31:12,687:INFO: Direct connected panel are producing 0.0W, trying to get 226.7W from hub.
2024-05-16 11:31:12,687:INFO: Based on time, solarpower (745.5W) minimum charge power (0W) and bypass state (False), hub could contribute 226.7W - Decision path: 1.1.
2024-05-16 11:31:12,687:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-16 11:31:12,688:INFO: Solarflow is willing to contribute 113.0W (per channel) of the requested 226.7!
2024-05-16 11:31:12,688:INFO: Not setting inverter output limit as it is identical to current limit!
2024-05-16 11:31:12,688:INFO: Sun: 05:09 - 20:59 Demand: 226.7W, Panel DC: (0.0W), Hub DC: (223.3W), Inverter Limit: 452.0W, Hub Limit: 800.0W
2024-05-16 11:31:12,688:INFO: HUB triggers limit function: 810.0 -> 745.5: executed
2024-05-16 11:31:13,726:INFO: HUB triggers limit function: 798.8 -> 733.4: skipped
2024-05-16 11:31:14,733:INFO: HUB triggers limit function: 787.8 -> 721.6: skipped
2024-05-16 11:31:15,743:INFO: HUB triggers limit function: 776.8 -> 717.5: skipped
2024-05-16 11:31:16,772:INFO: HUB triggers limit function: 766.0 -> 713.5: skipped
2024-05-16 11:31:17,786:INFO: HUB triggers limit function: 755.2 -> 709.5: skipped
2024-05-16 11:31:18,809:INFO: HUB triggers limit function: 744.5 -> 700.0: skipped
2024-05-16 11:31:19,834:INFO: HUB triggers limit function: 733.9 -> 696.2: skipped
2024-05-16 11:31:20,852:INFO: HUB triggers limit function: 723.5 -> 692.5: skipped
2024-05-16 11:31:31,049:INFO: HUB: S:625.5W [ 781.6,781.6,733.2,685.1,658.2,625.5 ], B: 99% (100|99), V:53.1V (53.3|53.0), C: 334W, P:False (manual, possible), F:17.9h, E:198.6h, H:220W, L:800W
2024-05-16 11:31:31,049:INFO: INV: AC:223.4W, AC_Prediction: 223.4W, DC:235.2W, DC_prediction: 235.2W (0.0|0.0|119.9|116.3), L:448W (112.0W/channel) [1600W]
2024-05-16 11:31:31,051:INFO: SMT: T:ShellyEM3 P:8.9W [ 10.6,10.6,11.6,13.6,13.6,13.8 ] Predict: 5.5W
2024-05-16 11:31:31,052:INFO: Direct connected panel are producing 0.0W, trying to get 228.2W from hub.
2024-05-16 11:31:31,052:INFO: Based on time, solarpower (625.5W) minimum charge power (0W) and bypass state (False), hub could contribute 228.2W - Decision path: 1.1.
2024-05-16 11:31:31,052:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-16 11:31:31,052:INFO: Solarflow is willing to contribute 114.0W (per channel) of the requested 228.2!
2024-05-16 11:31:31,052:INFO: Setting inverter output limit to 456W (1 min moving average of 114W x 4)
2024-05-16 11:31:31,053:INFO: Sun: 05:09 - 20:59 Demand: 228.2W, Panel DC: (0.0W), Hub DC: (224.4W), Inverter Limit: 456.0W, Hub Limit: 800.0W
2024-05-16 11:31:31,053:INFO: HUB triggers limit function: 658.2 -> 625.5: executed
2024-05-16 11:31:32,069:INFO: HUB triggers limit function: 655.0 -> 619.0: skipped
2024-05-16 11:31:34,128:INFO: HUB triggers limit function: 644.1 -> 608.8: skipped
2024-05-16 11:31:35,146:INFO: HUB triggers limit function: 637.2 -> 602.3: skipped
2024-05-16 11:31:36,163:INFO: HUB triggers limit function: 630.3 -> 595.8: skipped
2024-05-16 11:31:37,183:INFO: HUB triggers limit function: 623.5 -> 589.6: skipped
2024-05-16 11:31:38,202:INFO: HUB triggers limit function: 616.9 -> 583.5: skipped
2024-05-16 11:31:39,222:INFO: HUB triggers limit function: 610.4 -> 577.6: skipped
2024-05-16 11:31:40,240:INFO: HUB triggers limit function: 604.0 -> 571.8: skipped
2024-05-16 11:31:41,259:INFO: HUB: S:566.2W [ 699.1,699.1,648.9,629.3,597.8,566.2 ], B: 99% (100|99), V:53.3V (53.4|53.2), C: 280W, P:False (manual, possible), F:17.9h, E:198.6h, H:227W, L:800W
2024-05-16 11:31:41,259:INFO: INV: AC:225.3W, AC_Prediction: 225.3W, DC:237.1W, DC_prediction: 237.1W (0.0|0.0|121.8|118.3), L:448W (112.0W/channel) [1600W]
2024-05-16 11:31:41,261:INFO: SMT: T:ShellyEM3 P:-2.4W [ 11.4,11.4,12.9,12.7,12.6,9.8 ] Predict: 9.6W
2024-05-16 11:31:41,261:INFO: Direct connected panel are producing 0.0W, trying to get 227.9W from hub.
2024-05-16 11:31:41,261:INFO: Based on time, solarpower (566.2W) minimum charge power (0W) and bypass state (False), hub could contribute 227.9W - Decision path: 1.1.
2024-05-16 11:31:41,262:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-05-16 11:31:41,262:INFO: Solarflow is willing to contribute 113.5W (per channel) of the requested 227.9!
2024-05-16 11:31:41,262:INFO: Not setting inverter output limit as it is identical to current limit!
2024-05-16 11:31:41,262:INFO: Grid feed in from hub solarpower: 0.2W!
2024-05-16 11:31:41,262:INFO: Sun: 05:09 - 20:59 Demand: 227.9W, Panel DC: (0.0W), Hub DC: (228.1W), Inverter Limit: 452.0W, Hub Limit: 800.0W
[... cut ...] s me...
My config is set to let the app regulate the BP mode, which also works fine. This is what I can tell. I am also running latest master. But as said, the app does not know this, somehow...

I sadly do not have the states from MQTT, as I am not at home atm and only can retrospectively check the logs from the app.
How does the app read the BP mode from the hub? Is there maybe anything wrong configured on my end? What data can I provide so that you can investigate this a bit more?

Best, MaVo

mavo, 2024-05-17:
New day, new comment 🙈 
I played around a bit more in an effort to fix this issue for me locally, but had no success as of yet. To me, it seems almost clear, that my hub2k does not report `pass = 1` in MQTT, although its in BP mode. So there is one question @floriandrott - do you have a hub2000or the older hub1200?
What I've tried is by modifying the source code in the container to set the BP state when it reaches 100% battery and keep this value. However, as there is the telemetry update every now ad then, the manual set value is lost after 2 mins again and is replaced by `pass = 0`.
The question is, does the hub2k for some reason set this value somewhat differently?

It's super annoying for me, as my battery is currently full at around 2pm due to sunshine without any cloud, but I still use some power from the grid as soon as the BP is on due to the fact that the app does not know and the other logic in the app eventually reduces the inverter ever so slightly every now and then which leads to the situation, that the hub cannot feed more, therefore reduces what it provides and the app, as it has no clue about BP and does not want to take something from the battery so it reduces the inverter even more.

Any help would be great 😢

reinhard-brandstaedter, 2024-05-18:
What are your BP settings in the Zendure app?
Does the BP kick in if you do not control it via sf-control?

floriandrott, 2024-05-18:
@mavo I use the hub2000 in my setup.

Yes @reinhard-brandstaedter, BP kicks in automatically if the battery reaches 100% and I see as @mavo that the switch in HA is turned on for BP. 

The setting in the Zendure app for BP is "always on" / "always off" depending if BP kicked in or not.

reinhard-brandstaedter, 2024-05-18:
The pass property is just read like any other property of the hub. I can’t see any reason why it should not be reflected in the logs then.
So it would be really helpful to know what the state in MQTT is.

mavo, 2024-05-18:
@reinhard-brandstaedter the BP actually kicks in. I kinda assume it's reported differently compared to hub1200.
But as I can only access my HA and the logs via ssh but not mqtt I cannot tell the mqtt values.
However, in one of the first comments of @floriandrott here you can see that "pass" is reported as 0 always.

@reinhard-brandstaedter how did you first discover the values and properties and how they are reported in mqtt, did you just check what the hub emitted and deducted what is useful?

floriandrott, 2024-05-19:
@reinhard-brandstaedter I can check the MQTT changes. Any specific values on which I should pay attention when the battery reaches 100%? I will then check via MQTT Explorer the next time (hopefully today).

floriandrott, 2024-05-19:
@reinhard-brandstaedter I reached again 100% battery. Hub2000 is in BP mode (but still solarflow-control is reporting BP=false). In MQTT following values have changed:
- `passMode=1` to `passMode=2`
- `packState=1` to `packState=0`

mavo, 2024-05-19:
@floriandrott "packState" is the indicator, if the battery is idle, charging or discharging.
So this is not the value we are looking for. I also would assume it's nothing that is yet reported in the prettified topics, rather a raw one from the hub directly.

But I don't know enough to say where to look.
Good thing, I am back home and can check mqtt values as well :-)

mavo, 2024-05-19:
Quick thought @reinhard-brandstaedter, if we do not find how the BP state is transferred in Hub2k when manual mode is used, wouldn't it be an option to use the logic of the passMode as indicator if the BP is on?
I mean, as the app controls it, setting the value to the equal of "always on" / "always off" is defining in which state it is. So if you would change the "hub.getBypass()" method to check (pseudo-code incoming)
```
if hub.productId == "hub2k" and sf-app.BpControl
  return passMode == 2
else
  return pass = 1
```

This approach assumes that the `pass` property is not set when `passMode != 0`. I will try to check, but the forecast is not very great atm, so lets see when I can test it actually.
Maybe @floriandrott can check it as well (setting sf-control to not rule the BP mode but rather use auto and let the hubs firmware decide).

floriandrott, 2024-05-20:
@mavo alright, I will check it as well. I switched off `control_bypass` and enabled `auto` in the Zendure app for BP. Let's see what happens.

floriandrott, 2024-05-20:
@mavo no change here. The log says
`HUB: S:47.6W [ 61.0,61.0,54.5,52.4,49.5,47.6 ], B:100% (100), V:53.6V (53.6), C: 39W, P:False (auto, possible), F:0.0h, E:55.8h, H: 0W, L:600W`

I saw also a strange behavior from the hub BP logic. 100% reached and afterwards the battery was discharging and charging which isn't the correct BP behavior. I turned on BP manually.

mavo, 2024-05-20:
Thanks for testing.

And in this case, I hope that @reinhard-brandstaedter would be so kind to implement something in the direction that I mentioned 😇

reinhard-brandstaedter, 2024-05-20:
@mavo @floriandrott the problem I see with BP Mode is that it is a _intent_ to switch only. The hubs firware decides when it really performs the switch.
Just because bypassmode is 2, still doesn’t mean it is actuslly on.
I think the slow switchting/no switching even in auto mode also confirms this.

mavo, 2024-05-20:
But what then? Just do nothing? Not really a good option I'd say.

Imo, the PassMode is the best we can do. Currently it's kinda broken. Why do you think it's bad to use the PassMode? What's the difference for the app? The PassMode should indicate where it should (and mostly will be in a short time...)

reinhard-brandstaedter, 2024-05-20:
I honestly doubt that the firmware is differen in this regard. Why should everything work identical, just the pass telemetry is reported in a different way. I’d further investigate this what really is differen on the hub side. Eventually its even a bug on SF side. Plus we haven’t a precise observation from the MQTT publish topic at the time of switching to BP yet.

mavo, 2024-05-20:
What kind of recording can I provide?
I have no good explanation why it behaves differently for the hub2k, the only thing I can tell is that it does.
Maybe it's a bug in the firmware, but there is no update available and only God knows when there will be one.

If you tell me what kind of data I should record, I can try to do, when the conditions are good.

For the meantime I probably will change the code locally to use the passMode property in am effort to not just be stuck while we try to find the proper solution.

reinhard-brandstaedter, 2024-05-20:
All variables reported to MQTT would be interesting. e.g by keeping MQTT explorer oben for a few hours and collect the topic tree reported by sf-control. And then a history of the pass topic before and after the expected BP switch.

floriandrott, 2024-05-22:
@reinhard-brandstaedter I have no clue how to store the data in MQTT Explorer so I used a different tool to collect the MQTT topic data. I subscripted for the sf-control data `solarflow-hub` and for the raw data from the hub `/A8yh63/XXXXXXXX/properties/report`.

Battery reached at 18:25:40 100% and the hub went into BP mode. Attached are two files (one 10min data after the switch and one with +20min data after the switch).

[data.json](https://github.com/reinhard-brandstaedter/solarflow-control/files/15406580/data.json)
[data2.json](https://github.com/reinhard-brandstaedter/solarflow-control/files/15406582/data2.json)

I hope this helps to find out more about the hub behavior. Please let me know if I can provide anything else to troubleshoot.

reinhard-brandstaedter, 2024-05-22:
Thanks, that confirms that the hub2000 doesn't report the bypass state via the ```pass``` topic.
When switched to ```passMode : 2``` it reduces the solarinput down to 0, just like the hub1200 enables BP silently and then turns on solarinput again.
So for hub2000 one can only assume that BP is on if all these conditions are met ```passMode == 2``` & ```solarInput > 0 ``` & ```outputPackPower != 0``` & ```outputHomePower > 0```.
What a bummer!

floriandrott, 2024-05-23:
@reinhard-brandstaedter thank you for the changes. I tried it out today without luck. Bypass wasn't activated by sf-control and also not displayed in HA. With the old DEV-build, BP was at least reported to HA. But now, bypass isn't seen sf-control + HA.

Unfortunately I haven't any logs this time. Tomorrow I will check it again.

floriandrott, 2024-05-23:
@reinhard-brandstaedter thank you for the changes. I was testing it with the latest DEV build but BP wasn't recognized from sf-control.

The MQTT topic logs are attached here:
[data3.json](https://github.com/reinhard-brandstaedter/solarflow-control/files/15420420/data3.json)

The battery reached 100% and the hub went into BP mode. But sf-control was still reporting BP=false

```
2024-05-23 18:56:45,856:INFO: HUB: S:337.2W [ 351.7,351.7,350.4,343.6,340.2,337.2 ], B: 98% (98), V:51.9V (51.9), C: 231W, P:False (manual, possible), F:0.5h, E:130.3h, H: 92W, L:600W
2024-05-23 18:56:56,013:INFO: Bypass control, turning on bypass!
2024-05-23 18:56:56,013:INFO: Turning hub bypass ON
2024-05-23 18:56:56,014:INFO: Reading last full time: 2024-05-23 18:56:56
2024-05-23 18:56:56,015:INFO: Reading battery target mode: discharging
2024-05-23 18:57:05,919:INFO: HUB: S:293.5W [ 322.8,322.8,315.1,308.1,300.6,293.5 ], B:100% (100), V:51.9V (51.9), C: 143W, P:False (manual, not possible), F:0.0h, E:130.3h, H: 67W, L:600W
```

Any idea why?

mavo, 2024-05-23:
The exception I had as well.
For me it was thrown when BP should have been disabled as the sun was covered by clouds and it was already passed the sun - offset point.

I guess.the log line reads BP from pass property, but based on the decision path you can see when it was used. The decision path starts with a 0 in case BP mode is determined by SF control.

reinhard-brandstaedter, 2024-05-23:
Because Hub2k does not only NOT report the bypass state properly, but it reports it even wrong, I still pushes pass=0 even though it is in bypass....
passMode = 2 is still just a intent, and eventually at some point the BP is on, but this is somewhat guesswork.

reinhard-brandstaedter, 2024-05-23:
This:
```2024-05-23 18:56:56,013:INFO: Turning hub bypass ON```
sends the hub the intent to turn bypass on. Then the hub takes a bit to really turn it on, but then sends a pass=0 which sf-control again interprets as BP is off. Finally when sf-control is able to determine by other means that the BP has kicked in, there could be another update to pass=0 which makes sf-control again think it is off.
So the only way to guess the true BP state is to completely ignore ```pass``` and use ```passMode == 2``` & ```solarInput > 0```  & ```outputPackPower == 0``` (hub has solarinput but doesn't charge the battery and has the passmode set - even outputHome seems to be 0 at this time). However this could also be true before the actual switch has happened.

Overall, bypass control on hub2k is not as precise as on hub1200

Try again tomorrow with a new build

mavo, 2024-05-25:
@reinhard-brandstaedter with the change to recognize the hub2k's weird BP state, I think the code needs a bit more adjustment.

The logic to leave the BP when time is right and demand is higher than supplied energy does not set the passMode, or at least I don't see how it's done.

This seems to be the reason why @floriandrott and I saw the exception thrown that the limit variable was not set.
This is in the decision path 0.1, where it does not set any new limit, but as the following condition if the hub is not in BP anymore is not true for the hub2k, limit is never set therefore python rightfully complains.

welli2009, 2024-05-26:
It works for me now, I simply wrote a Homeassistant automation that sets the inverter limit to 100% as soon as the hub has reached 100% and passMode 2 has been published. This is repeated every 2 seconds until two hours before sunset or until passMode 1 is published.

floriandrott, 2024-05-26:
@reinhard-brandstaedter today I was able to test the bypass mode again. Finally it worked 🥳 Thank you.

sf-control switched to bypass:
```
2024-05-26 16:40:45,997:INFO: HUB: S:175.7W [ 191.5,191.5,193.6,191.8,183.2,175.7 ], B: 98% (98), V:52.4V (52.4), C: 140W, P:False (manual, possible), F:68.5h, E:200.0h, H: 14W, L:600W
2024-05-26 16:40:45,997:INFO: INV: AC:25.0W, DC:26.3W (12.0|13.0), L: 20W (10.0W/channel) [800W]
2024-05-26 16:40:45,998:INFO: Based on time, solarpower (175.7W) minimum charge power (290W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2024-05-26 16:41:00,974:INFO: Bypass control, turning on bypass!
2024-05-26 16:41:00,974:INFO: Turning hub bypass ON
2024-05-26 16:41:00,974:INFO: Reading last full time: 2024-05-26 16:41:00
2024-05-26 16:41:01,015:INFO: Reading battery target mode: discharging
2024-05-26 16:41:07,470:INFO: HUB: S:143.0W [ 177.4,177.4,172.2,160.8,146.8,143.0 ], B:100% (100), V:53.3V (53.3), C: 113W, P:False (manual, not possible), F:0.0h, E:200.0h, H: 0W, L:600W
2024-05-26 16:41:07,470:INFO: INV: AC:23.4W, DC:24.6W (7.8|8.7), L: 20W (10.0W/channel) [800W]
2024-05-26 16:41:07,470:INFO: Based on time, solarpower (143.0W) minimum charge power (290W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
```

The bypass state in the last log message isn't correct but in the next one it is :)
```
2024-05-26 16:42:02,031:INFO: HUB: S:41.5W [ 43.9,43.9,33.4,29.3,34.1,41.5 ], B:100% (100), V:52.7V (52.7), C: 0W, P:False (manual, not possible), F:0.0h, E:200.1h, H: 0W, L:600W
2024-05-26 16:42:02,032:INFO: Based on time, solarpower (41.5W) minimum charge power (290W) and bypass state (True), hub could contribute 600.0W - Decision path: 0.2.
```

The log message about the `HUB` information still says for BP `false` but I assume that's correct due to our workaround.

Turning off the bypass is still an issue. sf-control turned off BP and then I got the following error:
```
2024-05-26 18:38:24,972:INFO: Turning hub bypass OFF
Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 513, in <module>
    main(sys.argv[1:])
  File "/solarflow/solarflow-control.py", line 510, in main
    run()
  File "/solarflow/solarflow-control.py", line 448, in run
    client.loop_forever()
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/solarflow/solarflow-control.py", line 133, in on_message
    smartmeter.handleMsg(msg)
  File "/solarflow/smartmeters.py", line 95, in handleMsg
    self.updPower()
  File "/solarflow/smartmeters.py", line 79, in updPower
    log.info(f'SMT triggers limit function: {previous} -> {self.getPower()}: {"executed" if self.trigger_callback(self.client,force=force_trigger) else "skipped"}')
                                                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 408, in limit_callback
    limitHomeInput(client)
  File "/solarflow/solarflow-control.py", line 342, in limitHomeInput
    sf_contribution = getSFPowerLimit(hub,hub_contribution_ask)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 257, in getSFPowerLimit
    log.info(f'Based on time, solarpower ({hub_solarpower:4.1f}W) minimum charge power ({MIN_CHARGE_POWER}W) and bypass state ({hub.getBypass()}), hub could contribute {limit:4.1f}W - Decision path: {path}')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'limit' where it is not associated with a value
```

And then following error message is recurring:
```
2024-05-26 18:39:07,488:INFO: Turning hub bypass OFF
Exception in thread Thread-597:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.11/threading.py", line 1401, in run
    self.function(*self.args, **self.kwargs)
  File "/solarflow/utils.py", line 25, in _run
    self.function(*self.args, **self.kwargs)
  File "/solarflow/solarflow-control.py", line 418, in deviceInfo
    limitHomeInput(client)
  File "/solarflow/solarflow-control.py", line 342, in limitHomeInput
    sf_contribution = getSFPowerLimit(hub,hub_contribution_ask)
[... cut ...]
    log.info(f'Based on time, solarpower ({hub_solarpower:4.1f}W) minimum charge power ({MIN_CHARGE_POWER}W) and bypass state ({hub.getBypass()}), hub could contribute {limit:4.1f}W - Decision path: {path}')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'limit' where it is not associated with a value
```
Due to this error sf-control is sending the maximum possible energy to the house without respecting the real consumption.

Anything what I can do to help?

floriandrott, 2024-05-27:
@reinhard-brandstaedter I updated to the last DEV build.

Here I saw that something is wrong with the detected BP mode by the hub:
`2024-05-27 15:04:49,095:INFO: HUB: S:289.3W [ 317.1,317.1,313.2,301.9,293.7,289.3 ], B: 64% (64), V:50.3V (50.3), C: 256W, P:True (manual, possible), F:20.4h, E:222.4h, H: 13W, L:600W`

`P=True` but this isn't correct. I double-checked the MQTT value and here sf-control is reporting `passMode=1` and also the battery is charging.

floriandrott, 2024-05-27:
@reinhard-brandstaedter I'm unsure if you changed something in regards of the `UnboundLocalError: cannot access local variable 'limit' where it is not associated with a value` issue in the DEV version from today. Just to let you know that I had this error again this evening.

reinhard-brandstaedter, 2024-05-28:
The BP mode not displayed in the log is more/less just a presentation problem (accessor not used that differentiates between hub2k and hub1.2k, fixed that.
The problem with the unbound variable hopefully shouldn't happen anymore, the hub2k special handling led to some kind of race condition with MQTT updates...that's the problem with the non-reliable state of the hub2k.

floriandrott, 2024-05-29:
@reinhard-brandstaedter yesterday I had the issue with the variable again. When the error occurs then somehow sf-control stops working and the battery gets discharged rapidly because the inverter limit stays with 600W (max value due to BP). Somehow the BP gets as well deactivated due to the issue so that the battery is discharging. I had expected if BP on and not enough solarpower is available, then the battery gets not discharged until BP is turned off.

floriandrott, 2024-05-30:
@reinhard-brandstaedter I was looking today a bit more into the error of the variable.

I have the setting configured that sunset is 95 minutes earlier. When sf-control is switching to `night mode` then the error occurs.

```
2024-05-30 19:37:41,019:INFO: HUB: S:23.7W [ 23.7,23.7 ], B:100% (100), V:50.6V (50.6), C: 0W, P:True (manual, not possible), F:0.0h, E:299.0h, H: 0W, L:600W
2024-05-30 19:37:41,020:INFO: INV: AC:15.8W, DC:1.8W (0.4|1.3), L:600W (300.0W/channel) [800W]
2024-05-30 19:37:41,020:INFO: SMT: T:Smartmeter P:175.0W [ 170.4,170.4,168.6,166.5,159.5,177.0 ]
2024-05-30 19:37:41,020:INFO: Direct connected panel are producing 0.0W, trying to get 137.0W from hub.
2024-05-30 19:37:41,020:INFO: Turning hub bypass OFF
Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 513, in <module>
    main(sys.argv[1:])
  File "/solarflow/solarflow-control.py", line 510, in main
    run()
  File "/solarflow/solarflow-control.py", line 448, in run
    client.loop_forever()
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/solarflow/solarflow-control.py", line 133, in on_message
    smartmeter.handleMsg(msg)
  File "/solarflow/smartmeters.py", line 95, in handleMsg
    self.updPower()
  File "/solarflow/smartmeters.py", line 79, in updPower
    log.info(f'SMT triggers limit function: {previous} -> {self.getPower()}: {"executed" if self.trigger_callback(self.client,force=force_trigger) else "skipped"}')
                                                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 408, in limit_callback
    limitHomeInput(client)
  File "/solarflow/solarflow-control.py", line 342, in limitHomeInput
    sf_contribution = getSFPowerLimit(hub,hub_contribution_ask)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 257, in getSFPowerLimit
    log.info(f'Based on time, solarpower ({hub_solarpower:4.1f}W) minimum charge power ({MIN_CHARGE_POWER}W) and bypass state ({hub.getBypass()}), hub could contribute {limit:4.1f}W - Decision path: {path}')
                                                                                                                                                                         ^^^^^
UnboundLocalError: cannot access local variable 'limit' where it is not associated with a value
```

With the next update interval the error goes to:
```
2024-05-30 19:37:42,960:INFO: Triggering telemetry update: iot/A8yh63/XXXXXXX/properties/read
2024-05-30 19:38:42,960:INFO: Triggering telemetry update: iot/A8yh63/XXXXXXX/properties/read
2024-05-30 19:38:42,993:INFO: HUB: S:23.7W [ 23.7,23.7 ], B:100% (100), V:50.6V (50.6), C: 0W, P:False (manual, not possible), F:0.0h, E:299.0h, H: 0W, L:600W
2024-05-30 19:38:42,993:INFO: INV: AC:15.8W, DC:1.8W (0.4|1.3), L:600W (300.0W/channel) [800W]
2024-05-30 19:38:42,993:INFO: SMT: T:Smartmeter P:175.0W [ 170.4,170.4,168.6,166.5,159.5,177.0 ]
2024-05-30 19:38:42,993:INFO: Direct connected panel are producing 0.0W, trying to get 137.0W from hub.
2024-05-30 19:38:42,994:INFO: Turning hub bypass OFF
Exception in thread Thread-9:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.11/threading.py", line 1401, in run
    self.function(*self.args, **self.kwargs)
  File "/solarflow/utils.py", line 25, in _run
[... 50 lines cut ...]
    smartmeter.handleMsg(msg)
  File "/solarflow/smartmeters.py", line 95, in handleMsg
    self.updPower()
  File "/solarflow/smartmeters.py", line 79, in updPower
    log.info(f'SMT triggers limit function: {previous} -> {self.getPower()}: {"executed" if self.trigger_callback(self.client,force=force_trigger) else "skipped"}')
                                                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 414, in limit_callback
    limitHomeInput(client)
  File "/solarflow/solarflow-control.py", line 342, in limitHomeInput
    sf_contribution = getSFPowerLimit(hub,hub_contribution_ask)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 257, in getSFPowerLimit
[... cut ...]
Now I was wondering when the error goes away .... at 20.30 I restarted again manually sf-control and then it was detected that the battery is by 97%. 

```
2024-05-30 20:12:17,000:INFO: HUB: S:10.5W [ 10.5,10.5 ], B: 97% (97), V:49.6V (49.6), C:-143W, P:True (manual, possible), F:0.4h, E:299.6h, H:148W, L:600W
```
And now the error is gone and sf-control works again. I assume that only when the battery drops at least one percentage (battery < 100%) the error goes away. For the detection of the correct `electricalLevel` a restart of sf-control is required.

Anything how I can help more?

reinhard-brandstaedter, 2024-05-31:
The responsible code for that behavior is here:
```
    if hub.getBypass():
        path += "0."
        # leave bypass after sunset/offset
        if (now < (sunrise + sunrise_off) or now > sunset - sunset_off) and hub.control_bypass and demand > hub_solarpower:
            hub.allowBypass(False)
            hub.setBypass(False)
            path += "1."
        else:
            path += "2."
            limit = hub.getInverseMaxPower()

    if not hub.getBypass():
        if hub_solarpower - demand > MIN_CHARGE_POWER:
            path += "1." 
            if hub_solarpower - MIN_CHARGE_POWER < MAX_DISCHARGE_POWER:
                path += "1."
                limit = min(demand,MAX_DISCHARGE_POWER)
            else:
                path += "2."
                limit = min(demand,hub_solarpower - MIN_CHARGE_POWER)
        if hub_solarpower - demand <= MIN_CHARGE_POWER:  
            path += "2."
            if (now < (sunrise + sunrise_off) or now > sunset - sunset_off): 
                path += "1."                
                limit = min(demand,MAX_DISCHARGE_POWER)
            else:
                path += "2."  
                #limit = 0 if hub_solarpower - MIN_CHARGE_POWER < 0 and hub.getElectricLevel() < 100 else hub_solarpower - MIN_CHARGE_POWER                                   
                limit = 0 if hub_solarpower - MIN_CHARGE_POWER < 0 else hub_solarpower - MIN_CHARGE_POWER
        if demand < 0:
            limit = 0
```

However I can't see why ```limit``` would not be set in any of those execution paths. I had this same error when I added the method calls for bypass determination on Hub2k, but with the recent change I haven't had it in the past days. Though with the current weather I only reached bypass twice....

reinhard-brandstaedter, 2024-06-01:
The subsequent getBypass() after path 0.1 should now also return false if the hub's state hasn't changed yet (using the internal bypass state of the solarflow class now. However if the hub2000 for some reason will not "accept" the switch we are working on a wrong state ... lets see how this proves in reality.

mavo, 2024-06-02:
Hello @reinhard-brandstaedter,
I pulled the latest dev from yesterday...

Now I see this:
```
2024-06-02 05:31:30,939:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-06-02 05:31:30,987:INFO: HUB: S:188.6W [ 224.0,224.0,220.0,206.1,195.5,188.6 ], B: 18% (18|18), V:48.6V (48.6|48.6), C:-111W, P:True (manual, possible), F:12.0h, E:49.4h, H:288W, L:800W
2024-06-02 05:31:30,988:INFO: INV: AC:286.1W, DC:301.0W (0.0|0.0|152.3|147.4), L:579W (144.8W/channel) [1600W]
2024-06-02 05:31:30,988:INFO: SMT: T:ShellyEM3 P:24.2W [ 12.2,12.2,12.8,13.9,15.8,15.4 ]
2024-06-02 05:31:30,988:INFO: Direct connected panel are producing 0.0W, trying to get 290.2W from hub.
2024-06-02 05:31:30,988:INFO: Turning hub bypass OFF
2024-06-02 05:31:30,989:INFO: Based on time, solarpower (188.6W) minimum charge power (0W) and bypass state (False), hub could contribute 290.2W - Decision path: 0.1.2.1.
2024-06-02 05:31:30,989:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-06-02 05:31:30,989:INFO: Solarflow is willing to contribute 145.0W (per channel) of the requested 290.2!
2024-06-02 05:31:30,989:INFO: Not setting inverter output limit as it is identical to current limit!
2024-06-02 05:31:30,989:INFO: Sun: 04:50 - 21:22 Demand: 290.2W, Panel DC: (0.0W), Hub DC: (284.8W), Inverter Limit: 580.0W, Hub Limit: 800.0W
2024-06-02 05:32:10,379:INFO: HUB: S:175.0W [ 185.3,185.3,180.5,177.6,176.5,175.0 ], B: 18% (18|18), V:48.6V (48.6|48.6), C:-117W, P:True (manual, not possible), F:12.0h, E:49.4h, H:287W, L:800W
2024-06-02 05:32:10,379:INFO: INV: AC:287.8W, DC:303.0W (0.0|0.0|154.5|149.6), L:579W (144.8W/channel) [1600W]
2024-06-02 05:32:10,379:INFO: SMT: T:ShellyEM3 P:130.9W [ 22.0,22.0,22.7,25.6,29.8,46.2 ]
2024-06-02 05:32:10,379:INFO: Direct connected panel are producing 0.0W, trying to get 325.1W from hub.
2024-06-02 05:32:10,380:INFO: Turning hub bypass OFF
2024-06-02 05:32:10,380:INFO: Based on time, solarpower (175.0W) minimum charge power (0W) and bypass state (False), hub could contribute 325.1W - Decision path: 0.1.2.1.
2024-06-02 05:32:10,380:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-06-02 05:32:10,380:INFO: Solarflow is willing to contribute 162.5W (per channel) of the requested 325.1!
2024-06-02 05:32:10,380:INFO: Setting inverter output limit to 648W (1 min moving average of 162W x 4)
2024-06-02 05:32:10,380:INFO: Sun: 04:50 - 21:22 Demand: 325.1W, Panel DC: (0.0W), Hub DC: (288.9W), Inverter Limit: 648.0W, Hub Limit: 800.0W
2024-06-02 05:32:10,380:INFO: SMT triggers limit function: 29.8 -> 46.2: executed
2024-06-02 05:32:10,382:INFO: SMT triggers limit function: 31.5 -> 48.8: skipped
2024-06-02 05:32:10,384:INFO: SMT triggers limit function: 33.1 -> 51.3: skipped
2024-06-02 05:32:12,393:INFO: SMT triggers limit function: 35.4 -> 60.4: skipped
2024-06-02 05:32:12,395:INFO: SMT triggers limit function: 37.0 -> 62.6: skipped
2024-06-02 05:32:12,397:INFO: SMT triggers limit function: 38.0 -> 63.7: skipped
2024-06-02 05:32:12,855:INFO: SMT triggers limit function: 39.7 -> 64.8: skipped
2024-06-02 05:32:12,857:INFO: SMT triggers limit function: 40.8 -> 65.9: skipped
2024-06-02 05:32:12,859:INFO: SMT triggers limit function: 41.8 -> 66.8: skipped
2024-06-02 05:32:14,402:INFO: SMT triggers limit function: 43.6 -> 75.0: skipped
2024-06-02 05:32:14,403:INFO: SMT triggers limit function: 44.5 -> 75.6: skipped
2024-06-02 05:32:14,405:INFO: SMT triggers limit function: 45.5 -> 76.4: skipped
2024-06-02 05:32:16,415:INFO: SMT triggers limit function: 50.4 -> 85.2: skipped
2024-06-02 05:32:16,416:INFO: SMT triggers limit function: 51.3 -> 85.8: skipped
2024-06-02 05:32:16,418:INFO: SMT triggers limit function: 52.2 -> 86.1: skipped
2024-06-02 05:32:16,817:INFO: SMT triggers limit function: 55.0 -> 89.7: skipped
2024-06-02 05:32:16,818:INFO: SMT triggers limit function: 57.8 -> 93.1: skipped
2024-06-02 05:32:16,821:INFO: SMT triggers limit function: 60.6 -> 96.5: skipped
2024-06-02 05:32:18,426:INFO: SMT triggers limit function: 69.1 -> 109.0: skipped
2024-06-02 05:32:18,428:INFO: SMT triggers limit function: 72.0 -> 112.1: skipped
2024-06-02 05:32:18,429:INFO: SMT triggers limit function: 74.7 -> 115.0: skipped
2024-06-02 05:32:20,438:INFO: HUB: S:172.4W [ 180.3,180.3,177.0,175.2,174.1,172.4 ], B: 18% (18|18), V:48.6V (48.6|48.6), C:-143W, P:True (manual, not possible), F:12.0h, E:49.5h, H:309W, L:800W
2024-06-02 05:32:20,438:INFO: INV: AC:288.6W, DC:303.8W (0.0|0.0|155.2|150.2), L:579W (144.8W/channel) [1600W]
2024-06-02 05:32:20,438:INFO: SMT: T:ShellyEM3 P:225.5W [ 45.5,45.5,51.1,61.1,87.0,128.4 ]
2024-06-02 05:32:20,438:INFO: Direct connected panel are producing 0.0W, trying to get 408.6W from hub.
2024-06-02 05:32:20,439:INFO: Turning hub bypass OFF
2024-06-02 05:32:20,439:INFO: Based on time, solarpower (172.4W) minimum charge power (0W) and bypass state (False), hub could contribute 408.6W - Decision path: 0.1.2.1.
2024-06-02 05:32:20,439:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2024-06-02 05:32:20,439:INFO: Solarflow is willing to contribute 204.0W (per channel) of the requested 408.6!
2024-06-02 05:32:20,439:INFO: Setting inverter output limit to 816W (1 min moving average of 204W x 4)
2024-06-02 05:32:20,439:INFO: Sun: 04:50 - 21:22 Demand: 408.6W, Panel DC: (0.0W), Hub DC: (290.2W), Inverter Limit: 816.0W, Hub Limit: 800.0W
[... cut ...]
2024-06-02 05:32:39,817:INFO: Setting inverter output limit to 600W (1 min moving average of 150W x 4)
2024-06-02 05:32:39,817:INFO: Grid feed in from battery: 150.6W!
2024-06-02 05:32:39,817:INFO: Sun: 04:50 - 21:22 Demand: 300.7W, Panel DC: (0.0W), Hub DC: (451.3W), Inverter Limit: 600.0W, Hub Limit: 800.0W
2024-06-02 05:32:39,817:INFO: SMT triggers limit function: -140.6 -> -140.6: executed
```

The paths are... strange... And my system decided at 7am, that BP is on and let full steam deplete the battery O.o

mavo, 2024-06-02:
I fixed this by removing the changes in your last commit.
This seems to be a problem, dunno why, but it is. I had It running the whole day with a small adjustment to not always drain 800w and instead only deliver what it required from the house, but I could see the weird paths all the time.
It also mentioned a lot of times "turning BP off" in the logs.

Now, that I reverted to the older version of the BP check line it seems to be back to normal...

reinhard-brandstaedter, 2024-06-02:
Which change exactly?

floriandrott, 2024-06-04:
@reinhard-brandstaedter with the DEV version 4 days ago, it was two days ago working with the bypass switch. Maybe the reason was that solar power was very low this time and then the switch was working?

Today I switched to the latest DEV version and directly I saw that sf-control thinks BP mode is on but it isn't.
```
2024-06-04 16:47:30,374:INFO: Turning hub bypass OFF
2024-06-04 16:47:30,376:INFO: Reading last full time: 2024-06-02 19:43:06
2024-06-04 16:47:30,376:INFO: Reading battery target mode: discharging
2024-06-04 16:47:30,376:INFO: Set ChargeThrough: False
2024-06-04 16:47:30,376:INFO: Solarflow set DryRun: False
2024-06-04 16:47:30,376:INFO: OpenDTU set DryRun: False
2024-06-04 16:47:30,376:INFO: Reading last empty time: 2024-05-18 08:38:41
2024-06-04 16:47:30,419:INFO: Solarflow set DryRun: False
2024-06-04 16:47:30,419:INFO: OpenDTU set DryRun: False
2024-06-04 16:47:48,102:INFO: Determined inverter's max capacity: 800.0
2024-06-04 16:49:30,374:INFO: HUB: S:833.5W [ 833.0,833.0,833.3,833.5 ], B: 95% (95), V:51.1V (51.1), C: 601W, P:True (manual, possible), F:45.1h, E:416.2h, H:199W, L:600W
2024-06-04 16:49:30,374:INFO: INV: AC:204.8W, DC:216.2W (108.1|108.1), L:202W (100.8W/channel) [800W]
2024-06-04 16:49:30,374:INFO: SMT: T:Smartmeter P:43.0W [ 44.6,44.6,44.9,43.2,42.1,42.0 ]
2024-06-04 16:49:30,374:INFO: Direct connected panel are producing 0.0W, trying to get 206.8W from hub.
2024-06-04 16:49:30,375:INFO: Based on time, solarpower (833.5W) minimum charge power (250W) and bypass state (True), hub could contribute 600.0W - Decision path: 0.2.
```

reinhard-brandstaedter, 2024-06-04:
What is your minSoc and socSet variables set to?

Is this log right after startup? Did you let it run for a bit? Did you start it at a time when the bypass was on or off?

I'm beginning to think your best options are at this time to:
a) not use sf-control's bypass control and let the hub itself do whatever it thiks is correct for BP
b) report a problem with Zendure, asking for the same reporting of the BP state like in Hub1200 in the ```pass``` property

floriandrott, 2024-06-05:
@reinhard-brandstaedter this morning it went directly into
```
2024-06-05 09:46:23,983:INFO: HUB: S:132.0W [ 132.0,132.0 ], B: 4% ( 4), V:44.0V (44.0), C:-491W, P:True (manual, not possible), F:13.7h, E:433.1h, H:600W, L:600W
2024-06-05 09:46:23,983:INFO: INV: AC:577.6W, DC:610.0W (304.9|306.3), L:600W (300.0W/channel) [800W]
2024-06-05 09:46:23,983:INFO: SMT: T:Smartmeter P:-388.0W [ -385.4,-385.4,-386.3,-386.5,-385.4,-384.2 ]
2024-06-05 09:46:23,983:INFO: Direct connected panel are producing 0.0W, trying to get 154.6W from hub.
2024-06-05 09:46:23,984:INFO: Based on time, solarpower (132.0W) minimum charge power (250W) and bypass state (True), hub could contribute 600.0W - Decision path: 0.2.
```

So, it occurs also when sf-control is running several hours. Yesterday I was starting sf-control when BP was off. And the log from yesterday was after the startup of sf-control. Now I was runnning sf-control over night without any restart and bypass state is unfortunately true again.
```
2024-06-05 05:14:25,931:INFO: HUB: S:0.0W [ 0.0 ], B: 19% (19), V:48.3V (48.3), C:-226W, P:True (manual, possible), F:9.2h, E:428.6h, H:218W, L:600W
2024-06-05 05:14:25,931:INFO: INV: AC:208.3W, DC:219.8W (123.1|123.2), L:226W (112.8W/channel) [800W]
2024-06-05 05:14:25,931:INFO: SMT: T:Smartmeter P:114.0W [ 80.2,80.2,85.0,96.8,109.5,139.9 ]
2024-06-05 05:14:25,931:INFO: Direct connected panel are producing 0.0W, trying to get 333.2W from hub.
2024-06-05 05:14:25,932:INFO: Turning hub bypass OFF
2024-06-05 05:14:25,932:INFO: Good morning! We have consumed 63% of the battery tonight!
2024-06-05 05:14:25,932:INFO: Syncing time of solarflow hub (UTC): 2024-06-05, 05:14:25
2024-06-05 05:14:25,932:INFO: Turning hub bypass OFF
2024-06-05 05:14:25,932:INFO: Based on time, solarpower ( 0.0W) minimum charge power (250W) and bypass state (False), hub could contribute 333.2W - Decision path: 0.1.2.1.
2024-06-05 05:14:25,932:INFO: Not setting solarflow output limit to 600.0W as it is identical to current limit!
2024-06-05 05:14:25,932:INFO: Solarflow is willing to contribute 166.5W (per channel) of the requested 333.2!
2024-06-05 05:14:25,933:INFO: Setting inverter output limit to 332W (1 min moving average of 166W x 2)
2024-06-05 05:14:25,933:INFO: Sun: 05:11 - 21:18 Demand: 333.2W, Panel DC: (0.0W), Hub DC: (233.3W), Inverter Limit: 332.0W, Hub Limit: 600.0W
2024-06-05 05:14:25,933:INFO: SMT triggers limit function: 109.5 -> 139.9: executed
2024-06-05 05:14:33,681:INFO: HUB: S:0.0W [ 0.0 ], B: 19% (19), V:48.2V (48.2), C:-268W, P:True (manual, possible), F:9.2h, E:428.6h, H:264W, L:600W
2024-06-05 05:14:33,681:INFO: INV: AC:258.3W, DC:272.7W (176.9|177.3), L:332W (166.0W/channel) [800W]
2024-06-05 05:14:33,681:INFO: SMT: T:Smartmeter P:-17.0W [ 72.8,72.8,78.9,90.2,95.1,89.4 ]
2024-06-05 05:14:33,681:INFO: Direct connected panel are producing 0.0W, trying to get 384.9W from hub.
2024-06-05 05:14:33,681:INFO: Turning hub bypass OFF
2024-06-05 05:14:33,681:INFO: Based on time, solarpower ( 0.0W) minimum charge power (250W) and bypass state (False), hub could contribute 384.9W - Decision path: 0.1.2.1.
```

The logs are telling me that sf-control is trying to disable BP. Fun fact is that the hub is not in BP mode so sf-control just thinks it is. This behavior started with the latest DEV version. With the version several days ago, it was working ... only the limit variable issue was present.

socSet = 100; minSoc = 3 are my settings.

floriandrott, 2024-06-05:
@reinhard-brandstaedter I now switched to the current master version and directly the bypass state is correctly detected.
```
2024-06-05 09:55:33,077:INFO: Turning hub bypass OFF
2024-06-05 09:57:33,076:INFO: HUB: S:97.5W [ 98.0,98.0,97.5 ], B: 4% ( 4), V:45.8V (45.8), C: 28W, P:False (manual, possible), F:13.9h, E:433.3h, H: 68W, L:600W
2024-06-05 09:57:33,079:INFO: INV: AC:82.0W, AC_Prediction: 82.0W, DC:86.5W, DC_prediction: 86.5W (42.8|43.7), L:100W (50.0W/channel) [800W]
2024-06-05 09:57:33,081:INFO: SMT: T:Smartmeter P:116.0W [ 120.8,120.8,120.5,123.2,118.6,117.2 ] Predict: 121.0W
2024-06-05 09:57:33,081:INFO: Direct connected panel are producing 0.0W, trying to get 159.2W from hub.
2024-06-05 09:57:33,082:INFO: Based on time, solarpower (97.5W) minimum charge power (250W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
```

reinhard-brandstaedter, 2024-06-05:
The current master is behind those changes done for hub2000. Let is run a few days with BP switches to see if it really works.
Or did you mean SF firmware?

floriandrott, 2024-06-06:
@reinhard-brandstaedter sure I can test this some days. I meant sf-control in regards of using the master branch right now.

I know that master hasn't included the hub2000 changes. Unfortunately I can't use the dev branch at all due to the wrong BP state. I switched to dev and directly sf-control thinks that the hub is in bypass.

```
2024-06-06 10:47:43,314:INFO: HUB: S:186.6W [ 199.5,199.5,197.7,188.6,180.7,186.6 ], B: 12% (12), V:48.3V (48.3), C:-281W, P:True (manual, possible), F:38.8h, E:458.2h, H:439W, L:600W
2024-06-06 10:47:43,314:INFO: INV: AC:429.9W, DC:448.1W (226.0|227.2), L:600W (300.0W/channel) [800W]
2024-06-06 10:47:43,314:INFO: SMT: T:Smartmeter P:-87.0W [ 85.3,85.3,88.6,38.5,-9.6,-54.8 ]
2024-06-06 10:47:43,314:INFO: Direct connected panel are producing 0.0W, trying to get 334.5W from hub.
2024-06-06 10:47:43,315:INFO: Based on time, solarpower (186.6W) minimum charge power (190W) and bypass state (True), hub could contribute 600.0W - Decision path: 0.2.
```

As @mavo mentioned in https://github.com/reinhard-brandstaedter/solarflow-control/issues/244#issuecomment-2143986380 there seems an issue with the BP state detection. Due to this sf-control sets the inverter limit to maximum all the time and empties the battery.

You you please check this change and then I would switch back to DEV. But right now if I would use the DEV branch, the battery just drained.

mavo, 2024-06-06:
You can modify the code yourself @floriandrott. Did the same.

How to:
1. ssh into the machine running your sf control app (the docker host)
2. docker ps (find the containerId)
3. `docker exec -it <containerID> /bin/sh`
4. `vi solarflow.py`
5. find method getBypass() and [change this line](https://github.com/reinhard-brandstaedter/solarflow-control/blob/dev/src/solarflow/solarflow.py#L438) into `return self.bypass_mode == 2`
6. write and quit
7. restart the docker container

Additionally, I also added setting the limit in decision path 0.1 for safety reasons to not get a problem with the undefined limit var if BP should be turned off at the end of the day.

MaVo

reinhard-brandstaedter, 2024-06-06:
Reverting that change however means that you could be in an undefined state until the hub decides to send an update via MQTT to the passMode topic, which means you will eventually for 2 minutes hang in an undefined state of path 0.1 with an arbitrary default limit variable. Not sure if this is desireable.

mavo, 2024-06-06:
> Reverting that change however means that you could be in an undefined state until the hub decides to send an update via MQTT to the passMode topic, which means you will eventually for 2 minutes hang [...]

I don't get this. Why would I end there? At which point in time? When BP is toggled on, or should be toggled off?
When the energy level reaches 100%, it sends a MQTT msg with passmode=2, so it would be updated from MQTT's side.

So it means this can only be for BP off scenarios? And there, as I also added in path 0.1 the line `limit = demand` would mean, it tries to provide what my house consumes... 🤔 
Where is the issue of my train of thoughts? Is there any?

mavo, 2024-06-06:
> Did anyone of you ever try to just the the hub run without sf-control steering the bypass? Apparently the hub2000 should do a much better job with the own bypass control. So it would be interesting how it behaves.

Actually, I did not. Maybe I give it another go... But this is kinda meh, as I cannot see its BP state in HA due to the fact it does not report it in the pass property properly.

floriandrott, 2024-06-06:
> Did anyone of you ever try to just the the hub run without sf-control steering the bypass? Apparently the hub2000 should do a much better job with the own bypass control. So it would be interesting how it behaves.

@reinhard-brandstaedter do you mean
a) not using sf-control at all and just limit the inverter in a different way or 
b) using sf-control if control_bypass = false and leaving the rest as it is?

Option b I tried it 2 weeks ago but here I saw that the bypass was activated / deactivated several times (everytime when solarpower was falling down under the inverter limit). But I can try this out again. Just let me know if option a or b.

reinhard-brandstaedter, 2024-06-06:
@floriandrott option b) this is what gives you exactly how Zendure intended the ByPass to work (or not work).
The whole sf-control steering the bypass was ever meant to be experimental, to get long term results if it can be done with less relay switching. For the hub1.2 this is now working well for some weeks, but the extra delays and hazzle with hub2k without an ability to test it properly is annoying.

floriandrott, 2024-06-06:
@reinhard-brandstaedter  interesting fact I have seen in MQTT explorer:

The hub is now with the setting bypass = auto. The hub switched to bypass = on and in MQTT Explorer I also see `pass=1` and `passMode=0`. Based on this sf-control thinks now that the hub is not in bypass:
```
2024-06-06 17:46:24,754:INFO: HUB: S:243.2W [ 242.7,242.7,243.2 ], B:100% (100), V:51.1V (51.1), C: 0W, P:False (auto, possible), F:0.0h, E:465.1h, H:238W, L:600W
2024-06-06 17:46:24,754:INFO: INV: AC:141.5W, DC:149.1W (74.2|74.9), L:144W (72.0W/channel) [800W]
2024-06-06 17:46:24,754:INFO: SMT: T:Smartmeter P:249.0W [ 188.6,188.6,202.2,221.5,226.8,238.3 ]
2024-06-06 17:46:24,754:INFO: Direct connected panel are producing 0.0W, trying to get 339.6W from hub.
2024-06-06 17:46:24,754:INFO: Based on time, solarpower (243.2W) minimum charge power (100W) and bypass state (False), hub could contribute 143.2W - Decision path: 2.2.
```

I think based on the changes for the hub2k, the bypass detection via `pass=1` is not working anymore. Really crazy how the hub reacts.

[17 more replies]

closed 2024-09-26 by reinhard-brandstaedter