reinhard-brandstaedter/solarflow-control #245: nach dem Wechsel des Wechselrichter wird DTU nicht mehr erkannt / DTU is no longer recognized after changing the inverter

frankkax, 2024-05-13:
Wegen eines Defektes des WR musste er getauscht werden, danach funktioniert das solarflow-control nicht mehr. Nach meiner Analyse liegt es an der Seriennummer des WR in der DTU. Der neue WR hat Buchstaben in der Seriennummer. Im dtus.py  ist die Seriennummer des WR aber als int deklariert.

259:    opts = {"base_topic":str ,"inverter_serial":int,"sf_inverter_channels":list}

Wäre es möglich diese Variable in str zu deklarieren?

Das Programmpaket hat bis jetzt hervorragend seinen Dienst gemacht.

Due to a defect in the WR, it had to be replaced, after which the solarflow-control no longer works. According to my analysis, it is due to the serial number of the WR in the DTU. The new WR has letters in the serial number. In dtus.py, however, the serial number of the WR is declared as int.

259:    opts = {"base_topic":str ,"inverter_serial":int,"sf_inverter_channels":list}

Would it be possible to declare this variable in str?

The program package has done an excellent job so far.

reinhard-brandstaedter, 2024-05-13:
Interesting, haven't seen any alphanumeric serials so far. Which Inverter are you using?
Shouldn't be a big change to read it as string.

frankkax, 2024-05-13:
ich habe jetzt einen HMS-1600-4T mit der Serienummer 114A00xxxxxx. Der defekte WR war ein HM-800-2T mit einer rein nummerischen Seriennummer.

I now have an HMS-1600-4T with the serial number 114A00xxxxxx. The defective WR was an HM-800-2T with a purely numerical serial number.

frankkax, 2024-05-13:
noch als Ergänzung, hier der Ausschnitt aus dem Protokoll nach dem Start des Programm:

As a supplement, here is the excerpt from the log after starting the program:

2024-05-13 13:51:28,807: INFO:
MIN_CHARGE_POWER = 125
2024-05-13 13:51:28,807: INFO:
MAX_DISCHARGE_LEVEL = 120
2024-05-13 13:51:28,808: INFO:
MAX_ INVERTER_LIMIT = 600
2024-05-13 13:51:28,808:INFO:
MAX
_INVERTER_
_INPUT = 475
2024-05-13 13:51:28,808: INFO:
SUNRISE_OFFSET = 60
2024-05-13 13:51:28,808:INFO:
SUNSET_OFFSET = 120
2024-05-13 13:51:28,811:INFO: Publishing Homeassistant templates...
2024-05-13 13:51:28,897:INFO: Published 26 Homeassistant templates.
2024-05-13 13:51:28,898: INFO: Triggering telemetry update: iot/73bkTV/875xxxxx/properties/read
Traceback (most recent call last):
File "/solarflow/solarflow-control.py", line 569, in ‹module > main (sys.argv [1:])
File "/solarflow/solarflow-control.py", line 566, in main
File "/solarflow/solarflow-control.py", line 491, in run
du_opts = getOpts (dtuType)
へへへへへへへへへへへへへへへへ
File "/solarflow/solarflow-control.py", line 450, in getopts opts.update/{opt:opt_type(converter(configtype._
_name_
_ lower (), opt) )})
AAA
AAAAAAAAAAA
File "/usr/local/lib/python3.11/configparser.py", line 834, in getint return self._get_conv(section, option, int, raw=raw, vars=vars,
へへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへ
File "/usr/local/lib/python3.11/configparser-py", line 824, in _get_conv return self._get(section, conv, option, raw=raw, vars=vars,
^^へへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへへ
File "/usr/local/lib/python3.11/configparser.py", line 819, in _get return conv(self.get(section, option, **kwargs))
へへへへへへへへへへへへへへへへへへへへへへへへへへへへ
へへへへ
ValueError: invalid literal for int() with base 10: '1164A00xxxxxx'

reinhard-brandstaedter, 2024-05-13:
fixed on latest :dev build

closed 2024-05-13 by reinhard-brandstaedter