reinhard-brandstaedter/solarflow-control #249: How to control multiple SF-Hubs

morlac, 2024-05-19:
Hey,
I have the following setup and want to control  two SF Hubs 
`
+---------+
| Panel 1 |------------+
+---------+            |
               +---------------+
               | SF Hub 1200 1 |----+
               +---------------+    |
               |   SF AB2000   |    |         +------+
               +---------------+    |         | Grid |
+---------+            |            |         +------+
| Panel 2 |------------+            |             |
+---------+                         |     +--------------+
                                    +-----| CH1          |
+---------+                               |              |
| Panel 3 |------------+            +-----| CH2          |
+---------+            |            |     |     HMS 2000 |
               +---------------+    |  +--| CH3          |
               | SF Hub 1200 2 |----+  |  |              |
               +---------------+       |  | CH4          |
               |   SF AB2000   |       |  +--------------+
               +---------------+       |         .
+---------+            |               |         .
| Panel 4 |------------+               |         .
+---------+                            |    +---------+
                                       |    | OpenDTU |
+---------+                            |    +---------+
| Panel 5 |----------------------------+         .
+---------+                                      .
                                                 .
             +-----------------+           +-----------+
             | Tasmota + HiChi | . . . . . | Mosquitto |
             +-----------------+           +-----------+
                                                 |
                                                 |
                                         +---------------+
                                         | HomeAssistant |
                                         +---------------+
`
should I start two instances or is there a better way to achieve my goal?

reinhard-brandstaedter, 2024-05-20:
I think you can get rid of one hub.
- 2 panels to the hub with 4 batteries
- 3 panels to the inverter direct
- use only one channel of inverter for the output of the hub

olomberg, 2024-05-20:
I agree with Reinhard with regards to use only 1 hub with 2 battery packs.

When it comes to the solar panel config I would propose to connect 4 panels to the SF hub. Otherwise it could be hard to load nearly 4KW into the batteries on non-sunny conditions. These are my findings using a Hub2000 with AB2000 when drilling into the data kept in my InfluxDB. My priority is the get the battery loaded to 100% to cover the night.

reinhard-brandstaedter, 2024-05-20:
Oh, it’s two Hub1200. Controlling two hubs with one smartmeter as source of steering is really tricky. There is no logic in sf-control for eg which one would be prioritized when it comes to discharging.
So if you keep both you could set one to a fixed schedule to cover baseload and control the other.

morlac, 2024-05-20:
The two Hub1200 (with currently one AB2000 each) should have both the same prio as the panels these are connected to are on the same balcony. 
So it looks like I have to spend some more time to extend solarflow-control @reinhard-brandstaedter =)

My idea is to get the current demand from smartmeter, substract what the single Panel directly connected to the inverter provides and set the remaining demand 50/50 as export-to-home on the two Hub1200.

Many thanks for your work so far!

reinhard-brandstaedter, 2024-05-21:
@morlac if it is two Hub1200 with one AB2000 each you really can get rid of one hub and just use the two batteries on one hub. That makes the whole setup a lot easier. I think with your panel config you should be able to charge the ~4kWh stack while using the remaining 3 panels for the demand throughout the day. That's about the same setup I use today.

reinhard-brandstaedter, 2024-05-21:
> My idea is to get the current demand from smartmeter, substract what the single Panel directly connected to the inverter provides and set the remaining demand 50/50 as export-to-home on the two Hub1200.

Unfortunately this sounds easier than it is...
The hub is (still) rather slow in steering and below 100W also not precise (only 30W steps). This is why sf-control today uses the inverter for precise/fast limiting of the sf-hub output to home. If you use the hub's output limit only you will likely get into fuzzy behavior.
Imagine your demand is 400W, your direct panel delivers 250W, so that you need to get 150W from the hubs.
That would mean 75W from each, but you can't set the limit to 75W, only to 90W or 60W.
On the other hand: if you need 400W and your panels deliver 150W you will need 125W from each hub. You will be able to set that output limit to 125W, but then your HMS2000 will likely start really bad "fluctuating" as I'm assuming you are using y-cables and every hub is connected to two channels of the inverter.

kutzera, 2024-05-23:
I have a similar setting. 2x 1200 Hubs connected to a HMS-2000-4T. Each of the Hubs have 4 batteries with an total of nearly 8kwh. The inverter is steering the output of both Hubs simultaneously. The differences between both hubs is small enough in summer to clear it out with a full charge in summer. The reason for this setup is the max output of 2kW. And because the inverter is controlling the output while the hubs are set to output 1000W each at all time the reaction time is quite good. the DTU on batterie firmware is synced with a Shelly Pro 3EM. They are working pretty well. in April I had just alsmost 9kwh grid import and in may until now i had 5kwh of input. 
@reinhard-brandstaedter it would be a fantastic if you would consider releasing a HASS add-on version or a HACS version for the guys without programming knowledge.

reinhard-brandstaedter, 2024-05-24:
@kutzera your goal is a different one than of sf-control. If your only source of power are batteries then you can easily limit only the inverter output. However sf-control is intended to support setups with panels directly connected to the inverter as well which requires to consider the different/changing input to the inverter (e.g. hoymiles evenly distributes the configured output limit to the inputs).
In your case if you limit the inverter to 2000W you will more or less draw 1000W from each hub (and cross the "legal" limit). Also you do not really need to control anything on the hubs, you can run them in auto-pilot mode and just do a simple inverterlimit to demand match (and maybe a charge-through logic for the winter) - that's where this project has evolved from.

As for the HASS add-on: I think someone in the community has already done this (how-to should be somewhere in the issues likely).

As for two-hub + direct panel support: I do not have two hubs and I won't buy a second one, so developing support for that setup is unlikely, I simply can't test it properly.

closed 2024-07-12 by reinhard-brandstaedter