reinhard-brandstaedter/solarflow-control #250: Idea / Question: Not limit the Inverter if the Huboutput is 0

busaku, 2024-05-22:
If we can see here, there is more pv possible than actual using. Sometimes it happens, that i will import energy, before sc will rise the inverter limit.
[image]

So, why we "limiting" the inverter if the output from the hub is null?
Maybe i dont get it, but it should be possible to set the limit to 100% until we are hitting the max output limit or did i miss something?

busaku, 2024-05-22:
[image]
here its more clear: its possible to rise the limit, bus it just move slow up to the limit
[image]

so if i power up some device, it takes a few seconds until the power consumption is coverd.

reinhard-brandstaedter, 2024-05-22:
Because the same is true in the "other direction" - if you draw power from the hub and turn off a device you are feeding in from the hub/battery. Limiting via inverter is faster than limiting via the hub, so we rather limit via inverter and not the hub. That mean if you suddenly need power from the hub and the inverter limit is high you could suddenly drain more than what you need. The priority is to not "give" away from the battery too easily.

busaku, 2024-05-22:
yes you are right, thats why i mean with "if hub output is set to 0". 
only then we should set the limit to max or some calculated value (used direct channels * max channel power (just an example)).

i think the possible effency benefit is not that much, but i should not have any negative effects or?

reinhard-brandstaedter, 2024-05-22:
As described above, you either get some power a bit later (unused solar) or you pot. loose some power from the hub to the grid upon sudden change in demand... so no efficiency gain at all I'd say.

tuxianerDE, 2024-10-31:
@busaku is this issue still a topic, the latest comment indicates "no need to change". otherwise propose and request to close.

If you see room for improvement we are happy to look over a PR