reinhard-brandstaedter/solarflow-control #252: Turn off ChargeThrough does not set hub output limit -> no power deliver to house

mavo, 2024-05-23:
Hi @reinhard-brandstaedter,
as my battery was not charged 100% for the last couple of days, I decided its time to do so. As there is also plenty of sun and my battery will be full soonish, I wanted to turn off CT and did this by setting the mqtt topic value.
This all worked, but nothing happened afterwards. 

There are 2 possible reasons:
1) the hub output level was not set because of unknown
2) the app recognised that the battery was not full for the configured amount of time from the config.

Here are the logs:
```
2024-05-23 10:34:49,957:INFO: HUB: S:1096.7W [ 1235.1,1235.1,1221.7,1207.7,1159.6,1096.7 ], B: 94% (94|93), V:51.0V (51.0|50.9), C: 855W, P:False (manual, possible), F:137.9h, E:6.1h, H: 0W, L: 0W
2024-05-23 10:34:49,957:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.0W, DC_prediction: 0.0W (), L: 0W (0.0W/channel) [1600W]
2024-05-23 10:34:49,959:INFO: SMT: T:ShellyEM3 P:408.5W [ 409.6,409.6,408.4,404.6,403.5,411.8 ] Predict: 416.8W
2024-05-23 10:34:49,959:INFO: HUB triggers limit function: 1159.6 -> 1096.7: executed
2024-05-23 10:34:50,973:INFO: HUB triggers limit function: 1137.6 -> 1077.9: skipped
2024-05-23 10:34:51,991:INFO: HUB triggers limit function: 1122.4 -> 1058.9: skipped
2024-05-23 10:34:53,008:INFO: HUB triggers limit function: 1099.0 -> 1039.8: skipped
2024-05-23 10:34:54,027:INFO: HUB triggers limit function: 1074.8 -> 1010.8: skipped
2024-05-23 10:34:55,047:INFO: HUB triggers limit function: 1049.9 -> 991.8: skipped
2024-05-23 10:34:56,064:INFO: HUB triggers limit function: 1024.4 -> 973.3: skipped
2024-05-23 10:34:57,083:INFO: HUB triggers limit function: 998.4 -> 955.4: skipped
2024-05-23 10:34:58,120:INFO: HUB triggers limit function: 971.9 -> 925.0: skipped
2024-05-23 10:34:59,140:INFO: HUB triggers limit function: 945.2 -> 893.5: skipped
2024-05-23 10:35:00,158:INFO: HUB: S:861.2W [ 1110.9,1110.9,1077.0,1010.6,918.0,861.2 ], B: 94% (94|93), V:51.0V (51.0|50.9), C: 546W, P:False (manual, possible), F:137.9h, E:6.1h, H: 0W, L: 0W
2024-05-23 10:35:00,158:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.0W, DC_prediction: 0.0W (), L: 0W (0.0W/channel) [1600W]
2024-05-23 10:35:00,161:INFO: SMT: T:ShellyEM3 P:412.6W [ 409.0,409.0,406.0,405.6,411.9,410.9 ] Predict: 412.2W
2024-05-23 10:35:00,161:INFO: HUB triggers limit function: 918.0 -> 861.2: executed
2024-05-23 10:35:01,181:INFO: HUB triggers limit function: 903.4 -> 828.5: skipped
2024-05-23 10:35:02,197:INFO: HUB triggers limit function: 889.7 -> 796.1: skipped
2024-05-23 10:35:04,215:INFO: HUB triggers limit function: 865.0 -> 745.5: skipped
2024-05-23 10:35:05,236:INFO: HUB triggers limit function: 853.4 -> 717.1: skipped
2024-05-23 10:35:06,256:INFO: HUB triggers limit function: 842.5 -> 691.1: skipped
2024-05-23 10:35:07,269:INFO: HUB triggers limit function: 832.1 -> 667.2: skipped
2024-05-23 10:35:07,653:INFO: Set ChargeThrough: False
2024-05-23 10:35:08,306:INFO: HUB triggers limit function: 809.0 -> 645.3: skipped
2024-05-23 10:35:09,323:INFO: HUB triggers limit function: 785.3 -> 625.1: skipped
2024-05-23 10:35:10,342:INFO: HUB: S:606.4W [ 981.9,981.9,899.3,814.4,761.0,606.4 ], B: 94% (94|93), V:51.0V (51.0|50.9), C: 490W, P:False (manual, possible), F:137.9h, E:6.1h, H: 0W, L: 0W
2024-05-23 10:35:10,343:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.0W, DC_prediction: 0.0W (), L: 0W (0.0W/channel) [1600W]
2024-05-23 10:35:10,345:INFO: SMT: T:ShellyEM3 P:412.4W [ 407.1,407.1,407.0,412.1,411.6,412.6 ] Predict: 401.1W
2024-05-23 10:35:10,346:INFO: HUB triggers limit function: 761.0 -> 606.4: executed
2024-05-23 10:35:11,361:INFO: HUB triggers limit function: 736.2 -> 589.3: skipped
2024-05-23 10:35:12,380:INFO: HUB triggers limit function: 711.3 -> 573.4: skipped
2024-05-23 10:35:13,400:INFO: HUB triggers limit function: 686.7 -> 558.8: skipped
2024-05-23 10:35:14,419:INFO: HUB triggers limit function: 662.8 -> 545.3: skipped
2024-05-23 10:35:15,438:INFO: HUB triggers limit function: 639.9 -> 532.7: skipped
2024-05-23 10:35:16,457:INFO: HUB triggers limit function: 618.4 -> 521.0: skipped
2024-05-23 10:35:17,476:INFO: HUB triggers limit function: 598.1 -> 510.1: skipped
2024-05-23 10:35:18,512:INFO: HUB triggers limit function: 579.1 -> 499.7: skipped
2024-05-23 10:35:19,531:INFO: HUB triggers limit function: 561.3 -> 489.7: skipped
2024-05-23 10:35:20,550:INFO: HUB: S:480.1W [ 806.9,806.9,724.5,675.5,544.8,480.1 ], B: 94% (94|93), V:51.0V (51.0|50.9), C: 367W, P:False (manual, possible), F:137.9h, E:6.1h, H: 0W, L: 0W
2024-05-23 10:35:20,550:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.0W, DC_prediction: 0.0W (), L: 0W (0.0W/channel) [1600W]
2024-05-23 10:35:20,552:INFO: SMT: T:ShellyEM3 P:408.2W [ 408.0,408.0,411.7,411.3,411.7,411.6 ] Predict: 400.4W
2024-05-23 10:35:20,553:INFO: HUB triggers limit function: 544.8 -> 480.1: executed
2024-05-23 10:35:21,575:INFO: HUB triggers limit function: 529.5 -> 470.5: skipped
2024-05-23 10:35:23,626:INFO: HUB triggers limit function: 507.2 -> 460.4: skipped
2024-05-23 10:35:24,645:INFO: HUB triggers limit function: 494.8 -> 449.8: skipped
2024-05-23 10:35:25,668:INFO: HUB triggers limit function: 483.2 -> 439.1: skipped
2024-05-23 10:35:26,682:INFO: HUB triggers limit function: 472.6 -> 428.3: skipped
2024-05-23 10:35:27,700:INFO: HUB triggers limit function: 462.9 -> 417.8: skipped
2024-05-23 10:35:28,810:INFO: HUB triggers limit function: 453.9 -> 407.8: skipped
2024-05-23 10:35:29,453:INFO: Triggering telemetry update: iot/A8yh63/ETFTke36/properties/read
2024-05-23 10:35:33,026:INFO: HUB: S:381.1W [ 665.4,665.4,580.7,478.1,432.0,381.1 ], B: 94% (94|93), V:50.5V (50.6|50.4), C: 323W, P:False (manual, possible), F:137.9h, E:6.1h, H: 0W, L: 0W
2024-05-23 10:35:33,026:INFO: INV: AC:0.0W, AC_Prediction: 0.0W, DC:0.0W, DC_prediction: 0.0W (), L: 0W (0.0W/channel) [1600W]
2024-05-23 10:35:33,028:INFO: SMT: T:ShellyEM3 P:407.1W [ 410.8,410.8,410.8,410.8,410.1,409.7 ] Predict: 411.8W
2024-05-23 10:35:33,028:INFO: HUB triggers limit function: 432.0 -> 381.1: executed
2024-05-23 10:35:37,098:INFO: HUB triggers limit function: 411.4 -> 358.5: skipped
2024-05-23 10:35:38,115:INFO: HUB triggers limit function: 401.3 -> 351.0: skipped
2024-05-23 10:35:39,134:INFO: HUB triggers limit function: 391.4 -> 343.9: skipped
2024-05-23 10:35:40,153:INFO: HUB triggers limit function: 382.1 -> 337.4: skipped
2024-05-23 10:35:41,172:INFO: HUB triggers limit function: 373.2 -> 331.2: skipped
[... 23 lines cut ...]
2024-05-23 10:40:04,716:INFO: SMT triggers limit function: 442.3 -> 455.7: skipped
2024-05-23 10:40:04,720:INFO: SMT triggers limit function: 443.3 -> 456.7: skipped
2024-05-23 10:40:04,775:INFO: SMT triggers limit function: 444.2 -> 457.7: skipped
2024-05-23 10:40:04,778:INFO: SMT triggers limit function: 445.2 -> 458.6: skipped
2024-05-23 10:40:04,782:INFO: SMT triggers limit function: 446.0 -> 459.5: skipped
2024-05-23 10:40:06,789:INFO: SMT triggers limit function: 449.0 -> 464.2: skipped
2024-05-23 10:40:06,793:INFO: SMT triggers limit function: 449.6 -> 464.6: skipped
2024-05-23 10:40:06,796:INFO: SMT triggers limit function: 450.1 -> 464.9: skipped
2024-05-23 10:40:08,801:INFO: SMT triggers limit function: 452.8 -> 469.6: skipped
2024-05-23 10:40:08,804:INFO: SMT triggers limit function: 453.3 -> 469.8: skipped
2024-05-23 10:40:08,807:INFO: SMT triggers limit function: 453.8 -> 470.0: skipped
2024-05-23 10:40:10,814:INFO: SMT triggers limit function: 456.3 -> 477.8: skipped
2024-05-23 10:40:10,817:INFO: SMT triggers limit function: 456.5 -> 477.6: skipped
2024-05-23 10:40:10,820:INFO: SMT triggers limit function: 456.8 -> 477.4: skipped
2024-05-23 10:40:11,714:INFO: SMT triggers limit function: 457.1 -> 477.2: skipped
2024-05-23 10:40:11,717:INFO: SMT triggers limit function: 457.2 -> 478.3: skipped
2024-05-23 10:40:11,720:INFO: SMT triggers limit function: 457.3 -> 478.0: skipped
2024-05-23 10:40:12,715:INFO: SMT triggers limit function: 459.9 -> 478.3: skipped
2024-05-23 10:40:12,718:INFO: SMT triggers limit function: 460.0 -> 477.9: skipped
2024-05-23 10:40:12,721:INFO: SMT triggers limit function: 460.1 -> 477.4: skipped
```

App restart did not help, only setting hubs output power manually to >0.

MaVo

reinhard-brandstaedter, 2024-05-23:
No, the reason for what you are seeing is that your inverter is not ready (offline, no power at all).
Since you basically interrupted the CT no power is fed to inverter. So sf-control still has to assume the inverter is not ready, hence no limit function is not executed.

mavo, 2024-05-23:
But why does this action (turn off CT) not start the inverter by setting hub output power?
I could not start the inverter, I needed to provide power which I did by using the HA controls.
Not ideal, imo.

reinhard-brandstaedter, 2024-05-23:
well, leaving CT mode without any direct panels wasn't an anticipated usecase :-)

mavo, 2024-05-23:
And might it be one for the future? 😇

mavo, 2024-05-23:
Thanks for adjustment of the app!

closed 2024-05-23 by mavo