reinhard-brandstaedter/solarflow-control #254: Bypass switch not available in HA anymore

mavo, 2024-06-02:
Hellow @reinhard-brandstaedter (and maybe @ricariel as I think you did the most for the HA integrations),

for some reason, since 3 days (shown by HA in the entity state) the switch for BP mode is not available anymore.
[image]

This used to work fine, and I was able to flip it which also worked, but now its gone. I did not update the app (until yesterday, in a desperate attempt of fixing it by updating the app). Do you have any idea what could be the cause of this?
Only thing I know that happened was a firmware update of the NAS where I run the containers. But as everything else works, I do not really consider this as the direct cause.

What kind of data / logs can I provide to investigate this?

Thanks in advance,
MaVo

reinhard-brandstaedter, 2024-06-02:
You should have “Bypass Mode” and “Bypass State” available, can you check?

[image]

mavo, 2024-06-03:
Ahhh, I have no Idea why I did not see this, but yeah. You're right, that exists.

The only downside of it, its not populated with its current state -> it shows an empty selection (like in your screenshot).
The older version did indicate its current value and updated it when the value changed in MQTT.

reinhard-brandstaedter, 2024-06-03:
I actually just fixed a bug in the HA template for that selector. It should now be properly updated  and propagated!

mavo, 2024-06-03:
Awesome, I'll give it a try and close when it works, thank you!

mavo, 2024-06-06:
Hello @reinhard-brandstaedter,

I now pulled latest dev build and I can see in the container, that the template is updated. However, HA still shows
[image]
Anything I can do?

MaVo

mavo, 2024-06-07:
Any update, anything I can?

reinhard-brandstaedter, 2024-06-09:
I added the passMode to sensors now to get the current state.
The select switch is not updating but works for changing/setting the passMode

[image]

mavo, 2024-06-10:
Thank you. I can see the new entity in HA 🙇

closed 2024-06-10 by mavo