reinhard-brandstaedter/solarflow-control #256: Feature request: enable/disable the control logic

xsived, 2024-06-08:
Hi,

It would be great to have a MQTT command, which disables/enables the control logic of SF-control.
My use case:
I want to keep my feed to the grid at max 800W (max-inverter-limit) at all times, but I have 2640W of installed panels 
There are consumers in my house (EV, water boiler, IR heating), which are using much more power than 800W.
It would be great if I could do the following through my home automation:
- stop solarflow-control logic
- start boiler
- set inverter(s) to 100%
- wait for the water to get hot
- set inverter(s) to something between 0 and 50 % (based on the smart meter output)
- stop boiler
- start solarflow-control logic again

I know, I could stop/start the docker instance, but in this case, I'm losing the SF MQTT status.

Another option would be to be able to set the max-inverter-limit through MQTT.

Best regards
ivan

reinhard-brandstaedter, 2024-06-08:
The maximum AC limit (max_inverter_limit) is there for a reason. You can of course set it as high as you want, but I do not want to be held responsible if you burn down your house because your AC output is too high.
There are currently no plans to support a "max feed in feature" (see also #201 )

xsived, 2024-06-11:
Hi Reinhard,
Thanks for the prompt answer. I know the reasoning behind the max_inverter_limit and I for sure don't advocate for a "max feed in feature". My idea was to just temporarily disable the inverter and sf control logic over mqtt :-)
Best regards
-i.

reinhard-brandstaedter, 2024-06-12:
You can switch ```dryRun``` on/off on the fly by publishing "ON" or "OFF" to ```solarflow-hub/<device id>/control/dryRun```. It would still run but not publish any commands to SF or the inverter.
However I don't get it why you would want to do that. sf-control is usually detecting higher demand anyway and will provide up to the max_inverter_limit power if needed and available. So why do this externally?

xsived, 2024-06-13:
Hi Reinhard,
I thought the "dryRun" applies only to the SF-hub (because of the publish topic). Now I see, it's also in the DTU.py.
I would need this, when I want to take over the control of the DTU, without having to stop the sf-control container.
Thanks for the great work!
best regards
-i.

closed 2024-06-13 by xsived