reinhard-brandstaedter/solarflow-control #259: Hub2000 not reporting properties

Hidri02, 2024-06-10:
I have put my Hub 2000 offline using [Solarflow Bluetooth Manager](https://github.com/reinhard-brandstaedter/solarflow-bt-manager) and configured it to send the data to my mosquitto MQTT broker hosted on my HomeAssistant instance. 

At first I had issues with authentication because there is no way to turn off authentication in the mosquitto addon in home assistant. I ended up sniffing the password it tries to use with a network trace and wireshark. (The password for mine ended up being "B7C2BB5717917F0D", I wonder if that is the same for everyone?)
After fixing the authentication issue I could see the hub connect successfully and exchanging constantly pings as well as data in regular intervals of about every minute to:
/A8yh63/qL6HahTa/event/device

It seems to send 3 payloads every minute with the main difference being the "mode"?
```
{
    "messageId": "123",
    "method": "device",
    "deviceId": "qL6HahTa",
    "timestamp": 1718032838,
    "offData": 0,
    "data": {
        "eventId": 3,
        "startTime": 1718032659,
        "endTime": 1718032838,
        "inout": 1,
        "mode": 3,
        "electric": 18448
    }
}
```

```
{
    "messageId": "123",
    "method": "device",
    "deviceId": "qL6HahTa",
    "timestamp": 1718032848,
    "offData": 0,
    "data": {
        "eventId": 3,
        "startTime": 1718032669,
        "endTime": 1718032848,
        "inout": 1,
        "mode": 1,
        "electric": 17544
    }
}
```

```
{
    "messageId": "123",
    "method": "device",
    "deviceId": "qL6HahTa",
    "timestamp": 1718032858,
    "offData": 0,
    "data": {
        "eventId": 3,
        "startTime": 1718032679,
        "endTime": 1718032858,
        "inout": 1,
        "mode": 2,
        "electric": 33645
    }
}
```

I can also see solarflow-control asking for updates with a getAll at iot/A8yh63/qL6HahTa/properties/read
This is being responded to with a reply from the hub at: /A8yh63/qL6HahTa/properties/read/reply
```
{
    "messageId": "123",
    "deviceId": "qL6HahTa",
    "timestamp": 1718032806,
    "success": 1,
    "properties": {
        "getAll": 1
    }
}
```

However I do not get any further data. Based on what I have seen from others I should receive data on /A8yh63/XXXXXXXX/properties/report.

Any idea on what I could be doing wrong? :/
Or could this to be an issue with the latest firmware? (I got my Hub freshly last week and updated it to the latest verson)

reinhard-brandstaedter, 2024-06-10:
Regarding the password: no it is not identical for everyone. The password is based on the device ID and differs for everyone (but it can be calculated ;-)

So your hub reports something already, that means the setup has worked.
In the past people have reported that it took some time until telemetry is reported. Have you tried restarting the hub?

Hidri02, 2024-06-10:
Ah interesintg, didn't know it can be calculated!

Looking at it just now it appears that the battery indicator on the Hub (furtherst right LED) is currently blinking red. According to the manual for the AB2000 battery this indicates a low / high temperature warning. This doesn't really make sense to me as the shed the battery is sitting in is sitting at a nice 20°C.

I have just tried turning it off (holding the button 6 seconds waiting for all lights to go out) and on again (holding the button for 2s). This didn't really change anything, still blinking red battery led and the same data. As there wasn't really much sun today and the system has been disconnected since yesterday, I wonder if it just somehow drained the battery almost completely causing the blinking led.

One thing I just noticed while looking back through my steps is that I input the wrong device-id while taking the hub offline with solarflow-bt-manager. Instead of the device-id I set "SF_DEVICE_ID" to the ID number reported in the app for the hub (which is different than the actual device id). I wonder if that explains my issues with the property data report not working.

I will try to redo the process once more another day and report back. Hopefully I did not brick my battery in the process...

reinhard-brandstaedter, 2024-06-10:
Don't worry, it's unlikely that you bricked your battery that way. But with the wrong device id the final confirmation command was definitely sent wrong.
Either just repeat the command to take offline with the correct product id and device id (restart hub or wait). Or reset the hub and then try again with the correct id(s).
If your hub is not responding to bluetooth, just restart it.

Hidri02, 2024-06-17:
Thank you, after adapting the device id to the correct one it started working properly! :)

closed 2024-06-17 by Hidri02

nfocke, 2024-07-08:
> Thank you, after adapting the device id to the correct one it started working properly! :)

@Hidri02
Just to confirm,  you are getting property report topics with actual readings now? Which Firmware are you on?

Asking because I have the exact same issue with a new Hub2000 and could not get it to report actual data... restart did not help either.

Hidri02, 2024-07-08:
@nfocke Yes, I have all the metrics working properly for me now. Initially I had issues with getting mqtt to work with my mosquito in HomeAssistant due to being unable to turn of authentication. I ended up sniffing the password it tries with wireshark and then adding the user with that password to mosquito for authentication.
Then I had it half working where I could control it but the report data didn't show it. There my issue was that I used the wrong id as device id when I took the Hub offline. I used the id from the app instead of grabbing the device id as explained in the guide. After repeating the steps for taking it offline with the right ID it started working.

As for versioni don't know if there is a new one now, I installed mine about a month ago and I can't check in the app easily now that my hub is disconnected. MQTT reports the following:

`{
    "messageId": "123",
    "product": "solarFlow",
    "deviceId": "qL6HahTa",
    "timestamp": 1720472392,
    "properties": {
        "remainOutTime": 97,
        "remainInputTime": 59940,
        "packState": 2,
        "hubState": 0,
        "masterSoftVersion": 12298,
        "masterhaerVersion": 0,
        "inputMode": 0,
        "blueOta": 1,
        "pvBrand": 1,
        "pass": 0
    }
}
`

nfocke, 2024-07-09:
@Hidri02 Thank you for the info. Will check with my device version. 

Just one side-note. If you want to have the hub "online" in your Zendure hub and are comfortable with writing some Python code, I forward my MQTT messages (from my custom broker) with using paho-MQTT to the official Zendure broker. When you have the username (your device id) and password (you seem to have found it out), you can log in there and forward the messages. Our can modify Reinhard's code somewhat to allow this forwarding (in the on_message function).

nfocke, 2024-07-10:
@Hidri02 Problem solved. Was a in issue with the response message in the broker runner. So, in the end the effect was the same as in your case. No correct register replay was sent to the hub and then it seems not to report but just sends limited messages. Thanks for your pointers